dresden-elektronik/deconz-rest-plugin #5324: Conbee 2 just won't to devices, has it DIED after 2months of usage?

SeanHama, 2021-09-26:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi 4
 - Running method: (Raspbian only, no docker
 - Firmware version: (latest 19/08/2021)
 - deCONZ version: (2.12.06
 - Device:  ConBee II 
 - Do you use an USB extension cable: -yes 
 - Is there any other USB or serial devices connected to the host system? yes, clipsal device

## deCONZ Logs

## Additional context

Smanar, 2021-09-26:
05:30:30:376 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
05:30:30:376 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF0743D3, start reconfiguration

You have setting issue
Try to restore an old backup

Or try
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-issues

Can compare your setting with the capture from this link too
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-and-configuration-restore-does-not-help

SeanHama, 2021-09-26:
Ok, ive tried all your suggestions.
Ive reflashed firmware, Im on latest version of deconz.

Any other ideas pls, my house needs lights

SeanHama, 2021-09-26:
The TC ADDRESS AND TC LINK MASTER KEY A ZERO? Is that the issue?

Smanar, 2021-09-26:
You have tried an olad backup ?
Or use the hidden config to set an old connexion setting ?
And you still have the same error message ?

Have you tried to compare your setting with the last link and set them correctly manually ?

SeanHama, 2021-09-26:
Yes Ive tried old backup, it doesnt work 

"Or use the hidden config to set an old connexion setting ?" this page is not working. How do I exactly getting this to appear. I followed the article but nothing.

"Have you tried to compare your setting with the last link and set them correctly manually ?" ---> how do I do that?

SeanHama, 2021-09-26:
Can we jump on a screenshare. ??

Smanar, 2021-09-26:
>this page is not working. How do I exactly getting this to appear. I followed the article but nothing.

You have used ALT + clic ?

>how do I do that?

On the GUI, on the menu, you have something like "network setting", just compare that you have with the capture on the link.

SeanHama, 2021-09-26:
Alt + click does not work? Have you confirmed it still works in the latest release?

SeanHama, 2021-09-26:
[image]
Alt+Click, look at what happens

SeanHama, 2021-09-26:
This is my Deconz, any ideas or suggestions?
[image]

SeanHama, 2021-09-26:
This is what is appearing now in my Network settings, its still not working?
[image]

Mimiix, 2021-09-26:
Hi, 

As this is not a bug report as described in #5113, this issue gets closed. 

Feel free to use the forum for user questions.

closed 2021-09-26 by Mimiix