dresden-elektronik/deconz-rest-plugin #5328: deCONZ looses connection on a frequent basis

philipp089, 2021-09-28:
## Describe the bug

 deCONZ looses connection to all bulbs, motion sensors and remote controls, sometimes it shows "firmware not connected". The issue can sometimes be fixed by going in to the advanced options of phoscon (alt+click) and loading an older configuration. Sometimes it can be fixed by searching for lights in Phoscon but sometimes none of these steps work and i am back at resetting bulbs with the dimmer switch which takes quite a while of time. This behaviour is not brand specific as it happens with Philips Hue and Ikea Floalt or Tradfri. *.dat-Files were send to mpi@dresden-elektronik.de, if you need more data like logs etc. please let me know.

## Steps to reproduce the behavior

Quite random appearance, sometimes after rebooting HA, sometimes from one second to the other.

## Expected behavior

No connection loss or firmware not connected status. I had the setup of deCONZ and Raspbee II running without issues for at least a year but since a month or so i run into this issue at least once a week.

## Screenshots

## Environment

 - Host system: Raspberry Pi 4B 4GB
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26520700
 - deCONZ version: 2.12.06
 - Device: RaspBee II
 - Do you use an USB extension cable: no because RaspBee II
 - Is there any other USB or serial devices connected to the host system? If so: Google Coral USB Accelerator

## deCONZ Logs

that is quite an endless list, can you provide an email to send that files? As the system is running at the moment it might be usefull to send that information when the next occurance happens as i am pretty sure, that this is just a matter of time.

Mimiix, 2021-09-28:
Hi there,

As this is not a bug report as described in #5113, i am closing this issue. All questions and problems regarding setups are handled on the forums: https://forum.phoscon.de.

Kind regards,

closed 2021-09-28 by Mimiix

philipp089, 2021-09-28:
opened an issue under the general support section in the forum as long as it is not a known bug:
https://forum.phoscon.de/t/deconz-looses-connection-on-a-frequent-basis/532

Thanks for pointing me to the forum for future purposes.

Cheers, Philipp