dresden-elektronik/deconz-rest-plugin #5330: Aqara Roller Shade Driver E1

hastime, 2021-09-28:
## Device

- Product name: Roller Shade Driver E1
- Manufacturer: LUMI
- Model identifier: lumi.curtain.acn002
- Device type : Please remove all unrelated device types. 
 
  - Other: Battery Powered roller shade driver

## Screenshots

[image]

### Basic

[image]

### Identify

[image]

### Alarms

[image]

### Device Temperature

[image]

### Groups

[image]

### Scenes

[image]

### On/Off

[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

[image]

[image]

[image]

[image]

[image]

Mimiix, 2021-09-29:
Hi,

Can you also make a post on the forum?

hastime, 2021-09-29:
> Hi,
> 
> Can you also make a post on the forum?
> 

Hi I have made a forum post now, thanks, it just needs to be approved by the moderators. 

Thanks!

Mimiix, 2021-09-29:
> Hi I have made a forum post now, thanks, it just needs to be approved by the moderators.

You triggered the spambot for some reason 😄. I approved it. Thanks!

hastime, 2021-09-29:
> > Hi I have made a forum post now, thanks, it just needs to be approved by the moderators.
> 
> You triggered the spambot for some reason 😄. I approved it. Thanks!

Thanks! :)

ads1230, 2021-10-11:
Any luck with this?

volantynys, 2021-11-01:
Same already have one, would be wonderful to use it to its fullest

Mimiix, 2021-11-04:
None, as no devs have picked it up.

sanderlv, 2021-11-16:
Same here... deconz shows it but not function. Allthough I might try to switch too z2m and say goodbye to deconz... https://zigbee.blakadder.com/Aqara_ZNJLBL01LM.html

Holbroser21, 2021-11-16:
@sanderlv
I was actually able to pair mine with deConz successfully over the weekend. It doesn't show battery, but otherworks works perfectly in Home Assistant as a blind controller!  I am not quite sure what I did, as I spent over an hour attempting to pair it in the end.  However, I think there are at least a few steps needed. 

First you must have the top and bottom points set on the controller (they can be anything, you can change them later), and obviously, it must be paired and visible in the deConz gui.

At this point it was showing up as a light for me in both Phoscon and Home Assistant, and I spent some time manually reading every cluster, multiple times both with Phoson in searching mode, and not.  

Finally, I think what seemed to work was when I used the deConz.configure service in Home Assistant to send data to it over the API. I was only aiming to re-name the entity in deConz, but from there, it seems to have worked without an issue. 

If you give it all a go, let me know how you get on

Mimiix, 2021-11-16:
@SwoopX Can you help on this?

sanderlv, 2021-11-16:
@Holbroser21 not sure waht you actually mean to do...

I tried to search the deConz.configure but I do not seem to find "what to" service to it?
[image]

I have set a top and bottom limit. But Id no not get any feedback/control:
[image]

Holbroser21, 2021-11-16:
@sanderlv I am just as confused myself to be honest, although obviously very happy it does work. I must have pushed every button/ reset every setting / clicked every box about 5 times while trying to set it up. 

The service call I made was just this, where I got the path of the controller from the Phoscon API page:
[image]

Then this is now what I see on the entitiy's page in HA:
[image]

I do have this set in deConz from when I was testing things, but the values apear to have no effect on anything:
[image]

Mimiix, 2021-11-16:
@Holbroser21 Me and @sanderlv tried some things on discord. I've noticed he iddn't had the Lumi specific cluster. That's prolly why it didn't work for him yet. He'll try tonight :)

lwsrbrts, 2021-11-16:
A couple of things I notice on my E1. My "firmware" date compared to @Holbroser21 is different.

[image]

And the "Lumi specific" node ID on @Holbroser21's is FCC0 whereas on mine it's the same ID but "Unknown" and selecting it does nothing.

[image]

I've tried resetting and re-discovering but I always end up with a dimmable light device type.

lwsrbrts, 2021-11-16:
Mine is: 26720700

[image]

Mimiix, 2021-11-16:
@lwsrbrts https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues Can you try  this?

lwsrbrts, 2021-11-16:
@Mimiix thanks for the idea. No change unfortunately.

Mimiix, 2021-11-16:
@Holbroser21 What deCONZ version are you on?

Holbroser21, 2021-11-16:
Nothing special, I'm using the deCONZ add-on in HA, version 6.10.0 (latest), and my Conbee is a V1, firmware 0x26400500 (latest stable)

kvanisterdael, 2021-11-17:
@lwsrbrts : I have the same issue; the E1 always shows up as a Dimmable Light:

[image]

However if I go into the cluster 'Window covering' I can execute up/down/stop:
[image]

sanderlv, 2021-11-17:
> 
> 
> @lwsrbrts : I have the same issue; the E1 always shows up as a Dimmable Light:
> 
> [image]
> 
> However if I go into the cluster 'Window covering' I can execute up/down/stop: [image]
> 
> My deConz version: [image]

Yes, exactly like all but one of us. Although your conbee stick firmware is older than most of us. If upgaded it exposes extra clusters in your aqara (not that it helps, but it's more than you have now :-))

kvanisterdael, 2021-11-17:
@sanderlv : I have a RaspBee 1 instead of a Conbee, so that's why the firmware is different; it is the latest firmware available...

SwoopX, 2021-11-17:
Guys, please check out the following:

1. Make sure your running the latest Conbee/Raspbee firmware https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog
2. If you run 2.13.0 or 2.13.1, make sure you update to 2.13.2
3. If any of the above was not given **before** you paired the device(s), reset and re-pair while being up to date
4. Pair in normal/hybrid mode (tab control)
5. If you followed 1-4, try restarting deconz
One thing worth to note: if the FCC0 cluster has the name of unknown, then there must be something off with the setup. That one got added many versions ago and is since then part of file `general.xml`

sanderlv, 2021-11-17:
>     1. Make sure your running the latest Conbee/Raspbee firmware https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog
-- Yes, I am
[image]
 
>     2. If you run 2.13.0 or 2.13.1, make sure you update to 2.13.2
-- Not sure how to/where to update since I am running the addon in home assistant...
> 
>     3. If any of the above was not given **before** you paired the device(s), reset and re-pair while being up to date
-- Was given

>     4. Pair in normal/hybrid mode (tab control)
-- Not sure where to do this.
>     5. If you followed 1-4, try restarting deconz
>        One thing worth to note: if the FCC0 cluster has the name of unknown, then there must be something off with the setup. That one got added many versions ago and is since then part of file `general.xml`
-- It's not unknown with me:
[image]

lwsrbrts, 2021-11-17:
Thanks for getting involved @SwoopX 

> One thing worth to note: if the FCC0 cluster has the name of unknown, then there must be something off with the setup. That one got added many versions ago and is since then part of file `general.xml`

The installation is from around March 2021 (this year) and is using the SD-card images available from https://phoscon.de/en/raspbee/sdcard at that time - so it's not _that_ old and I have to assume the image doesn't introduce anything _off_ through normal use and updates.

The SD-card image is the stable version rather than beta and all updaes will have been performed through `apt`. 

Flashing the firmware on the ConBee II will have been done using the instructions available here: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian

It's worth me mentioning that I'm not running beta and am therefore on 2.12.06. If I need to run 2.13.02 that's fine but it's not particularly clear that version is _required_. Is it?

lwsrbrts, 2021-11-17:
After a complete reboot of the RasPi and re-connecting the blind controller, I've now got a `Lumi specific` FCC0 identifier, alas, the blind controller is still not showing in Home Assistant as a controllable covering.

The API says:

```json
{
    "etag": "9e579fdd4e0b18a0988d654801c8482b",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-11-17T12:47Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.curtain.acn002",
    "name": "Office Blind",
    "state": {
        "bri": 254,
        "lift": 100,
        "on": true,
        "open": false,
        "reachable": true
    },
    "swversion": "05-12-2021",
    "type": "Dimmable light",
    "uniqueid": "54:ef:44:10:00:1d:74:be-01"
}
```

sanderlv, 2021-11-17:
Mine says on API:

{
   "etag":"575386d68fe8a2f7ae09605e0e8f0767",
   "hascolor":false,
   "lastannounced":"2021-11-16T14:20:30Z",
   "lastseen":"2021-11-17T18:02Z",
   "manufacturername":"LUMI",
   "modelid":"lumi.curtain.acn002",
   "name":"aqara 1",
   "state":{
      "bri":15,
      "lift":6,
      "on":true,
      "open":true,
      "reachable":true
   },
   "swversion":"05-18-2021",
   "type":"Dimmable light",
   "uniqueid":"54:ef:44:10:00:2e:09:5d-01"
}

SwoopX, 2021-11-23:
I had a look around on this device and it appears that Xiaomi is (once again) not following the standards here.

Can anyone please try setting attribute 0x0055 of the analog output cluster (0x000D) to have it opened/closed? Should have a range from 0 - 100.

thomasloven, 2021-11-23:
That works. And reading from it reports the position when manually changed.

sanderlv, 2021-11-23:
what should I punt where?

[image]

closed 2021-11-26 by manup

kvanisterdael, 2021-11-26:
I pulled the latest commits and recompiled my deConz, but the E1 is still visible as a dimmable light in the Phoscon application. 
What can I do to make this work?

SwoopX, 2021-11-26:
Use the REST API with `state/lift`

kvanisterdael, 2021-11-29:
Thanks for your help, but I get following reply:

```
PUT /api/A53342B49D/lights/3/state HTTP/1.1
Content-Type: text/plain
User-Agent: PostmanRuntime/7.28.4
Accept: */*
Cache-Control: no-cache
Postman-Token: 67062b3a-07b7-4084-9fe0-202526ac5633
Host: 192.168.0.4
Accept-Encoding: gzip, deflate, br
Connection: keep-alive
Content-Length: 22
 
{
"lift": 50,
}
 
HTTP/1.1 200 OK
Access-Control-Allow-Origin: *
Content-Type: application/json; charset=utf-8
Content-Length: 199
ETag: "dc196c0835cf5ced44c11600f9f93fe7"
 
[{"error":{"address":"/lights/3/state","description":"parameter, lift, not available","type":6}},{"error":{"address":"/lights/3/state","description":"missing parameter to set light state","type":5}}]
```

Weird, because the lift state IS available on device 3:

```
GET /api/A53342B49D/lights/3 HTTP/1.1
User-Agent: PostmanRuntime/7.28.4
Accept: */*
Cache-Control: no-cache
Postman-Token: 1f9b897b-033f-4dea-97cb-a773583bbdfd
Host: 192.168.0.4
Accept-Encoding: gzip, deflate, br
Connection: keep-alive
 
HTTP/1.1 200 OK
Access-Control-Allow-Origin: *
Content-Type: application/json; charset=utf-8
Content-Length: 370
ETag: "dc196c0835cf5ced44c11600f9f93fe7"
 
{"etag":"dc196c0835cf5ced44c11600f9f93fe7","hascolor":false,"lastannounced":"2021-11-29T05:41:40Z","lastseen":"2021-11-29T08:00Z","manufacturername":"LUMI","modelid":"lumi.curtain.acn002","name":"Dimmable light 3","state":{"bri":0,"lift":0,"on":false,"open":true,"reachable":true},"swversion":"05-18-2021","type":"Dimmable light","uniqueid":"54:ef:44:10:00:2e:12:0d-01"}
```

Arjan-21, 2021-12-06:
> Use the REST API with `state/lift`

Thanks for all your help. read the whole topic about this.

I've the same problem as noted above. Also trying to control the blinds with changing the state.

Arjan-21, 2021-12-07:
When I execute from deconz to go up or down it works and "lift" is changing the value. Same happend when I'm passing manually on the device.

kvanisterdael, 2021-12-13:
@SwoopX : I just installed beta 2.13.03 but it still show up like a dimmable light and the HTTP response still says "parameter, lift, not available".
Could you have another look on what could be wrong?

Mimiix, 2021-12-13:
Did you re pair?

Arjan-21, 2021-12-13:
> Did you re pair?

I did and restart deconz and the pc.

kvanisterdael, 2021-12-14:
Re-pair, restart deconz, updated deconz, re-pair... still no luck

SwoopX, 2021-12-14:
Hm, that now was a serious amount of digging to find anything... Looks like there must be another place taken into consideration for explicit whitelisting as the device doesn't care about any zigbee spec compliance. Again, just 1 line but responsible for the code jumping to and handling the lift item stuff.

kvanisterdael, 2021-12-15:
@SwoopX : thanks a lot! The REST command now works and I can make it move to the correct 'lift' position.

Another question: is it possible to send a STOP command to stop moving immediately via the REST api?

Arjan-21, 2021-12-15:
Do I need to add the code somewhere?
I use deConz now for 2/3 weeks. so I've never done this.

Thanks for your help @SwoopX

SwoopX, 2021-12-15:
@kvanisterdael That should work as well, you might want to try the below
[image]

@Arjan-21 No, not really. Compiling the linked branch and then exchanging the plugin should be sufficient
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing

kvanisterdael, 2021-12-16:
@SwoopX : thanks very much for your help! This works perfectly fine...

Arjan-21, 2021-12-19:
@SwoopX Just updated to 2.13.4 and it works really great. Thank you very much!!

Pete1979, 2021-12-20:
> @SwoopX Just updated to 2.13.4 and it works really great. Thank you very much!!

Did you do anything but update deconz? I just migrated from deconz to ZHA yesterday to try to get this roller to work as intended, dident manage to get it working in ZHA tho. But if all i need to do it use deconz again i guess i can migrate back.

Mimiix, 2021-12-20:
@Pete1979 ZHA != deconz. you need deCONZ for this.

Pete1979, 2021-12-20:
> @Pete1979 ZHA != deconz. you need deCONZ for this.

I know, thats why im thinking of migrating back to deconz, since i could not get the roller working in ZHA.

Arjan-21, 2021-12-20:
@Pete1979 Yes just update to a 2.13.4 (Beta) and now it works with the lift command

sanderlv, 2021-12-20:
Mine broke after update

[s6-init] making user provided files available at /var/run/s6/etc...exited 0. [s6-init] ensuring user provided files have correct perms...exited 0. [fix-attrs.d] applying ownership & permissions fixes... [fix-attrs.d] done. [cont-init.d] executing container initialization scripts... [cont-init.d] firmware.sh: executing... [15:35:34] INFO: GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh Path | Vendor | Product | Serial | Type -----------------+--------+---------+------------+------- /dev/ttyACM0 | 0x1CF1 | 0x0030 | DE2194613 | ConBee II /dev/ttyACM1 | 0x1CF1 | 0x0030 | | ConBee II /dev/ttyUSB0 | 0x0403 | 0x6001 | AO3RXEHS | Generic FTDI [cont-init.d] firmware.sh: exited 0. [cont-init.d] nginx.sh: executing... [cont-init.d] nginx.sh: exited 0. [cont-init.d] novnc.sh: executing... [cont-init.d] novnc.sh: exited 0. [cont-init.d] done. [services.d] starting services [services.d] done. [15:35:34] INFO: Websockify waiting for VNC to start [15:35:34] INFO: Running the OSRAM LEdvance OTA updater... [15:35:34] INFO: Running the IKEA OTA updater... [15:35:34] INFO: Running the deCONZ OTA updater... [15:35:34] WARNING: Your direct VNC access is not protected! [15:35:34] INFO: Starting VNC server (local/no)... [15:35:34] INFO: Starting websockify... WebSocket server settings: - Listen on 127.0.0.1:5901 - Flash security policy server - Web server. Web root: /usr/share/novnc - No SSL/TLS support (no cert file) - proxying from 127.0.0.1:5901 to 127.0.0.1:5900 [15:35:37] INFO: deCONZ waiting for VNC to start [15:35:37] INFO: Starting the deCONZ gateway... libpng warning: iCCP: known incorrect sRGB profile libpng warning: iCCP: known incorrect sRGB profile [15:35:38] INFO: Starting Nginx... 2021/12/20 15:35:38 [notice] 314#314: using the "epoll" event method 2021/12/20 15:35:38 [notice] 314#314: nginx/1.14.2 2021/12/20 15:35:38 [notice] 314#314: OS: Linux 5.10.83 2021/12/20 15:35:38 [notice] 314#314: getrlimit(RLIMIT_NOFILE): 1048576:1048576 2021/12/20 15:35:38 [notice] 314#314: start worker processes 2021/12/20 15:35:38 [notice] 314#314: start worker process 1457 15:35:37:841 HTTP Server listen on address 0.0.0.0, port: 40850, root: /usr/share/deCONZ/webapp/ 15:35:37:854 COM: use stable device path /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2194613-if00 15:35:37:864 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2194613-if00 / serialno: DE2194613, ConBee II 15:35:37:864 ZCLDB init file /data/.local/share/dresden-elektronik/deCONZ/zcldb.txt 15:35:37:890 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin 15:35:39:804 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin 15:35:40:826 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2194613-if00 / serialno: DE2194613, ConBee II 15:35:42:292 Device firmware version 0x26720700 ConBee II 15:35:42:301 unlocked max nodes: 512 15:35:42:414 Device protocol version: 0x010E 15:35:42:496 Current channel 15 15:35:42:524 CTRL ANT_CTRL 0x03 15:35:42:528 CTRL ZDP_RESPONSE handler 0x0001 15:35:42:567 Device protocol version: 0x010E 15:35:42:671 CTRL ANT_CTRL 0x03 15:35:42:674 CTRL ZDP_RESPONSE handler 0x0001 parse error: Invalid numeric literal at line 1, column 34 15:36:37:812 Device TTL 4789 s flags: 0x7 15:37:37:810 Device TTL 4729 s flags: 0x7 15:38:37:812 Device TTL 4669 s flags: 0x7 15:39:37:814 Device TTL 4608 s flags: 0x7

Mimiix, 2021-12-20:
@sanderlv this is not the place to post that kind of stuff.

sanderlv, 2021-12-20:
I don't know since I am the OP and wanted to say that for me this update broke it.

AlmightyCZ, 2021-12-25:
I'm sorry to write here, but I'm not sure where to ask, as I'm not sure if it's a bug. And I also have some features related questions.

I have deCONZ 2.13.4 and the device pairs as a Dimmable light, is this the usual way how are blinds appearing in the deCONZ?

I can still control the device through window covering cluster (lift percentage only), however the device is not reporting current position (always 0). Is it not supported by the device, or is this a bug?

There is also an option for Window Covering mode, but it's read only and set to Drapes. The original aqara app supports some device mode for window blinds, with tilt up/down button, these do a micro movement up or down in order to change the tilt of blinds blades. Is this possible to achieve through API? Or does it has to be done with a up/down+pause+stop? I suppose the tilt options of the Window Covering cluster are not supported as this method of tilting depends on the previous blind movements and has no "tilt level" feedback, right?

ebaauw, 2021-12-28:
Got myself an Aqara Roller Shade Driver E1 (I suppose the EU version) in another attempt to automate my venetian blinds.  I think we can improve on the current support:
- The device advertises as an _On/Off Light_, but the API exposes it as _Dimmable Light_?  I think the API should expose it with a `type` of `Window covering device`.  The _On/Off_ cluster doesn't do anything.
- The _Window Covering_ commands _Open_, _Close_, _Stop_, and _Go to Lift Percentage_ all seem to work (with percentage meaning: percentage open).  Even the _Reversed_ bit in _Mode_ seems to work.  Indeed the _Current Position Lift Percentage_ doesn't do anything.  The API doesn't issue any of these commands, though.  It does configure attribute reporting on _Current Position Lift Percentage_  though.
- The percentage is reported as _Present Value_ in the _Analog Output_ cluster.  The report also includes attribute 0xF000 (u32) with value `0`.  Haven't seen any other values, yet.  Writing _Present Value_ makes the device move; this is what the API issues on setting `state.lift` or `state.open`.
- The actual device state is reported as _Present Value_ in the _Multistate Output_ cluster: `2` for stopped; `0` for moving downwards; and `1` for moving upwards.  This report also includes the 0xF000 (u32) attribute, with different value.  Not sure what these represent.  It doesn't seem to report the position while moving.
- The device supports the _Identify_ commands, but the API doesn't expose `state.alert`.
- The Lumi specific cluster looks interesting.  When setting _Device Operation Mode_ to 0, the device no longer reports _Analog Output_ nor _Multistate Output_.  Various reviews online mention that the device supports different modes in the app (amongst which one for Venetian blinds), I suppose that's some magic setting in this cluster.  Maybe there's a magic value to enable _Current Position Lift Percentage_ or _Current Position Tilt Percentage_, depending on the configuration?
- No _Power Configuration_ cluster, as we've come to expect from Xiaomi devices.  It doesn't report the _Power Descriptor_ either.   I did capture some special attribute report for 0xFCC0/0x00F7, with octec string (0x41) value: `03:28:64:05:21:08:00:09:21:00:01:0a:21:00:00:0b:20:00:0c:20:01:0d:23:19:0e:00:00:11:23:01:00:00:00:64:20:00:65:20:65:66:20:80:67:20:04:69:20:02:6a:21:f9:1f:6b:20:00`.  That looks suspiciously familiar; will try and decode that later.  I suppose battery voltage and/or percentage is in there, as well as charging state.  Battery should be exposed as a separate ZHABattery resoirce; we'll have to think of something to report the charging state.
- Also captured an attribute report for 0xFCC0/0x00EE with u32 value of 3609 (0x00000e19).  No clue what that means.
- I haven't seen any traffic from the (client) _Time_ cluster.  I don't think the device supports on-device schedules, so I was wondering what that cluster is doing here.
- The _OTAU_ cluster seems functional, but without an actual firmware image we cannot know for sure.

ebaauw, 2021-12-28:
Here's my understanding of the special attribute report:
```
                                            1                        2                        3                        4                         5
                     1 2  3  4 5  6 7  8 9  0 1  2 3  4 5  6 7  8  9 0  1  2 3  4 5 6 7  8 9  0 1 2 3  4 5  6  7 8  9  0 1  2  3 4  5  6 7  8  9 0  1 2  3 4  5
         attr tp  l    temp   RSSI dB              parent       ?       ?             ?             ?    lift       ?       ?       ?       ?         ?       ?
         ---- -- -- ------- --------- --------- --------- ------- ------- ------------- ------------- ------- ------- ------- ------- ------- --------- -------
         00f7 41 37 0328 19 0521 0800 0921 0001 0a21 0000 0b20 00 0c20 01 0d23 190e0000 1123 01000000 6420 00 6520 62 6620 80 6720 04 6920 02 6a21 f51f 6b20 00
e1:              55    25°C  u16       u16       u16        u8      u8     u32           u32            u8      u8      u8      u8      u8     u16        u8
```

It is already parsed by the REST API plugin, although not yet fully understood:
```
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:344 APS-DATA.indication srcAddr: 0x4564, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xFCC0, lqi: 255, rssi: -44 
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:345         asdu: 1c5f11e00af700413703281905210800092100010a2100000b20000c20010d23190e00001123010000006420006520626620806720046920026a21f51f6b2000
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:345 APS-DATA.indication from child 0x4564
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:346 Node data 0x54ef44100033af31 profileId: 0x0104, clusterId: 0xFCC0
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:348 0x54EF44100033AF31 extract Xiaomi special attribute 0x00F7
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:348         03 Device temperature 25 °C
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:349         05 RSSI dB (?) 8 (0x0008)
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:349         09 unknown 256 (0x0100)
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:350         0a Parent NWK 0 (0x0000)
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:350         0b unknown 0 (0x00)
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:351         0c unknown 1 (0x01)
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:351         0d unknown 3609 (0x00000E19)
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:352         11 unsupported tag (data type 0x23)
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:352         64 lift 0 (100%)
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:353         64 smoke/gas density 0 (0x00)
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:353         65 unknown 98 (0x62) 
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:353         66 unknown 128 (0x80) 
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:354         67 unsupported tag (data type 0x20)
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:354         69 unsupported tag (data type 0x20)
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:355         6A unsupported tag (data type 0x21)
Dec 28 12:32:17 pi5 deCONZ[22342]: 12:32:16:355         6B unsupported tag (data type 0x20)
```
When providing power to the USB-C port, the device sends a report for 0xFCC0/0x0409 with u8 value 0x01.  When disconnecting the power, or when charging has finished (as witnessed by the light turning off), it sends value 0x02.  I think this corresponds to tag 69 in the special attribute report.

According to https://notenoughtech.com/home-automation/aqara-roller-shade-driver-e1-zigbee2mqtt/, the device contains a 7.4V 1000mAh battery.  Typically Xiaomi reports battery voltage using tag 01 with a u16 value in mV.  The only value that comes close would be tag 6A, suggesting 8.247V for a fully charged battery.  Seems a bit high, but I'll monitor it for a couple of days.  Good thing: you can force a special attribute report by briefly pressing the reset button.

ebaauw, 2021-12-28:
> Also captured an attribute report for 0xFCC0/0x00EE with u32 value of 3609 (0x00000e19). No clue what that means.

Which corresponds to tag 0d.

ebaauw, 2021-12-29:
Still todo: figure out battery voltage or percentage.

I ordered a Xiaomi hub, hoping to be able to sniff its interaction with the E1.

ebaauw, 2021-12-29:
A bit late for Xmas, but got my Aqara M1 hub today.  My sniffer is now working overtime ;-)

Aqara app uses _Present Value_ on _Multistate Output_ cluster to send commands:
- On _Stop_ it writes value 2 (which we already saw reported when the driver stopped moving;
- On _Rotate Up_ (step up), it writes value 6.  On _Rotate Down_, it writes value 5.  The driver response with a _Write Attributes Response_ (obviously), followed by a _Report Attributes_ of _Multistate Output_'s _Present Value_ with the value just written, followed by _Report Attributes_ of _Multistate Output_'s _Present Value_ with value 2, followed by a _Report Attribute_s of _Analog Output_'s _Present Value_ with the new position.

This calls for write-only `up` and `down` commands (cf. `stop`).

Changing the Curtain Type in the app doesn't do anything.  This is veneer only in the app.

Setting _Reverse direction_ writes bool attribute 0xFCC0/0x0400.  Not much value to expose this over the API, as you would only set this on installation.  Tag 66 changes value from 0x80 to 0xC0 when reversed is set.  This also inverts the current position.  Smells like a bitmap.

Setting positions:
- On clearing the current position, it writes false to bool attribute 0xFCC0/0x0402;
- On setting the highest position, it writes 1 to u8 attribute 0xFCC0/0x0407;
- On setting the lowest position, it writes 2 to  u8 attribute 0xFCC0/0x0407;

Again, you would only set these on installation, not much value exposing this over the API.  Damn those heretics that don't run the GUI.

Adjusting the speed writes u8 attribute 0xFFC0/0x0408.  2 = High, 1 = Medium, 0 = Low.  This doesn't seem to be reflected in the special attribute.  Could make sense to expose over the API (e.g. slower = less sound at night), preferably through a `config.speed`, but `/lights` resources have no `config` object.

The app reports battery percentage and charging state.  Battery now reports 99% after some playing around.  Latest special attribute value is:
```
03:28:14:05:21:01:00:09:21:00:01:0a:21:00:00:0b:20:00:0c:20:01:0d:23:19:0e:00:00:11:23:01:00:00:00:64:20:64:65:20:63:66:20:80:67:20:09:69:20:02:6a:21:f5:1f:6b:20:00
```
During charging, it's, still 99%:
```
03:28:14:05:21:01:00:09:21:00:01:0a:21:00:00:0b:20:00:0c:20:01:0d:23:19:0e:00:00:11:23:01:00:00:00:64:20:64:65:20:63:66:20:80:67:20:09:69:20:01:6a:21:f5:1f:6b:20:00
```
As the charging light goes off, the app reports 100% and the driver sends:
```
03:28:14:05:21:01:00:09:21:00:01:0a:21:00:00:0b:20:00:0c:20:01:0d:23:19:0e:00:00:11:23:01:00:00:00:64:20:64:65:20:64:66:20:80:67:20:09:69:20:01:6a:21:7c:20:6b:20:00
```
This looks too good to be true, but it would seem tag 65 is the battery percentage!  No voltage to percentage conversion needed.  6a also changes value - still thinking this might be the battery voltage, but with 65 reporting the percentage, we don't care.

ebaauw, 2021-12-29:
And the device supports OTA firmware upgrades.  The Aqara Home app is upgrading it from 0.0.0_1425 to 0.0.0_1427.  Not sure where they got the firmware file from, or how the version is reported over Zigbee.

ebaauw, 2021-12-29:
Hm. looks like the new firmware no longer reacts to the _Window Covering_ commands.  It does send a _Deafult Response_ with status _Success_, but nothing happens.

[25 more replies]