dresden-elektronik/deconz-rest-plugin #5331: Tuya Door Magnetic Sensor Model ZM-CG205

FHEMUser, 2021-09-30:
## Device

- Product name: Zigbee Door Sensor
- Manufacturer: Tuya ?
- Model identifier: ZM-CG206
- Device type : Sensor

## Screenshots

[image]
### Basic
[image]

### Identify
[image]

### Other clusters that are not mentioned above
[image]

[image]

onebavarian, 2021-11-07:
Hello, is there any update on that or has anyone a working DDF file? 
With beta 2.13.01 it gets recognized as _TZ3000_cn58yln1 with Model ID TS0203. 
But somethings wrong, it doesn`t work.

Smanar, 2021-11-07:
All the discution is here ^^

Can you make a try with DDF file ?

onebavarian, 2021-11-07:
I have no idea what I am doing but I'm trying.

Right now the state open/closed is working with Phoscon and HomeAssistant and after 10 minutes battery state too.

I edited the DDF Phoscon showed me for device _TZ3000_cn58yln1.
Copied and pasted lines 9 to 129 from https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/wiser/contact_sensor_cct591011_as.json into "Preview", overwriting the original lines.
Changed under "state/open" the following values to:   
 "at": "0x0002",
 "cl": "0x0500",

Right now, I'm not sure if "Model ID" and "Product" should be "ZM-CG205" and what I should use for "Manufacturer Name"

I attached the DDF File as txt.

[contact_sensor_thi_tuya.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7492706/contact_sensor_thi_tuya.txt)

Smanar, 2021-11-07:
Model id  : TS0203
manufacturername : _TYZ ..... (make drag and drop from the attribute 0x0004 on the 3th capture.

I m surprised it have worked with bad values ....

onebavarian, 2021-11-07:
Changed:
"manufacturername": "_TYZB01_xph99wvr",
"modelid": "TS0203",
"product": "TS0203",

[contact_sensor_thi_tuya.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7492749/contact_sensor_thi_tuya.txt)
 
I am surprised too.

Smanar, 2021-11-07:
The DDF seem fine, it s not working ?
The device still have 0xXXXX as name on the GUI ?

onebavarian, 2021-11-07:
The DDF is working - added 2 more devices and it shows me "OpenClose" and a number. 

I'm happy right now.

Smanar, 2021-11-07:
NP, thx for the DDF, will help other users.

FHEMUser, 2021-11-08:
I load the DDF file to the device and a hit "hot reload".
But what is the next step ?
I do not see new devices in the phoscos app.

Smanar, 2021-11-08:
The device can be invisible in phoscon.
If the name change 0xXXXX to "OpenClose X" it s ok.

onebavarian, 2021-11-08:
With one device I had to pair it again, then it worked.

FHEMUser, 2021-11-09:
I have 3 items of this.
At the start you see one device have Basic Cluster, now no device have this Basic info.
Manufacturer and Model Identifier always unknown.
I use FHEM as Homeautomation and the device must be visible on Phoscon to use it.

I don´t understand why all devices have no basic info.
I delete all, do a new pairing with long press (10 seconds) and i see the new device only with no basic info.

[image]

FHEMUser, 2021-11-09:
Hey it works !
I hit the rest pin long (10 seconds) and the LED flash. I think thats a reset
Then i hit the pin again until the LED flash (5 seconds) and the device will be created.
Now i have the basic info.

Then again reset an pair and the device will show on phoscon !
I think reset and pair must be do more than 1 time.

Now i am happy again. The aqara devices are more and more expensive. This tuya devices are some cheaper.
Ok, i add the other two and will test it !

Thanks !

onebavarian, 2021-11-09:
Just wanted to write the same. :) 

With my third device I had to pair again without deleting it first. I think the first 2 worked because I paired them mutliple times. 

I'm glad it worked!

FHEMUser, 2021-11-10:
> Just wanted to write the same. :)

Do you see the devices in phoscon app ?

onebavarian, 2021-11-10:
Yes, I see the devices in Phoscon. (I edited the numbers)
[image]

I have no idea about the refresh.interval. As I said, I just merged 2 files and it worked. 

> Right now the state open/closed is working with Phoscon and HomeAssistant and after 10 minutes battery state too.
>
> I edited the DDF Phoscon showed me for device _TZ3000_cn58yln1. Copied and pasted [...]

lazarus72, 2021-12-06:
Hello FHEMUser
Hello onebavarian

i also try to get this sensor connected, but i don't know ?
but I do not know what this ddf file is and where I can find it?
I tried several times with longer press the pin and then again. i get nothing displayed.
My version is 2.13.02 / 10.11.2021
@Smanar : will this sensor maybe be implemented in the next version? Because I have 12 of these :-)

best regards

FHEMUser, 2021-12-07:
You must set to normal mode:

“Panels → Control → Normal mode” needs to be enabled.

This must be set every restart.
Then right Klick on the devices and "edit DDF".
Open the file that's posted here in the new window. Then hit "hot reload" on files menu or something in this new window. Close the window.

Now learn again the devices. I must do learning 2 times. Then it will show. In the phoscon app I don't see the new devices. But I don't need this.

Take a look here
https://forum.phoscon.de/t/door-magnetic-sensor-zm-cg205-by-tuya-paired-but-not-work/546/10

lazarus72, 2021-12-07:
:-( i only have headless mode and i can't get the gui to work after the installation.
is there another way? directly in the console?

Smanar, 2021-12-08:
Nope, else you can recompile the code with some personnal change, but it s more work to do.

lazarus72, 2021-12-09:
hmm, i can'T connect to the gui.
i try to disable and stop deconz
and the to enable and start deconz-gui
but when i start deconz-gui nothing works anymore, no phoscon, i get an connection error message inside my domoticz and no switch is working.
also the vnc connection didn't work

what can i do?

@Smanar: is it foreseeable when this sensor is implemented in deconz?

Smanar, 2021-12-09:
>also the vnc connection didn't work

You need to have a desktop to use the GUi version, if not it s probably for that VNC don't work too.

>is it foreseeable when this sensor is implemented in deconz?

I have perhaps a workaround, I m asking to be sure, but it seem on last version, the v12.3.3-beta, even with all by defaut DDF will be enabled for gold status.
So in theory, you just need to set the status to "gold" and put the file in the good directory.

lazarus72, 2021-12-09:
what does this mean:  set the status to "gold" and put the file in the good directory.?

Smanar, 2021-12-10:
For the status https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
And if it not work there is a way to force DDF be nabled with command line https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-command-line-parameters#dev-test-managed

lazarus72, 2021-12-30:
Hi Smanar, sorry for the late feedback. I hope you had some pleasant holidays.
Thanks for your support and tips. I got it working now thanks to your help.
I installed the xserver and the session manager, and after adjusting the resolution I finally had access via VNC.
After stopping deconz headless I started the interface with deconz-gui.
After that, with the tips from you, FHEMUser and onebavarian, I was able to learn the sensors quite easily.
Thanks again to all of you and a happy new year

Smanar, 2021-12-30:
NP, and happy Holidays too :)

crazykiwi2003, 2022-06-08:
Will that device be included in one of the next Deconz/Phoscon releases so it can be paired without the DFF-workarounds?

closed 2022-06-08 by Mimiix

crazykiwi2003, 2022-06-09:
I would like to try, but unfortunately I have no idea how I can… :-/

Smanar, 2022-06-10:
What is your OS ?
You just need to create a text file in a "devices" folder and call it whatyouwant.json

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

crazykiwi2003, 2022-06-20:
Thanks for the explanation. I just tried - unfortunately it does not work. Sensor cannot be found...