dresden-elektronik/deconz-rest-plugin #5334: New device: Tuya smart plug 16a

tomvanderduin, 2021-09-30:
### Device
Product name: Tuya smart plug 16a
Manufacturer: _TZ3000_dpo1ysak
Model identifier: TS011F

Device type : Smart plug with power metering

### Screenshots
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2021-09-30:
It's possible this PR was enought to enable the power measurement for it, if you can make a try ?
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5206/

hcpalka, 2021-09-30:
I have the same device and installed it in home assistant -but no energy metering sensor was  
created - any help or ideas how to remedy this?

jfparis, 2021-10-05:
@Smanar I built and install the PR and it made no difference
* Switched are recognised as a "light"
* power statistics are not available

Smanar, 2021-10-05:
Can you share the device JSON ?
You can have it in http://phoscon.de/pwabeta Help / API information

Smanar, 2021-10-10:
Have added the mac adress for the first one 
>const quint64 silabs3MacPrefix    = 0xec1bbd0000000000ULL;

But your is already in code
>const quint64 YooksmartMacPrefix  = 0x84fd270000000000ULL;

Can you check if your manufacture number is 0x1002 too ? in node info panel.

jfparis, 2021-10-10:
Manufacturer code is 0x1002.

Just to make sure I test properly
* I pull the branch https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5206
* Compile and install 
* Delete the device in the Phoscon web interface
* repair it 

Should I see the power reading in the Phoscon interface as well ?

Smanar, 2021-10-10:
>Should I see the power reading in the Phoscon interface as well ?

No, I don't think Phoscon display power measurement device, but like you have done previously is fine, take a llok in the json direclty.

For me all is fine, the device is in whitelist

```
{ VENDOR_EMBER, "TS011F", YooksmartMacPrefix }, // Tuya Plug Blitzwolf BW-SHP15
VENDOR_EMBER = 0x1002
YooksmartMacPrefix  = 0x84fd270000000000ULL;
```

Smanar, 2021-10-11:
The code is now in the master branch, just validated, and yes the deconz-dev package is updated ATM because ddf file

diibify, 2021-11-05:
I have a `_TZ3000_cphmq0q7` Is it also in it?

febalci, 2021-11-11:
As i see from zigpy, [they also handled ](https://github.com/zigpy/zha-device-handlers/pull/1057/commits/39d9ebf9362f7de1afb91522e64fc2ff80bfc509) both the restore of powerOnOff and power readings. Is there a way to create a DDF file for this?  I am using  _TZ3000_ew3ldmgx and only On/Off works on this. Thanks...

Smanar, 2021-11-12:
The power reading is workng on last version on ?
For the restore of poweronoff can try this edition on the xml file

Just after
```
            <attribute id="0x8001" name="Light mode" type="enum8" default="0" required="m" access="r">
                <value name="Mode 1" value="0"></value>
                <value name="Mode 2" value="1"></value>
                <value name="Mode 3" value="2"></value>
            </attribute>
```

Add
```
            <attribute id="0x8002" name="Power on state" type="enum8" default="0" required="m" access="r">
                <value name="Off" value="0"></value>
                <value name="On" value="1"></value>
                <value name="Last state" value="2"></value>
            </attribute>
            <attribute id="0x8004" name="Switch mode" type="enum8" default="0" required="m" access="r">
                <value name="Command" value="0"></value>
                <value name="Event" value="1"></value>
            </attribute>
```

If it work pls make a PR.

febalci, 2021-11-12:
@Smanar, thanks for showing me the way. 
I use deconz 12.3.2.beta. My device is a TS011F with manufacturer code  _TZ3000_ew3ldmgx. I did a re-inclusion; still only on/off works. Power readings 0x505, 0x508 and 0x50B are all '0' all the time. Although i do not know the refresh period of these values i happened to test it for short periods. I am gonna test it hourly tomorrow.

This works perfectly for the Power On State:
`            <attribute id="0x8002" name="Power on state" type="enum8" default="0" required="m" access="rw">
                <value name="Off" value="0"></value>
                <value name="On" value="1"></value>
                <value name="Last state" value="2"></value>
            </attribute>`

I had to change access="r" to access="rw" in order to be able to change the setting. Although i didn't had any use for 0x8004 - "Switch Mode" for this particular plug, i put it there anyway.

Smanar, 2021-11-13:
>i didn't had any use for 0x8004 - "Switch Mode" for this particular plug, i put it there anyway.

As you want it's your PR, but it was for other devices, it s a same place, it's just to avoid making 2 PRs.

jasimancas, 2021-11-19:
Same error here with _TZ3000_dpo1ysak
[image]

jasimancas, 2021-11-20:
I have used the DDF files from the latest beta version and got it to report measurement values, but I have a problem.
In the cluster (Wh):
[image]
In Node-red/HA:
[image]
[image]

480 vs 4800.

Is it possible to solve this directly from the DDF?

It seems to return 4 digits instead of 3, which should be "0482" instead of "4820".

thomas-svrts, 2021-11-20:
Are you sure?
The value 482 reported by Tuya means 4,82kWh or 4820 Wh.

jasimancas, 2021-11-20:
They seem to me to be very high values, for the tests I have been doing with it yesterday and today.

But for values, it is possible that it is what you say.

Any way to test it to find out for sure?

Smanar, 2021-11-20:
And yes it s possible, you have a field "Expression" with something like "Item.val = Attr.val" you can add a *10 or / 10.

oywino, 2021-11-22:
I just received the Tuya 16A Smart Plug and connected it to Conbee II and deConz 2.13.01 and HA 2021.11.5
This is my device ID:
[image]
The electric measurement is present and correct in deConz:
[image]
But does not exist in HA. Instead, I note that the same error (phantom device registrations) appear with this new Tuya Smart Plug as with the old Tuya Smart Plug model:

Old Plug (the big one):
[image]

New Plug (with the yellow collar):
[image]

As you can see, the new (yellow) plug does not offer any electric measurement data, but it suffers from the same number of phantom switches.  This seems to be a consistent issue with all Tuya devices.  Av have seeral other Smart Plugs, (Develco, Aqara, Apex) and none of them have such issues.
I wonder what is it that Tuya does, to make so much trouble (incompatibilities) ?

Smanar, 2021-11-22:
There is a strange issue on this device ATM https://forum.phoscon.de/t/new-tuya-smart-plug-nas-wr01b-or-tz3000-w0qqde0g/819/13

It seem reporting is not working.

oywino, 2021-11-23:
Perhaps a stupid question, byt why is it that electric measurement data is not present in the event message from the REST API, event though it does show in HA ?

```
{
    "17:31:13:495": {
        "attr": {
            "id": "45",
            "lastannounced": "2021-11-20T22:07:00Z",
            "lastseen": "2021-11-23T16:31Z",
            "manufacturername": "LUMI",
            "modelid": "lumi.plug.maeu01",
            "name": "Smart plug Aqara",
            "swversion": "09-10-2019",
            "type": "Smart plug",
            "uniqueid": "04:cf:8c:df:3c:82:39:9c-01"
        },
        "e": "changed",
        "id": "45",
        "r": "lights",
        "t": "event",
        "uniqueid": "04:cf:8c:df:3c:82:39:9c-01"
    },
    "17:31:48:306": {
        "e": "changed",
        "id": "45",
        "r": "lights",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "t": "event",
        "uniqueid": "04:cf:8c:df:3c:82:39:9c-01"
    }
}
```

[image]

I was expecting to be able to see all data in the Event stream ?

Smanar, 2021-11-23:
It s because the value is not updated in the API, and I think the api is configured to send only changed value on your side.
You can try to look direclty in the api http://phoscon.de/pwabeta in help / api information.

oywino, 2021-11-23:
That's exactly what I was doing. The above event stream was picked up from the _help / API_ information.

Smanar, 2021-11-23:
Lol, oups, sorry, haven't reconized it ^^.

But it s the notification, according to your setting it can send only changed value, on same place you can see the complete json, And remember the Consumption is a "sensor", the one on your post is the "light" device. You have at least 1 light + 1 sensor (or more) for your device.

oywino, 2021-11-23:
Aha - to my big surprise, it does show up as a sensor:
//192.168.XX.XX:6080/api/CC04FEXXXX/sensors/32

```
{
    "config": {
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "388765bb6598093b31862ede4f1fc8ec",
    "lastannounced": null,
    "lastseen": "2021-11-23T21:28Z",
    "manufacturername": "_TZ3000_cphmq0q7",
    "modelid": "TS011F",
    "name": "Consumption 32",
    "state": {
        "consumption": 8650,
        "lastupdated": "2021-11-23T21:28:32.503",
        "power": 0
    },
    "type": "ZHAConsumption",
    "uniqueid": "84:71:27:ff:fe:8d:96:1d-01-0702"
}
```

However, this "sensor" is not listed in Phoscon under "Sensors"  !
And of course, no sensors are visible in Phoscon for the Tuya Plug - although they are visible in deConz.
You did "something" with the configuration to make this work for the "old" Tuya Plug. 
Now the problem is back again for this "new" version of the Tuya Plug.  
Will it be like this every time ?  and only for Tuya products ?

Can you fix this also, just like you did for the other Tuya plug ?

Smanar, 2021-11-24:
But ZHAConsumption are never displayed in phoscon. I don't understand what is the issue ?

oywino, 2021-11-24:
That was my mistake, I'm sorry for the confusion.
You answered my question perfectly.

The remaining "issue" is how to get electric measurements from the "new" Tuya Plug visible in HA.
I just hope this can be fixed ?

Smanar, 2021-11-24:
Ha the link above ?

I realy don't have idea for the moment, and it s same code for all tuya plug, so this one have something specal, but no idea yet.

oywino, 2021-12-10:
After updating deConz to 2.13.02 the new Tuya works as expected with electric measurements.  Moving onwards to deCONZ 2.13.03 and it still works 😊

However, my new Tuya Power strips fails to provide correct electric measurments.
[image]
https://www.aliexpress.com/i/4001148859962.html

This is what is shown in the API:
```
{
    "18:44:14:416": {
        "config": {
            "on": true,
            "reachable": false
        },
        "e": "changed",
        "id": "38",
        "r": "sensors",
        "t": "event",
        "uniqueid": "60:a4:23:ff:fe:80:2b:dd-01-0b04"
    },
    "18:45:01:575": {
        "attr": {
            "id": "38",
            "lastannounced": null,
            "lastseen": "2021-12-10T17:45Z",
            "manufacturername": "_TZ3000_o005nuxx",
            "modelid": "TS011F",
            "name": "Power 38",
            "swversion": null,
            "type": "ZHAPower",
            "uniqueid": "60:a4:23:ff:fe:80:2b:dd-01-0b04"
        },
        "e": "changed",
        "id": "38",
        "r": "sensors",
        "t": "event",
        "uniqueid": "60:a4:23:ff:fe:80:2b:dd-01-0b04"
    }
}
```
Why is a device listed as "unreachable" and still it produces an output (?) Doesn't the fact that the output triggers the API imply that the device somehow is reachable?

Here's the node info:

[image]

0000 Basic Cluster info:
[image]

0006 On/Off cluster:
[image]

This is the cluster list for 1 of the 4 identical output sockets.
[image]

It's interesting to notice that there is no information anywhere about electrical measurements. Still, two entities shows up both in the API as well as in HA:
[image]

Is it somehow possible to 'fix' this so that electric measurements becomes usable?
Many thanks in advance.

Smanar, 2021-12-10:
>Why is a device listed as "unreachable" and still it produces an output (?) Doesn't the fact that the output triggers the API imply that the device somehow is reachable?

Yes, but I have the impression seing this issue all the time ATM. But the code is moving a lot.

Now the bad news, this device don't use classic zigbee cluster for power and consumption, IDK how you can have them in the json. Its totally new cluster, not used by deconz yet.

oywino, 2021-12-10:
So not even DDF could solve this?

Smanar, 2021-12-11:
No, need firt to sniff the traffic to find how work this new cluster (if it s not already done on other project) them add it in deconz code.

oywino, 2021-12-11:
Are you using Google Translate??
Your sentence makes no sense at all. 
What does "firt" mean?
"them add in"  what does that mean?

Smanar, 2021-12-11:
Lol, not but a bad english.

was "first", and the rest mean we need to add something in the deconz core to manage this cluster.

oywino, 2021-12-11:
Wasn't the main purpose of the DDF to allow adding almost everything - via end-user configuration ?
Thus, no longer the need to make more PR for core modification and recompiling?  Or am I wrong?

Smanar, 2021-12-11:
>Wasn't the main purpose of the DDF to allow adding almost everything - via end-user configuration ?
Thus, no longer the need to make more PR for core modification and recompiling? Or am I wrong?

Yes exactly, but the support in DDF core need to be added feature by feature, the next step will be the battery switches, then the tuya cluster.

If this device use "normal working mode", you can use DDF file even with a not supported cluster in deconz. But for that you need to know with one cluster is used 0xE000 or 0xE001 and wich one attribute need to be read.
If the device use the same working mode than the tuya cluster, it will be not possible yet using DDF.

torandreroland, 2021-12-12:
I added a DDF for Neo NAS-WR01B (i.e. model TS011F with manufacturer __TZ3000_w0qqde0g_) in [this pull request](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5577).

JustNZ, 2021-12-14:
@torandreroland the ddf worked - thanks

but in Home Assistant I can see a refresh rate for example on the power consumption of 5min

oywino, 2021-12-14:
I can't help it, but I have a distinct feeling that DDF could be the magic wand that can solve my problem. 
The problem is that I hardly understand anything about what, how and why when it comes to DDF.
You guys seem to understand exactly how to exploit DDF.  Could you point me in the right direction to something readable to begin my learning process?

JustNZ, 2021-12-14:
@oywino 
I personaly don't know that much about ddf. The only thing I know is that they point the data from those devices to the right directions.

For Installing:
I copied @torandreroland ddf file to /usr/share/deCONZ/devices/neo/NAS-WR01B.json and readded my devices - and it worked

oywino, 2021-12-14:
What exactly do you mean by "and added my device" ?
How did you add your device to his DDF file?
Were your device identical to his?

JustNZ, 2021-12-14:
So for me I don't had to assign this ddf to my devices - deCONZ knows that this ddf exists and that my device can/should use it

oywino, 2021-12-14:
I dont' understand (?)
First you write that you did add your device, and now you write that you did not have to add your device ?!
Could you please provide an explanation that is possible to understand?

JustNZ, 2021-12-14:
sorry for the confusion - what I meant with adding was pairing it to my conbeeII

So if already paired - remove and repair after placing the ddf in the correct directory

oywino, 2021-12-14:
Ok, I see.
But the DDF should be edited to match the device with the correct settings, right?
Or did you have an identical device to @torandreroland so no editing was needed?

JustNZ, 2021-12-14:
I did not edit anything - I copied the file like it is and it was working right away

In this ddf is the manufacturername set which matches with my smart-plug
`"manufacturername": "_TZ3000_w0qqde0g",`

oywino, 2021-12-14:
So perhaps just changing the manufacturer name is enough ?

JustNZ, 2021-12-14:
What you have do to is:
1. remove the device if already paired
2. place the ddf /usr/share/deCONZ/devices/neo/NAS-WR01B.json
3. what I also did is changed deconz to hybrid mode (don't know if that is required)
4. pair your device again

JustNZ, 2021-12-14:
yes and also the `"modelid": "TS011F",`

oywino, 2021-12-14:
Exactly, so the steps should be:

1. remove the device
2.  edit the manufacturer name and model-ID in the DDF file
2. place the DDF in  /usr/share/deCONZ/devices/neo/NAS-WR01B.json
3. pair the device again

But how do you change deCONZ to "hybrid" mode ?

JustNZ, 2021-12-14:
for step 2. I'm not sure - as I said I'm not an expert and never created/edited any ddf 

for the hybrid mode:
1. open deconz (vnc)
2. click on panel and control
3. select hybrid (will be changed back to default after a reboot)

oywino, 2021-12-14:
What made you think that "hybrid" was needed, and what does it actually mean?

torandreroland, 2021-12-18:
Hi! Sorry for not responding for a while.

I think you need the latest beta version of the deconz-rest-plugin (see https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.13.3-beta) because of changes to DDF not least to which DDF-statuses are loaded automatically (see https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5565).

For the time being the DDF has a status of silver, meaning that it's tested but more testing should be done. You can set this up in the Control-panel in the Deconz GUI. If that is too complicated, you can also edit the status yourself in the json from silver to gold, and then on v2.13.3-beta of the deconz-rest-plugin, the DDF should be loaded automatically.

Glad to hear that it works for some at least.

Off topic, I got a "lemon" of the smart plug with a range of about 10 centimeter, so I decided to remove it from my setup.

oywino, 2021-12-18:
> And BTW, are you sure this device support power measurement ? it s not present in z2m too https://www.zigbee2mqtt.io/devices/SM-SO306EZ-10.html

Officially it doesn't support electrical measurements. That's why I was somewhat surprised to see that the following sensor is reported:
```
{
    "config": {
        "on": true,
        "reachable": false
    },
    "ep": 1,
    "etag": "ed437d31b36cae52190012d354a502c4",
    "lastannounced": null,
    "lastseen": "2021-12-18T10:25Z",
    "manufacturername": "_TZ3000_o005nuxx",
    "modelid": "TS011F",
    "name": "Consumption 39",
    "state": {
        "consumption": null,
        "lastupdated": "none",
        "power": null
    },
    "type": "ZHAConsumption",
    "uniqueid": "60:a4:23:ff:fe:80:2b:dd-01-0702"
}
```

And I have more than one power strip, and they all report the same sensor but with different names (Consumption XX).
But as you can see, they are all reported as "unreachable".  So these sensors can be seen in HA, but as unreachable.

Smanar, 2021-12-18:
if your device is this one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5334#issuecomment-991181369

The sensor created are not for the one n the capture, not the same mac adress

If the device have the good cluster, the sensor will be created, even not working, but without report it will be marked as "unreacheable" naturally.

oywino, 2021-12-18:
As I said: I have three identical units.  Of course they have different MAC-adresses.

Smanar, 2021-12-18:
Ok, so your device don't have cluster 0x0702 and 0x0b04 but still create the sensors entries, you have  a special setting on DDF stuff side ?

oywino, 2021-12-18:
I have not done any DDF stuff at all (yet).
And I cannot find any cluster info to explain how they seem to have power and consumption sensors. But they do!

Smanar, 2021-12-18:
I have checked the code again, no clue, perhaps clusters are hidden but the device use them at a moment ...

oywino, 2021-12-18:
So there's nothing to be done?
We just have to accept that these "orphan" sensors exist, but are unreachable?

[39 more replies]

closed 2022-03-19 by github-actions[bot]