dresden-elektronik/deconz-rest-plugin #5337: Device name

majorsl, 2021-10-02:
## Device

- Product name: THIRDREALITY Zigbee Motion Sensor, Pet Friendly 3RMS16BZ
- Manufacturer: unknown
- Model identifier: unknown
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots

### Basic
[image]

### Other clusters that are not mentioned above
[image]

[image]
[image]

stacecom, 2021-10-03:
Very interested in this myself.

majorsl, 2021-11-16:
Ping.

Smanar, 2021-11-22:
Hello, can you show too the cluster list ?

majorsl, 2021-11-23:
> Hello, can you show too the cluster list ? To see if it s a classic sensor.

Is this it?
[image]

majorsl, 2021-11-23:
Node Info too:
[image]

Smanar, 2021-11-24:
Its realy look classic. Will be a small/fast PR.

I can't promise it will be merged, but if you have a linux machine with full OS, I can make a code modification ?

majorsl, 2021-11-24:
> Its realy look classic. Will be a small/fast PR.
> 
> I can't promise it will be merged, but if you have a linux machine with full OS, I can make a code modification ?

I run deconz in Docker.  That's okay, if it gets merged for the next release or even a stable beta, I can wait.  I appreciate that I'll get to use it someday and get some more as needed.

Smanar, 2021-11-25:
The PR is made, it s realy a small change, but untested and you know the situation for new device.

But it will be easy to add it with DDF file, when the core will be able to support IAS cluster (Swoop have already made PR)

majorsl, 2021-12-17:
Ping.

Mimiix, 2022-01-04:
@Smanar can't we use DDF?

Smanar, 2022-01-05:
Heu, yes it s possible, even there is already a Pr in waiting list.
If @majorsl wana remake a try using DDF ?

majorsl, 2022-01-05:
I honestly would have no idea where to even begin to create a DDF.

Smanar, 2022-01-06:
Can try this one https://pastebin.com/Pjt9jSSY

Not tested, made with editing another sensor, so I can be wrong, but not far way the reality ^^

In the code there is

`  "path": "/devices/3RMS16BZ.json",`

so use this name for the json file

majorsl, 2022-01-06:
I'll give it a try this weekend.  I assume with Docker that these DDFs are inside the container?

Smanar, 2022-01-07:
Yep HA have updated the deconz version.

majorsl, 2022-01-08:
Any idea how I can get into my running container? I usually do docker exec -it (container has) /bin/bash or whatever the shell is.  I tried bash, tcsh, zsh, fish... of course, I'm assuming the container even has a shell bin as built.

Smanar, 2022-01-08:
Ha, to edit the text file ?
I never use docker for this reason, I m not engineer, you are searching way to create and edit a file ?

majorsl, 2022-01-08:
Yes. Since docker containers are self-contained operating systems, you need to "log in" to the contain as you would any remote system.  You just have to tell docker the shell to use that the container is expecting. Once logged in, I can test the DDF by editing the file in the expected location. It could be this container isn't using a shell or not allowing it, in which case I will not be able to test anything until the next release if the DDF is included.

majorsl, 2022-01-08:
Okay, I got in with just a simple /bin/sh.  What is the proper path for these DDF files?  I found three "/devices" directories in the container:

/opt/deCONZ/devices - this directory is empty.

/usr/share/deCONZ/devices - this directory contains:
button_maps.json  dresden_elektronik  generic  ikea  immax  innr  lutron  philips  samsung  tuya  wiser  xiaomi

/usr/share/deCONZ/webapp/pwa/js/devices - this directory contains:
constants-installcode.js     installcode-functions.js  machine-actions.js  scan-actions.js
constants.js		     load-page.js	       machine.js	   scan-functions.js
generic-helper-functions.js  load-svg.js	       make-machine.js	   scan-states.js

Smanar, 2022-01-08:
`/opt/deCONZ/devices`
I think this one is the better

`/usr/share/deCONZ/devices`
This one will work, but can change at every deconz update

Can put the file on both.
You will have the path written on the window title when selectiong the node then right clic "edit DDF"

majorsl, 2022-01-09:
I was able to add the file to the devices. I re-paired the sensor and looked at the Node, edit DDF, and the new DDF was there. So, it did find the scheme.

Unfortunately, it does not appear in Phoscon which is likely why it does not appear to Home Assistant either.

Mimiix, 2022-01-09:
It would show in ha if it works. Phoscon needs to add own support.

Smanar, 2022-01-10:
ha ?
And how was the node name ? 0xXXXX or a normal name ?
I can be wrong in the DDF, I have made it with a text editor.

majorsl, 2022-01-10:
Some screenshots that might be helpful:
[image]
[image]
[image]
[image]

Smanar, 2022-01-10:
All seem fine, what is the node title ? The same thing like that

[image]

It's something like 0xXXXX or a real name ?
Have you tried "Hot relaod" ? in the menu of the DDF editor.
If it s still not working can you check in log when make the "hot reload" with the flag "ddf" ?

majorsl, 2022-01-10:
The node title looks like your screenshot (no real name).

When I Hot Reload and watch the log, it does not seem to give any errors and says that a DDF was found for the address of my sensor.

I reloaded the integration in Home Assistant and it isn't in there.  Also from Developer Tools, Services called deconz.device_refresh and still not showing.

Smanar, 2022-01-11:
> The node title looks like your screenshot (no real name).

OK, the device is not reconized, don't need to search far away. If the device is reconised and added in the API, you will have other thing than 0xXXXX.

>When I Hot Reload and watch the log, it does not seem to give any errors and says that a DDF was found for the address of my sensor.

Hu ? why deconz don't reconise the device if it can find the DDF ? You can share a capture of the logs ?
I have perhaps made a typo in the manufacture name (use "," instead of "." or "L" instead of "I") but if the log say it have find something ....

majorsl, 2022-01-11:
With VNC into the deconz Docker, I can only get to the debug view. Logs don't seem to be saved anywhere. I acn screen shot the VNC window if we're looking for something when I hot reload it.  Does that help.

Smanar, 2022-01-11:
Yep, a picture can be enought, because for the moment all seem right.
I don't understand why the api don't reconise it if the DDF is good.

majorsl, 2022-01-14:
@Smanar I really thank you for all the help.  For now, could you link me to the DDF documentation and example file(s)?  I want to tinker with this myself and see if I can figure it out and learn.  The file does not look to complicated for this one so it might be a good one for me to start with.  Maybe I'll have a question or two.

Smanar, 2022-01-15:
As you have the application, I think the easier is using the DDF editor, all is done with drag & drop https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
But it seem somes user have issue to maximize/minize windows to do that on the HA VNC stuff.

As sample I have used this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/samsung/samjin_multi_sensor.json

It s a mutli sensor that use too the IAS cluster (0x0500), like your device

But I think the easier is starting with the DDF I have gived, you can remove/add stuff on it using the editor, but there is some thing harder to do if you start from scratch, like the part with fingerprint (not possible only with the editor)

But I think we are missing something because the result is strange, if the DDF is good why the device is not reconized ...

majorsl, 2022-01-15:
Thanks or those, I'll take a look.  Here is the debug screenshot where you can see it does find the device and loads the DDF.  Let me know if any of the other debug options are needed and I'll re-test.
[image]

Smanar, 2022-01-16:
Have you tried to make the same thing but with phoscon with the permit join enabled ? (add new sensor)
I m asking to other devs where we can miss something, but it s perhaps because even the DDF core can create new entry if the permit join is not enabled.

majorsl, 2022-01-16:
When we first created the DDF, I removed the sensor from deconz.  I then used Phoscon to "add new sensor" and paired it.  Of course, Phoscon did not see it, but it did re-appear in deconz.

I tried that process again just a little while ago to make sure and it still does not work.  I also did it right next to the Conbee II to be sure it wasn't going through another node during the add and having a problem there.

If I should try it another way, I can give it a go.

Smanar, 2022-01-17:
So no more information yet ...
A fast thing to try.
- put the actual json in an other folder
- start a DDF from scratch
- Just set the manufacture name / product id using drag and drop
- Just set a ZHAPresence (without configuration)

You will have a not working DDF, but I hope the device will be reconized and included.

Smanar, 2022-01-18:
BTW there is someone making a DDF for this sensor https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5681

majorsl, 2022-01-18:
If I get a chance, I'll try that DDF from the PR after work. I'll add your battery schema if the author hasn't updated it yet too.

Smanar, 2022-01-18:
And don't forget to edit the "manufacturername".

devWaves, 2022-01-18:
Hey guys. I am trying to also add the Thirdreality water leak sensor

I got it somewhat working by copying most of @Smanar PR but replacing for water leak and the rest api finds it

if DDF support can be added for water leak sensor that would be cool too. I dont know how to do that atm

The thirdreality water leak sensor is 3RWS18BW

Mimiix, 2022-01-18:
@devWaves can you please use a dedicated git issue for that? If there's no request yet, please create a proper device request.

devWaves, 2022-01-19:
> @devWaves can you please use a dedicated git issue for that? If there's no request yet, please create a proper device request.

I created a separate git request for the ThirdReality Water Leak sensor. It looks very similar to the changes for the motion sensor
#5682

majorsl, 2022-02-10:
Ping.

Smanar, 2022-02-10:
Ha yes here we have a problem, it's not possible to set the manufacture name using the editor, because of the comma.
It's perhaps possible using direclty a text editor.
Someone is trying to make a DDF ?

majorsl, 2022-03-05:
Ping.

Smanar, 2022-03-27:
Hello, If I make a DDF to test, can you add it in your OS ?
Or try creating a fake one, to be able to edit it later.

majorsl, 2022-03-27:
Yes, I'll try again!  Should I wait for the next release?  I am only on the last stable release right now, not the beta.

majorsl, 2022-04-05:
Hurray! That one seems to have worked.  I'm seeing the presence sensor and battery level. Thanks!

Smanar, 2022-04-06:
Ha finally ^^.
If all is working can you make a PR or give the final DDF for I make it, to close the issue.

majorsl, 2022-04-06:
It is the exact DDF you did above about 10 days ago. I didn't change anything at all, it worked right away. Go ahead and make the PR, it's all your hard work! Thanks again!

closed 2022-05-06 by github-actions[bot]

justin-caldicott, 2023-03-10:
Sorry to resurrect an old issue, but I'm wondering if you're still able to pair your Third Reality motion sensors @majorsl?

I have the same sensor, 3RMS16BZ, and am using version 2.19.3 of deconz. But it's refusing to pair. I'm doing:
* Phoscon, add sensor, other
* Hold the sensor near the ConBee 2 dongle
* Press and hold the small button on the PCB for a few seconds

The LED goes red, then quickly flashes blue, but then slowly flashes red 3 times and nothing is added. Any other tips for how you got yours to pair? It looks like it should work, given the PR was merged.

Smanar, 2023-03-10:
Hello, have you the GUI to compare the model id and the manufacture name ?

Mimiix, 2023-03-10:
For pairing questions, please reside to the forums.

Locking this up.