dresden-elektronik/deconz-rest-plugin #5341: Device name

dev-zero, 2021-10-03:
## Device

- Product name: Nexentro/Insta Switching Actuator Mini ("Schaltaktor Mini")
- Manufacturer: Insta
- Model identifier: Generic UP Device
- Device type : Please remove all unrelated device types. 
  - Light
  - Switch

## Screenshots

### Basic

[image]
[image]
[image]

dev-zero, 2021-11-08:
The stale bot on device requests is really annoying...

Mimiix, 2021-11-08:
@dev-zero There's a reason for it:) If it isn't there, it wouldn't get any attention every 30 days by me or others involved. Now we get a heads-up every time instead of forgetting about it. Clever, right?

In this case, that's great as i seem to have missed your issue. There seems to be another request open: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5341

In there, the guy seem to have DDF working for this device (afaik). Either way, as it's a double post, i'll close this issue and ask you to stick to that issue :)

Kind regards,

closed 2021-11-08 by Mimiix

dev-zero, 2021-11-09:
Do you mean #4619 (different device by the same company)? The linked issue is this one.

Mimiix, 2021-11-09:
> @Mimiix Do you mean #4619 (different device by the same company)? The linked issue is this one.
> 
My bad on the URL. Indeed, 4619. 

> And wrt stable bot: I would appreciate a reminder if a bug has not been triaged yet, but as soon as an issue has been verified there is IMHO no point in marking it stale (and auto-closing), it only generates noise.

If a bug is confirmed, the bot doesn't come in anymore. Do you have a example of a bug that doesn't have confirmed bug at this point? I might have forgotten to add the label there.  It becoming stale always triggers me to act on it and I always ping people in the backend to act on it.

dev-zero, 2022-02-11:
Ok, I don't know what else to do: the linked issue is for a different device by the same company, and despite running deconz v2.13.4 the behaviour in the rest api is still that when I try to turn the device on it just toggles. I can configure the device via bluetooth into 3 different modes: pushbutton, switch, sensor, but I have the same behaviour within the REST api for all of them.