dresden-elektronik/deconz-rest-plugin #5342: Innopro S-LUX-SB Device name

AlphaDE, 2021-10-03:
## Device

- Product name: Innopro S-LUX-ZB.
- Manufacturer: unknown
- Model identifier: unknown
- Device type : RFD
  - Sensor

## Screenshots

[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

https://githubmemory.com/repo/Koenkk/zigbee2mqtt/issues/8036

https://zigbee.blakadder.com/Tuya_S-LUX-ZB.html

https://de.aliexpress.com/item/1005002828041023.html

Mimiix, 2021-10-03:
Missing all screens.

AlphaDE, 2021-10-03:
@Mimiix I tried to get more screens but I found no way to reach those shown in the linked instructions.

Mimiix, 2021-10-03:
Check the wiki on how to read clusters

AlphaDE, 2021-10-04:
Thanks @Mimiix 
[image]
[image]
[image]

[image]
[image]
[image]

closed 2021-11-25 by github-actions[bot]