dresden-elektronik/deconz-rest-plugin #5343: Legrand Wireless Smart Dimmer with Netatmo WNRL63

stevenwfoley, 2021-10-03:
## Device
- Product name: Wireless Smart Dimmer with Netatmo
- Manufacturer: Legrand
- Model identifier: Remote Dimmer Switch
- Device type : Switch
https://www.legrand.us/wiring-devices/radiant-collection/dimmers/smart-tru-universal-dimmer-with-netatmo/p/wnrl63wh

## Screenshots
[image]
[image]
[image]
[image]
[image]
[image]

stevenwfoley, 2021-10-03:
FYI, this Legrand/Netatmo device also appears to require zigbee channel 11 as with other Legrand devices [883](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/883). My deconz instance was on channel 25 and the switch paired and worked for a couple minutes but then completely stopped working. I tried repairing several times but experienced the same exact behavior. I changed deconz to channel 11 and the device instantly started working properly and consistently.

Smanar, 2021-10-04:
I think it will be easy to add it, it's just a battery powered switch with 2 positions "up" and "down" ? No special features ?

stevenwfoley, 2021-10-04:
Yes, battery powered dimmer switch, so it has 4 buttons total: on, off, dim-up, dim-dowm.
Thank you!

Smanar, 2021-10-04:
4 buttons or 4 features on only 2 buttons ?
The actual one have only 2 positions on/off and it's the long press that make the dum up and dim down.

stevenwfoley, 2021-10-04:
4 buttons.

stevenwfoley, 2021-10-05:
I'm not seeing a single event from the button presses of this device in the deconz debug log view. I've captured some wireshark data though. I can send it if its helpful.

Smanar, 2021-10-05:
We will see, but I think it's standard, Legrand is realy zigbee compliant.
I haven't asked, you have a linux OS machine with full OS to test code modification ?

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch Legrand_1 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

But you need to edit the json file too, do you know where to find it ? On raspberry it's /usr/share/deCONZ/devices/button_maps.json

Changes are just https://github.com/Smanar/deconz-rest-plugin/commit/95eb0b8b366b350b514b58e99a8d216ac283f57b#diff-a3adce467fe93b9c612c198c187dad34f5284d6c9adf66aa28127f0f619979b5

It can work on first try, else you will have some debug line starting by "[INFO] - Button" when using a button (with the "info" flag)

stevenwfoley, 2021-10-06:
I compiled the dev REST plugin, re-joined the switch to deconz, and the device was now identified by deconz and phoscon; although the phoscon gui shows the device as only having two buttons. The unfortunate part is that the buttons do not function, all I see with INFO and INFO_L2 debug view is this:
- Warning duplicated zcl.cmd: 0x09, cluster: 0xFC01 with zcl.seq: 1 for Legrand / Remote dimmer switch
- [WARNING] - Missing cluster in sensor fingerprint: 0x646266FFFE611E7C - 0xFA3B (Remote dimmer switch), endpoint: 0x01, cluster: 0xFC01, payload: 7C1E61FEFF66626401, zclSeq: 1
- [INFO] - No button handler for: Remote dimmer switch, unicast to: 0x0000, endpoint: 0x01, cluster: 0xFC01, command: 0x09, payload: 7C1E61FEFF66626401, zclSeq: 1

stevenwfoley, 2021-10-06:
FYI, I confirmed with Legrand that this specific device is zigbee 3.0 compatible. https://developer.legrand.com/forums/topic/wnrl63-and-zigbee-3-0/

Smanar, 2021-10-06:
>the phoscon gui shows the device as only having two buttons

Phoscon is just an app using the API, I haven't impact on it, but BTW I think you will need a third app to use this device or wait for phoscon will support it too.

>[WARNING] - Missing cluster in sensor fingerprint: 0x646266FFFE611E7C - 0xFA3B (Remote dimmer switch), endpoint: 0x01, cluster: 0xFC01, payload: 7C1E61FEFF66626401, zclSeq: 1

Up to now I never add the Legrand Cluster to the fingerprint, I never see an use for it

>[INFO] - No button handler for: Remote dimmer switch, unicast to: 0x0000, endpoint: 0x01, cluster: 0xFC01, command: 0x09, payload: 7C1E61FEFF66626401, zclSeq: 

Yeah it s exactly the log I need, but BTW they are all on cluster 0xFC01 ?
I need this log but for all 4 buttons.
But It s strange you have no logs on classic clusters, the 0x0008 and 0x0006

stevenwfoley, 2021-10-07:
I had to manually bind cluster 6 and 8. I've tried repairing many times and the device never binds the clusters. Is there a way to compensate for this in deconz?

Here are the log entries after I bound the clusters.
On: [INFO] - Button 1002 - Remote dimmer switch, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 0
Off: [INFO] - Button 2002 - Remote dimmer switch, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: Off, payload: None, zclSeq: 0
Dim-Up: [INFO] - Button 1001 - Remote dimmer switch, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Dimm up, payload: 00FF, zclSeq: 0
Dim-Up: [INFO] - Button 1003 - Remote dimmer switch, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Dimm up stop, payload: None, zclSeq: 1
Dim-Down: [INFO] - Button 2001 - Remote dimmer switch, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Dimm down, payload: 01FF, zclSeq: 3
Dim-Down: [INFO] - Button 2003 - Remote dimmer switch, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Dimm down stop, payload: None, zclSeq: 4

Smanar, 2021-10-13:
>Do you still want the inclusion logs?

No, if it s working.
Just tell me if you miss something else I will make the PR

stevenwfoley, 2021-10-15:
I just noticed that the only entity showing up in Home Assistant is the battery level. Is this an issue with deconz not exporting/exposing the capabilities of the switch or is this an issue with Home Assistant?

closed 2021-11-13 by github-actions[bot]

stevenwfoley, 2022-02-10:
I just noticed a nuance with this hardware and I'm not sure if it matters, but the manufacturer and model text strings are technically both preceded with a single space. If you look at the Cluster Info image, the Manufacturer Name is " Legrand" and the Model Identifier is " Remote dimmer switch". I'm not sure if this effects the string matching logic, or if somewhere the strings are trimmed of whitespace before matching, but just FYI.

Smanar, 2022-02-11:
>or if somewhere the strings are trimmed of whitespace before matching,

Good question, I know last chars are trimmed, but never tested for firsts chars.

But if you have the device included, can just take a look in the device json (for exemple in phoscon / help / api Information / light ...)