dresden-elektronik/deconz-rest-plugin #5344: HA Integration, Windows deCONZ not Pushing

mrrodge2020, 2021-10-04:
## Describe the bug

Any comms from HA to deCONZ seem to work but nothing ever comes back. If I turn on a light/switch the command gets through and all works fine, as does calling the deconz.device_refresh service (this updates all my temperature sensor readings).

It seems the API isn't pushing anything back to my HA integration.  Temperature sensors in HA are now showing readings from 3 days ago but are up to date in Phoscon.  If I manually poll by calling the deconz.device_refresh service they'll update just the once.  If I turn on a light/switch, HA issues the command and the switch/light turns on/off as desired but the state doesn't register in HA.

Could not log in to Phoscon from any machine other than host, checked just now and cannot log in from host.  Unsure if this is separate issue.
## Steps to reproduce the behavior

## Expected behavior

Local push, as per HA integration IOT class.
## Screenshots

N/A
## Environment

 - Host system: Windows Server 2022, Conbee II connected via direct USB.
 - Running method: Virtualised HA with deCONZ integration connected via IP to the host.  Port 80 is open on host's Windows Firewall to enable connection.
 - Firmware version: 0x264a0700
 - deCONZ version: 2.12.6
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? Nothing other than keyboard/mouse.

## deCONZ Logs

Can provide if needed - please specify what to log.
## Additional context

Kane610, 2021-10-04:
Sounds like your firewall doesn't allow connections to deconz websocket

mrrodge2020, 2021-10-04:
> 
> 
> Sounds like your firewall doesn't allow connections to deconz websocket

Which firewall?  I have Windows Firewall turned off now and still no joy - for pushed stuff, do I have to allow something through the HASSOS firewall?

mrrodge2020, 2021-10-04:
OK thanks.  Should I be able to see some sort of connection failure in HA's logs?  Windows firewall is completely off, still no joy.  Can only log in to Phoscon web app using loopback and not host IP - is this normal?

Kane610, 2021-10-04:
You should probably see something. If not try enabling debug for the integration

mrrodge2020, 2021-10-04:
Getting somewhere now.

` Logger: pydeconz.websocket
Source: /usr/local/lib/python3.9/site-packages/pydeconz/websocket.py:106
First occurred: 30 September 2021, 17:40:08 (6 occurrences)
Last logged: 11:38:36
Websocket is not accessible (75.0.0.3)`

How do I view/change what the websocket port is/is there a list of ports other than 80 that I need to open?  Weird though as the Windows Firewall is off.

mrrodge2020, 2021-10-04:
Firewall off, websocket log errors have stopped, suggesting connection is OK.  Still not getting anything pushed through though.

mrrodge2020, 2021-10-04:
Where do I look?!

Kane610, 2021-10-04:
Home assistant logs

mrrodge2020, 2021-10-04:
Only have the default settings so working in configuration.yaml now to set debug on deCONZ.

mrrodge2020, 2021-10-04:
OK so set debug and got the following, but no line 80 in your link:

`2021-10-04 14:43:59 DEBUG (MainThread) [homeassistant.components.deconz] deCONZ event created: lumi_ctrl_neutral2_2
2021-10-04 14:43:59 DEBUG (MainThread) [homeassistant.components.deconz] deCONZ event created: kitchen_undercabinet_switch
2021-10-04 14:43:59 DEBUG (MainThread) [homeassistant.components.deconz] deCONZ event created: cbedside
2021-10-04 14:43:59 DEBUG (MainThread) [homeassistant.components.deconz] deCONZ event created: jbedside
2021-10-04 14:43:59 DEBUG (MainThread) [homeassistant.components.deconz] deCONZ event created: office_power_button
2021-10-04 14:44:06 DEBUG (MainThread) [homeassistant.components.deconz] deCONZ SSDP discovery {'CACHE-CONTROL': 'max-age=100',
'GWID.phoscon.de': '00212EFFFF0483E5',
'UDN': 'uuid:fd75b167-f714-43e4-af59-b8c8567e23f1',
'_host': '75.0.0.3',
'_location_original': 'http://75.0.0.3:80/description.xml',
'_port': 1900,
'_source': 'search',
'_timestamp': datetime.datetime(2021, 10, 4, 14, 43, 59, 714478),
'_udn': 'uuid:fd75b167-f714-43e4-af59-b8c8567e23f1',
'deviceType': 'urn:schemas-upnp-org:device:Basic:1',
'friendlyName': 'Phoscon-GW (75.0.0.3)',
'hue-bridgeid': '00212EFFFF0483E5',
'iconList': {'icon': {'depth': '24',
'height': '48',
'mimetype': 'image/png',
'url': 'hue_logo_0.png',
'width': '48'}},
'manufacturer': 'Royal Philips Electronics',
'manufacturerURL': 'http://www.dresden-elektronik.de',
'modelDescription': 'Philips hue compatible Personal Wireless Lighting',
'modelName': 'Philips hue bridge 2015',
'modelNumber': 'BSB002',
'modelURL': 'http://www.dresden-elektronik.de',
'presentationURL': 'index.html',
'serialNumber': '00212E0483E5',
'ssdp_ext': '',
'ssdp_location': 'http://75.0.0.3:80/description.xml',
'ssdp_server': 'Linux/3.14.0 UPnP/1.0 IpBridge/1.26.0',
'ssdp_st': 'upnp:rootdevice',
'ssdp_usn': 'uuid:fd75b167-f714-43e4-af59-b8c8567e23f1::upnp:rootdevice'}
2021-10-04 14:44:06 DEBUG (MainThread) [homeassistant.components.deconz] Preparing linking with deCONZ gateway {'host': '75.0.0.3', 'port': 80}
2021-10-04 14:44:06 DEBUG (MainThread) [homeassistant.components.deconz] deCONZ SSDP discovery {'CACHE-CONTROL': 'max-age=100',
'GWID.phoscon.de': '00212EFFFF0483E5',
'UDN': 'uuid:fd75b167-f714-43e4-af59-b8c8567e23f1',
'_host': '75.0.0.3',
'_location_original': 'http://75.0.0.3:80/description.xml',
'_port': 1900,
'_source': 'search',
'_timestamp': datetime.datetime(2021, 10, 4, 14, 43, 59, 845008),
'_udn': 'uuid:fd75b167-f714-43e4-af59-b8c8567e23f1',
'deviceType': 'urn:schemas-upnp-org:device:Basic:1',
'friendlyName': 'Phoscon-GW (75.0.0.3)',
'hue-bridgeid': '00212EFFFF0483E5',
'iconList': {'icon': {'depth': '24',
'height': '48',
'mimetype': 'image/png',
'url': 'hue_logo_0.png',
'width': '48'}},
'manufacturer': 'Royal Philips Electronics',
'manufacturerURL': 'http://www.dresden-elektronik.de',
'modelDescription': 'Philips hue compatible Personal Wireless Lighting',
'modelName': 'Philips hue bridge 2015',
'modelNumber': 'BSB002',
'modelURL': 'http://www.dresden-elektronik.de',
'presentationURL': 'index.html',
'serialNumber': '00212E0483E5',
'ssdp_ext': '',
'ssdp_location': 'http://75.0.0.3:80/description.xml',
'ssdp_server': 'Linux/3.14.0 UPnP/1.0 IpBridge/1.26.0',
'ssdp_st': 'uuid:fd75b167-f714-43e4-af59-b8c8567e23f1',
'ssdp_usn': 'uuid:fd75b167-f714-43e4-af59-b8c8567e23f1'}
2021-10-04 14:44:06 DEBUG (MainThread) [homeassistant.components.deconz] deCONZ SSDP discovery {'CACHE-CONTROL': 'max-age=100',
'GWID.phoscon.de': '00212EFFFF0483E5',
'UDN': 'uuid:fd75b167-f714-43e4-af59-b8c8567e23f1',
'_host': '75.0.0.3',
'_location_original': 'http://75.0.0.3:80/description.xml',
'_port': 1900,
'_source': 'search',
'_timestamp': datetime.datetime(2021, 10, 4, 14, 43, 59, 906920),
'_udn': 'uuid:fd75b167-f714-43e4-af59-b8c8567e23f1',
'deviceType': 'urn:schemas-upnp-org:device:Basic:1',
'friendlyName': 'Phoscon-GW (75.0.0.3)',
'hue-bridgeid': '00212EFFFF0483E5',
'iconList': {'icon': {'depth': '24',
'height': '48',
'mimetype': 'image/png',
'url': 'hue_logo_0.png',
'width': '48'}},
'manufacturer': 'Royal Philips Electronics',
'manufacturerURL': 'http://www.dresden-elektronik.de',
'modelDescription': 'Philips hue compatible Personal Wireless Lighting',
'modelName': 'Philips hue bridge 2015',
'modelNumber': 'BSB002',
'modelURL': 'http://www.dresden-elektronik.de',
'presentationURL': 'index.html',
'serialNumber': '00212E0483E5',
'ssdp_ext': '',
'ssdp_location': 'http://75.0.0.3:80/description.xml',
'ssdp_server': 'Linux/3.14.0 UPnP/1.0 IpBridge/1.26.0',
'ssdp_st': 'urn:schemas-upnp-org:device:basic:1',
'ssdp_usn': 'uuid:fd75b167-f714-43e4-af59-b8c8567e23f1::urn:schemas-upnp-org:device:basic:1'}
2021-10-04 14:45:57 DEBUG (MainThread) [homeassistant.components.deconz] Preparing linking with deCONZ gateway {'host': '75.0.0.3', 'port': 80}
2021-10-04 14:46:05 DEBUG (MainThread) [homeassistant.components.deconz] Preparing linking with deCONZ gateway {'host': '75.0.0.3', 'port': 80}
2021-10-04 14:46:05 DEBUG (MainThread) [homeassistant.components.deconz] deCONZ event created: lumi_ctrl_neutral2_2
2021-10-04 14:46:05 DEBUG (MainThread) [homeassistant.components.deconz] deCONZ event created: kitchen_undercabinet_switch
2021-10-04 14:46:05 DEBUG (MainThread) [homeassistant.components.deconz] deCONZ event created: cbedside
2021-10-04 14:46:05 DEBUG (MainThread) [homeassistant.components.deconz] deCONZ event created: jbedside
2021-10-04 14:46:05 DEBUG (MainThread) [homeassistant.components.deconz] deCONZ event created: office_power_button`

mrrodge2020, 2021-10-04:
Holy moly I've got it (I think).  Firewall I thought was off actually wasn't.  Feel so stupid!

Thanks so much for your patience.

90% of the way there now.  I just can't work out what port(s) I need to open.

mrrodge2020, 2021-10-04:
Done it!  Was 8081.  Looked in the HTTP debug view of deCONZ, saw the port logged there.

Thanks so much.

closed 2021-10-04 by mrrodge2020