dresden-elektronik/deconz-rest-plugin #5346: IKEA Symfonisk controller doesn't send events?

fribse, 2021-10-04:
## Describe the bug

I've had some issues with the zigbee network, half of it is suddenly unresponsive.
So I restarted the deconz to make it better, and I've now walked around the house and 'rebooted' lights (screw them out, and back in) and most of it is back alive.
The Symfonisk controller does however not send any events. I've tried removing it and readding it, and all looks good (as shown in the images).
But I simply do not see any events in deconz_event in home assistant.

## Steps to reproduce the behavior

## Expected behavior

It should send events for presses and rotation.
## Screenshots

[image]
[image]
[image]

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

TheNON75, 2021-10-04:
Hi @fribse 
I had a similar issue some weeks ago after a battery swap in one of my symfonisk controller.
I’ve added it at least twice until it started to work.
Can you do some more attempts? (And keep clicking with the knob until it is full read)

Mimiix, 2021-10-04:
This seems like it's a user question as it's not reproducable and not pointing to code (as explained in #5113).

Please use the forum for this question.

closed 2021-10-04 by Mimiix

fribse, 2021-10-04:
I upgraded the firmware to the latest in the USB stick, and after doing that, it works again?

TheNON75, 2021-10-04:
Dear @fribse 

please consider that this bug report is closed now due to the reasons referred to by Mimiix
I am kindly asking you to use the forum for this question.
Sorry for the inconvenience

der-david, 2022-01-06:
@fribse If you still have problems using the Symfonisk, maybe my [comment on #3969](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3969#issuecomment-1007008377) can help.

fribse, 2022-01-07:
No, it's solved, thanks!