dresden-elektronik/deconz-rest-plugin #5347: Device name

igreg, 2021-10-04:
The device does not show in the Photoscon app. I used the app to make the node join the cluster; to be able to query the data with the DeCONZ application. There seem to be other manufacturer for this model (TS0012) so perhaps it is just a mapping that's missing somewhere?

## Device

- Product name: qs-zigbee-s04-2c-l (made by Lonsonho)
- Manufacturer: _tz3000_llfaquvp
- Model identifier: TS0012
- Device type :
  - Light
  - Switch (2 Gang, no neutral, wired switch)

## Screenshots

[image]

[image]

### Basic

[image]

### Identify

[image]

### Alarms

n/a

### Device Temperature

n/a

### Groups

[image]

### Scenes

[image]

### On/Off

01
[image]
02
[image]

SwoopX, 2021-10-05:
Could you please change the title to something more talkative? 😉

Smanar, 2021-10-05:
I think it s because your device is an end device and the other a router.
Have you a linux machine with full OS to test code modifications ?

igreg, 2021-10-06:
I actually have the same exact problem with the 1-gang version of the switch.
 * Product name: 1CH Zigbee Switch Module-L (no neutral line)
 * Manufacturer: _TZ3000_jir4araar
 * Model identifier: TS0011
 * Device type :
     * Light
     * Switch (1 Gang, no neutral, wired switch)

Pretty much the same than above except it does not have the On/Off "Light mode 2"

Smanar, 2021-10-06:
Ok so first try

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch tuya_switch_7 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

This code just add the manufacture 0x1141 in whitelist to permit "light" device creation for end device.

closed 2021-11-04 by github-actions[bot]

blazejpawlak, 2021-12-02:
Is anyone planning on adding this switch to supported ones in the upcoming release?

Smanar, 2021-12-02:
Nop, have not found someone to test the last change, and ATM adding new device using c++ code is disabled, waiting for DDF core.

blazejpawlak, 2021-12-02:
I can do that. Is [this](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5347#issuecomment-936747855) enough to complete it?

Smanar, 2021-12-03:
If you have a recent deconz version I can help you to make a DDF file to use this device (I think the core is advanced enought for this kind of device).

The problem is ATM all new device addition are blocked, to not add code that will be removed when ddf will be ready. So even If I finish the PR, not sure it will be validated.

blazejpawlak, 2021-12-04:
That would be great. I'm running deconz v2.12.06 / 8/19/2021 with firmware 26680700. 

About adding the device to a releasable version — is it because some major deconz overhaul is coming?

Smanar, 2021-12-04:
>About adding the device to a releasable version — is it because some major deconz overhaul is coming?

Yes https://dresden-elektronik.github.io/deconz-dev-doc/modules/ddf/

Are you able to update your deconz version to the lasted beta to test DDF file ? (V v2.13.2)

blazejpawlak, 2021-12-08:
I'll have to make a clone of my RPi's sdcard and try. Soonest I can try is after the 20th Dec. I'll keep you posted.

koenheemskerk, 2022-03-04:
Have you made progress?

blazejpawlak, 2023-02-20:
I can confirm that this switch now works with deconz in version 2.20.01 / 9/19/2022 with Conbee II running firmware in version 26780700.

One funny thing though, when I press both buttons of the physical "dumb" switch connected to this device via L1 and L2 only one of them turns on connected to L2. When I press only one switch (connected to either L1 or L2) and then another one both lights come up. 
So if I press both buttons of the switch (connected to L1 and L2), only L2 light comes up. Individually (pressing one button of the switch, waiting a second, then pressing the other button) both lights come up.

Ideas?

Smanar, 2023-02-21:
Yep, an other issue was created here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6325

For your issue IDK, have made a search on z2m, it don't seem this device have something special (a tuya unlock sequence for exemple)