dresden-elektronik/deconz-rest-plugin #5349: LIDL SilverCrest Smart Radiator Thermostat

synatis, 2021-10-05:
## Device

- Product name: LIDL SilverCrest Smart Radiator Thermostat
- Manufacturer: _TZE200_chyvmhay
- Model identifier: TS0601
- Device type : Thermostat

## Screenshots

### Endpoint and Node Info
[image]

### Basic
[image]

### Groups
[image]

### Scenes
[image]

### Thermostat
[image]

synatis, 2021-10-08:
Let me know if more info is required!

holzmichel6, 2021-10-11:
Hallo, bei mir lassen sich diese thermostate  auch nicht steuern ...

Smanar, 2021-10-13:
Ok so first try

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch lidl_trv https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

The code make lot of thing, but still missing some feature like the "windows open", just tell me that is working or missing.

Smanar, 2021-10-15:
"Preset" can't work, It s not used for TRV (but I think we will need it for this one, because of it s special features), It s only used for thermostat, not valve.
If I m right you haven't it in the json ?

And for auto, there is a problem I don't have HA, so idk how it work
>If it's set to Off and I change the temp. it switches to "Auto" but doesn't push the new temp. on it. Same if it's already in Auto-mode - no changes.

I can't say if it s the device that switch itself to "auto" when receve a temperature, or if it s from HA.

If you still are on the GUI, can you take a look in logs with "info" "info_l2" and trying the same procedure.
I need logs with "tuya" inside pls.
It can depend of order, if HA send first the temperature before the mode, so perhaps I need to memorise it somewhere because there is not the same issue on other devices with HA too, something specific to this one ...

synatis, 2021-10-17:
Hey, are those the json you need to see?
[image]

### Info for switching temp while device is set to off

>19:03:44:636 create node 0xCC86ECFFFE4EAF03 seen by 2 neighbors
19:03:44:969 Send Tuya request 0xB4E3F9FFFEC74CBF : Dp_type: 0x02, Dp_identifier 0x10, data: 00000016
19:03:45:115 create node 0x5C0272FFFECA638F seen by 2 neighbors
19:03:45:324 Tuya debug 4 : Address 0xB4E3F9FFFEC74CBF Payload 00101002000400000016
19:03:45:324 Tuya debug 5 : Status: 0 Transid: 16 Dp: 528 (0x02,0x10) Fn: 0 Data 22
19:03:45:445 Tuya debug 4 : Address 0xB4E3F9FFFEC74CBF Payload 00101002000400000016
19:03:45:445 Tuya debug 5 : Status: 0 Transid: 16 Dp: 528 (0x02,0x10) Fn: 0 Data 22
19:03:47:036 create node 0x5C0272FFFEF5B403 seen by 2 neighbors
19:03:47:516 create node 0x5C0272FFFED1D6EE seen by 2 neighbors
19:03:47:997 create node 0xCC86ECFFFE4EAF03 seen by 2 neighbors
19:03:48:476 create node 0x5C0272FFFECA638F seen by 2 neighbors
19:03:49:916 create node 0x5C0272FFFED1D6EE seen by 2 neighbors
19:03:50:396 create node 0x5C0272FFFEF5B403 seen by 2 neighbors
19:03:50:854 Device TTL 2220 s flags: 0x7
19:03:52:316 create node 0xCC86ECFFFE4EAF03 seen by 2 neighbors
19:03:52:326 Tuya debug 4 : Address 0xB4E3F9FFFEC74CBF Payload 00020204000100
19:03:52:326 Tuya debug 5 : Status: 0 Transid: 2 Dp: 1026 (0x04,0x02) Fn: 0 Data 0
19:03:52:437 Tuya debug 4 : Address 0xB4E3F9FFFEC74CBF Payload 00020204000100
19:03:52:437 Tuya debug 5 : Status: 0 Transid: 2 Dp: 1026 (0x04,0x02) Fn: 0 Data 0
19:03:52:796 create node 0x5C0272FFFECA638F seen by 2 neighbors
19:03:54:715 create node 0x5C0272FFFEF5B403 seen by 2 

### Info_l2

>19:10:27:733 Erase task req-id: 160, type: 37 zcl seqno: 137 send time 3, profileId: 0x0104, clusterId: 0xEF00
19:10:27:837 CTRL skip creating node 0x5C0272FFFECA638F while permit join is disabled
19:10:27:917 Tuya debug Request : Address 0xB4E3F9FFFEC74CBF, Endpoint 0x01, Command 0x02, Payload 00020204000102
19:10:29:278 CTRL skip creating node 0x5C0272FFFED1D6EE while permit join is disabled
19:10:29:757 CTRL skip creating node 0x5C0272FFFEF5B403 while permit join is disabled
19:10:30:833 Daylight now: sunsetEnd, status: 200, daylight: 0, dark: 1
19:10:31:228 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0xE1A7
19:10:31:228 neigbor 0x0c4314fffe619d00 is unknown child
19:10:31:677 CTRL skip creating node 0xCC86ECFFFE4EAF03 while permit join is disabled
19:10:32:144 Websocket 192.168.178.25:43764 send message: {"config":{"battery":100,"heatsetpoint":1300,"locked":false,"mode":"off","offset":0,"on":true,"preset":"auto","reachable":true,"schedule":{},"schedule_on":false,"setvalve":false},"e":"changed","id":"2","r":"sensors","t":"event","uniqueid":"b4:e3:f9:ff:fe:c7:4c:bf-01-0201"} (ret = 273)
19:10:32:156 CTRL skip creating node 0x5C0272FFFECA638F while permit join is disabled
19:10:32:636 CTRL skip creating node 0x0C4314FFFE619D00 while permit join is disabled
19:10:33:286 Erase task req-id: 178, type: 37 zcl seqno: 138 send time 1, profileId: 0x0104, clusterId: 0xEF00
19:10:33:479 Tuya debug Request : Address 0xB4E3F9FFFEC74CBF, Endpoint 0x01, Command 0x02, Payload 0010100200040000001a
19:10:33:655 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:10:34:077 CTRL skip creating node 0x5C0272FFFEF5B403 while permit join is disabled
19:10:34:556 CTRL skip creating node 0x5C0272FFFED1D6EE while permit join is disabled
19:10:35:036 CTRL skip creating node 0xCC86ECFFFE4EAF03 while permit join is disabled
19:10:35:516 CTRL skip creating node 0x5C0272FFFECA638F while permit join is disabled
19:10:36:956 CTRL skip creating node 0x5C0272FFFED1D6EE while permit join is disabled
19:10:37:436 CTRL skip creating node 0x5C0272FFFEF5B403 while permit join is disabled
19:10:37:832 DB save zll database items 0x00000080
19:10:37:832 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('2', 'Thermostat 2', 'ZHAThermostat', 'TS0601', '_TZE200_chyvmhay', 'b4:e3:f9:ff:fe:c7:4c:bf-01-0201', '', '{"lastupdated":"2021-10-14T17:10:22.331","on":false,"temperature":2190}', '{"battery":100,"heatsetpoint":1300,"locked":false,"mode":"off","offset":0,"on":true,"preset":"auto","reachable":true,"schedule":null,"schedule_on":false,"setvalve":false}', '{"d":769,"ep":1,"in":[61184,0,513,61184],"p":260}', 'normal', '1', '2021-10-13T17:44Z', '')
19:10:37:839 DB saved in 6 ms
19:10:38:912 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0xE1A7
19:10:38:912 neigbor 0x0c4314fffe619d00 is unknown child
19:10:39:356 CTRL skip creating node 0xCC86ECFFFE4EAF03 while permit join is disabled
19:10:39:836 CTRL skip creating node 0x5C0272FFFECA638F while permit join is disabled

synatis, 2021-10-17:
Set mode to manual directly on the device and turned up the value from 13°C to 20°C
It's amazing what you can do, just by looking at logs :'D

> 15:37:35:069 Idle timer triggered
15:37:35:292 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0xE1A7
15:37:35:292 neigbor 0x0c4314fffe619d00 is unknown child
15:37:35:741 create node 0xCC86ECFFFE4EAF03 seen by 2 neighbors
15:37:35:741 CTRL skip creating node 0xCC86ECFFFE4EAF03 while permit join is disabled
15:37:36:220 create node 0x5C0272FFFECA638F seen by 2 neighbors
15:37:36:221 CTRL skip creating node 0x5C0272FFFECA638F while permit join is disabled
15:37:36:701 CTRL skip creating node 0x0C4314FFFE619D00 while permit join is disabled
15:37:37:720 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
15:37:38:141 create node 0x5C0272FFFED1D6EE seen by 2 neighbors
15:37:38:142 CTRL skip creating node 0x5C0272FFFED1D6EE while permit join is disabled
15:37:38:620 create node 0x5C0272FFFEF5B403 seen by 2 neighbors
15:37:38:621 CTRL skip creating node 0x5C0272FFFEF5B403 while permit join is disabled
15:37:39:101 create node 0x5C0272FFFECA638F seen by 2 neighbors
15:37:39:101 CTRL skip creating node 0x5C0272FFFECA638F while permit join is disabled
15:37:39:568 DB save zll database items 0x00000084
15:37:39:569 DB sql exec REPLACE INTO auth (apikey, devicetype, createdate, lastusedate, useragent) VALUES ('19426F394B', 'pydeconz', '2021-10-13T18:07:01', '2021-10-17T13:37:15', 'HomeAssistant/2021.10.4 aiohttp/3.7.4.post0 Python/3.9')
15:37:39:569 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('2', 'Thermostat 2', 'ZHAThermostat', 'TS0601', '_TZE200_chyvmhay', 'b4:e3:f9:ff:fe:c7:4c:bf-01-0201', '', '{"lastupdated":"2021-10-17T13:37:03.589","on":false,"temperature":1600}', '{"heatsetpoint":2000,"locked":false,"mode":"heat","offset":0,"on":true,"preset":"auto","reachable":true,"schedule":null,"schedule_on":false,"setvalve":false}', '{"d":769,"ep":1,"in":[61184,0,513,61184],"p":260}', 'normal', '1', '2021-10-13T17:44Z', '')
15:37:39:574 DB saved in 5 ms
15:37:39:575 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
15:37:39:582 create node 0xCC86ECFFFE4EAF03 seen by 2 neighbors
15:37:39:583 CTRL skip creating node 0xCC86ECFFFE4EAF03 while permit join is disabled
15:37:39:589 Master: read param with arg 0x19
15:37:39:604 Device TTL 3259 s flags: 0x7
15:37:40:286 Tuya debug Request : Address 0xB4E3F9FFFEC74CBF, Endpoint 0x01, Command 0x02, Payload 00101002000400000028
15:37:40:286 Tuya debug 4 : Address 0xB4E3F9FFFEC74CBF Payload 00101002000400000028
15:37:40:286 Tuya debug 5 : Status: 0 Transid: 16 Dp: 528 (0x02,0x10) Fn: 0 Data 40
15:37:40:286 Debug tuya 77 : 1
15:37:40:287 Debug tuya 77 : 2
15:37:40:389 Tuya debug Request : Address 0xB4E3F9FFFEC74CBF, Endpoint 0x01, Command 0x02, Payload 00101002000400000028
15:37:40:389 Tuya debug 4 : Address 0xB4E3F9FFFEC74CBF Payload 00101002000400000028
15:37:40:389 Tuya debug 5 : Status: 0 Transid: 16 Dp: 528 (0x02,0x10) Fn: 0 Data 40
15:37:40:389 Debug tuya 77 : 1
15:37:40:390 Debug tuya 77 : 2
15:37:41:020 create node 0x5C0272FFFED1D6EE seen by 2 neighbors
15:37:41:020 CTRL skip creating node 0x5C0272FFFED1D6EE while permit join is disabled
15:37:41:501 create node 0x5C0272FFFEF5B403 seen by 2 neighbors
15:37:41:501 CTRL skip creating node 0x5C0272FFFEF5B403 while permit join is disabled
15:37:42:974 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0xE1A7
15:37:42:974 neigbor 0x0c4314fffe619d00 is unknown child
15:37:43:420 create node 0xCC86ECFFFE4EAF03 seen by 2 neighbors
15:37:43:421 CTRL skip creating node 0xCC86ECFFFE4EAF03 while permit join is disabled
15:37:43:901 create node 0x5C0272FFFECA638F seen by 2 neighbors
15:37:43:902 CTRL skip creating node 0x5C0272FFFECA638F while permit join is disabled
15:37:44:381 CTRL skip creating node 0x0C4314FFFE619D00 while permit join is disabled
15:37:45:275 Tuya debug Request : Address 0xB4E3F9FFFEC74CBF, Endpoint 0x01, Command 0x02, Payload 001818020004000000dc
15:37:45:275 Tuya debug 4 : Address 0xB4E3F9FFFEC74CBF Payload 001818020004000000dc
15:37:45:275 Tuya debug 5 : Status: 0 Transid: 24 Dp: 536 (0x02,0x18) Fn: 0 Data 220
15:37:45:277 Websocket 192.168.178.25:57266 send message: {"e":"changed","id":"2","r":"sensors","state":{"lastupdated":"2021-10-17T13:37:45.276","on":false,"temperature":2200},"t":"event","uniqueid":"b4:e3:f9:ff:fe:c7:4c:bf-01-0201"} (ret = 175)
15:37:45:397 Tuya debug Request : Address 0xB4E3F9FFFEC74CBF, Endpoint 0x01, Command 0x02, Payload 001818020004000000dc
15:37:45:397 Tuya debug 4 : Address 0xB4E3F9FFFEC74CBF Payload 001818020004000000dc
15:37:45:397 Tuya debug 5 : Status: 0 Transid: 24 Dp: 536 (0x02,0x18) Fn: 0 Data 220
15:37:45:413 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
15:37:45:820 create node 0x5C0272FFFED1D6EE seen by 2 neighbors
15:37:45:821 CTRL skip creating node 0x5C0272FFFED1D6EE while permit join is disabled
15:37:45:885 Node data 0x5c0272fffe03af80 profileId: 0x0104, clusterId: 0x0702
15:37:45:886 No consumption sensor found for 0x5C0272FFFE03AF80, endpoint: 0x01
15:37:45:886 ZCL attribute report 0x5C0272FFFE03AF80 for cluster: 0x0702, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
15:37:45:886 	payload: 000025573400000000
[... cut ...]
15:37:49:180 create node 0x5C0272FFFEF5B403 seen by 2 neighbors
15:37:49:181 CTRL skip creating node 0x5C0272FFFEF5B403 while permit join is disabled
15:37:49:569 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
15:37:50:646 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0xE1A7
15:37:50:646 neigbor 0x0c4314fffe619d00 is unknown child
15:37:51:101 create node 0xCC86ECFFFE4EAF03 seen by 2 neighbors
15:37:51:102 CTRL skip creating node 0xCC86ECFFFE4EAF03 while permit join is disabled

Smanar, 2021-10-17:
Lol, it s because I know well tuya stuff now

>15:37:40:286 Tuya debug 5 : Status: 0 Transid: 16 Dp: 528 (0x02,0x10) Fn: 0 Data 40

So it s that, we realy need to divide by 2.
But on this try the "heatsetpoint" was already 2000 (and not 1300) before the change, so the value is not updated.

> Changing temperature works in all 3 modes (the buttons below: "Auto", "Heat" and "Off" from left to right) but only pushes it on the device if mode is set to "Heat".
> If it's set to Off and I change the temp. it switches to "Auto" but doesn't push the new temp. on it. Same if it's already in Auto-mode - no changes.

So with new code it will work for "heat" and "auto", IDK yet for "off", but as it switch itself from "off" to "auto" it can work, perhaps not on the first seconds, but on next notification from device.

Have to removed "preset" and "valve", but you need to restart or reinclude the device for the json be updated, so we will check that later.

synatis, 2021-10-17:
In HA:
Switching modes works now correctly. Off is something like holiday mode, shown as a suitcase.
Each mode remembers their value. 

E.g.:
- Off = 17°C
- Heat ("Manu" on the device display) = 18°C
- Auto = 19°C

HA always shows the value I last set. Let's say I was in auto and changed to 20
[image]

Off is still 17. Heat 18. Auto 20. Now when I switch to Off via HA, it shows:
[image]

which is incorrect because the device shows 17.

Switches automatically from Off to Auto if value is changed via HA. Doesn't switch from Heat/Manu to Auto, which is nice.

preset is gone as you said 👍 

Let me know if more logs are needed. Awesome work so far!

Smanar, 2021-10-19:
>Max. temp the device can according to the manual is 29,5°C. In case that's something you need to know as well.

I m using the standard one used for all termostats, min 5 max 32, So the API can accept 30 but the device will return 29.5 at maximum if it can go over.

So you need something more to code ?

synatis, 2021-10-19:
> So you need something more to code ?

Not sure :/ Other than that:

> * It seems like the temps are saved in Sun, Moon and Suitcase.
> * If I set it to "off" on the device - it switches to auto after 4-5 sec.

pillemats, 2021-10-19:
I have just ordered some of this and can support if you want! I can test or something else. Hopefully they will supported asap :-) 
Thanks for the great support team here and on discord!!!!

rikroe, 2021-10-20:
As just mentioned sun & moon usually only indicate that the current temperature (manual or not) is either set to the  high or low temperature set points. At least that's what all my other electric valves do.

I've also ordered some valves and can test hopefully on the weekend.

Would it make sense to also buy the Lidl Smart Home gateway? I.e. To sniff the traffic somehow? 
The options shown in this are basically scheduling high/low set points, offset, automatic/manual mode, vacation mode and some settings for open windows detection: https://youtu.be/LJ6zOvOaXoo (German video)

Smanar, 2021-10-20:
> It seems like the temps are saved in Sun, Moon and Suitcase.

And it s 'not good ?

> If I set it to "off" on the device - it switches to auto after 4-5 sec.

Here it s not my fault ^^, look at your log, you are setting the "off" mode, and it s the device itself that return to "auto", 

But according to the documentation "manual" and "auto" or not good too.
The "manu" mode is the "standard auto", you set 1 temperature and the device use it for regulation.
The "auto" mode is a custom mode that need 2 temperatures.
The "off" mode is totaly not existant.

Here it seem the guys use 60 and 0 value for "off" and "heat" https://github.com/Lavisx/lidl-valve-z2m/blob/main/lidl_radiator_valve.js

To keep a "standard" will be nice having same feature for all TRV
BTW if the "auto" use 2 temperatures, we are still missing one.

Manu use 0x02 0x010
Auto use 0x02 0x69 and another one.

>Would it make sense to also buy the Lidl Smart Home gateway? I.e. To sniff the traffic somehow?

No I don't think as we can have all command using the device manualy and looking in logs for command value.

rikroe, 2021-10-29:
Ok, did some further debugging (not sure if I'll be able to do any one the weekend) and I think I found the issue regarding `heat` instead of `off`. 

It really seems that a heatsetpoint = 0 is the culprit. I don't really now how to get around this. 
After changing all `heatsetpoint` related temperatures in rest_sensors.cpp and tuya.cpp to `500` (or `\x0A`), a config read via REST API shows a `heatsetpoint` of `500` and finally mode `off`.
Unfortunately the device **now shows 5.0° instead of `OFF`** as we had before with 0.

Probably thats something we have to take up with manup? Or who would be the best person to talk to?

Smanar, 2021-10-30:
About the c++, try with

>DBG_Printf(DBG_INFO, "Tuya debug heatpoint 3. mode: %s\n", qPrintable(mode));

Item is a class, not something displayable.

>DBG_Printf(DBG_INFO, "Tuya debug heatpoint 4. item2test: %s\n", item2test);

item2test is not a string but a bool, so perhaps using instead %d.

> 
> I can set a heatsetpoint of 500 and will get [{"success":{"/sensors/7/config/heatsetpoint":500}}] returned from REST API. TRV also shows 5.0.
> With heatsetpoint of 500 and will get an empty response (but still HTTP code 200) returned from REST API, and TRV shows 4.0. But when just looking at the sensor via REST (GEE /api/API_KEY/sensors/7), then heatsetpoint is still 500.

Here I m not sure to understand, on the first line you have a return and not on the second one for the same command "heatpoint" = 500.

And I have found some explanation, heatsetpoint is locked
>rItemDescriptors.emplace_back(ResourceItemDescriptor(DataTypeInt16, QVariant::Double, RConfigHeatSetpoint, 500, 3200));

And have similar problem here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5153

This device haven't "off", and too much mode possible.
So instead of trying to have same mode than classic TRV, I have proposed him to use the "config/preset".

rikroe, 2021-10-30:
Thanks, figured out the `qPrintable` by myself already :)

I agree with you that maybe `config/mode` isn't the right value and preset might be better suited.

I checked the display, basically the modes on the TRV are `auto`, `manu` and `away` (instead of `off`). So if we implement vacation mode at some point, that would be the third mode.

And for `OFF` or `BOOST` (the current `heat` mode) we could use `config/preset`. This also seems to be used by the guys at https://github.com/Lavisx/lidl-valve-z2m/blob/main/lidl_radiator_valve.js you mentioned already.

I will take a look next week, won't have time over the weekend.

Smanar, 2021-10-30:
>I will take a look next week, won't have time over the weekend.

NP, I can do it for monday, will be fast, if you are agree, I will remove the config/mode and instead use config/preset with all mode.

- manu : that is not manual but auto with 1 temperature
- program with moon and sun
- boost
- away/holiday 

(The "off" is not a real mode, and not specified on the manual so idk yet if we keep it, but if you still want it, better to keep it in config/mode)

And after that finish all mode, for exemple the away, I think the "program" is already able to work.

jonny-379, 2021-10-30:
Hi, I don't know if this might help. I looked the Lidl device up on the ZigBee library https://zigbee.blakadder.com/Lidl_TVR01.html it says it is compatible with ZIGBEE2MQTT. Maybe there are some information that might help to debug the values and function from the thermostat.

Smanar, 2021-10-30:
I have all values, it s not a problem, the problem is more how to integrate it. We can't ask to third app devs to rewrite all their code to make it compatible with all new device.

And trying to have "standard/genric" commands, used by all devices.

rikroe, 2021-10-31:
I think working via presets is a smart idea. However at least on the device, we would still need to set corrsponding `mode` values (and therefore also report them in REST API).

This is what I think we could need on the LIDL TRV (other TRVs might of course need different settings on device or don't expose everything):

| TRV | preset | mode | heatsetpoint |
| - | --- | --- | - |
Auto <br />(using   programmed schedule) | `auto` | `auto` <br />`mode` = 0 | from schedule |
Manu<br /> (manual heatsetpoint until next   scheduled timer) | `manual` |  `heat`<br />`mode` = 1 | set via REST API |
Vacation mode <br />(keep one temperature until vacation mode is <br />finished or preset switched back) | `away` |  `off`<br />`mode` = 2 | set via vacation schedule | 
OFF mode on device | `off` | `heat`| 0.0° | 
Boost mode |  `boost` | `heat`| 30°/50°

Smanar, 2021-10-31:
There not the "off" in preset ^^. This mode is on the config/mode.
And the device don't have preset and mode, it have only one of them.

When it s TRC with only heat+ off + auto > "mode"
When it s a on wall thermostat with 10 feature > "preset"
But here ...

So if I use your tab, I need to do

- preset = auto > send Tuya mode = 0 and set deconz mode to "auto"
- preset = manual > send Tuya mode =1 and set deconz mode to "auto"
- preset = away > send Tuya mode = 2 and set deconz mode to "auto"

- deconz mode = off > send Tuya mode = 1 and set temperature to 0°C
- deconz mode = heat > send Tuya mode = 1 and set temperature to 60°C
- deconz mode = auto > send Tuya mode = 1 and "make magic"

Except for the last situation all is clear for me.

rikroe, 2021-11-01:
Looks fine 👍 

On the last one I would also just go back to default scheduled mode:
> - deconz mode = auto > send Tuya mode = 0 and do not set temperature and preset to "auto"

Also, if a `heatsetpoint` is manually send to the TRV, it should switch to preset = `manual`

Smanar, 2021-11-01:
Ok so new version online, idk what can be the result.

Mode: on off auto
Preset: auto manual away

Trying to reproce the tab, have used 0°C and 60 °C for off and heat.

steff75, 2021-11-05:
Here is the complete instruction manual online as PDF, because the printed manual from the package is not complete.
[https://stesbintegrationprod.blob.core.windows.net/public/articlemanual/2f64c92c-6d3c-4d50-9c05-7442f5da6107.pdf](https://stesbintegrationprod.blob.core.windows.net/public/articlemanual/2f64c92c-6d3c-4d50-9c05-7442f5da6107.pdf)

I have now also bought 2 of these thermostats and compiled the deconz-rest-plugin from the lidl_trv branch. I'm not sure what the status is now, but currently when I change the heatsetpoint with the rest API, the device display value is not changed yet.

Thanks already for the progress so far with the thermostat integration!

Smanar, 2021-11-05:
>when I change the heatsetpoint with the rest API, the device display value is not changed yet

You mean the heat setpoint is not working ? Your device have same manufacture name ?

steff75, 2021-11-06:
Yes, you are right. In auto-mode I can change the heatsetpoint with the API. If I then switch to the preset "manual", then no more. The heat-mode sets the heatsetpoint to 60 °C and then also does not accept any change of the degree via the API in this mode.

I admit that the configuration with modes and presets still confuses me a bit with this thermostat. So far I had only one smart thermostat, the Danfoss Ally. There I am used to it, that this gets a heatsetpoint from the smart home and holds this temperature until it receives the next value.

What mode/preset combination is closest to that with the Silvercrest thermostat?

Smanar, 2021-11-06:
For the moment, not comfirmed yet, but

>         deconz preset = auto > send Tuya mode = 0 and set deconz mode to "auto"
>         deconz preset = manual > send Tuya mode =1 and set deconz mode to "auto"
>         deconz preset = away > send Tuya mode = 2 and set deconz mode to "auto"
> 
>         deconz mode = off > send Tuya mode = 1 and set temperature to 0°C and preset to "manual"
>         deconz mode = heat > send Tuya mode = 1 and set temperature to 60°C and preset to "manual"
>         deconz mode = auto > send Tuya mode = 1 and set temperature to "sun" temperature and preset to "manual"

So if I m right, if you set the preset = auto, it s the mode controlled by the schedule, not the set heatpoint.
It was possible previously to set the heaeatpoint, so it's possible, but If I remember it was disabled to use instead the sun and moon temperature

```
config/comfort_heatsetpoint
config/eco_heatsetpoint
```

Element2, 2021-11-06:
I believe the sun and moon temperature icons are only used to indicate any low and high temperature in a daily/weekly schedule. 
Low being a trivial (any) temperature which is lower then the high temperature, thus marked by a moon icon.
High being a trivial (any) temperature which is higher then the low temperature and thus marked by a high icon.
So the icons are schedule indications of high and low and NOT a specific value of temperature.
 
This schedule, including times and week days when it must heat high and when it must heat low can be programmed by hand on the device, using the buttons.
However, a schedule can also be run automatically in the LIDL app. I am not sure if this is the same schedule as can be programmed on the device self.

It would be interesting to know if the LIDL app 'sends' the schedule to the device and it then runs it as an automated program on the device (if so THAT would be a nice feature to have) or that the LIDL app at the scheduled times in the LIDL app send a high temp and low temp setting to the device. If that is the case, the device is basically a dumb thermostat device and the LIDL app does the work. 

The point of the above being. I believe the device is using two basic modes:
Manual: The api sends only 1 heatpoint. 
Programmed: The api sends a schedule with week days and times and high and low heatpoints and then let the device run it OR a schedule can only be programmed by hand on the device via the buttons and then runs on the device as a program.

rikroe, 2021-11-06:
I have a local version where I am using `preset` to build the boost/heat mode, therefore freeing heat alone to be activated when changing the temperature.

It's not finished, work got in the way but I should be able to get it some of it done this weekend and share with you for discussion. 

The problem really is that the options you have on the device don't align with what home automation systems expect.

Element2, 2021-11-06:
@rikoe: Agreed. To transfer a schedule to the device is not a high priority. But, if you want to control a lot of individual heaters (I have 10+) you need a lot of thermostat devices. In that case the api/appication is generating a lot of  traffic to manage all the individual schedules on the devices.

Smanar, 2021-11-06:
From the doc here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5349#issuecomment-947478023

- Manual ( tuya mode = 1) use only 1 heatpoint.
- Auto (tuya mode = 0) "the device regulate the temperature beetwwen 2 temperatures (Moon and Sun I presume)

You can set the program and it worl on stand alone mode.

>The problem really is that the options you have on the device don't align with what home automation systems expect.

Yeahh :(

>Also we need to figure out how to read the schedule from the device

I don't think this a problem, the code have already a base, and we can find the command used on google.
I m waiting for the rest is working before starting to put the schedule mode

rikroe, 2021-11-06:
Almost 😀 maybe putting emphasis on heat & off temperatures was not smart as this is not really how the TRV is working. 

**@all**: are OFF and BOOST modes needed? Or should they be excluded for now? 

On __auto__ (tuya mode 0) the device will work according to schedule. Sun & moon temperatures do not matter because you can schedule any temperature for a given weekday/time.

On __manual__ (tuya mode 1) the device will keep a specific temperature *until the next change from schedule*, changing back to auto.

Tuya mode 3 is vacation mode but  I think this is prio 3 after getting basic functionality + schedules working.

Smanar, 2021-11-06:
>are OFF and BOOST modes needed? Or should they be excluded for now?

Its 2 of the 3 modes used natively on third app ^^.

>On auto (tuya mode 0) the device will work according to schedule. Sun & moon temperatures do not matter because you can schedule any temperature for a given weekday/time.

Ha, So on this mode the device don't use them ?
So on wich one mode it use them ?

ChristianH21220, 2021-11-06:
think  Sun & moon (sun=standard heating temp / moon= Energie safe temp [night temp]) is used by the scheduler in the device.

As this device have a Button on the front of the devices, this can used to override the actice schedule (from the device) and use the selected heatpoint (sun or moon). At the next schedule (from the device)  the selected mode is changed to the Mode from the shedule (from the device) sun or moon.

If we have one heatoint that can changed by the controller (smarthome hub/third app) then there is no need to use the scheduler from the device. The third app/controller do the job for schedule und heatpoint change.
@Smanar: this singel heatpoint is the sun heatpoint - correct ?

A boost Mode is not needed (from my point of view) as the device don't have a boost Button like for example the eurotronic spirit TVR.

Vacation Mode is a nice Feature when the device scheduler is in use. Without the device scheduler this job can done as well by the controller/third App.

rikroe, 2021-11-06:
> Its 2 of the 3 modes used natively on third app ^^.

Not really in my opinion, as the TRV itself does much more than heat only (manual mode [tuya mode = 1] is keeping a temperature automatically) or just being off (vacation mode icon on the device instead of OFF for tuya mode = 2).

> this singel heatpoint is the sun heatpoint - correct ?

You can already set the current target temperature via the `heatsetpoint` config. This is different from the sun heatpoint.

Sun & moon temperatures are used as default values for schedule, but you can overwrite them when setting a schedule.
I'd keep them because they do no harm and one can quickly change between them by clicking a button on the device.
Basically if you press the button on the device, the `heatsetpoint` attribute will change to either `comfort_heatsetpoint` or `eco_heatsetpoint`.

Smanar, 2021-11-06:
So the next step is command to set the shedule for the "auto" mode ?

pillemats, 2021-11-06:
> **@ALL**: are OFF and BOOST modes needed? Or should they be excluded for now?
 
In my case off will be nice…. If its not possible to integrate now it is enough to temp low for „off“-mode

Hopefully to get work the trv in deconz because the 3rd app from lidl works terrible

Many thanks to all developers that work on that!!!

rikroe, 2021-11-06:
> So the next step is command to set the shedule for the "auto" mode ?

I'll try to check your changes on my TRV tonight. But from what you describe in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5349#issuecomment-962423760, I think we mixed some things up:

## modes
**tuya mode = 0**
TRV shows auto. When send via deconz, just change tuya mode, device should switch back to scheduled temperature on its own.

**tuya mode = 1**
TRV shows Manu. 3rd parties will show "heat". Used when setting a manual (i.e. non-schedule temperature). Usually not set via deconz but by changing `heatsetpoint` to some other temperature.

**tuya mode = 2**
TRV shows vacation mode. 3rd parties will show "off". Not set directly by setting mode but with a special vacation mode command that includes end date/time and temperature during vacation mode.

## presets
**eco** (or named **off**)
Set tuya mode = 1 and `heatsetpoint` = 0. We might need to check with manup if we can remove the heatsetpoint reset.

**boost**
Set tuya mode = 1 and `heatsetpoint` = 60. We might need to check with manup if we can remove the heatsetpoint reset.

That is the best understanding I have for this TRV, also based on what has been commented. Please correct me if you disagree.

@Smanar could you help implementing this logic? Thanks for your effort!

ChristianH21220, 2021-11-06:
@rikroe: thx for the explanation regarding the shedule and sun/moon (comfort/eco heatpoint).

Agree to the modes as you wrote 👍 but why the presets should set eco to heatsetpoint 0 ? This is not eco, this is off. 
Eco should be the eco point (energy-heat)
Off is separate  and like you wrote -> heatpoint:0 without heatpoint reset
[image]

Boost (full power):  fine but will this not better have a heatpoint reset after some minutes (maybe 5 min)  to either comfort_heatsetpoint or the former value of the heatsetpoint before boost was called. Because if the device don't handle this via own firmware setting , it will produce a bad surprise if boost stay permanent on.

rikroe, 2021-11-06:
The problem with the off mode you showed is that it the switch between the three is currently mapped to mode that can be one of auto, manual or vacation on the device itself.

@Smanar maybe could we change how preset and mode are handled on the device?

So tuya mode is mapped to deconz preset and vice versa?
Deconz preset would be automatic, manual & away (for tuya modes 0, 1, 2).
Deconz mode could be automatic, off, boost which set tuya mode 0 or 1 and a specific temperature for the latter two? 

Agree that boost mode would be best with a timer, but don't know if that is possible.

Smanar, 2021-11-06:
Yeah the timer is possible for boost from my memory, but it will another command, ATM we hare using heatpoint = 60°C and have see a boost command.

But tuya mode will be invisibles, just used by code. User need to see only deconz mode and deconz preset. Mode are not tuya mode, but deconz mode.
In my mind all presets are automatic, away, schedule or sun/moon are auto for me. So mode can be off/ boost  or auto (and auto mean enabling preset)

Note:
As we have a config/eco_heatsetpoint, can use this value with the "manu" preset for the preset=eco

rikroe, 2021-11-06:
I have updated the table below. I think tuya modes don't matter that much, except for the first 3 (as they map to actual modes on the device).

| TRV | deconz<br />preset | deconz<br /> mode | tuya <br />mode | heatsetpoint |
| - | --- | --- | - | - |
Auto <br />(using   programmed schedule) | `auto` | `auto` | 0 | from schedule or set preset or mode to auto |
Manu<br /> (manual heatsetpoint until next   scheduled timer) | `manual` |  `auto` | 1 | set by setting a temperature manually <br /> if no temp given, use sun temp
Vacation mode <br />(keep one temperature until vacation mode<br /> is finished or preset switched back) | `away` |  `auto` |  2 | set via vacation schedule command | 
Eco mode | `eco` | `auto` | 0 | use moon temp |
OFF mode on device | - | `off`  | 1 | 0.0° | 
Boost mode | - |  `heat`| 1 | 60° for 5 minutes

Some findings from your current version:
- Changing `heatsetpoint` to be visible on TRV only works if TRV is in Auto mode.
- Deconz mode `heat` set TRV to 60° and Manu, but still shows deconz mode `auto`. deconz preset is `manual`
- Deconz mode `heat` set TRV to OFF and Manu, but still shows deconz mode `auto`. deconz preset is `manual`
- When setting deconz mode back to `auto`, Manu is still shown on TRV but with current scheduled temperature. deconz preset is still `manual`

Smanar, 2021-11-07:
>Changing heatsetpoint to be visible on TRV only works if TRV is in Auto mode.

Was because I m using the setpointauto command only if deconz mode = auto.
Now I m using the setpoint auto if deconz mode = auto AND deconz preset = auto, else use standard setpoint command

>Setting deconz mode back to auto, seems not to work. Temperature is last temp before heat/off, but TRV still shows Manu. deconz preset is still manual

New version to test.

Have added "eco" preset, but I don't think it will be displayed in the API, it will change the mode to "auto" and set the temperature.

> Deconz mode heat set TRV to 60° and Manu, but still shows deconz mode auto. deconz preset is manual
> Deconz mode heat set TRV to OFF and Manu, but still shows deconz mode auto. deconz preset is manual

Here we have a problem ..
You set the device to mode = off, but after a time it will send a periodic report for his tuya mode = 1, and on the code I m using the second line of your tab.

Will try to set something in code, ignore all report for tuya mode if deconz mode is not "auto"

rikroe, 2021-11-07:
Please find the my testing results and some tuya debug logs here: https://gist.github.com/rikroe/c06efa1f9c50147bd1c53db6f1866635#file-tuya_debug_2021-11-17t15_with_comment
Full tuya debug logs (just grepped 'Tuya debug' to exclude all other spam): https://gist.github.com/rikroe/c06efa1f9c50147bd1c53db6f1866635#file-tuya_debug_2021-11-17t15_full

Unfortunately, reporting for OFF and boost is still broken. 

> Will try to set something in code, ignore all report for tuya mode if deconz mode is not "auto"

This will make it impossible to change mode on the device itself right (i.e. changing back from manu or vacation mode to auto on device keys)? I really think that we have to rely on the heatsetpoint, otherwise it will never be valid all the time.

Also, when changing modes, temperature on the TRV changes depending on the mode. But it seems the new temp data isn't send so I think we have to poll the temperature after tuya mode change.

> Have added "eco" preset, but I don't think it will be displayed in the API, it will change the mode to "auto" and set the temperature.

Maybe automatically set mode "eco" in deconz if TRV temp = eco_heatsetpoint?

Smanar, 2021-11-07:
>This will make it impossible to change mode on the device itself right (i.e. changing back from manu or vacation mode to auto on device keys)?

Arf, yep, right.

>I really think that we have to rely on the heatsetpoint, otherwise it will never be valid all the time

But the problem is the device send 2 reports, one for the temperature with setpoint = 0 °C, so ok its fine but later it send only tuya mode = 1
Can use a special preset = "forced" that can be set only by the device, and not the API to remember thoses 2 special mode ?

>Maybe automatically set mode "eco" in deconz if TRV temp = eco_heatsetpoint?

Yep but will have same problem than for "off" and "heat" when the device make a report just with tuya mode.

>Also, when changing modes, temperature on the TRV changes depending on the mode. But it seems the new temp data isn't send so I think we have to poll the temperature after tuya mode change

Yes I m reading your logs ...
How many temperature can be memorised by the device for different mode ? 1 heat Setpoint and 1 Auto Setpoint ( I say that because the device realy use 2 working mode for them) ?
Someone have found a way to trigger a report (We can't pool tuya cluster) using the GUI with the tuya cluster, with mimic the request.
```
status 0
transid 0
dp = 0x0269 or 0x0210
fn = 0
and that you want as data.
```

Else the tuya application can work with that, so it s possible, we are already storing eco and comfort temperature, can store too 2 heatpoints.

rikroe, 2021-11-07:
> But the problem is the device send 2 reports, one for the temperature with setpoint = 0 °C, so ok its fine but later it send only tuya mode = 1 Can use a special preset = "forced" that can be set only by the device, and not the API, to remember thoses 2 special mode ?

Maybe in all 3 tuya attributes (heatsetpoint, mode, preset) we have to check against all 3 deconz values (again heatsetpoint, mode, preset) to figure out what is the correct combination in the end? 
So create a function that is always called when one of the three attributes from the TRV is changed?

> How many temperature can be memorised by the device for different mode ?

I think 3: 1 for current value of auto (which gets overwritten from schedule), 1 for current value of manu and 1 for current value of vacation.
When changing through the TRV modes, different memorized temperatures are displayed.

> Someone have found a way to trigger a report (We can't pool tuya cluster) using the GUI with the tuya cluster, with mimic the request.

Is there any way to test this? I.e. to be able to send a command like this via REST API?

Smanar, 2021-11-07:
Why it so complicated, was so easy on his code https://github.com/Lavisx/lidl-valve-z2m/blob/main/lidl_radiator_valve.js
Perhaps he manage the TRV only with code and never manualy.

>So create a function that is always called when one of the three attributes from the TRV is changed?

But the problem is they are not changed in same time.

>1 for current value of vacation

Lol, where this one is from ? I don't see it in tuya command. Perhpas it s from the vacation program ?

But if I m right the device make report, so even if the temperature is not updated in the API when changing the mode manualy, it will be updated after some time during the periodic report ?

>Is there any way to test this? I.e. to be able to send a command like this via REST API?

He have do that using the GUI, I can reproduce the request with the code, but I don't remember wich one command he have used.

rikroe, 2021-11-07:
Checked the Javascript code too and I think it should be quite the same easy way for us.
- OFF and BOOST mode are only set via deconz and not on TRV. Receiving `0x0210` with either 0 or 120 will not update the deconz temperature, but just set deconz mode (and deconz preset for `boost`). 0 or 60° are not set to API, heatsetpoint stays.
- If either 0x0210 (Manu), 0x0269 (Auto) or the AwayTemp (to be figured) arrive, all three will overwrite deconz heatsetpoint.
- Tuya mode switches set both deconz mode and deconz preset.
- There is no `eco` preset

Smanar, 2021-11-07:
>will not update the deconz temperature

It s already the case no ?

>but just set deconz mode (and deconz preset for boost)

mode = "heat" or "off" but for preset ?

>There is no eco preset

yes, I think the "away" preset is enought

>Maybe we just have to poll different attributes depending on mode changed

Not possible poll Tuya attribute :( It s for that the user have find this command to force the TRV making a report.

rikroe, 2021-11-07:
sending mode = off will result in TRV display OFF, tuya mode 1, 0x0210 = 0 deconz mode auto and deconz preset manual.
sending mode = auto will result in TRV diplay 60.0, tuya mode 1, 0x0210 = 120, deconz mode auto and deconz preset manual.

Also, we should use a heatsetpoint of 30.0° for mode heat, as the TRV will display ON when on 30. When using the wheel on the device it has a range of 0 (OFF) to 30 (ON).

The Javascript sets the following for 60° (we should use 30°): mode `heat` and preset `boost`. 
mode `heat` can also be set by the manu mode on TRV but if the temperature is 30°, then it still is preset `boost`. 

The same goes for mode `off`. If device temperature reports 0°, then mode is `off`, doesn't matter which tuya mode it is in. 

I tried around using deconz GUI to poll. But I am always setting some default values on the device. It sometimes looks as if it works but then next time it doesn't. :(

Smanar, 2021-11-08:
You have some logs to check for command order ?

If we have first 0x0210 = 0
deconz mode need to be set to "off"
So if after we have 0x0402 = 0 It need to be skipped by
```

                            //Do the device is in "auto" mode
                            item = sensorNode->item(RConfigMode);
                            if (item && item->toString() != QLatin1String("auto"))
                            {
                                return;
                            }
```

Or on log you can see websocket notification, so you can see if the config/mode is modified (even it s just for 0.5s)
Because I don't see what is bad on the code ....

rikroe, 2021-11-08:
You are right, deconz mode will be set to `off` for a short period of time (starting at 21:43:46:311). It is reset at 21:43:52:045 but only receives tuya mode 1 (for manu) or temperature 0 before from the device. Hope this log helps a little. 

```
21:43:45:310 poll node 00:0b:57:ff:fe:4f:d0:6a-01
21:43:45:310 Poll light node Esszimmer 1
21:43:45:359 read attributes of 0x000B57FFFE4FD06A cluster: 0x0006: [ 21:43:45:360 0x0000 21:43:45:360 ]
21:43:45:361 add task 451758 type 19 to 0x000B57FFFE4FD06A cluster 0x0006 req.id 236
21:43:45:361 Poll APS request 236 to 0x000B57FFFE4FD06A cluster: 0x0006
21:43:45:460 Poll APS confirm 236 status: 0x00
21:43:45:461 Erase task req-id: 236, type: 19 zcl seqno: 58 send time 0, profileId: 0x0104, clusterId: 0x0006
21:43:45:492 Node data 0x000b57fffe4fd06a profileId: 0x0104, clusterId: 0x0006
21:43:45:493 0x000B57FFFE4FD06A: update ZCL value 0x01/0x0006/0x0000 after 0 s
21:43:46:262 don't create binding for attribute reporting of sensor Thermostat 7
21:43:46:305 Send Tuya request 0x0C4314FFFE73C758 : Dp_type: 0x04, Dp_identifier 0x02, data: 01
21:43:46:306 add task 451764 type 37 to 0x0C4314FFFE73C758 cluster 0xEF00 req.id 243
21:43:46:306 Send Tuya request 0x0C4314FFFE73C758 : Dp_type: 0x01, Dp_identifier 0x10, data: 00000000
21:43:46:307 add task 451765 type 37 to 0x0C4314FFFE73C758 cluster 0xEF00 req.id 244
21:43:46:307 delay sending request 244 dt 0 ms to 0x0C4314FFFE73C758, ep: 0x01 cluster: 0xEF00 onAir: 1
21:43:46:308 delay sending request 244 ep: 0x01 cluster 0xEF00 to 0x0c4314fffe73c758 onAir 1
21:43:46:308 delay sending request 244 dt 0 ms to 0x0C4314FFFE73C758, ep: 0x01 cluster: 0xEF00 onAir: 1
21:43:46:309 delay sending request 244 ep: 0x01 cluster 0xEF00 to 0x0c4314fffe73c758 onAir 1
21:43:46:310 delay sending request 244 dt 0 ms to 0x0C4314FFFE73C758, ep: 0x01 cluster: 0xEF00 onAir: 1
21:43:46:310 delay sending request 244 ep: 0x01 cluster 0xEF00 to 0x0c4314fffe73c758 onAir 1
21:43:46:311 Websocket 172.17.0.1:37852 send message: {"config":{"battery":100,"comfort_heatsetpoint":2200,"eco_heatsetpoint":1700,"heatsetpoint":2150,"locked":false,"mode":"off","offset":0,"on":true,"preset":"auto","reachable":true,"schedule":{},"schedule_on":false},"e":"changed","id":"7","r":"sensors","t":"event","uniqueid":"0c:43:14:ff:fe:73:c7:58-01-0201"} (ret = 308)
21:43:46:312 Websocket 192.168.91.5:57049 send message: {"config":{"battery":100,"comfort_heatsetpoint":2200,"eco_heatsetpoint":1700,"heatsetpoint":2150,"locked":false,"mode":"off","offset":0,"on":true,"preset":"auto","reachable":true,"schedule":{},"schedule_on":false},"e":"changed","id":"7","r":"sensors","t":"event","uniqueid":"0c:43:14:ff:fe:73:c7:58-01-0201"} (ret = 308)
21:43:46:410 delay sending request 244 dt 0 ms to 0x0C4314FFFE73C758, ep: 0x01 cluster: 0xEF00 onAir: 1
21:43:46:411 delay sending request 244 ep: 0x01 cluster 0xEF00 to 0x0c4314fffe73c758 onAir 1
21:43:46:501 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x3808
21:43:46:509 delay sending request 244 dt 0 ms to 0x0C4314FFFE73C758, ep: 0x01 cluster: 0xEF00 onAir: 1
21:43:46:510 delay sending request 244 ep: 0x01 cluster 0xEF00 to 0x0c4314fffe73c758 onAir 1
21:43:46:514 poll node 7c:b0:3e:aa:00:ad:5e:50-03
21:43:46:515 Poll light node Osram Steckdose
21:43:46:563 Poll APS request to 0x7CB03EAA00AD5E50 cluster: 0x0006 dropped, values are fresh enough
21:43:46:609 delay sending request 244 dt 0 ms to 0x0C4314FFFE73C758, ep: 0x01 cluster: 0xEF00 onAir: 1
21:43:46:610 delay sending request 244 ep: 0x01 cluster 0xEF00 to 0x0c4314fffe73c758 onAir 1
21:43:46:710 delay sending request 244 dt 1 ms to 0x0C4314FFFE73C758, ep: 0x01 cluster: 0xEF00 onAir: 1
21:43:46:711 delay sending request 244 ep: 0x01 cluster 0xEF00 to 0x0c4314fffe73c758 onAir 1
21:43:46:809 delay sending request 244 dt 1 ms to 0x0C4314FFFE73C758, ep: 0x01 cluster: 0xEF00 onAir: 1
21:43:46:810 delay sending request 244 ep: 0x01 cluster 0xEF00 to 0x0c4314fffe73c758 onAir 1
21:43:46:910 delay sending request 244 dt 1 ms to 0x0C4314FFFE73C758, ep: 0x01 cluster: 0xEF00 onAir: 1
21:43:46:911 delay sending request 244 ep: 0x01 cluster 0xEF00 to 0x0c4314fffe73c758 onAir 1
21:43:47:009 delay sending request 244 dt 1 ms to 0x0C4314FFFE73C758, ep: 0x01 cluster: 0xEF00 onAir: 1
21:43:47:010 delay sending request 244 ep: 0x01 cluster 0xEF00 to 0x0c4314fffe73c758 onAir 1
21:43:47:110 delay sending request 244 dt 1 ms to 0x0C4314FFFE73C758, ep: 0x01 cluster: 0xEF00 onAir: 1
21:43:47:110 delay sending request 244 ep: 0x01 cluster 0xEF00 to 0x0c4314fffe73c758 onAir 1
21:43:47:210 delay sending request 244 dt 1 ms to 0x0C4314FFFE73C758, ep: 0x01 cluster: 0xEF00 onAir: 1
21:43:47:210 delay sending request 244 ep: 0x01 cluster 0xEF00 to 0x0c4314fffe73c758 onAir 1
21:43:47:310 delay sending request 244 dt 1 ms to 0x0C4314FFFE73C758, ep: 0x01 cluster: 0xEF00 onAir: 1
21:43:47:311 delay sending request 244 ep: 0x01 cluster 0xEF00 to 0x0c4314fffe73c758 onAir 1
21:43:47:410 delay sending request 244 dt 1 ms to 0x0C4314FFFE73C758, ep: 0x01 cluster: 0xEF00 onAir: 1
21:43:47:410 delay sending request 244 ep: 0x01 cluster 0xEF00 to 0x0c4314fffe73c758 onAir 1
21:43:47:510 poll node 00:0b:57:ff:fe:4c:01:f4-01
21:43:47:511 Poll light node Wohnzimmer 1
21:43:47:511 delay sending request 244 dt 1 ms to 0x0C4314FFFE73C758, ep: 0x01 cluster: 0xEF00 onAir: 1
21:43:47:511 delay sending request 244 ep: 0x01 cluster 0xEF00 to 0x0c4314fffe73c758 onAir 1
[... cut ...]
21:44:01:100 read attributes of 0x000B57FFFE4FD06A cluster: 0x0006: [ 21:44:01:101 0x0000 21:44:01:101 ]
21:44:01:102 add task 451840 type 19 to 0x000B57FFFE4FD06A cluster 0x0006 req.id 104
21:44:01:102 Poll APS request 104 to 0x000B57FFFE4FD06A cluster: 0x0006
21:44:01:151 Poll APS confirm 104 status: 0x00
21:44:01:152 Erase task req-id: 104, type: 19 zcl seqno: 76 send time 0, profileId: 0x0104, clusterId: 0x0006
21:44:01:199 Node data 0x000b57fffe4fd06a profileId: 0x0104, clusterId: 0x0006
21:44:01:200 0x000B57FFFE4FD06A: update ZCL value 0x01/0x0006/0x0000 after 0 s
```

Smanar, 2021-11-09:
I realy don't understand

```
21:43:46:311 Websocket 172.17.0.1:37852 send message: {"mode":"off","preset":"auto"}
21:43:51:787 Tuya debug 5 : Status: 0 Transid: 2 Dp: 1026 (0x04,0x02) Fn: 0 Data 1
21:43:52:042 Tuya debug 5 : Status: 0 Transid: 16 Dp: 528 (0x02,0x10) Fn: 0 Data 0
21:43:52:045 Websocket 172.17.0.1:37852 send message: {"mode":"auto","preset":"auto"}
```

on 0x02 0x10 The value can't move because temp = 0
on 0x04 0x02 The vaue can't move because mode = "off"

And there is nothing both state.

>21:43:46:306 Send Tuya request 0x0C4314FFFE73C758 : Dp_type: 0x01, Dp_identifier 0x10, data: 00000000

Have corrected that, the type was not good, we need 0x02 0x10.
Have set the heat temperature to 30°C
Have added some more debug line.

I realy hope it will be solved, bored, I prefer make the schedule feature.

rikroe, 2021-11-10:
Can totally understand you being bored. Unfortunately it didn't help. Please find the logs here:
- [heat log](https://gist.github.com/rikroe/d187e44e39e29db2e32f947414d641ca#file-tuya_debug_2021-11-10_heat-log)
- [off log](https://gist.github.com/rikroe/d187e44e39e29db2e32f947414d641ca#file-tuya_debug_2021-11-10_off-log)

If you prefer to work on schedule first, I have set a schedule on the device and received the following on Tuya debug:
https://gist.github.com/rikroe/d187e44e39e29db2e32f947414d641ca#file-schedule

Smanar, 2021-11-10:
Haaaaaa ^^
I think I have found the problem

```
                            QString mode = QLatin1String("auto");
                            if (temp == 0) { QString mode = QLatin1String("off"); }
                            if (temp == 6000) { QString mode = QLatin1String("heat"); }
```

I re-create a new variable mode.

Code is online, after your validation I will update the whole code to the last official and start to retreive the schedule in the schedule field.

rikroe, 2021-11-11:
So I just quickly tested your code (and think I found out how to filter only relevant log entries - basically only lines with hex device ID, Tuya or unique_id. Let me know if you need more).
`docker logs -f deconz | grep --line-buffered -E 'Tuya|0x0C4314FFFE73C758|0c:43:14:ff:fe:73:c7:58-01-0201'`

OFF does work 👍 Auto still works 👍 Unfortuately, heat still jumps back to Auto :(
Logs for full switch (Heat > Auto > Off > Auto) can be found here: https://gist.github.com/rikroe/d187e44e39e29db2e32f947414d641ca#file-tuya_debug_2021-11-11_heat_auto_off_auto-log

Smanar, 2021-11-11:
NP, it s corrected, have updated the code with last official.

Starting the make the schedule

rikroe, 2021-11-11:
Yes, `mode` `heat`, `off` and `auto` are working now! Cool! 👍 

I do get a schedule back, however only the first two time points instead of 5. Also it seems as if the decimal places in `heatsetpoint` are ignore. 
Maybe they also should be `1800` and `2150` instead? Also not sure about `schedule_on`, maybe I have to wait for the TRV to automatically go to the next scheduled time.

```json
"schedule": {
    "W111": [
        {
            "heatsetpoint": 18,
            "localtime": "T06:00"
        },
        {
            "heatsetpoint": 21,
            "localtime": "T12:45"
        }
    ]
},
"schedule_on": false
```

Logs are here: https://gist.github.com/rikroe/a2e21d8f64d09bb0bc0707b333343a69

steff75, 2021-11-12:
> * it seems as if the decimal places in `heatsetpoint` are ignored. Maybe they also should be `1800` and `2150` instead?

Yes, in other examples I have seen only four-digit values so far

> * what does `W111` stand for?

[https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/tuya.h](url)

> // Monday = 64, Tuesday = 32, Wednesday = 16, Thursday = 8, Friday = 4, Saturday = 2, Sunday = 1
            // If you want your schedule to run only on workdays, the value would be W124. (64+32+16+8+4 = 124)
            // The API specifies 3 numbers, so a schedule that runs on Monday would be W064.
            //
            // Workday = W124
            // Not working day = W003
            // Saturday = W002
            // Sunday = W001
            // All days = W127

This should be all days except Wednesday. The [documentation for thermostat schedules](https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/sensors/#supported-config-attributes_1) is unfortunately still missing.

Thanks for your work on the thermostat!

Smanar, 2021-11-12:
Lol another missing documentation ?
From my memory it s a normalized codification rule, but I don't find it again. Or was just from deconz and the first schedule code ...

"schedule_on" is not used yet on this device, you need it ?

Have made some correction and add a debug line because you have some strange days, like W15 ...

[109 more replies]