dresden-elektronik/deconz-rest-plugin #5351: Device name

grrrmml, 2021-10-05:
## Device

- Product name: STARKVIND Air purifier (E2007 and E3007 [Table integrated])
- Manufacturer: IKEA of Sweden
- Model identifier: STARKVIND Air purifier
- Device type : 
  - Sensor: Air Quality (unknown if exposed and readable)
  - Other: Fan Control: Off/Low/Medium/High/On/Auto/Smart ( on the device, there is a wheel with levels Auto/1-5 and On/Off [push] )

## Screenshots
[image]

[image]

### Basic
[image]

### Groups
[image]

### Scenes
[image]

### Fan Control
[image]

### Unknown (FC57)
### Unknown (FC7D)
 nothing visible for both

dusopn, 2021-10-16:
> Waiting for this device request until I buy one 👍

Exactly me too 👍

sjorge, 2021-11-10:
I've been poking this device using z2m, sadly no gateway to compare but I did manage to get something out of 0xfc7d

This is the cluster definition I added to z2m, I'm not 100% sure PM25 but it's extremely likely. It follows the same attribute layout as far as I could tel for the Ember zigbee additional measurement clusters.

```
    manuSpecificIkeaPM25Measurement: {
        ID: 0xfc7d,
        manufacturerCode: manufacturerCode_1.default.IKEA_OF_SWEDEN,
        attributes: {
            measuredValue: { ID: 0x0000, type: dataType_1.default.uint16 },
            measuredMinValue: { ID: 0x0001, type: dataType_1.default.uint16 },
            measuredMaxValue: { ID: 0x0002, type: dataType_1.default.uint16 },
            measuredTolerance: { ID: 0x0003, type: dataType_1.default.uint16 },
        },
        commands: {},
        commandsResponse: {},
    },
```

Reading this gets the following readReponse:
```
Zigbee2MQTT:info  2021-11-10 21:23:22: Read result of 'manuSpecificIkeaPM25Measurement': {"measuredValue":2287,"measuredMinValue":0,"measuredMaxValue":259200,"measuredTolerance":0}
```

measuredValue probably needs a `/100` I think as I had it in auto made and it went to the lowest speed so I don't think it is considering the air dirty and 2287ug/m3 would be extremely high if I am not mistaken.

I was unable to configure reporting on this, but I might be doing something wrong.

I sadly do not own the ikea gateway to compare it with the values in the tradfri app.

ebaauw, 2021-11-14:
I received the Starkvind (standing model) I ordered and paired it to the TRAFDR hub without any issues.  It uses regular pairing, so I have finally captured the hub's network key, enabling me to sniff the Zigbee communication between the Starkvind and the hub.  The bastards are using channel 25, same as my production deCONZ network, but nothing that filerting on PANID cannot solve.

The hub seems to control the air purifier exclusively by writing to manufacturer-specific attributes in the 0xFC7D cluster.  The Strakvind sends attribute reports in addition to the _Write Attributes Response_.  The _Report Attributes_ is answered by the Hub with a _Default Response_.  The hub queries the _Scenes_ cluster.

So far, I've found out the following attributes.

Attribute | Data Type | Use
-- | -- | --
0x0003 | bool | Disable LEDs
0x0004 | u16 | Air Quality.  0xffff when the purifier is off
0x0005 | bool | Lock Controls
0x0006 | u8 | Target Mode | Off: 0 (0x00) or fixed fan speed between 10 and 50 (0x0a - 0x32)
0x0007 | u8 | Current Mode | Same values as 0x0006

_Target Mode_ takes the following values:

Value | Meaning
-- | -- 
0x00 | Off
0x01 | Auto
0x0a - 0x32 | Fixed speed, between 10 and 50, corresponding to the LEDs on the Starkvind

_Current Mode_ reports the same values, except for 0x01.  The Smart Home app and the dial on the purifier only allow setting 10, 20, 30, 40, 50, but I've also seen _Current Mode_ being reported as 25.  The is also reflected in the app.

I've observed the following values for _Air Quaily_:

Value | Meaning
-- | --
0x0002 | Good.
0xFFFF | No data available, because device is off.

I failed to capture resetting the filter date, but otherwise the filtering logic seems to be handled by the hub.

The hub exposes the purifier to HomeKit natively, as a single accessory with an _Air Quality Sensor_ and an _Air Purifier_ service.  It only uses standard HomeKit characteristics, and doesn't expose disabling the LED.  _Lock Physical Controls_ isn't supported by Home, but works from Eve (which display it as _Child Lock_).

I'll update `general.xml` with these findings, so we can control the purifier from the GUI.  For REST API support, we need to make from hard decisions:
- I think the actuator function should be exposed as a `/lghts` resource, since the device supports (and seems to use) groups and scenes.  However, historically, we made the distinction between target and current modes using `config` vs `state` attributes.  Current mode could be exposed as `state.speed`, which is already used for the Hampton Bay fan controller.  I don't think we already expose other air purifiers?  Haven't figured out yet what attributes the _Scenes_ cluster supports, let alone how to support these in the API.
- I think the air quality sensor should be exposed as a ZHAAirQuality `/sensors` resource, but without `state.airqualityppb`, as the purifier doesn't seem to report the raw value.

ebaauw, 2021-11-14:
[image]

SwoopX, 2021-11-14:
I'd say so. You might want to check this out for further reference

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/3338039cb8a498f90f5000f0b118ded7ef963a84/utils/utils.cpp#L346-L358

ebaauw, 2021-11-14:
Yeah, saw that.  I have no other air quality sensor to compare, but it would fit the u16.  Maybe I should light a celebratory cigar… Been eyeing the Eve Room, but want a version supporting Thread before spending that amount of money.

ebaauw, 2021-11-14:
The Starkvind accepts, but ignores the _Identify_ command.  It has implemented _Trigger Effect_, but not all values.  _Blink_ and _Breathe_ work, and trigger the LED on the pairing button.

michaelnimbs, 2021-11-14:
Hello, I implemented the Starkvind support for the pytradfri library for home-assistant mentioned above in this thread:
[https://github.com/home-assistant-libs/pytradfri/issues/350#issue-1013418968](https://github.com/home-assistant-libs/pytradfri/issues/350#issue-1013418968)
I can confirm that it shows the raw value for the air quality sensor:
```
Indicators for air quality (PM 2.5):
Green: 0-35 / Good
Amber: 36-85 / OK
Red: 86- / Not good
```
I confirmed this by blowing out some candles and letting the smoke get inside the Starkvind, the value went up above 200 and the Ikea app showed the correct corresponding states and the fan went to full power.

ebaauw, 2021-11-14:
Thanks, @michaelnimbs.  Do you happen to know in which unit the Starkvind reports the air quality, or is this just a raw sensor value.  Most air quality sensor I’ve seen (as in: read about - I don’t have any) use five levels of quality instead of three, and I’d like to do the same, exposing the Starkvind through the deCONZ API.

> I confirmed this by blowing out some candles and letting the smoke get inside the Starkvind

That is a much better approach than picking up smoking!  I was already tempted to light a cigar…

michaelnimbs, 2021-11-14:
@ebaauw Maybe μg/m3, but I don't know for sure.

grrrmml, 2021-11-14:
First of all, thanks to everybody for the work so far.
The unit should be μg/m3, and I found following values (for PM2.5 and comparison to PM10).
This is my second air quality measuring device, but first with such values.

Maybe following table helps (taken from a texas instruments datasheet).

PM2.5 			PM10		AQI			AQI Category
0.0 to 12.0 	0 to 54 	0 to 50 	Good
12.1 to 35.4 	55 to 154 	51 to 100 	Moderate
35.5 to 55.4 	155 to 254 	101 to 150 	Unhealthy for sensitive groups
55.5 to 150.4 	255 to 354 	151 to 200 	Unhealthy
150.5 to 250.4 	355 to 424 	201 to 300 	Very unhealthy
250.5 to 350.4 	425 to 504 	301 to 400 	Hazardous
350.5 to 500.4 	505 to 604 	401 to 500 	Hazardous

SwoopX, 2021-11-14:
Whoever might be interested in this, some further reading on VOC levels https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/9_Gas_Sensors/Sensirion_Gas_Sensors_SGP3x_TVOC_Concept.pdf

Interestingly, the danish company Develco seems to have chosen the german approach, which is the current implementation 🤔

ebaauw, 2021-11-14:
This is all new to me.  What I understand from https://www.epa.gov/pm-pollution/particulate-matter-pm-basics and https://www.wkcgroup.com/tools-room/micrograms-per-cubic-meter-parts-per-billion-converter/ and @SwoopX 's link above.

There's different ways of how to measure a pollutant:
- By **density** (weight per volume) of the pollutant in µg/m³ (microgram per cubic meter) or mg/m³;
- By **level** (as part of the volume) of the pollutant in ppb (parts per billion) or ppm (parts per million);
- You cannot generally convert density (µg/m³) to level (ppb) or vice versa, as this is dependent on the specific weight of the pollutant and the temperature (high school physics: pV = nRT).  You can only convert for a given pollutant and at a given temperature (at standard air pressure);

There's different types of pollutants:
- PM2.5 and PM10 look at particulate matter (dust), PM2.5 looks at fine inhalable particles, with a diameter of 2.5µm; PM10 looks at inhalable particles, with a diameter of 10µm;  In think these are usually measured by density;
- VOC looks at Volatile Organic Compounds (gasses).  TVOC is all VOC gasses combined.  This seems to be measured by density as well.
- Other harmful gasses like CO (carbon monoxide) are usually measures by level.

I suppose depending on what pollutant you're looking at (PM2.5, PM10, TVOC, CO, ...), different density and level thresholds apply for computing air quality.  HomeKit supports different characteristics for the different pollutants, explicitly specifying density vs level in the name, see "Air Quality and Smoke Detection" on https://developer.apple.com/documentation/homekit/hmcharacteristic/characteristic_types, but the TRADFRI hub only exposes _Air Quality_ to HomeKit.  Without knowing what pollutant(s) the sensor in the Starkvind actually measures and how (level vs density), the value reported in attribute 0x0004 cannot be compared with values reported by other sensors.

grrrmml, 2021-11-14:
@ebaauw Taking your observation under regular conditions ("Air Quality now reports 7") and michaelnimbs "letting the smoke get inside the Starkvind, the value went up above 200" > that corresponds quite well with the figures I added above.
0.0 to 12.0 > Good
150.5 to 250.4 > Very unhealthy (when smoking candles for 24hours/day)

(Unfortunately I cannot contribute much more at the moment. I installed the deconz beta, and compiled and replaced the rest-api on my RPi4, but cannot yet get it working, even after deleting the device and re-adding. I will maybe try again after PR is merged. If I miss something important, I'd be happy for a hint...)

ebaauw, 2021-11-15:
You need to copy the `general.xml` file from my PR to `/usr/share/deCONZ/zcl` and restart deCONZ.  I haven’t yet made any changes to the REST API plugin code.

ebaauw, 2021-11-15:
So we need a `state.pm2_5` to expose the value of 0x0004 (in µg/m³).

There seem to be multiple standards of Air Quality, with different number of levels, names for those levels, and thresholds for the PM2.5 density:
- The current implementation by @SwoopX follows the **German** standard from the link in his other comment, with values excellent, good, moderate, poor, and unhealthy.  I haven't been able to find a mapping of PM2.5 to these levels;
- The **US** standard uses six levels, as mentioned by @grrrmml.  The mapping threshold mentioned there are consistent with https://www.airnow.gov/aqi/aqi-calculator-concentration/.
- The **EU** standard uses six levels: good, fair, moderate, poor, very poor, and extremely poor.  A mapping table is on https://www.eea.europa.eu/themes/air/air-quality-index (click on **Legend Explained**).
- HomeKit uses five levels: excellent, good, fair, inferior, poor.  It also adds an unknown value (which makes sense, as the Starkvind reports 0xFFFF this when it's off).

I would propose to stick to the current implementation and use the EU mapping, so we would get:

PM2.5 | EU Index | `state.airquality`
-- | -- | --
0-10 | Good | `excellent`
10-20 | Fair | `good`
20-25 | Moderate | `moderate`
25-50 | Poor | `poor`
50-75 | Very poor | `unhealthy`
75-1000 | Extremely poor | `out of scale`
65535 | -- | `unknown`

We might want to rename `out of scale` to `hazardous`.

SwoopX, 2021-11-17:
@ebaauw I guess we then should have different rating implementations that can be chosen? Maybe something like `QString getAirQualityStringPm25(quint16 level) ` (note to myself: the data type should be corrected to quint16 in the current implementation)

I'm currently playing with esposing dedicated DDF parsing functions for air quality. Works out quite fine so far, but still need to align with manup on the implementation. Based on the above, we probably need 2 functions to choose from...

ebaauw, 2021-12-09:
Ping

ebaauw, 2021-12-09:
@Mimiix, Why doesn't the bot remote the stale label?

Mimiix, 2021-12-09:
@ebaauw It checks once a day at 3 AM. So it removes it tonight :)

yurgh, 2021-12-14:
Came here from this thread: https://forum.phoscon.de/t/how-to-add-starkvind-air-cleaner/800 
I am finally running the community docker deCONZ 2.13.03. The Starkvind is paired and networking nicely with my other devices,  but I'm still a bit lost on how to expose it to Home Assistant? Is that not possible at the moment?

nicoscholl, 2021-12-14:
@yurgh how did you add it to the deconz? It not a light nor a switch or a sensor

yurgh, 2021-12-14:
> @yurgh how did you add it to the deconz? It not a light nor a switch or a sensor
Using VNC. One way is to enable Control from the Panel menu, open the Pairing tab and click "enable permit join". Then hold the pairing button on the Starkvind for a few seconds until it pulses. It will then show up as a light resource.
[image]

ebaauw, 2021-12-14:
Currently it's visible only in the GUI and not exposed over the API.

Still not decided how best to expose it by the API.  It seems natural to expose the PM2.5 level as ZHAAirQuality sensor (with a new attribute for PM2.5 instead of VOC).  As it supports grouping, a `/lights` resource seems appropriate for the fan, but as it supports target vs actual speed, a `/sensors` resource seems more appropriate.  Since both the fan control and PM2.5 level are on the same cluster, a single resource/subdevice seems natural.

MTrab, 2021-12-25:
Any news? Just godt one for christmas, and would love to integrate it to my Home Assistant

rowdy238, 2021-12-27:
Is the device already supported and if not, can it be expected in the next version?
I got one for Christmas...
I would like to thank you in advance for the work you have done.

Ascariota, 2022-01-04:
@yurgh : I use Home Assistant also, how did you do to add to Home Assistant interface ?
Thx

> @yurgh how did you add it to the deconz? It not a light nor a switch or a sensor
> Using VNC. One way is to enable Control from the Panel menu, open the Pairing tab and click "enable permit join". [...]

yurgh, 2022-01-05:
> @yurgh : I use Home Assistant also, how did you do to add to Home Assistant interface ? Thx
>
I never got it to show up in Home Assistant using Deconz. It need to be supported by the REST-interface first, which I think is underway.

Personally I just used the opportunity to try out zigbee2mqtt, where Starkvind is supported

BlankStarBE, 2022-01-25:
> Is the device already supported and if not, can it be expected in the next version?
> I got one for Christmas...
> I would like to thank you in advance for the work you have done.

Just got one and sure it will be included in a (near) future release!

TheNON75, 2022-02-09:
With the help of DDF, some limited functionalities can already be implemented, such as:
- config/fanmode off/auto/low/medium/high
- config/ledindicator (no update is possible atm)
- config/locked (no update is possible atm)
- state/airqualityppb
- state/speed:
- state/on: to be derived

ebaauw, 2022-02-09:
Note that the Xiaomi reports VOC level in ppb, which is exposed as `airqualityppb`.  The Starkvind reports PM2.5 density in µg/m³.  This cannot (shouldn’t) be exposed as the same resource attribute, even if you were to convert PM2.5 density to PM2.5 level in ppb (as it’s a different quantity).

Kllrv, 2022-02-19:
> With the help of DDF, some limited functionalities can already be implemented, such as:
Have you got such a working DDF available somewhere?

MTrab, 2022-02-27:
Would be nice to see some progress to this issue/feature.

What do you need to speed up this one?

kimosak, 2022-03-18:
Nothing new with Starkving on v2.15.0-beta deconz-rest-plugin, is anything needed to speed it up?
Thanks

MTrab, 2022-03-18:
Gave up waiting - moved to Zigbee2MQTT

andiyk, 2022-04-08:
is still being worked on?

kimosak, 2022-04-08:
Same question here, Waiting for v2.15.+ to test it

mayanigrosh, 2022-04-13:
@ebaauw How can the rest of us help this along?

manup, 2022-04-13:
Hi we have the Starkvind on the order list expected to arrive in May to be able to create the DDF.
If anybody is keen to create one earlier I'm happy to help out.

TheNON75, 2022-05-02:
Hi @manup,
Considering it is already May, I am not sure if it makes sense, but here I am

manup, 2022-05-04:
The Starkvind is ordered, so I hope to have a DDF ready shortly after its arrives.

MichelEhmen, 2022-05-20:
Hello, Are there already any updates?

manup, 2022-05-20:
I have the Starkvind now running in my living room (vanilla without Zigbee, boy this thing can be loud..). The last days kept me a bit busy with bug fixes, I'm starting now to build the DDF and check what needs to be extended on the C++ side based on @ebaauw research.

closed 2022-06-07 by manup

oywino, 2023-04-07:
I've had my IKEA STARKVIND for some moths, integrated into Home Assistant using REST-API via deConz without any issues - until quite recently when it suddenly and for no apparent reason reports Air Quality Out-of-Scale and the speed automatically increases to maximum:

[image]

Then after a while it seems to autocorrect itself again.  This has never happened before, but recently it happens almost weekly and randomly.  
Does anyone know why?

Mimiix, 2023-04-07:
Hi,

For any questions on devices, please use the forums. If you think a bug is involved, open a new issue. 

Locking this up.