dresden-elektronik/deconz-rest-plugin #5352: Frient intelligente KeyPad

Maxcrouz73, 2021-10-06:
Hello there, this is my first contribution to this great software, hope i do this correctly.

## Device

- Product name: intelligent keypad.
- Manufacturer: frient A/S
- Model identifier: KEPZB-110
- Device type :  
- Other: Keypad

[image]

[image]

[image]

### Basic
[image]

### Identify

[image]

### Poll Contro
[image]
l

### IAS Zone
[image]

### Power Configuration
[image]

Smanar, 2021-10-06:
Hello.
Yes it's fine ^^.
But BTW keypad are something new in deconz, do you know how to use them ? There is a new feature in the HA plugin, but the HA OS is realy not a good OS for compilation.

And have you a linux machine with full OS to test code modification ?

Smanar, 2021-10-06:
In deconz It s not exactly the same working mode, there is a stand alone mode and a more complex one https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/alarmsystems/

>I guess i must send you the GUI when i push the buttons on the keypad?

I don't think, will use standard code to start, give me 1/2 days

Smanar, 2021-10-07:
Ok so first try

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch frient_keypad https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

I have set all by defaut (and not enabled battery yet).
To check the working mode I think the better way is using the logs with "info" "ias" and "info_l2"
Not sure you will have all informations displayed on the device Json (visible here http://phoscon.de/pwabeta in help/api information)

Maxcrouz73, 2021-11-05:
> Ha yes I have it too, try
> 
> `sudo apt install qtdeclarative5-dev`

Hi,

Sorry for the late answer...

i have try `sudo apt install qtdeclarative5-dev`

But i run this issue:

```
device.cpp: In function ‘void DEV_InitStateHandler(Device*, const Event&)’:
device.cpp:231:24: error: ‘DBG_DEV’ was not declared in this scope
             DBG_Printf(DBG_DEV, "DEV Init no node found: 0x%016llX\n", event.deviceKey());
                        ^~~~~~~
device.cpp:231:24: note: suggested alternative: ‘DBG_OTA’
device.cpp: In function ‘void DEV_NodeDescriptorStateHandler(Device*, const Event&)’:
device.cpp:289:24: error: ‘DBG_DEV’ was not declared in this scope
             DBG_Printf(DBG_DEV, "ZDP node descriptor verified: 0x%016llX\n", device->key());
                        ^~~~~~~
device.cpp:289:24: note: suggested alternative: ‘DBG_OTA’
device.cpp:338:20: error: ‘DBG_DEV’ was not declared in this scope
         DBG_Printf(DBG_DEV, "read ZDP node descriptor timeout: 0x%016llX\n", device->key());
                    ^~~~~~~
device.cpp:338:20: note: suggested alternative: ‘DBG_OTA’
device.cpp: In function ‘void DEV_ActiveEndpointsStateHandler(Device*, const Event&)’:
device.cpp:353:24: error: ‘DBG_DEV’ was not declared in this scope
             DBG_Printf(DBG_DEV, "ZDP active endpoints verified: 0x%016llX\n", device->key());
                        ^~~~~~~
device.cpp:353:24: note: suggested alternative: ‘DBG_OTA’
device.cpp:399:20: error: ‘DBG_DEV’ was not declared in this scope
         DBG_Printf(DBG_DEV, "read ZDP active endpoints timeout: 0x%016llX\n", device->key());
                    ^~~~~~~
device.cpp:399:20: note: suggested alternative: ‘DBG_OTA’
device.cpp: In function ‘void DEV_SimpleDescriptorStateHandler(Device*, const Event&)’:
device.cpp:426:24: error: ‘DBG_DEV’ was not declared in this scope
             DBG_Printf(DBG_DEV, "ZDP simple descriptors verified: 0x%016llX\n", device->key());
                        ^~~~~~~
device.cpp:426:24: note: suggested alternative: ‘DBG_OTA’
device.cpp:472:20: error: ‘DBG_DEV’ was not declared in this scope
         DBG_Printf(DBG_DEV, "read ZDP simple descriptor timeout: 0x%016llX\n", device->key());
                    ^~~~~~~
device.cpp:472:20: note: suggested alternative: ‘DBG_OTA’
device.cpp: In function ‘bool DEV_ZclRead(Device*, ResourceItem*, deCONZ::ZclClusterId_t, deCONZ::ZclAttributeId_t)’:
device.cpp:537:20: error: ‘DBG_DEV’ was not declared in this scope
         DBG_Printf(DBG_DEV, "DEV not reachable, skip read %s: 0x%016llX\n", item->descriptor().suffix, device->key());
                    ^~~~~~~
device.cpp:537:20: note: suggested alternative: ‘DBG_OTA’
device.cpp:545:20: error: ‘DBG_DEV’ was not declared in this scope
         DBG_Printf(DBG_DEV, "TODO cluster 0x%04X not found: 0x%016llX\n", device->key(), static_cast<quint16>(clusterId));
                    ^~~~~~~
device.cpp:545:20: note: suggested alternative: ‘DBG_OTA’
device.cpp: In function ‘void DEV_BasicClusterStateHandler(Device*, const Event&)’:
device.cpp:602:24: error: ‘DBG_DEV’ was not declared in this scope
             DBG_Printf(DBG_DEV, "Failed to read %s: 0x%016llX\n", it.suffix, device->key());
                        ^~~~~~~
device.cpp:602:24: note: suggested alternative: ‘DBG_OTA’
device.cpp:612:24: error: ‘DBG_DEV’ was not declared in this scope
             DBG_Printf(DBG_DEV, "DEV modelId: %s, 0x%016llX\n", qPrintable(device->item(RAttrModelId)->toString()), device->key());
                        ^~~~~~~
device.cpp:612:24: note: suggested alternative: ‘DBG_OTA’
device.cpp:637:20: error: ‘DBG_DEV’ was not declared in this scope
         DBG_Printf(DBG_DEV, "DEV received %s: 0x%016llX\n", event.what(), device->key());
                    ^~~~~~~
device.cpp:637:20: note: suggested alternative: ‘DBG_OTA’
device.cpp:643:20: error: ‘DBG_DEV’ was not declared in this scope
         DBG_Printf(DBG_DEV, "DEV read basic cluster timeout: 0x%016llX\n", device->key());
                    ^~~~~~~
[... 89 lines cut ...]
device.cpp:1469:24: error: ‘DBG_DEV’ was not declared in this scope
             DBG_Printf(DBG_DEV, "DEV Poll Busy %s/0x%016llX ZCL response seq: %u, status: 0x%02X\n",
                        ^~~~~~~
device.cpp:1469:24: note: suggested alternative: ‘DBG_OTA’
device.cpp: In function ‘void DEV_DeadStateHandler(Device*, const Event&)’:
device.cpp:1488:20: error: ‘DBG_DEV’ was not declared in this scope
         DBG_Printf(DBG_DEV, "DEV enter passive state 0x%016llX\n", event.deviceKey());
                    ^~~~~~~
device.cpp:1488:20: note: suggested alternative: ‘DBG_OTA’
device.cpp: In function ‘void mergeBindingReportConfigs(DDF_Binding&, const DDF_Binding&)’:
device.cpp:1757:24: error: ‘DBG_DEV’ was not declared in this scope
             DBG_Printf(DBG_DEV, "DEV add reporting cluster: 0x%04X, attr: 0x%04X\n", b.clusterId, br.attributeId);
                        ^~~~~~~
device.cpp:1757:24: note: suggested alternative: ‘DBG_OTA’
device.cpp: In member function ‘void Device::addBinding(const DDF_Binding&)’:
device.cpp:1775:20: error: ‘DBG_DEV’ was not declared in this scope
         DBG_Printf(DBG_DEV, "DEV add binding cluster: 0x%04X,  0x%016llX\n", bnd.clusterId, d->deviceKey);
                    ^~~~~~~
device.cpp:1775:20: note: suggested alternative: ‘DBG_OTA’
make[1]: *** [Makefile.Release:1156: release/device.o] Error 1
make[1]: *** Attente des tâches non terminées....
make[1] : on quitte le répertoire « /home/pi/deconz-rest-plugin »
make: *** [Makefile:40: release] Error 2

```

Maxcrouz73, 2021-11-05:
> Ha yes I have it too, try
> 
> `sudo apt install qtdeclarative5-dev`

Hello don't know if you see my reply i delete it, cause the probleme is solved.
So, i want to display the armed_mode and witch keys are type on keyboard but i don't know where to find the json to add cmds on jeedom.
On jeedom i have 👍 
```
{
    "7": {
        "config": {
            "battery": null,
            "enrolled": 1,
            "on": true,
            "pending": [],
            "reachable": true
        },
        "ep": 44,
        "etag": "eeec8fa3ca94d2694759d1cba107bb7a",
        "lastannounced": null,
        "lastseen": "2021-11-05T13:30Z",
        "manufacturername": "frient A\/S",
        "modelid": "KEPZB-110",
        "name": "AncillaryControl 7",
        "state": {
            "action": "not_ready",
            "lastupdated": "2021-11-05T13:28:41.813",
            "lowbattery": false,
            "panel": null,
            "seconds_remaining": 0,
            "tampered": false
        },
        "swversion": "2021-05-21 14:59",
        "type": "ZHAAncillaryControl",
        "uniqueid": "00:15:bc:00:43:00:0d:16-2c-0501"
    }
}
```

Smanar, 2021-11-05:
>Hello don't know if you see my reply i delete it, cause the probleme is solved.

He nope ^^, have missed it, and I haven't make the PR, waiting for return.

Keypad was added recently with a new alarm system https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/alarmsystems/ But if I m right keypad can stil lworking in stand alone mode.

>The read-only attribute state.action contains the last action a user invoked on the keypad.

>The read-only attribute state.panel represents the state machine and mirrors the alarm system state.armstate attribute. It reflects what is shown on the panel (when activated by the keypad’s proximity sensor).

From my memory, no, the code is never displayed, for security reason.

I don't think you need to create an alarm system

>This is only needed than more than one alarm system is needed. The “default” alarm system with id “1” is created automatically.

But try to set a "code0" with the id = 1

tpjanssen, 2022-01-18:
Not quite sure what the current status is... Is the device currently supported? If not, will it be on the short term?

Smanar, 2022-01-18:
I don't have keypad to answer.
I know some users have issue with the new alarm system, some other are using DDF https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5520#issuecomment-1010928122

But yes If an user with a full working keypad can answer ....

tpjanssen, 2022-01-18:
> I don't have keypad to answer. I know some users have issue with the new alarm system, some other are using DDF [#5520 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5520#issuecomment-1010928122)
> 
> But yes If an user with a full working keypad can answer ....

OK. Thank you for your quick reply! I will purchase one.

How you prefer to tackle this? I'm using Home Assistant with the Deconz add-on.

ZeppDK, 2022-01-18:
I have one… i managed to get it into deconz, but i cant get anything usefull over in HA, so right now i just have an expensive paperweight 😂

tpjanssen, 2022-01-18:
@ZeppDK can you supply @Smanar with the information he needs?

ZeppDK, 2022-01-18:
I have no idea what he needs, and as said, cant male it do anything 😅

tpjanssen, 2022-01-18:
> I have no idea what he needs, and as said, cant make it do anything

@Smanar I hope you can help us with this?

Smanar, 2022-01-18:
>@Smanar I hope you can help us with this?

Can try, but for the moment, I known some users have tried, but not sure they have succed.

@ZeppDK where are you with your device ATM ? He is reconized in the API or not ?

ZeppDK, 2022-01-18:
Nope, KEYZB-110 is nowhere to find in the API

Smanar, 2022-01-18:
Yep, this device is not yet in the API, need to use DDF.

ZeppDK, 2022-01-18:
i have addet the DFF file and restartet, now i have the following if i search in the API...

[image]

tpjanssen, 2022-01-18:
> i have addet the DFF file and restartet, now i have the following if i search in the API...
> 
> [image]

So your paperweight is turning into something useful! 😅

ZeppDK, 2022-01-18:
its a step in the right direction... i still only get this in HA 
[image]

Smanar, 2022-01-19:
Yeah no the "unknow part" the alarm system stuff.
IDK if there is something in HA to help you to do that, so sorry to use the basic method https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/alarmsystems/

Have you an alarm system ready ? GET /api/<apikey>/alarmsystems
`GET /api/<apikey>/alarmsystems`

I think yes, it s by defaut

Set a pin0 code, else nothing can work https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/alarmsystems/#set-alarm-system-configuration

I can give you curl request If you don't have a REST client ?

ZeppDK, 2022-01-19:
im in the deep end of the pool here, im a HA noob, and not a programmer :-)
if you send som curl i will try running it, 
and ofcourse add thoose lines to ZHAAncillaryControl in the json file

ZeppDK, 2022-01-19:
uuhh, and i also need to have a look at https://www.home-assistant.io/integrations/manual i see

Smanar, 2022-01-19:
Nope ^^, it depend more of deconz than HA.

Here you have how to use the API https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/
But I can give you curl command to type.

But ATM the json is not complete, and you can't edit it, it's deconz that need to make a better one.
The missing line are in the DDF, so they need to be in the json, it's for that I think something have missed.

You have the DDF stuff here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
- select your node
- take "edit ddf"
- check if the fine in the new windows title is the path you have used
- make "hot reload" in the menu
- check again the json.

ZeppDK, 2022-01-20:
i succeded in adding the 2 options.
[image]

ZeppDK, 2022-01-20:
and i can see that HA updated to this.
[image]

Smanar, 2022-01-20:
>i succeded in adding the 2 options.

Ha nice, what you have done to have them ?

So to check alarm system, just use in a browser
`http://IP:PORT/api/KEY//alarmsystems`

To add a pin code with curl
curl -H 'Content-Type: application/json' -X PUT -d '{"code0": 1234}' http://IP/PORT/api/KEY/alarmsystems/0/config

For IP and PORT, they are the same you have used to access phosocn.
For the API key, IDK if you can see it in HA else you can retreive one https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

ZeppDK, 2022-01-20:
i added this to the KEPZB-110.json file

`       {
          "name": "state/panel",
          "default": "exit_delay",
          "public": true
        },
`

ZeppDK, 2022-01-20:
from the browser:
[image]

but this is from the curl:
[image]

Smanar, 2022-01-20:
My bad, the ID is not 0 but 1 `/alarmsystems/1/config` (it s the one you have see on the browser)

ZeppDK, 2022-01-20:
aahh, that makes sense
[{"success":{"/alarmsystems/1/config/configured":true}}]

Smanar, 2022-01-20:
Ok so  next step (I m discovering the doc in same time than you ^^) 

> → After pairing a keypad as sensor, it’s automatically added to the default alarm system with id “1”.
> 
>     Keypads use the the alarm system PIN code (see Update alarm system configuration), which is verified each time the keypad arms or disarms the alarm system.
>     If a valid PIN code is entered on the keypad, the alarm system’s state changes according to the requested arm or disarm command.
>     Important: For REST-API clients a keypad is a read-only sensor. The alarm system automatically takes care of: controlling the keypad’s panel state, entry and exit delays, and state transitions.

So if there is not issue you can use the pin0 code "1234" and something need to change on `http://IP:PORT/api/KEY//alarmsystems`

But I will ask too Robban to know if there is some stuff in HA to manage alarm system.

ZeppDK, 2022-01-20:
i have restartet deconz, reloaded the HA plugin, and run deconz.device_refresh , nothings changed...

but i cant seem to find any key combination including the 1234 pin that does anything, there is no change in the http://IP:PORT/api/KEY/alarmsystems

ZeppDK, 2022-01-20:
aaahh,
after using this string the keypad is accepting the pin
[image]

ZeppDK, 2022-01-20:
yatsy... it seems to work with HA now also, after reload of the plugin... now only thing im missing, is to find out how to se the ID of an rfid tag hold up to the panel, then everything works ad expected! :-)
 
[image]

tpjanssen, 2022-01-20:
Looks like serious progress has been made in the last three days!

@Smanar What is the next step now? To add the device officially? How can add we add the device it after this? Just add a new sensor in Phoscon?

tpjanssen, 2022-01-21:
What @ZeppDK probably means is that he doesn't know how to receive feedback when a RDIF tag is held against the terminal. The terminal has an integrated RFID reader.

Smanar, 2022-01-21:
Haa .... You mean the keypad have a RFID reader ?
Ha **** I don't think this part it s ready in the code, can you catch logs (with info and aps) when using a RFID card/badge ?

ZeppDK, 2022-01-21:
Yes, it have an RFID reader inside :-)

What log files? 
i have lots of tags, so if i know where to look, i Can test

Smanar, 2022-01-22:
Deconz one, using "info" and "IAS".
But if I m right you have a RFID to arm and 1 RFID to disarm for exemple ? so you will have same result than when you type the code.

ZeppDK, 2022-01-24:
cant seem to find til ID for the tags, but i get the following in the log when holding a tag to the keypad...
[image]

Smanar, 2022-01-24:
Your forget the flag "ias" it s the flag used by keypad.
But from the code the RFID device just ask for panel status, and deconz need to send an aswer.
But It's strange it's not a arm or disarm request ? You need to program the RFID ? or all have a different feature ?

ZeppDK, 2022-01-24:
RFID tags are just a "key", so they all have a unik ID, nothing else... normaly you code an alarm system to allow the tags to arm & disarm, so you dont have to have 1 tag for disarming, and another for arming

ZeppDK, 2022-01-24:
with the flag "IAS" i only get this...

[image]

Smanar, 2022-01-24:
So all seem fine 0x00 mean IAS_ACE_PANEL_STATUS_PANEL_DISARMED
If you do the same thing with the alarm armed the result will be 0x01 0x02 or 0x03

But perhaps the tagts just prevend you using a code ? so you need to press armed or disarmed after having using the tga no ?

ZeppDK, 2022-01-24:
if i just press lock or unlock button it says code length 0
but if i use a tag first, and then press lock or unlock it says code length 9
so i guess your right, but i still dont now how to capture that code, so that i can add them to the allowed codes :)
[image]

Smanar, 2022-01-24:
Ha yessssss ^^, good catch.
There is an hidden code somewhere to reconize good and bad rfid. 
There is nothing on the documentation ? to add or remove a RFID ? I don't think it s something coded hardware.

For security reason, the code never appear. I can make a change to display it if you can compile the code.
But I think we can catch it using the GUI, need to check.

But the problem is the same, need to be implemented in the api to be used by third app.

Can you check if you have information in the documentation ? I think there is something to add or remove a RFID so perhaps a way to add code with an invisible/securised way.

ZeppDK, 2022-01-24:
I have looked more than once in the documentation, and keyforbs are mentioned multiple times, but i cant find any hint on how to add/configure them in deconz

ZeppDK, 2022-01-24:
I Can find some indikation that it should work with smartthings, and maybe something called “squid.link 2x”… but only part interesting info was this datasheet https://www.develcoproducts.com/media/2032/kepzb-110-technical-manual-keypad.pdf

ZeppDK, 2022-01-24:
But its still not saying much about the rfid reader

Smanar, 2022-01-25:
I need to ask manup the direction to take.

- ATM you can set only the "master" pin, so not possible to add other RFID, so need to improve the code first
- Can make a "display code" command.
- Can add the code with an invisible way, with a "memorise code" command.

ZeppDK, 2022-01-25:
It would be nice to have multiple pins (and maybe somehow asign them to a user, so you Can see who is disabling the alarm, and easy disable a user/code if its not used anymore)

Yes, i Will post the DFF

ZeppDK, 2022-01-27:
my DFF
`{
  "schema": "devcap1.schema.json",
  "manufacturername": "Develco Products A/S",
  "modelid": "KEPZB-110",
  "product": "KEPZB-110",
  "sleeper": false,
  "status": "Draft",
  "path": "/devices/KEPZB-110.json",
  "subdevices": [
    {
      "type": "ZHAAncillaryControl",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x2c",
        "0x0501"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0401",
        "endpoint": "0x2C",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ],
        "out": [
          "0x0501"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "description": "The current device battery level in 0&ndash;100&thinsp;%."
        },
        {
          "name": "config/enrolled",
          "public": false,
          "description": "State of IAS enrollment process."
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending",
          "description": "Pending tasks to configure the device."
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/action",
          "public": false
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "description": "True when the device battery runs low."
        },
        {
          "name": "state/panel",
          "default": "exit_delay",
          "public": true
        },
        {
          "name": "state/tampered"
        }
      ]
    }
  ]
}
`

Smanar, 2022-01-27:
So I need to take a look on code, but I will make a command for the alarm system will be able to learn code (this one will be invisible for security reason)

BTW I haven't asked, are you able to compile the code for tests ? (Need a linux machine with full/real OS)

ZeppDK, 2022-01-28:
yep, i have multiple linux hosts...

Smanar, 2022-01-29:
Perfect, so it will step by step because I m asking which one request to use for the API (don't want to re write all because of a bad choice)

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz (The code is based on realy last one, so need a recent version):

```
sudo apt install deconz-dev
git clone --branch keypad_rfid https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2"

So what do this code :
- more debug (with IAS, info and info_l2), for exemple you wil be able to see the code send by the rfid (you can hide it in logs)
- Added the request `PUT /api/<apikey>/alarmsystems/<id>/config/code` with as body {index,code} , I m not sure for the synthax but probably {1,"12345"} index 0 is reserved for master pin

Can try to catch the code used by the RFID, try to add it in index 1, and use it, you will have lots of usefull logs.

ZeppDK, 2022-01-31:
okay, i can see the code:
[image]

but i cant seem to get the syntax for putting in the code ...

i have tried what you sugested
[image]

i also tried the same way i added code0
[image]

ZeppDK, 2022-01-31:
this worked, but i cant arm or disarm the panel, not with code1 or code0

[image]

ZeppDK, 2022-01-31:
i have now added 2 rfid tags in position 1 and 2, and can use them both, and also the pin, so it seems to be working.

Smanar, 2022-01-31:
Lol, serioulsy ? on first try ?
BTW you have hidded the RFID code ^^, but there is something common to them ? like the start by "+" ? It's realy something new, first time I m seing that.

Now need to make that naturally, I will ask if we need to put the keypad in "learn mode" or " the alarm system.

ZeppDK, 2022-01-31:
Yes, they start with “+”, and so far 8 numbers after the +

[85 more replies]

closed 2024-01-15 by github-actions[bot]