dresden-elektronik/deconz-rest-plugin #5370: essentials Smart Home Heizkörperthermostat

r4nc0r, 2021-10-13:
Hi,

I found this cheap thermostat and it would be really great if deconz would support it, the Premium is already [supported](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3381),):
https://www.wesmartify.de/heizung/heizkoerperradiator/333/essentials-smart-home-heizkoerperthermostat/

## Device

- Product name: essentials Smart Home Heizkörperthermostat 
- Manufacturer: _TZE200_i48qyn9s
- Model identifier: TS0601
- Device type : Please remove all unrelated device types. 
  - Thermostat

## Screenshots

[image]

[image]

[image]

Mimiix, 2021-10-13:
@Smanar

r4nc0r, 2021-11-04:
Any update on this?

Smanar, 2021-11-04:
Have some problem on the lidl one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5349

r4nc0r, 2021-11-29:
Ping :)

r4nc0r, 2021-12-18:
Anything new?

Smanar, 2021-12-18:
Not realy, all new devices are blocked for the moment, I think we need to wait for tuya cluster integration in DDF core.

r4nc0r, 2022-09-15:
Can someone reopen this? I still thinks this would be a great addition to deconz, because this is a really cheap and good thermostat.

r4nc0r, 2022-09-15:
Thanks, can I help in any way?

Smanar, 2022-09-16:
No, sorry thx for help, we will need it later, but for the moment still waiting for a PR be merged to have a "base"

mat-l, 2023-02-14:
Hey @Smanar, are there update about the Hama thermostat? 🔥

Smanar, 2023-02-14:
I think it will be possible soon https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6674
The PR will be in the 2.21.1 version (if no problem) so will be possible to add it using DDF.

mat-l, 2023-03-08:
Ping ✌🏽

Smanar, 2023-03-08:
Ha yes its now possible, which one deconz version are you using ?

Can you try this DDF, it's the same used than for the _TZE200_thbr5z34, some dp are same, so probably the whole device ...

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_i48qyn9s",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "essentials TRV",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "meta": {
        "values": {
          "config/mode": {"auto": 0, "heat": 1, "off": 2}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 34, "eval": "Item.val = Attr.val * 20;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 16, "eval": "Item.val = Attr.val * 50;"},
          "write": {"fn": "tuya", "dpid": 16, "dt": "0x2b", "eval": "Item.val / 50;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 30, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 30, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/mode",
          "values": [
              ["auto", 0], ["heat", 1], ["off", 2]
          ],
          "parse": {"fn": "tuya", "dpid": 2, "eval": "if (Attr.val == 0) { Item.val = 'auto' } else if (Attr.val == 1) { Item.val = 'heat' } else { Item.val = 'off' }"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x30", "eval": "if (Item.val == 'auto') { 0 } else if (Item.val == 'heat') { 1 } else { 2 }"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 24, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        }
      ]
    }
  ]
}
```

closed 2023-04-08 by github-actions[bot]