dresden-elektronik/deconz-rest-plugin #5372: SALUS SQ610RF

alexskra, 2021-10-14:
## Device

- Product name: SQ610RF (AJSQ610RF)
- Manufacturer: SALUS
- Model identifier: SQ610RF
- Device type: 
  - Thermostat

## Screenshots

### Node info
[image]

### Basic
[image]

### Power Configuration
[image]

### Thermostat
[image]
[image]
[image]
[image]

[image]
[image]
[image]
[image]
[image]
[image]

### Diagnostics
[image]

Mimiix, 2021-11-30:
@Smanar something for you?>

Smanar, 2021-12-01:
IDK.
Not sure DDF can support thermostat, and it seem there is some "specialities" on this device https://github.com/Koenkk/zigbee-herdsman-converters/issues/1524#issuecomment-963996011

A special action suite to make a perfect inclusion, but from the capture it seem @alexskra have done it. Have you tried to send some command using the cluster 0x0201 ? Do you have a recent deconz version that support DDF ?

closed 2021-12-31 by github-actions[bot]

xyz4tech, 2022-08-01:
Any update on this?

bimusiek, 2022-12-15:
Would be amazing to crack it, this is one of the best thermostats in Poland

Smanar, 2022-12-15:
Can you test id the device need special inclusion or not, with testing directly the device on the Cluster 0x0201 ?

Smanar, 2022-12-16:
There is some starting way on z2m issue, but I don't think generic zigbee projects are supporting thoses devices yet.

bimusiek, 2022-12-22:
I have used now the API of Salus and currently using their gate. If someone is interested, the homebridge plugin is available here: https://github.com/Hernas/homebridge-salus-sq610