dresden-elektronik/deconz-rest-plugin #5375: Add Support for LIVARNO LUX FB20-002 Remote Control

thommycgn, 2021-10-16:
## Device

- Product name: Livaro Lux FB20-002 remote
- Manufacturer: LIVARNO LUX
- Model identifier: FB20-002, 328031_2001
- Device type : 
  - Remote
- Date: 06/2020

I was using it already with zigbee2mqtt. Now I changed over to deconz and the remote is not supported here. :(
Can you please support it? Maybe you can use this?
https://github.com/Koenkk/zigbee-herdsman-converters/blob/ffc473f3a790dbf950d83791d6b135a3eb267802/devices.js#L15630

## Screenshots

[image]

[image]

closed 2021-11-17 by github-actions[bot]

frontibit, 2022-03-07:
I'm not the origilan owner but here are some more screenshots. Can we reopen the ticket as this Wireless controll is still not supported. I can only read the battery information but no switches are possible.

[image]

[image]

Mimiix, 2022-03-07:
@frontibit I'd rather have you create a new request and add all screenshots in there.