dresden-elektronik/deconz-rest-plugin #5394: MOES TRV HY368 TS0601 : wrong reporting of current temperature (2 degrees)

SebLz, 2021-10-18:
## Describe the bug

Multiple MOES HY368 TRV report wrong "current temperature values" of 2 degrees. Sometimes it updates to the right value, sometimes to 2.
Similar to this closed issue: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5384 ?
## Steps to reproduce the behavior

## Expected behavior

Display same temperature values as on the TRV device
## Screenshots

[image]
[image]

## Environment

 - Host system: Odroid C4
 - Running method: Home Assistant deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.12.06 / 19/08/2021
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system: no

## deCONZ Logs

## Additional context

REST API:

```
{
    "config": {
        "heatsetpoint": 2000,
        "locked": false,
        "mode": "auto",
        "offset": 0,
        "on": true,
        "preset": "manual",
        "reachable": true,
        "schedule": {
            "W124": [
                {
                    "heatsetpoint": 20,
                    "localtime": "T06:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T08:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T11:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T12:30"
                },
                {
                    "heatsetpoint": 20,
                    "localtime": "T17:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T22:00"
                }
            ],
            "W3": [
                {
                    "heatsetpoint": 20,
                    "localtime": "T134:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T08:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T11:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T12:30"
                },
                {
                    "heatsetpoint": 20,
                    "localtime": "T17:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T22:00"
                }
            ]
        },
        "schedule_on": false,
        "setvalve": false,
        "windowopen_set": false
    },
    "ep": 1,
    "etag": "0fe205de324967817d4ae8b739e9ca30",
    "lastannounced": "2021-10-18T20:11:43Z",
    "lastseen": "2021-10-18T20:11Z",
    "manufacturername": "_TZE200_ckud7u2l",
    "modelid": "TS0601",
    "name": "Radiateur chambre à coucher",
    "state": {
        "lastupdated": "2021-10-18T20:13:12.031",
        "lowbattery": false,
        "on": true,
        "temperature": 200,
        "valve": 85
    },
    "type": "ZHAThermostat",
    "uniqueid": "5c:02:72:ff:fe:8e:f7:e9-01-0201"
}
```

SebLz, 2021-10-21:
After quite some struggles, I've managed to build your version in a new container on my odroidc4 and imported my deconz config (config.ini, zll.db and zlcdb.txt) in it. Then I did some rest calls to capture the values (the test container is not connected to homeassistant). I have not noticed wrong temperatures anymore, so it seems your fix works, thanks !
Here is the log with the new version compiled from your branch:

```
23:07:28:094 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ac802020004000000a0,
23:07:28:095 Tuya debug 5 : Status: 10 Transid: 200 Dp: 514 (0x02,0x02) Fn: 0 Data 160,
23:07:28:166 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ac902020004000000a5,
23:07:28:167 Tuya debug 5 : Status: 10 Transid: 201 Dp: 514 (0x02,0x02) Fn: 0 Data 165,
23:07:28:286 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0aca03020004000000cd,
23:07:28:287 Tuya debug 5 : Status: 10 Transid: 202 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:28:405 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0acb03020004000000cd,
23:07:28:406 Tuya debug 5 : Status: 10 Transid: 203 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:28:530 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0acc02020004000000aa,
23:07:28:531 Tuya debug 5 : Status: 10 Transid: 204 Dp: 514 (0x02,0x02) Fn: 0 Data 170,
23:07:28:650 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0acd03020004000000cd,
23:07:28:651 Tuya debug 5 : Status: 10 Transid: 205 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:28:770 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ace03020004000000cd,
23:07:28:771 Tuya debug 5 : Status: 10 Transid: 206 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:28:890 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0acf02020004000000af,
23:07:28:891 Tuya debug 5 : Status: 10 Transid: 207 Dp: 514 (0x02,0x02) Fn: 0 Data 175,
23:07:29:012 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ad003020004000000cd,
23:07:29:013 Tuya debug 5 : Status: 10 Transid: 208 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:29:135 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ad103020004000000cd,
23:07:29:136 Tuya debug 5 : Status: 10 Transid: 209 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:29:253 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ad202020004000000b4,
23:07:29:254 Tuya debug 5 : Status: 10 Transid: 210 Dp: 514 (0x02,0x02) Fn: 0 Data 180,
23:07:29:376 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ad303020004000000cd,
23:07:29:377 Tuya debug 5 : Status: 10 Transid: 211 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:29:497 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ad403020004000000cd,
23:07:29:498 Tuya debug 5 : Status: 10 Transid: 212 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:29:620 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ad502020004000000b9,
23:07:29:621 Tuya debug 5 : Status: 10 Transid: 213 Dp: 514 (0x02,0x02) Fn: 0 Data 185,
23:07:29:739 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ad603020004000000cd,
23:07:29:740 Tuya debug 5 : Status: 10 Transid: 214 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:29:862 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ad703020004000000cd,
23:07:29:863 Tuya debug 5 : Status: 10 Transid: 215 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:29:982 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ad802020004000000be,
23:07:29:983 Tuya debug 5 : Status: 10 Transid: 216 Dp: 514 (0x02,0x02) Fn: 0 Data 190,
23:07:30:105 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ad903020004000000cd,
23:07:30:106 Tuya debug 5 : Status: 10 Transid: 217 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:30:227 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ada03020004000000cd,
23:07:30:228 Tuya debug 5 : Status: 10 Transid: 218 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:30:350 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0adb02020004000000c3,
23:07:30:351 Tuya debug 5 : Status: 10 Transid: 219 Dp: 514 (0x02,0x02) Fn: 0 Data 195,
23:07:30:472 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0adc03020004000000cd,
23:07:30:473 Tuya debug 5 : Status: 10 Transid: 220 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:30:592 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0add03020004000000cd,
23:07:30:593 Tuya debug 5 : Status: 10 Transid: 221 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:30:715 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ade0404000102,
23:07:30:716 Tuya debug 5 : Status: 10 Transid: 222 Dp: 1028 (0x04,0x04) Fn: 0 Data 2,
23:07:30:835 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0adf02020004000000c8,
23:07:30:836 Tuya debug 5 : Status: 10 Transid: 223 Dp: 514 (0x02,0x02) Fn: 0 Data 200,
23:07:30:960 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ae003020004000000cd,
23:07:30:961 Tuya debug 5 : Status: 10 Transid: 224 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:31:082 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ae103020004000000cd,
23:07:31:083 Tuya debug 5 : Status: 10 Transid: 225 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:31:206 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ae202020004000000cd,
23:07:31:206 Tuya debug 5 : Status: 10 Transid: 226 Dp: 514 (0x02,0x02) Fn: 0 Data 205,
23:07:31:322 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ae303020004000000cd,
23:07:31:323 Tuya debug 5 : Status: 10 Transid: 227 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
23:07:31:444 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0ae403020004000000cd,
23:07:31:444 Tuya debug 5 : Status: 10 Transid: 228 Dp: 515 (0x02,0x03) Fn: 0 Data 205,
[... cut ...]
23:07:47:963 Tuya debug 5 : Status: 11 Transid: 13 Dp: 372 (0x01,0x74) Fn: 0 Data 0,
23:07:48:084 Tuya debug 4 : Address 0x5C0272FFFE8EF7E9 Payload 0b0e7502000400000004,
23:07:48:085 Tuya debug 5 : Status: 11 Transid: 14 Dp: 629 (0x02,0x75) Fn: 0 Data 4,
23:07:48:360 GW firmware version is up to date: 0x26660700,
23:07:52:690 Tuya debug 1 : Time sync request,
23:07:52:692 Send Tuya command 0x24, data: 029e6171d6a86171f2c8,
23:07:54:703 ZCL attribute report 0x842E14FFFE139AEF for cluster: 0x0702, ep: 0x01, frame control: 0x08, mfcode: 0x0000 ,
```

fi-sch, 2021-11-01:
I am experiencing the same issue. Unfortunately, I will need to wait for next stable to test the fix, since I'm usiany deCONZ as Home Assistant add-on.

closed 2021-11-06 by manup

SebLz, 2021-11-10:
@Smanar : since I'm running on [this branch](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5436), the bug seems to have reappeared (or maybe it has never been completely fixed... in the tests I made before I could only use the REST API, now it's running 24/7 and connected to HA and I notice that after a while the 2 degrees appears again).

Smanar, 2021-11-11:
The branch only update the "mode", if you speak about this PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5471

The PR was validated 5 days ago, so can you try with the last code from official github or wait for the version v2.13.3

Edit:
I m stupid, it was normal, the branc on your previous link was created before the PR from this issue was validated, so it's n't inside.
I have updated the branch for the "auto" issue, you can update and recompile.

SebLz, 2021-11-14:
Tested it for 2 days, no more 2 degrees issue, thanks 👍