dresden-elektronik/deconz-rest-plugin #5403: Aqara E1 Door & Window Sensor (MCCGQ14LM)

a-jey, 2021-10-20:
## Device

- Product name: Aqara E1 MCCGQ14LM.
- Manufacturer: .
- Model identifier: lumi.magnet.acn001.
- Device type :  
  - Sensor

## Screenshots

Node
[image]

Node Info
[image]

Basic
[image]

Identify
[image]

Power Configuration
[image]

OTAU
[image]

IAS Zone (1)
[image]

IAS Zone (2)
[image]

Package
[image]

[image]

As you can see on the two IAS Zone screenshots: the basic function is working.
Manufacturer field is empty, but that's also on other LUMI Sensors which are already supported.

SwoopX, 2021-10-29:
> 
> 
> > My bad, change the status from draft to anything else. Eventually, you need to restart deconz
> 
> No, your json got bronze as status. Even if change it from draft to bronze on that mask and save the file, overwriting yours, on restarting the app it's set to draft again.

That very much sound like the editor takes an auto generated one. Have you stored my file where the editor suggests to save the file? And have you changed the file type to .json?

Just to be clear: with restarting the app, you meant deconz and not just the editor?

a-jey, 2021-10-29:
Just to go thge save way I uninstalled deCONZ, deleted the folders 
AppData\Local\dresden-elektronik\deCONZ
AppData\Local\deCONZ

Then I installed deCONZ again
generated the folder AppData\Local\dresden-elektronik\deCONZ\devices
(that's exactly the folder which the DDF editor showed in the save menu before)
placed the renamed lumi.magnet.acn001.json in that folder
started deCONZ
set DDF handling to normal
started pairing

The node appears and gets Manufacturer Name and Model Identifier from your File.

Maybe I wrote it a bit unclear before: if I then rightclick the node and select edit DDF the status shows as draft (maybe that's the normal appearance?).
If I open your DDF file in a text editor the status is set to bronze.

If I open the Phoscon App through deCONZ menu and go to sensor->connect new sensor->other
the sonsor got not registered, no matter if I just trigger or reset the sensor.
Checked in several browsers with the same result.

SwoopX, 2021-11-01:
I just tried it in a Windows 10 VM and for me it works. I've taken the file from the PR above and saved it to `AppData\Local\deCONZ\devices\xiaomi`. During sensor search, it got properly recognized and added.

a-jey, 2021-11-01:
> 
> 
> I just tried it in a Windows 10 VM and for me it works. I've taken the file from the PR above and saved it to `AppData\Local\deCONZ\devices\xiaomi`. During sensor search, it got properly recognized and added.

recognized and added in deCONZ or in Phoscon WebApp?

I got no problem to get it recognized in deCONZ but in Phoscon WebApp it also doesn't work if I use the save location you mentioned. Tried in different browsers.

SwoopX, 2021-11-02:
> recognized and added in deCONZ or in Phoscon WebApp?

Both, full service.

Are you running in normal mode?

[image]

a-jey, 2021-11-02:
Yeah, I'm running it in normal mode.

a-jey, 2021-11-02:
Just tried it on a Windows 10 Laptop, the same result:
no problem in deCONZ, but doesn't find the sensor in Phoscon.

Mimiix, 2021-11-02:
How does it show in deCONZ?

a-jey, 2021-11-02:
> 
> 
> How does it show in deCONZ?

This is how it shows with 2 Sensors I tried:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5403#issuecomment-956821785

Mimiix, 2021-11-02:
> > How does it show in deCONZ?
> 
> This is how it shows with 2 Sensors I tried: [#5403 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5403#issuecomment-956821785)

Sorry, missed that comment.

lboue, 2021-11-02:
Hello,

My lumi.magnet.acn001 is recognized after a uploading the 'xiaomi_mccgq14lm_openclose_sensor.json' file to this directory:

Running config:

- deCONZ_ConBeeII_0x26720700.bin.GCF
- deCONZ plugin version 2.13.01 / 06/10/2021

C:\Users\%username%\AppData\Local\deCONZ\devices\xiaomi
[image]

a-jey, 2021-11-02:
> 
> 
> Hello,
> 
> My lumi.magnet.acn001 is recognized after a uploading the ['xiaomi_mccgq14lm_openclose_sensor.json](https://raw.githubusercontent.com/dresden-elektronik/deconz-rest-plugin/010938bf929362b535b0cf420e1e60796cf4f1f3/devices/xiaomi/xiaomi_mccgq14lm_openclose_sensor.json)' file to this directory: C:\Users%username%\AppData\Local\deCONZ\devices\xiaomi
> 
> Running config:
> 
>     * deCONZ_ConBeeII_0x26720700.bin.GCF
> 
>     * deCONZ plugin version 2.13.01 / 06/10/2021
> 
> 
> [image]

Yes yes yes!!!

With your linked file it immediately works like a charm.
Thanks a lot!

a-jey, 2021-11-02:
> 
> 
> I just tried it in a Windows 10 VM and for me it works. I've taken the file from the PR above and saved it to `AppData\Local\deCONZ\devices\xiaomi`. During sensor search, it got properly recognized and added.

The reason it didn't work on my side:
I missed the detail to use the new mentioned file.
I used the first one mentioned in this thread.

febalci, 2021-11-05:
Tried the 'xiaomi_mccgq14lm_openclose_sensor.json'  ddf file on ubuntu. It shows 'closed' all the time; not getting any status update from the sensor.

Glennesink, 2021-11-05:
Any way to add this to a Raspberry? Or wait for an update?

febalci, 2021-11-06:
Updated deconz firmware from 26660700 to 26720700. Phoscon  web app shows the details now, including the battery. So the sensor details page is just like the ones above. Good, however the sensor never updates the state; it is always 'open' whether i use the magnet to open or close it.

closed 2021-11-06 by manup

Glennesink, 2021-11-06:
> Any way to add this to a Raspberry? Or wait for an update?

Got it done with this instruction
https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware

mcormick, 2021-11-11:
> [...] however the sensor never updates the state; it is always 'open' whether i use the magnet to open or close it.
Same here. The status is only updated when I manually press the little button.

Myth5656, 2021-11-11:
same for me:  I am running my ConBee2 on Rpi4
2.13.02 / 10.11.2021
26720700

I have 10 sensors here. I Managed to connect one. Update for open/close was diplayed instantly in Phoscon App. I added a second sensor. Sice I did this, values only gets refreshed when I push the "connect" button on the sensor.

Mimiix, 2021-11-12:
@Myth5656  How many sensors / routers do you have in total?

Myth5656, 2021-11-12:
> @Myth5656 How many sensors / routers do you have in total?

I have 10 sensors (just connected 2) and use one Gateway

Is there anything that I can provide to support fixing the bug?

Mimiix, 2021-11-12:
@Myth5656 To make sure it's not a user question, i suggest opening a forum post and we see from there. Or join the discord and check there. 

If you just have these sensors, and nothing else, we can't be really sure that its the device or not.

Myth5656, 2021-11-12:
> @Myth5656 To make sure it's not a user question, i suggest opening a forum post and we see from there.

@Mimiix 
Just did.. 
https://forum.phoscon.de/t/aqara-door-sensor-mccgq14lm-not-updating/865

Myth5656, 2021-11-16:
@Mimiix 
Did you follow the [Thread](https://forum.phoscon.de/t/aqara-door-sensor-mccgq14lm-not-updating/865)? Is it a "user problem" or is there a bug that can be fixed?

Mimiix, 2021-11-17:
I did.  Manuel mentioned something up with Xiaomi stuff pairing, but i am not sure if i understood correctly.

SwoopX, 2021-11-17:
The odd thing over here is that it worked and works flawlessly. However, there's obviously a couple of things to follow:

1. Make sure your running the latest Conbee/Raspbee firmware https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog
2. If you run 2.13.0 or 2.13.1, make sure you update to 2.13.2
3. If any of the above was not given **before** you paired the device(s), reset and re-pair while being up to date
4. Pair in normal/hybrid mode (tab control)
5. 5. If you followed 1-4, try restarting deconz

KebapBoy, 2021-11-17:
I just paired 10 sensors sucessfully and their state is updating seemlessly now when triggered.
I followed @SwoopX s guidelines and had to restart the deconz container otherwise the sensors were indeed shown in Phoscon but only with the correct state after I pressed the little button on a sensor.
Is this the intended behaviour to have to restart deconz or is this due to any bug?

pipiche38, 2021-11-26:
> @SwoopX do you know what binds/enrolling commands deconz uses to get this device working? For z2m Ive tried both writing the ias + enroll response and a bind of the ias cluster but no luck. I think this info could also be used for ZHA.

@Koenkk did you make it working on z2m ? getting similar request and problem with zigate

Koenkk, 2021-11-26:
@pipiche38 yes, coordinator needs to respond with the Xiaomi manufacturer code when the sensor sends a node descriptor to it. (requires a fw update)

pipiche38, 2021-11-27:
> @pipiche38 yes, coordinator needs to respond with the Xiaomi manufacturer code when the sensor sends a node descriptor to it. (requires a fw update)

Thanks very much for the hint, I'll check with the firmware guys. I beleive the challenge is to differentiate a Aqara Door Sensor to an other device doing the Node Desriptor request. Do you use the mac@ prefix for that ?

Koenkk, 2021-11-28:
@pipiche38 yes, you can find the changeset I did to the TI firmware here: https://github.com/Koenkk/Z-Stack-firmware/blob/f29b4525bb7bfdc28c91139d0926469bb5754ab9/coordinator/Z-Stack_3.x.0/firmware.patch#L533

nukleuz80, 2021-12-09:
Hey guys,

i got it to work now:

Setup:
RPI3B+ with Raspbian Buster Desktop (all updates incl.)
RaspBee I with actual firmware: **26400500**
DeConz: 2.13.02 / 10.11.2021

I copied the DDF in this folder: /home/pi/.local/share/dresden-elektronik/deConz/devices
and restarted deConz - now the sensor is changing from close to open...

Thanks to all of you guys!

Zahnweh, 2021-12-14:
Up to now 2.13.02 can only be installed manually on native Deconz. At least that's what I could find out. But what about the Deconz HA addon?

nukleuz80, 2021-12-15:
Can anyone tell me how to include more than one of these sensors pls? From today i‘m on 2.13.3 Beta and i can‘t pair the second sensor. Phoscon cant find it like the first one. In DeConz the Sensor is visible.

thanks for your help!

ShlomiPorush, 2021-12-20:
> > Up to now 2.13.02 can only be installed manually on native Deconz. At least that's what I could find out. But what about the Deconz HA addon?
> 
> We need to wait for the stable release and then for the update for the Home Assistant Addon because as far as I know the Addon is always based on a stable release.

The Home Assistant added updated today and I still can't add any of the Aqara sensors.

Mimiix, 2021-12-20:
@ShlomiPorush you can't add them or they don't show up?

ShlomiPorush, 2021-12-20:
> @ShlomiPorush you can't add them or they don't show up?

they don't show up in Phoscon only in deCONZ as an unidentified sensor. 
[image]

Mimiix, 2021-12-20:
> > @ShlomiPorush you can't add them or they don't show up?
> 
> they don't show up in Phoscon only in deCONZ as an unidentified sensor. [image]
> 
> I do have other Aqara door sensors and no problem with them. just the sensor MCCGQ14LM

Did you re pair.?

Mimiix, 2021-12-20:
And to add to thar: are you on the latest firmware?

ShlomiPorush, 2021-12-20:
Yes. on version 2.13.04. did try to delete the node did not help. 

[image]
[image]

DoomiW, 2021-12-20:
Just had the same issue one hour ago after updating deCONZ to 2.13.04
First I noticed I was also not on the latest Firmware (should be 26720700 for Conbee 2). After updating the Firmware I had to switch deCONZ to hybrid mode and delete the node which I tried to pair before. Another try following these steps worked fine - the node appears in deCONZ and also within Phoscon.

ShlomiPorush, 2021-12-20:
so my firmware 26660700 is not the latest? how do I go about updating it

Zahnweh, 2021-12-20:
> Just had the same issue one hour ago after updating deCONZ to 2.13.04
> 
> First I noticed I was also not on the latest Firmware (should be 26720700 for Conbee 2). After updating the Firmware I had to delete the node which I tried to pair before and switch deCONZ to hybrid mode (as SwoopX already guided [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5403#issuecomment-971378048)). Another try following these steps worked fine - the node appears in deCONZ and also within Phoscon.
> 
> 

How do I enter hybrid mode?

woytekbode, 2021-12-20:
@ShlomiPorush Easiest for me is to just use Windows 10.

@DoomiW I have the same problem, device only shows in Deconz app. I have the latest Deconz and Conbee II versions. I'm using the HA addon on HA OS. Not sure how to switch Deconz to Hybrid mode, is that required for this device?

DoomiW, 2021-12-20:
> so my firmware 26660700 is not the latest? how do I go about updating it

Take a look on [those steps](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually) depending on your host system. The latest Firmware can be checked [here](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog)

kanuni89, 2021-12-20:
Sinze the DDF file is in Bronze status.. you should activate in Deconz to allow the states..
Go to: Panel -> Control -> on the DDF tab check the Bronze checkbox.. After that it worked for me

DoomiW, 2021-12-20:
> @ShlomiPorush Easiest for me is to just use Windows 10.
> 
> @DoomiW I have the same problem, device only shows in Deconz app. I have the latest Deconz and Conbee II versions. I'm using the HA addon on HA OS. Not sure how to switch Deconz to Hybrid mode, is that required for this device?

Same environment for me (and a long time to wait for stable deCONZ version :D ). At least for me it just worked in hybrid mode
deCONZ -> Panels -> Check control
On the bottom you'll see a new tab 'Control' where you can set the DDF Mode to hybrid

woytekbode, 2021-12-20:
@kanuni89 HERO! Works like a charm now! Can see it in Phoscon too, very snappy. @DoomiW thanks!

oli737, 2021-12-20:
I see a openclose123 in deconz but not in Phoscon. What i am doing wrong?

Mimiix, 2021-12-20:
@oli737 phoscon probably didn't add support yet. It should show in rest api / home assistant.

kanuni89, 2021-12-20:
> I see a openclose123 in deconz but not in Phoscon. What i am doing wrong?

have you put the DDF file into the data folder ? If yes, try to delete the node and restart the deconz, maybe after that it will work...

kanuni89, 2021-12-20:
> @oli737 phoscon probably didn't add support yet. It should show in rest api / home assistant.

for me it works... i see it in deconz and phoscon

DoomiW, 2021-12-20:
openclose123 sounds normally good as i had the same before manually renaming it via Phoscon. Could be worth restarting deCONZ

oli737, 2021-12-20:
i we try again pair the sensor

oli737, 2021-12-20:
Thanks guys, it works lika a charm.

ShlomiPorush, 2021-12-20:
So I updated deCONZ to firmware 26720700, in deCONZ I can see the type of the device "Open Close" but in Phoscon the status of the sensor is Not reachable.

[image]
[image]

dfrommi, 2021-12-20:
I have the same problem as @ShlomiPorush 

I have 2 of those Aqara devices and both are behaving strangely... the blue light flashes continuously with a certain pattern:
flash -> 2 sec pause -> flash -> 2 sec pause -> flash flash flash -> 2 sec pause .... and again from the beginning. 
Is that normal? Or is this some kind of reset loop?
Already removed the battery for a while, put it back in, same again.

a-jey, 2021-12-20:
Works like a charm now!

Thank you all for your work :-)

a-jey, 2021-12-20:
> So I updated deCONZ to firmware 26720700, in deCONZ I can see the type of the device "Open Close" but in Phoscon the status of the sensor is Not reachable.
> 
> [image] [image]

You changed the mode in deCONZ to Hybrid?

ShlomiPorush, 2021-12-20:
> You changed the mode in deCONZ to Hybrid?

Yes. did not help

[42 more replies]