dresden-elektronik/deconz-rest-plugin #5426: Samjin - Multipurpose sensor - vibration always false and orientation have no values

nukleuz80, 2021-10-29:
Hi there,

## Describe the bug

Two hours ago i included the samjin multipurpose sensor.
It is working, but the vibration (boolean) is always false and the orientation without values (0,0,0)

## Steps to reproduce the behavior

1. Reset the device
2. connect mode on device
3. connect mode in phoscon (sensor)
4. got the device in phoscon, deconz and api

## Expected behavior

Please let me know what i can do to get it work.

## Screenshots
from iobroker objects after include the device twice:
[image]

## Environment

DeConz: 

 - Host system: (Raspberry Pi 3b+)
 - Running method: (Raspbian)
 - Firmware version: (26390500)
 - deCONZ version: (2.13.01 / 6.10.2021 beta)
 - Device: (RaspBee I)

Thanks

SwoopX, 2021-10-29:
What makes you so sure that the root cause is deconz and not iobroker not processing the sent values correctly? 😃 

Please provide the REST API output of the affected sensor and share some logs from deconz while the corresponding functionalities are triggered.

nukleuz80, 2021-10-29:
A good question :)

So here is the output from REST API:
`{
    "config": {
        "battery": null,
        "enrolled": 1,
        "on": true,
        "pending": [],
        "reachable": true
    },
    "ep": 1,
    "etag": "410ce51d6d6dd38285314c7b96f9b7e9",
    "lastannounced": "2021-10-29T18:54:03Z",
    "lastseen": "2021-10-29T20:12Z",
    "manufacturername": "Samjin",
    "modelid": "multi",
    "name": "Multisensor",
    "state": {
        "lastupdated": "2021-10-29T19:27:14.228",
        "open": false
    },
    "type": "ZHAOpenClose",
    "uniqueid": "28:6d:97:00:01:12:5f:0a-01-0500"
}`

`{
    "config": {
        "battery": null,
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "b1fea47f40caba43c78da477f36915a2",
    "lastannounced": "2021-10-29T18:54:03Z",
    "lastseen": "2021-10-29T20:13Z",
    "manufacturername": "Samjin",
    "modelid": "multi",
    "name": "Temperature 69",
    "state": {
        "lastupdated": "2021-10-29T20:13:35.333",
        "temperature": 2046
    },
    "type": "ZHATemperature",
    "uniqueid": "28:6d:97:00:01:12:5f:0a-01-0402"
}`

`{
    "config": {
        "battery": null,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "b1fea47f40caba43c78da477f36915a2",
    "lastannounced": "2021-10-29T18:54:03Z",
    "lastseen": "2021-10-29T20:13Z",
    "manufacturername": "Samjin",
    "modelid": "multi",
    "name": "Vibration 70",
    "state": {
        "lastupdated": "none",
        "vibration": null
    },
    "type": "ZHAVibration",
    "uniqueid": "28:6d:97:00:01:12:5f:0a-01-fc02"
}`

And some log from DeConz (debug view - Info and Info_L2):

`22:16:54:107 Node data 0x286d970001125f0a profileId: 0x0104, clusterId: 0x0402
22:16:54:112 0x286D970001125F0A: update ZCL value 0x01/0x0402/0x0000 after 0 s
22:16:54:119 [INFO] - No button map for: multi, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0402, command: 0x0A, payload: 0000290308, zclSeq: 68
22:16:54:123 ZCL attribute report 0x286D970001125F0A for cluster: 0x0402, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
22:16:54:127 	payload: 0000290308
22:16:54:131 binding for attribute reporting of ep: 0x01 cluster 0x0402 seems to be active
22:16:54:144 Websocket 192.168.0.23:64153 send message: {"attr":{"id":"68","lastannounced":"2021-10-29T18:54:03Z","lastseen":"2021-10-29T20:16Z","manufacturername":"Samjin","modelid":"multi","name":"Multisensor","swversion":null,"type":"ZHAOpenClose","uniqueid":"28:6d:97:00:01:12:5f:0a-01-0500"},"e":"changed","id":"68","r":"sensors","t":"event","uniqueid":"28:6d:97:00:01:12:5f:0a-01-0500"} (ret = 336)
22:16:54:150 Websocket 192.168.0.56:48690 send message: {"attr":{"id":"68","lastannounced":"2021-10-29T18:54:03Z","lastseen":"2021-10-29T20:16Z","manufacturername":"Samjin","modelid":"multi","name":"Multisensor","swversion":null,"type":"ZHAOpenClose","uniqueid":"28:6d:97:00:01:12:5f:0a-01-0500"},"e":"changed","id":"68","r":"sensors","t":"event","uniqueid":"28:6d:97:00:01:12:5f:0a-01-0500"} (ret = 336)
22:16:54:168 Websocket 192.168.0.23:64153 send message: {"attr":{"id":"69","lastannounced":"2021-10-29T18:54:03Z","lastseen":"2021-10-29T20:16Z","manufacturername":"Samjin","modelid":"multi","name":"Temperature 69","swversion":null,"type":"ZHATemperature","uniqueid":"28:6d:97:00:01:12:5f:0a-01-0402"},"e":"changed","id":"69","r":"sensors","t":"event","uniqueid":"28:6d:97:00:01:12:5f:0a-01-0402"} (ret = 341)
22:16:54:174 Websocket 192.168.0.56:48690 send message: {"attr":{"id":"69","lastannounced":"2021-10-29T18:54:03Z","lastseen":"2021-10-29T20:16Z","manufacturername":"Samjin","modelid":"multi","name":"Temperature 69","swversion":null,"type":"ZHATemperature","uniqueid":"28:6d:97:00:01:12:5f:0a-01-0402"},"e":"changed","id":"69","r":"sensors","t":"event","uniqueid":"28:6d:97:00:01:12:5f:0a-01-0402"} (ret = 341)
22:16:54:194 Websocket 192.168.0.23:64153 send message: {"e":"changed","id":"69","r":"sensors","state":{"lastupdated":"2021-10-29T20:16:54.116","temperature":2051},"t":"event","uniqueid":"28:6d:97:00:01:12:5f:0a-01-0402"} (ret = 165)
22:16:54:200 Websocket 192.168.0.56:48690 send message: {"e":"changed","id":"69","r":"sensors","state":{"lastupdated":"2021-10-29T20:16:54.116","temperature":2051},"t":"event","uniqueid":"28:6d:97:00:01:12:5f:0a-01-0402"} (ret = 165)
22:16:54:218 Websocket 192.168.0.23:64153 send message: {"attr":{"id":"70","lastannounced":"2021-10-29T18:54:03Z","lastseen":"2021-10-29T20:16Z","manufacturername":"Samjin","modelid":"multi","name":"Vibration 70","swversion":null,"type":"ZHAVibration","uniqueid":"28:6d:97:00:01:12:5f:0a-01-fc02"},"e":"changed","id":"70","r":"sensors","t":"event","uniqueid":"28:6d:97:00:01:12:5f:0a-01-fc02"} (ret = 337)
22:16:54:224 Websocket 192.168.0.56:48690 send message: {"attr":{"id":"70","lastannounced":"2021-10-29T18:54:03Z","lastseen":"2021-10-29T20:16Z","manufacturername":"Samjin","modelid":"multi","name":"Vibration 70","swversion":null,"type":"ZHAVibration","uniqueid":"28:6d:97:00:01:12:5f:0a-01-fc02"},"e":"changed","id":"70","r":"sensors","t":"event","uniqueid":"28:6d:97:00:01:12:5f:0a-01-fc02"} (ret = 337)
22:16:54:310 Master: read param with arg 0x19
22:16:54:352 Device TTL 2001 s flags: 0x7`

nukleuz80, 2021-10-31:
Ok, i see there are no orientation values in the API. But the vibration value is NULL. 

I restart the raspberry yesterday - but nothing change… 

Any suggestions?

Thanks

SwoopX, 2021-10-31:
Hm, was that sensor newly added to the network or has it worked before? Currently, I'd put my bet on a missed binding. You might try resetting the sensor and then pair it again. For that, do not touch it in Phoscon so the sensors keep their IDs.

nukleuz80, 2021-11-01:
The sensor was newly added - never worked before. I resetted the sensor and did´nt touch it in Phoscon or DeConz.
But nothing changed - hmmmm...

nukleuz80, 2021-11-01:
Now i re-included it 5 times and there is the same result... Tomorrow i will try z2m for sensor tests... to see if it´s working.

nukleuz80, 2021-11-01:
I tested with iobroker zigbee (conbee2) which is based on z2m and there are all informations i missed in deconz:

[image]

All values are changing and working very well...

The yellow marked options are missing in deconz - i think tampered, too.

Here are all datapoints i received from z2m:
[image]

I want this in deconz - thanks! :)

nukleuz80, 2021-11-13:
Is there something i can do?

closed 2021-12-13 by github-actions[bot]

nukleuz80, 2021-12-14:
FYI: Now it is working fine: Now i´m on 2.13-2beta on Raspbian Buster.