dresden-elektronik/deconz-rest-plugin #5427: Device name

elpablillo, 2021-10-30:
## Device

- Product name: VIKEFON
- Manufacturer: _TZ3000_cphmq0q7
- Model identifier: TS011F
- Device type : 
  - Switch

## Screenshots

### Node info
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]

### Scenes
[image]

### On/Off
[image]

### Simple Metering
[image]

### Electrical Measurement
[image]

Smanar, 2021-10-30:
Same strange cluster than https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5373

elpablillo, 2021-11-22:
> Same strange cluster than #5373

So, any chance those power measure will appear in HA?

Smanar, 2021-11-22:
Hu ... I think not ATM sorry.
We are converting legacy code to DDF, so I don't think we will add C++ code to support new cluster for the moment

CarmeloSena, 2021-12-03:
Do you already know if there is a way to monitor the power of this socket?

Smanar, 2021-12-04:
No more information on my side.

closed 2022-01-02 by github-actions[bot]