dresden-elektronik/deconz-rest-plugin #5429: Where are green links?

LeoeLeoeL, 2021-10-30:
## Describe the bug
I updated from 12.6 to 13.1. Now lots of green (strong) links are vanished.
I read about the new LQI filter but I tried to set 1 and 0 too!!

## Steps to reproduce the behavior

## Expected behavior
 I expect to watch strong links as before.
There are devices very close each other and they should be green.

## Screenshots
 An old screenshot grabbed on 2.11.1-beta:
[image]

Now:
[image]

We can see the difference focusing on the device called ON/OFF Plugin Unit 46 (upper picture) and Multipresa 1 (lower picture) in the lower right..
It's the same device in the same place.
But also Presa Esterna Strada and Irrigazione Strada in the upper left; The distance between them is 15 cm and in previous versions the link was green

## Environment
0x26720700

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26720700
 - deCONZ version: (2.13.01)
 - Device: ConBee II
 - Do you use an USB extension cable: yes -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which? APC UPS

## deCONZ Logs

## Additional context

Smanar, 2021-10-30:
Hello https://forum.phoscon.de/t/network-routing-optimisation-question-deconz-2-13-01/615/8

But I have same reaction on my network (and less green if I enable source routing), but all is working same than before, for me it s just cosmetic.

LeoeLeoeL, 2021-10-30:
Ok, I read about the different calculation but I lost green links among devices that "kiss" each other!!

Smanar, 2021-10-30:
Have you check the LQI value to compare ?

LeoeLeoeL, 2021-10-31:
No I didn't.
Anyway, if this modification has be done to help to discover weak links it would be useful to understand which is the weaker device in the link. I.E. if I see 128/255 I would like to know which is the weaker between Lidl and Osram smartplugs.
Two more questions:
- What about when 110/0 ? I should not see that link if I'm filtered 30 or more;
- Why the weakest links (for instance 4/14) aren't red as before? Now everything is a pale orange.

Last thing:
[image]
The thin red line under LUCE SALA CINEMA and MULTIPRESA 1 is a glitch?

Mimiix, 2021-10-31:
I am not sure if this is a bug as per #5113.

Mimiix, 2021-10-31:
From the discord

Zigbee network troubleshooting

New UI features have been added to find routing issues in a Zigbee network.

The line colors between nodes have more contrast to better show weak links.

Routing depends on how good devices can "see" each other.
The driving parameter here is the  Link Quality Indicator (LQI).

For example, given two lights A and B, there are two LQI values in play: (i) How A sees B, and (ii) how B sees A. When the LQI button in deCONZ is enabled these values are shown.

LQI value meanings:

- 255 excellent
- - 200 good
- 130 flaky but can work
- - < 100 problematic

Always aim for LQI values of 200 and above to get a stable network!

Under "Panels > Source Routing" is a new checkbox Minimum LQI display which hides links below the given LQI value.

When setting a new minimum LQI value, after a few seconds weak links are hidden in the UI, to easier spot areas where a router should be placed.

New Dependency

Under Linux the libqt5qml5 needs to be installed.

Please refer to the related version to get and overview of respective changes.
The detailed changelog can be found at GitHub release page v2.13.0.

Community message 

Please note that the DDF functionality is in beta and we do not provide active support for it. It is meant for advanced users. Any questions about I should be asked here: https://forum.phoscon.de/c/deconz/general-support/12 but no promises here. 

A special thanks to all contributors of the deCONZ Github community.

- Dennis / @Mimiix / https://discord.gg/3XGEYY9

LeoeLeoeL, 2021-10-31:
So, Houston we ha a problem (of connectivity) :-)
I have bulbs in the same chandelier and they don't have a line if I set 130!
And... Ikea extenders work worse than a bulb as routers. Practically useless.

pillemats, 2021-11-01:
Same Problem here. In 12.6 all works very well. At 13.x same problem with the connection

Mimiix, 2021-11-01:
Would be useful to see the lqi values in 2.12.6 and the latest beta.

Is there differences there?

LeoeLeoeL, 2021-11-01:
Probably not.
It could be nice to have both option to choose for the representation: "(LQI A + LQI B) / 2" and "Minimum of LQI A and LQI B".

Mimiix, 2021-11-01:
> Probably not.

You mean that there's not a difference between the LQI Values in 2.12.6 and the latest beta? Just that the colors are different?

LeoeLeoeL, 2021-11-01:
I suppose LQI values are given by devices so that the Deconz version should be irrelevant. Somebody who still has 2.12.6 can verify when he will update.
But losing so many green lines has a bad impact. You realize Zigbee is not so reliable.

Mimiix, 2021-11-01:
> But losing so many green lines has a bad impact. You realize Zigbee is not so reliable.

Well it isn't really. It just made it more clear. If the lines changed color, but the functunality stayed the same: What is the issue then? It's even more helpful now. It's mostly a color/gfx change. 

As you are the reporter of a bug, you need to make sure it complies to #5113. I'm happy to wait a few days, but if nobody comes up with anything reproducable / proof i have to close this issue.

LeoeLeoeL, 2021-11-01:
Ok, let's just wait few days in order to see if somebody notices differences in LQI values updating from 2.12.6 to 2.13.x.

Mimiix, 2021-11-01:
Can't you revert yourself? Just to test out?

manup, 2021-11-01:
> Probably not. It could be nice to have both option to choose for the representation: "(LQI A + LQI B) / 2" and "Minimum of LQI A and LQI B".

This is indeed on the todo list to have get different views for the network. v2.13.x approach looks now way more drastic in some networks, as above. But this is the most realistic picture to really figure out weak spots.

The devil is in the details, the best is always to have A and B LQI as high as possible, so sending and receiving is solid.
In my experience showing  "(LQI A + LQI B) / 2" is misleading since a connection might only be good in one direction, while the line is shown more greenish.

LeoeLeoeL, 2021-11-01:
> ... it would be useful to understand which is the weaker device in the link. I.E. if I see 128/255 I would like to know which is the weaker between Lidl and Osram smartplugs.

Do you think it's possible to do that?

Mimiix, 2021-11-02:
@jjansen85 As i said on Tweakers, their pure indicative. In this case: What happens after a restart or if you click LQI? Either way: please post this on the forum as it is more a user question/own setup issue than a general bug.

jjansen85, 2021-11-02:
Sorry, thought this might have been related to this issue.
Restart or clicking LQI does not change anything.

Mimiix, 2021-11-02:
The difference here is that you are on Docker. 

Perhaps the VNC is not showing it properly 🤔.

I'll send you a PM on Tweakers, perhaps we can test some things.

pillemats, 2021-11-02:
I am on win10 vm and have the same. Also the devices won’t connect! Not only I can’t see the lines….

so I think this is not a docker problem

Mimiix, 2021-11-02:
> lso the devices won’t connect!

Different issue, as devices are working in the above cases.
Please open a forum post on this and make sure to add logs.

Smanar, 2021-11-02:
And if you set lower value for LQI ? < 130 ? On the last version deconz seem more psychorigid  ^^

Mimiix, 2021-11-02:
@Smanar Lets keep it on topic here please. 1 issue at the time.

pillemats, 2021-11-02:
> > lso the devices won’t connect!
> 
> Different issue, as devices are working in the above cases. Please open a forum post on this and make sure to add logs.

sorry, it was my mistake! devices works correctly but not shown in deconz as connected line
[image]

Mimiix, 2021-11-02:
@pillemats I do see links.

It sometimes can take a while and somtimes they dont show for a while. Sometimes you have to move them to make them show.

pillemats, 2021-11-02:
Yes I now but in 12.6 they will connect much faster. 1 h is ago after starting 13.1 and many devices are not connected and don’t work. Also many routines I set.
In 12.6 all devices are connected after 10 minutes

LeoeLeoeL, 2021-11-02:
A strange thing in my network under 13.1 (the 2nd picture in this thread) is the only "long" green line is between the coordinator and an end device (Tradfri shortcut button) and that device is farer than many routers .

manup, 2021-11-02:
Can you check the *deCONZ > Menu > Panels > Source Routing > Min LQI display* value if the links are weak and fall below this value they are not shown.

In that case you can set it to 0 to disable the filter.

The "Fast neighbor discovery" checkbox speeds up the query of the neighbor links (but should be disabled after that).

pillemats, 2021-11-02:
i cant set lqi max down to 75

closed 2021-12-01 by github-actions[bot]

felker, 2022-03-12:
@Mimiix is there a way to figure out which nodes are A,B in the ordering of the LQI values in https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/LQI-explained ?

Smanar, 2022-03-12:
Lol, ha yes good question.
When there is 2 routers with 80/70, how to know wich one is A or B ?

Mimiix, 2022-03-12:
The LQI is shown Per line between 2 devices.

felker, 2022-03-13:
But is there anyway to know which line is which direction? E.g. in this example the two Lamps are mains-powered devices, but the switch is battery-powered so the LQI is nonexistent in one direction, presumably Lamp to Switch? So it can be inferred that the 255 LQI corresponds to A->B A=Lamp, B=Switch, and the 0 LQI is B->A ?

But unclear how you could figure it out for the LQI values between the two lamps...

[image]