dresden-elektronik/deconz-rest-plugin #5435: frient Smart Plug Mini

juliendragz, 2021-11-01:
## Device

- Product name: frient Smart Plug Mini
- Manufacturer: frient A/S
- Model identifier: SPLZB-132
- Device type : 
  - Other: smart plug on / off

## Screenshots

[image]

### Node Info panel
[image]

### Basic
[image]
[image]

### Identify
[image]

### Alarms
[image]

### Device Temperature
[image]

### Electrical Measurement
[image]
[image]

### On/Off
[image]
[image]

### Simple Metering
[image]
[image]
[image]
[image]
[image]
[image]

## Infos
Hi,
I searched in if the device was not already added I found this model which is added in 2019 which looks like:
https://github.com/dresden-elektronik/deconz-rest-plugin/commit/efae9b2b134f4a5e824b45980baad9135537f606

I also tried to add the device with DDF, by adding the measurement data I have to get this information in the websocket but not in the rest api.
[ddf without edit.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7453367/ddf.without.edit.txt)
Api with ddf edited:
[On_off_smart_plug .txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7453216/On_off_smart_plug.txt)
[image]
[image]

And if I understood correctly the use of DDF I cannot activate it when starting deconz? Is it a tool that is used to verify the mapping of the device, to then integrate it into deconz?

SwoopX, 2021-11-01:
Closing this one as it is already supported and 100% the same like the Develco smart plug (its even the same company).

closed 2021-11-01 by SwoopX

juliendragz, 2021-11-01:
My bad, I updated deconz without resetting the device. I convinced myself that it was not necessary 🤦‍♂️. Thank you