dresden-elektronik/deconz-rest-plugin #5437: Lights are not a color controllable light but the type is and it reports being in colormode hs

kmplngj, 2021-11-02:
## Describe the bug

Philips Hue Lux E27
IKEA TRADFRI bulb E14 W op/ch 400lm

Both support dimming, both no color, both no color temperature.

The lights are not a color controllable light but the type is and it reports being in colormode hs to home assistant.

## Steps to reproduce the behavior

See home assistant log. When I try to turn the light on/off an error is logged in Home Assistant.

[home-assistant.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7458527/home-assistant.log)

## Expected behavior

## Screenshots

## Environment

 - Host system: VM
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26720700
 - deCONZ version: deCONZ 2.12.6
 - Device: ConBee II
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs

## Additional context

https://github.com/home-assistant/core/issues/58805

kmplngj, 2021-11-24:
Ping

MaStahl84, 2021-12-08:
Hello,
same problem here with two "Tuya Dimm LED Controller _TZ3000_4whigl8i - TS0501B".
Both are controllable with Phoscon or HueEssentials, but don't have the right state in Home Assistant.

Environment
Host system: VMWare ESXI
Running method: Home Assistent deCONZ Ubuntu aon VMWare ESXI
Firmware version: 26720700
deCONZ version: deCONZ 2.12.6
Device: ConBee II
Do you use an USB extension cable: yes
Is there any other USB or serial devices connected to the host system? If so: Which? No

MaStahl84, 2021-12-21:
Hi, 
with Version 2.13.4 i could solve this problem with a new DDF file.

I set colormode to static "xy". The lights are now controllable with Home Assistant.
All other only dimmable lights had the same settings.

I didn't try to remove the attribute colormode, maybe that works too.

kmplngj, 2022-01-12:
Ping

Mimiix, 2022-01-12:
This can be fixed with the DDF as @MaStahl84 mentioned

kmplngj, 2022-01-12:
@Mimiix Should this not be fixed as a default also for new users? 

Will my edit survive an update of the deconz Home Assistant add-on?

Mimiix, 2022-01-12:
We can publish the DDF afterwards for other users :)

berendhaan, 2022-01-24:
Is this being fixed in an upcoming release, or do I just have to "patch" my DDF?

Mimiix, 2022-01-24:
@berendhaan needs to be done in your ddf.

closed 2022-02-23 by github-actions[bot]