dresden-elektronik/deconz-rest-plugin #5438: Aqara motion sensors not consistent

benblb, 2021-11-02:
## Describe the bug
Occasionally an Aqara motion sensor will not detect motion. It normally fixes itself 30s - 1m later and then triggers which causes me to believe there is an issue with the Aqara sensor... however they were working perfectly before I upgraded my HA server and restored deCONZ so I suspect there may be a deCONZ problem. Additional context at bottom has more info. 

## Steps to reproduce the behavior
- Setup Aqara motion sensor
- Use sensor and note that sometimes it misses detection

Happens several times a day, around once every hour if I walk by. 

## Expected behavior
Motion sensor should always detect and notify deCONZ

## Screenshots
N/A and no errors in logs. 

## Environment
 - Host system: NAS (Chromebox running Debian 11)
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.12.06 / 8/19/2021
 - Device: ConBee II
 - Do you use an USB extension cable: yes, USB 2.0
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
Only weird thing I see in logs is "...(lumi.sensor_motion.aq2): disable presence"

## Additional context
Troubleshooting has been completed.

- New batteries have been put into the 3 problem motion sensors
- Disabled all powered devices and re-paired motion sensors to ensure they are not using a router (all directly connected to Conbee II)
- Rebooted host system
- USB extension added (then another reboot)
- Verified nearby WiFi is not conflicting with Zigbee channel 25 (in use)

I do want to note that everything was working perfectly when I was running an "unsupported" Home Assistant install on Gallium OS (now Debian 11). I accidentally bricked that "NUC", so I swapped to one that is identical in every way *except* I upgraded the 16GB SSD to 128GB and this one has 8GB of RAM instead of 4. 
The old server and this went in the same spot in my closet and I restored my HA backup including deCONZ. 

It's possible this is not a deCONZ issue, but nothing had changed with the Aqara sensors before the problems began.

Mimiix, 2021-11-02:
Hi there,

As this is not a bug as described in #5113, i am closing this issue. Please use the forums for help/setup questions.

Kind regards,

closed 2021-11-02 by Mimiix

benblb, 2021-11-02:
I am aware of the forums @Mimiix. How do you know for sure this is not a bug?

Mimiix, 2021-11-02:
For the same reason we can't be sure its a user specific issue: We want github to have issues that are 100% provable bugs.

benblb, 2021-11-02:
Ah, that's fair. Thanks very much.