dresden-elektronik/deconz-rest-plugin #5439: Avatto / Tuya Zigbee Thermostat ME81H

radeonite, 2021-11-02:
## Device

- Product name: Avatto / Tuya Zigbee Thermostat ME81H
- Manufacturer: _TZE200_ye5jkfsb
- Model identifier: TS0601
- Device type : 
  - Thermostat

## Screenshots

### Basic
[image]

### Identify
[image]

### Tuya specific cluster (empty)
[image]

tinloaf, 2021-11-08:
Is this the same as #5436 ? I have another one of those TS0601, with a different manufacturer ID (`_TYST11_jeaxp72v`, "wesmartify Essentials"), which is probably also the same?

Smanar, 2021-11-08:
Not possible to guess, I don't find information about this one.
You can compare used tuya command

radeonite, 2021-11-09:
Some more info about the product => https://www.aliexpress.com/item/1005003228429571.html
Is there something I can do from my side with the product itself to get more info?

I tried seeing if there are any Tuya log prints, but I only saw this one (process was pairing the device):
18:04:02:752 Tuya debug 1 : Time sync request
18:04:02:755 Send Tuya command 0x24, data: 0006618aaa02618ab812

Smanar, 2021-11-09:
To make it short, the new device integration is blocked for the moment (deconz is making a big transformation to use DDF)

So if you are able to compile the code (you need a linux machine with full OS), I can give you the modification to do to have at least the temperature and the setpoint, but the code will be only for you

And I have take a look, it seem deconz don't use clone of this device yet.

radeonite, 2021-12-05:
I'll be setting up a VM to compile the code. I guess it's only possible to compile for Linux, not Windows?

Smanar, 2021-12-05:
Yep, but it s for your personnal use ? Else remember lot of PR are blocked ATM.

radeonite, 2021-12-09:
Alright, I'll just wait it out and see what the future brings with DDF :)

closed 2022-01-07 by github-actions[bot]

Apfelsafft, 2022-10-28:
Hi,
is there any update? 
I do have the same device (ME81AH). It shows up in deconz but I can't get any data from the thermostat.
Is there a DDF available?

radeonite, 2022-10-28:
With the latest release I don't see a difference, so I don't think there's any DDF yet for this device. I've been thinking of trying to create a DDF myself, but I'm not sure where to start :)

Smanar, 2022-10-28:
Not possible to add new Tuya TRV for the moment.

Apfelsafft, 2022-10-29:
I played around with the DDF editor but had no luck - and no knowledge ;-)
If someone has a DDF, maybe even for a similar device that could be adjusted then, that would be awesome.

But until then I guess, I'll need to wait.
Thanks for the replies!!

Smanar, 2022-10-29:
It's not possible for the moment to add tuya TRV, the PR that can permit that is still in waiting list.

Apfelsafft, 2022-11-02:
Thanks for the info. 
Could please state which PR it is so that I can watch it?

GunnarThorset, 2022-11-16:
Hello. I also have got the ME81AH. But I cant figure out how to put in pairing mode, so I can connect it to the Smart Life app, as a start. Does anyone know?

Apfelsafft, 2022-11-19:
Hi Gunnar,
switch the Thermostat off and then press up and down keys simultaneously
until it starts the pairing mode after a few seconds.
You can also google for the user manual. Can't remember where I found it
but it contains some more infos on pairing. Compared to the one pager that
was delivered with the device.

Wish you success ;-)

Best,
René

radeonite, 2023-06-05:
@Smanar Any idea if the changes of https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6674 might help get support for this device as well? Or how can I help to debug and get it working? :)

Smanar, 2023-06-06:
Sure, can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_ztvwu4nk", "_TZE200_5toc8efa", "_TZE200_ye5jkfsb", "_TZE200_u9bfwha0"],
  "modelid": ["TS0601", "TS0601", "TS0601", "TS0601"],
  "vendor": "Tuya",
  "product": "Tuya TRV (MOE)",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "meta": {
        "values": {
          "config/mode": {"auto": 0, "heat": 1, "off": 3}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 16, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 16, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 40, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 40, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/mode",
          "values": [
              ["auto", 0], ["heat", 1], ["off", 3]
          ],
          "parse": {"fn": "tuya", "dpid": 2, "eval": "if (Attr.val == 0) { Item.val = 'auto' } else if (Attr.val == 1) { Item.val = 'heat' } else { Item.val = 'off' }"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x30", "eval": "if (Item.val == 'auto') { 0 } else if (Item.val == 'heat') { 1 } else { 3 }"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 24, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        }
      ]
    }
  ]
}
```

Smanar, 2023-06-09:
I don't see something specific for the _TZE200_ye5jkfsb except for the local temperature.

> But if I then read again, the written value is returned

How you read it ? It s not possible to read attribute on tuya device.
Thoses device use same id for read and write value so if the read is working and not the write, it's probably more an isue in the "write" part, wrong type for exemple.
It's perhaps visible on log when you make the request (with "aps" and "aps_l2" but lot of talkative)

radeonite, 2023-06-09:
As expected, a case of PEBCAK :). I was reading it via the REST API so indeed nothing was being read from the device.

The issue with the temperature was due to sending an invalid value. I assumed it would be similar to another Tuya thermostat I have, but instead of f.e. 3000 for 30 degrees, I have to use 300 (that or divide by 100 iso 10 in the JSON file).

The "mode" has 2 options, one is the "manual" mode (0) and the other is "schedule" (1). 

Additionally, DP 0x01 allows to turn on/off the device (bool type, 0 off and 1 on). 

There's something strange with the temperature read-out though (DP 0x18) . It doesn't seem to be updated, the state keeps sitting at "lastupdated": "none" and the "temperature" at 0. Using a data request in Deconz on 0x18 also doesn't produce anything in the logs. But I'll try some more things tomorrow.

radeonite, 2023-06-10:
Temperature reporting is ok now, I gave the device a reboot. Probably I confused it somehow by maybe sending incorrect data :)

Smanar, 2023-06-10:
>Additionally, DP 0x01 allows to turn on/off the device (bool type, 0 off and 1 on).

Ha yes I reconise the device now, and why the DDF don't have battery field.
Can try to add on the DDF

```
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "none"}
        },
```

>As expected, a case of PEBCAK :). I was reading it via the REST API so indeed nothing was being read from the device.

So you are not reading the value on the device, and if you can read the new value in it, it mean the device have already send the new value.
Deconz need to send a websocket notification when it receive the new value

> If I write to it, the value isn't taken over by the thermostat. But if I then read again, the written value is returned. So that sounds a bit like this thermostat isn't using the same datapoint ID?

So yes, you are looking the value inside the API, not on the device, a test you can do is changing the value on the device, you will see if deconz update it in the API too (mean the dpid is good, so the request is bad or the device is too "lazy")

>EDIT: or maybe not, the first couple of minutes the temperature gets updated, but after that there are no more updates. Hmm, I wonder why...

Lot of issue that are like connexion issues, but on a wired device ....
You can heat the sensor, and look for debug message in logs, for the dpid 0x18 you can have a convertion issue, I can see one but there is so much clone on tuya, I can't be sure for this model.
You can heat the device, and look for logs in deconz/help/debug view with flags "info"

`"TY_DATA_%s: seq %u, dpid: 0x%02X, type: 0x%02X, length: %u, val: %d\n"`

radeonite, 2023-06-11:
I had the JSON file in a random directory, but now I put it together with the other TUYA devices under "devices\tuya\". That and I tried increasing the temperature. Now I do see the device reporting the current temperature! I do notice this thermostat is less accurate than the other model and seems to only send updates per full degree so maybe that's why I thought there were no updates as the temperature is quite stable here. Either way, temperature read-out is ok.

Then the state/on, I updated the DDF with the piece you shared. If I change the on/off state at the thermostat, then indeed the state/on is toggling between true/false when read via the REST API. But if I try to set the state/on via the REST API I get an error stating this parameter is not available.  Or should I try setting it differently?

Body for state/on:
`{
    "state": {
        "on": false,
    }
}`

Full DDF as used now:
`{
  "schema": "devcap1.schema.json",
  "manufacturername": [
    "_TZE200_ztvwu4nk",
    "_TZE200_5toc8efa",
    "_TZE200_ye5jkfsb",
    "_TZE200_u9bfwha0"
  ],
  "modelid": [
    "TS0601",
    "TS0601",
    "TS0601",
    "TS0601"
  ],
  "vendor": "Tuya",
  "product": "Tuya TRV (MOE)",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "meta": {
        "values": {
          "config/mode": {
            "manual": 0,
            "schedule": 1
          }
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "read": {
            "fn": "tuya"
          },
          "write": {
            "dpid": 16,
            "dt": "0x2b",
            "eval": "Item.val / 100;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 16,
            "eval": "Item.val = Attr.val * 100;",
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "config/locked",
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 40,
            "dt": "0x10",
            "eval": "Item.val;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 40,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": false
        },
        {
          "name": "config/mode",
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 2,
            "dt": "0x30",
            "eval": "if (Item.val == 'manual') { 0 } else { 1 }",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 2,
            "eval": "if (Attr.val == 0) { Item.val = 'manual' } else { Item.val = 'schedule' }",
            "fn": "tuya"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
		{
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 24,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 0
        }
      ]
    }
  ]
}`

Smanar, 2023-06-12:
Use as body `{"on": false}` on endpoint `http://IP:PORT/api/API_KEY/sensors/ID/state`