dresden-elektronik/deconz-rest-plugin #5443: Aqara P1 Door and Window Sensor

MiAutomations, 2021-11-03:
## Device
-Link: https://s.click.aliexpress.com/e/_9AzK9T
- Product name: Aqara P1 Door and Window Sensor
- Manufacturer: LUMI
- Model identifier: lumi.magnet.ac01
- Device type :
  - Sensor
 

## Screenshots

[image]

[image]

Mimiix, 2021-11-03:
Please add all screenshots.

MiAutomations, 2021-11-03:
Please all the screen shots added Thank you

[image]
[image]
[image]
[image]

SwoopX, 2021-11-03:
Sorry to say, but the screenshots are not usable as all the attributes aren't read. Push the physical button and then immediately press the read button.

ads1230, 2021-12-17:
@SwoopX did my screenshots help?

closed 2022-01-15 by github-actions[bot]