dresden-elektronik/deconz-rest-plugin #5446: Device name

megawubs, 2021-11-04:
## Device

- Product name: YPHIX ZIGBEE LED DRIVER 350-1400MA 50W
- Manufacturer: Philips.
- Model identifier: DIM Lightning.
- Device type : Please remove all unrelated device types. 
  - Light (LED DRIVER)

## Screenshots

 [image]
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]

### Scenes
[image]

### On/Off
[image]

### Level Control
[image]

### Color Control
[image]

### Diagnostics
[image]

### Other clusters that are not mentioned above

#### OATU
[image]

#### Green Power
[image]

This is my first device request issue. I've tried to do my best and keep to the issue template. Please let me know if something is missing or if more info is needed. Also, if someone is up to it, i would like to know how to add support for this device myself, but have no idea where to start. Some guidance would be nice!

Mimiix, 2021-11-04:
Looks fine to me :)

What is not working currently? As it seems to be recognized correctly?

megawubs, 2021-11-04:
I've described the problem here: https://github.com/home-assistant/core/issues/58174#issuecomment-956513886 and was told to open a device request. TLDR: The device reports values it doesn't typically support, causing Home-Assistant to crash and not display the correct device state.

megawubs, 2021-11-04:
And, since a few weeks this device looses connection randomly.

megawubs, 2021-11-27:
Is there anyone with more knowledge about the plugin that can look at this? From what I understand the plugin gives back values to Home-Assistant that the device does not support.

closed 2021-12-26 by github-actions[bot]

tomdegoeij, 2022-11-16:
@megawubs Were you able to resolve your issues with this device? I'm planning on purchasing a few of those but want to be sure they are supported using Deconz through HA. Thanks!