dresden-elektronik/deconz-rest-plugin #5447: Yooksmart (Yoolax) D10110 roller shade - Lift percentage is reported incorrectly (reversed)

teladog01, 2021-11-04:
## Describe the bug

[image]

This roller shade 👆[(Amazon link)](https://www.amazon.com/gp/product/B08LV5ZGM4/) is having its lift percentage incorrectly reported by the REST API. The value reported is opposite of the expected value.

Fully open should be `"lift": 0`, but is reported as `"lift": 100`. 
Fully closed should be `"lift": 100`, but is reported as `"lift": 0`.
10% closed should be `"lift": 10`, but is reported as `"lift": 90`.
90% closed should be `"lift": 90`, but is reported as `"lift": 11`.

Controls all work as expected, just the lift is reported incorrectly.

## Steps to reproduce the behavior

Control the device. Query the REST API. Review the JSON. Observe the "lift" value is opposite of the expected value.

## Expected behavior

Correct reporting of lift percentage from the REST APi.

## Screenshots

### Fully Open:
[image]
[image]
```
{
  "etag": "ce361ccec9fddea1596987b92a95aab0",
  "hascolor": false,
  "lastannounced": null,
  "lastseen": "2021-11-04T20:45Z",
  "manufacturername": "yooksmart",
  "modelid": "D10110",
  "name": "South Window Shade",
  "state": {
    "bri": 254,
    "lift": 100,
    "on": true,
    "open": false,
    "reachable": true
  },
  "swversion": "20201120",
  "type": "Window covering device",
  "uniqueid": "8c:f6:81:ff:fe:de:c1:a1-01"
}
```

### Fully Closed:
[image]
[image]
```
{
  "etag": "f4d78394e3c2e0dbab8f4ba4f57579f7",
  "hascolor": false,
  "lastannounced": null,
  "lastseen": "2021-11-04T20:47Z",
  "manufacturername": "yooksmart",
  "modelid": "D10110",
  "name": "South Window Shade",
  "state": {
    "bri": 0,
    "lift": 0,
    "on": false,
    "open": true,
    "reachable": true
  },
  "swversion": "20201120",
  "type": "Window covering device",
  "uniqueid": "8c:f6:81:ff:fe:de:c1:a1-01"
}
```
### 10% Closed:
[image]
[image]
```
{
  "etag": "32def84048c1357f2881c21f477bef28",
  "hascolor": false,
  "lastannounced": null,
  "lastseen": "2021-11-04T20:58Z",
  "manufacturername": "yooksmart",
  "modelid": "D10110",
  "name": "South Window Shade",
  "state": {
    "bri": 228,
    "lift": 90,
    "on": true,
    "open": true,
    "reachable": true
  },
  "swversion": "20201120",
  "type": "Window covering device",
  "uniqueid": "8c:f6:81:ff:fe:de:c1:a1-01"
}
```
### 90% Closed:
[image]
[image]
```
{
  "etag": "ef5470b21f8f7eb00fe317d9a0097007",
  "hascolor": false,
  "lastannounced": null,
  "lastseen": "2021-11-04T21:00Z",
  "manufacturername": "yooksmart",
  "modelid": "D10110",
  "name": "South Window Shade",
  "state": {
    "bri": 27,
    "lift": 11,
    "on": true,
    "open": true,
    "reachable": true
  },
  "swversion": "20201120",
  "type": "Window covering device",
  "uniqueid": "8c:f6:81:ff:fe:de:c1:a1-01"
}
```
## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26400500
 - deCONZ version: (2.13.01)
 - Device: RaspBee I
 - Is there any other USB or serial devices connected to the host system? USB mouse and keyboard

## deCONZ Logs

## Additional context

Mimiix, 2021-11-05:
@Smanar

Mimiix, 2021-11-05:
Funny one: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4257#issuecomment-855281308

Someone already said it was reversed and now you are saying its reversed again 😅

teladog01, 2021-11-05:
> Funny one: [#4257 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4257#issuecomment-855281308)
> 
> Someone already said it was reversed and now you are saying its reversed again 😅

Interesting. Similar I suppose. But also different.

The first post screenshots show different Manufacturer, Yookee vs yooksmart, and Date Code, 20200808 vs 202011. Model Identifier is the same, D10110. So the OP appears to be a different device.

But then a few posts down, a different user re-opened the issue, but that user's screenshots show same values I have. So that thread is apparently discussing two different devices?

Mimiix, 2021-11-05:
> Interesting. Similar I suppose. But also different.
>
> The first post screenshots show different Manufacturer, Yookee vs yooksmart, and Date Code, 20200808 vs 202011. Model Identifier is the same, [...]

That can be, or you (or the other users) have mounted it incorrectly on the walls. 

If you mount it the otherway around, up becomes down and down becomes up.

teladog01, 2021-11-05:
> That can be, or you (or the other users) have mounted it incorrectly on the walls.
> 
> If you mount it the otherway around, up becomes down and down becomes up.

I'm aware of this possibility, and the motors have a built-in "reverse" function for the situation of mounting on the other side. The reverse function is performed using the included remote. I have tested this out, and when I perform the reverse procedure, then the status becomes correct, but the controls are backwards.

So it seems the motor functions as expected, but there is a problem with deconz's interpretation of the data?

Mimiix, 2021-11-05:
> I'm aware of this possibility, and the motors have a built-in "reverse" function for the situation of mounting on the other side. The reverse function is performed using the included remote. I have [...]

So in that case, the other user has it wrong , right?

teladog01, 2021-11-05:
> So in that case, the other user has it wrong , right?

Well, all I can say is that there is certainly a discrepancy, if indeed these are the exact same devices.

Mimiix, 2021-11-05:
Makes sense.

Lets wait what @Smanar brings to the table :)

Mimiix, 2021-11-05:
@jcarusone1650 What's your  side on things. Did you reverse it with the remote?

teladog01, 2021-11-05:
I found this in deconz:

[image]

Unfortunately, checking the "Reversed" box, then pressing the "write" button, seems to have no effect. The motor behaves the same. Nothing is reversed.

jcarusone1650, 2021-11-05:
I did not have to reverse the motors, the standalone/delivered remote/motors are working as delivered from manufacturer. I have 18 of them in the house. I will fire up deconz today to confirm they were not delivered in a reverse configuration.

teladog01, 2021-11-05:
I just noticed that my device has the "Tuya specific Cluster", whereas the [other thread screenshot](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4257#issuecomment-805135176) does not show this.

So something is different in deconz for these two seemingly identical devices (yooksmart D10110).

teladog01, 2021-11-05:
> I did not have to reverse the motors, the standalone/delivered remote/motors are working as delivered from manufacturer. I have 18 of them in the house. I will fire up deconz today to confirm they were not delivered in a reverse configuration.

Please also check if you have the "Tuya specific Cluster".

Smanar, 2021-11-05:
You have no more attribute available on the cluster 0x0102 ? On your capture they are all disabled, except the 0x0017

Have check on z2m too, and they are not using the "reversed" setting too, they only reverse the return from the device, like deconz on line 168 (windows_covering.cpp file)

For me the problem is more from the mouting procedure.

>The reverse function is performed using the included remote. I have tested this out, and when I perform the reverse procedure, then the status becomes correct, but the controls are backwards

Control on the remote or on the API ?

If I m right you have 4 mouting mode possible.
- The motor on the left/right.
- The layer is behind/front of the tube.

teladog01, 2021-11-05:
> You have no more attribute available on the cluster 0x0102 ? On your capture they are all disabled, except the 0x0017
> 
I'm not sure how to answer other than to say what you see in the screenshot is correct, 0x0102 is the only attribute that is access="rw" and also clickable (not grayed). Is that not normal? Sorry, I have nothing to compare to. I have two of these roller shades and they both look the same.

> > The reverse function is performed using the included remote. I have tested this out, and when I perform the reverse procedure, then the status becomes correct, but the controls are backwards
> 
> Control on the remote or on the API ?

Both are backwards after the remote reverse procedure.

> If I m right you have 4 mouting mode possible.
> 
> * The motor on the left/right.
> * The layer is behind/front of the tube.

Yes this is true. One of the shades, the motor is on the left. The other, the motor is on the right. Both are shade material in front of the tube. This is how I configured them from Yoolax. Both exhibit the same reversal of lift percentage.

> But the presence of the tuya can be a sign it's a different device. You have "Tuya log" on deconz when using the device ? with "info" and "info_l2"

Where do I look for "Tuya log"? To be clear, I am not, nor have I ever, used anything Tuya related.

Smanar, 2021-11-05:
In all case deconz can't make better than the official method with the remote.

> the motor is on the left. The other, the motor is on the right, Both are shade material in front of the tube

And both have the same problem ? Not logic, if for all the shade is on the front, the direction can't be the same natively.

>Where do I look for "Tuya log"? To be clear, I am not, nor have I ever, used anything Tuya related.

If the device use tuya cluster, you will see lines starting by

> Tuya debug 4 ......
> Tuya debug 5 ......

Honestly,  I m shy to reverse the device using the code, because it s not possible for me making something compatible for all users.
But can try to reverse it using DDF, have you already take a look on it ?

Can you give the DDF for the actual device ?

teladog01, 2021-11-05:
> > the motor is on the left. The other, the motor is on the right, Both are shade material in front of the tube
> 
> And both have the same problem ? Not logic, if for all the shade is on the front, the direction can't be the same natively.

This fact hasn't sat well with me either. Doesn't make sense; one of the shades should be correct and one should be reversed. 🤔
...
...
...And I think I've just had a light bulb moment just now💡🧠

Did Yoolax put the motors on the wrong shades? If they are, then I think that explains this whole thing. But that would mean Yoolax is doing a different firmware on their motors depending on the orientation of installation???

Off to test...

teladog01, 2021-11-05:
Ok, well, never mind on all that... I swapped the motors in the two shades, but after performing the reverse procedure with the remote and then re-calibrating full open and full closed, then the behavior was exactly the same as before...control correct but status reversed. 🤷‍♂️

> Honestly, I m shy to reverse the device using the code, because it s not possible for me making something compatible for all users. But can try to reverse it using DDF, have you already take a look on it ?
> 
> Can you give the DDF for the actual device ?

I've not messed at all with the DDF stuff, so wouldn't know where to start with that. I am willing to do testing if you can guide me through it.

Smanar, 2021-11-06:
>but after performing the reverse procedure with the remote and then re-calibrating full open and full closed,

Do you think it s possible it s the calibration that check for covering side and set it direclty ?

Here you have a sample with capture https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5248#issuecomment-945174475

I just need your DDF file for this device as complete as possible to (try to) edit command to reverse it

jcarusone1650, 2021-11-06:
Confirmed my motors are not in reverse @teladog01
[image]

Smanar, 2021-11-06:
You have used the "read" button ? You have lot of more setting available than him ....

jcarusone1650, 2021-11-06:
Yes, still not reversed

jcarusone1650, 2021-11-06:
One thing I do remember, was when the shades were reversed, the open/close worked correctly, but the lift for partial open/close was reversed

Smanar, 2021-11-07:
> One thing I do remember, was when the shades were reversed, the open/close worked correctly, but the lift for partial open/close was reversed

Yeah, it s the worse situation. it seem this devive natively reverse bri/lift compared to open/on

teladog01, 2021-11-07:
> Confirmed my motors are not in reverse @teladog01
> [image]

And no Tuya specific cluster. So it seems these are really different devices, even though they have the same identification in deconz.

teladog01, 2021-11-07:
> Do you think it s possible it s the calibration that check for covering side and set it direclty ?

Yes I suppose that is the obvious takeaway here. The motor orientation must be inferred from the calibration routine.

> Here you have a sample with capture [#5248 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5248#issuecomment-945174475)
> 
> I just need your DDF file for this device as complete as possible to (try to) edit command to reverse it

Here's the DDF... Thanks much for your help!

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "yooksmart",
  "modelid": "D10110",
  "product": "D10110",
  "sleeper": false,
  "status": "Draft",
  "subdevices": [
    {
      "type": "$TYPE_BATTERY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0001"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0202",
        "endpoint": "0x01",
        "in": [
          "0x0001"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/battery",
          "description": "The current device battery level in range 0&ndash;100&nbsp;%.",
          "refresh.interval": 3600
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_WINDOW_COVERING_DEVICE",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/bri",
          "description": "The current brightness.",
          "refresh.interval": 5
        },
        {
          "name": "state/lift",
          "description": "The lift state of a roller blind."
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/open",
          "description": "The open/close state."
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0102",
      "report": [
        {
          "at": "0x0008",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "groupcast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x6E61"
    }
  ]
}
```

Smanar, 2021-11-07:
To make a try
Select for exemple "state/lift" on the central part you will see 

> expression : Item.val = Attr.val

Try with replacing it by

> expression : Item.val = 100 - Attr.val

teladog01, 2021-11-07:
I suppose I also need to apply this somehow. Is that done by setting Status to something other than Draft? What are differences in Bronze, Silver, and Gold? Sorry, can't find any documentation on that...

Smanar, 2021-11-07:
Ha yes right have skipped this part.
You need to use other thing than "draft" so "bronze" is enought, then "hot relaod" in menu.

teladog01, 2021-11-07:
> To make a try Select for exemple "state/lift", on the central part you will see (all the rest need to be pre-filled)
> 
> > expression : Item.val = Attr.val

I don't see that...🤔
[image]

teladog01, 2021-11-07:
I guess I should set "Parse" to "zcl"?
[image]

Smanar, 2021-11-07:
Arf, need to complete all, there is no starting point for covering.

Yep, ZCL for both
Endpoint auto
Cluster 0x0102
Attribute 0x0008

teladog01, 2021-11-07:
Attribute "state/bri" has things pre-filled. But the Cluster and Attributes are different than the values you gave. Do I need to change these on all the state attributes?
[image]

Smanar, 2021-11-07:
Yep was the state/bri by defaut for bulbs, your device don't use the Cluster 0x0008

and for state/bri the expression is `Item.val = 255 - Attr.val * 254 / 100`

teladog01, 2021-11-07:
I'm having a hard time getting the DDF settings to take effect. I've set DDF handling to Normal on the Control panel. I've set the device Status to Bronze in the DDF editor. Please confirm location to save the DDF file. Here is the default, is it correct
[image]
?

Smanar, 2021-11-08:
I think the location don't have impact.
Just don't restart deconz.
Nothing change if you use "hot reload" ?

teladog01, 2021-11-08:
> Nothing change if you use "hot reload" ?

Correct. Nothing changes as far as I can tell. Lift percentage is still backwards. Here is the API return when fully open (same as before):
```
{
  "etag": "62533ea47f004660ed6332d67f7589d2",
  "hascolor": false,
  "lastannounced": "2021-11-07T21:29:19Z",
  "lastseen": "2021-11-07T21:29Z",
  "manufacturername": "yooksmart",
  "modelid": "D10110",
  "name": "South Window Shade",
  "state": {
    "bri": 255,
    "lift": 100,
    "on": true,
    "open": false,
    "reachable": true
  },
  "swversion": "20201120",
  "type": "Window covering device",
  "uniqueid": "8c:f6:81:ff:fe:de:c1:a1-01"
}
```
[image]

### d10110.json
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "yooksmart",
  "modelid": "D10110",
  "product": "D10110",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/d10110.json",
  "subdevices": [
    {
      "type": "$TYPE_BATTERY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0001"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0202",
        "endpoint": "0x01",
        "in": [
          "0x0001"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/battery",
          "refresh.interval": 3600
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_WINDOW_COVERING_DEVICE",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/bri",
          "refresh.interval": 700,
          "read": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 0,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 0,
            "eval": "Item.val = 255 - Attr.val * 254 / 100",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lift",
          "refresh.interval": 700,
          "read": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 0,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 0,
            "eval": "Item.val = 100 - Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "state/on",
          "refresh.interval": 700,
          "read": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 0,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 0,
            "eval": "Item.val = (255 - Attr.val * 254 / 100) > 0",
            "fn": "zcl"
          }
        },
        {
          "name": "state/open",
          "parse": {
            "eval": "Item.val = (100 - Attr.val) < 100",
            "fn": "zcl"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0102",
      "report": [
        {
          "at": "0x0008",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

teladog01, 2021-11-08:
I don't know if this is helpful, but I am seeing entries like these repeating every five minutes or so in the Debug view, with DDF checked:

```
12:44:35:398 8c:f6:81:ff:fe:de:c1:a1-01/state/on expression: Item.val = (255 - Attr.val * 254 / 100) > 0 --> true
12:44:35:404 8c:f6:81:ff:fe:de:c1:a1-01/state/lift expression: Item.val = 100 - Attr.val --> 95
12:44:35:439 8c:f6:81:ff:fe:de:c1:a1-01/state/bri expression: Item.val = 255 - Attr.val * 254 / 100 --> 242.3
12:45:37:445 8c:f6:81:ff:fe:d5:ad:b9-01/state/on expression: Item.val = (255 - Attr.val * 254 / 100) > 0 --> true
12:45:37:466 8c:f6:81:ff:fe:d5:ad:b9-01/state/lift expression: Item.val = 100 - Attr.val --> 100
12:45:37:489 8c:f6:81:ff:fe:d5:ad:b9-01/state/bri expression: Item.val = 255 - Attr.val * 254 / 100 --> 255
```

Smanar, 2021-11-08:
Nice, if you see it in logs mean something is happen ^^.

You don't have the websocket notification returned in same time than this line for exemple ?

>12:44:35:404 8c:f6:81:ff:fe:de:c1:a1-01/state/lift expression: Item.val = 100 - Attr.val --> 95

To check if the 95 value is used.

teladog01, 2021-11-08:
With DDF, INFO, INFO_L2 checked:
```
14:44:04:015 8c:f6:81:ff:fe:de:c1:a1-01/state/on expression: Item.val = (255 - Attr.val * 254 / 100) > 0 --> true
14:44:04:023 8c:f6:81:ff:fe:de:c1:a1-01/state/lift expression: Item.val = 100 - Attr.val --> 95
14:44:04:030 8c:f6:81:ff:fe:de:c1:a1-01/state/bri expression: Item.val = 255 - Attr.val * 254 / 100 --> 242.3
14:44:04:052 0x8CF681FFFEDEC1A1: update ZCL value 0x01/0x0102/0x0008 after 0 s
14:44:04:054 ZCL attribute report 0x8CF681FFFEDEC1A1 for cluster: 0x0102, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
14:44:04:055 	payload: 08002005
14:44:04:059 Websocket 10.3.11.14:51044 send message: {"e":"changed","id":"5","r":"lights","state":{"bri":241,"lift":95,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"8c:f6:81:ff:fe:de:c1:a1-01"} (ret = 158)
14:44:04:061 Websocket 10.3.11.13:51062 send message: {"e":"changed","id":"5","r":"lights","state":{"bri":241,"lift":95,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"8c:f6:81:ff:fe:de:c1:a1-01"} (ret = 158)
14:44:04:064 Websocket 10.3.11.11:62487 send message: {"e":"changed","id":"5","r":"lights","state":{"bri":241,"lift":95,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"8c:f6:81:ff:fe:de:c1:a1-01"} (ret = 158)
```

teladog01, 2021-11-08:
I used the DDF editor to modify the expressions for the state attributes to remove the inverting arithmetic ("255 - " and "100 - "). Now the Item.val is evaluated to the correct values in the first three lines below, but the websocket messages that follow still show the incorrect/inverted values, as if the websocket is getting its info from somewhere else.

```
14:59:00:101 8c:f6:81:ff:fe:de:c1:a1-01/state/on expression: Item.val = (Attr.val * 254 / 100) > 0 --> true
14:59:00:109 8c:f6:81:ff:fe:de:c1:a1-01/state/lift expression: Item.val = Attr.val --> 5
14:59:00:133 8c:f6:81:ff:fe:de:c1:a1-01/state/bri expression: Item.val = Attr.val * 254 / 100 --> 12.7
14:59:00:152 0x8CF681FFFEDEC1A1: update ZCL value 0x01/0x0102/0x0008 after 0 s
14:59:00:153 ZCL attribute report 0x8CF681FFFEDEC1A1 for cluster: 0x0102, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
14:59:00:155 	payload: 08002005
14:59:00:157 Websocket 10.3.11.14:51044 send message: {"e":"changed","id":"5","r":"lights","state":{"bri":241,"lift":95,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"8c:f6:81:ff:fe:de:c1:a1-01"} (ret = 158)
14:59:00:159 Websocket 10.3.11.13:51062 send message: {"e":"changed","id":"5","r":"lights","state":{"bri":241,"lift":95,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"8c:f6:81:ff:fe:de:c1:a1-01"} (ret = 158)
14:59:00:161 Websocket 10.3.11.11:62487 send message: {"e":"changed","id":"5","r":"lights","state":{"bri":241,"lift":95,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"8c:f6:81:ff:fe:de:c1:a1-01"} (ret = 158)
```

Smanar, 2021-11-09:
Ok the problem is probalby this one https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5465

Need to bypass better the legacy code when using DDF.

teladog01, 2021-11-10:
>We will have a new version soon, I realy hope It will be inside.

Me, too 🙂🤞

teladog01, 2021-12-09:
Not solved. I believe we are waiting for v12.3.3-beta

Mimiix, 2021-12-09:
@teladog01 Please bump it when the bot posts the first message😅

Smanar, 2021-12-09:
The v12.3.3-beta is available. Can you say if you still have the same issue ?

Smanar, 2021-12-11:
So I have a covering too, and it still not working on my side :(

Need to check code.

Smanar, 2021-12-11:
Ok so I have a tuya covering device too.

It work now but need to use the "strict" DDF mode, "hybrid" still not wroking for me.

teladog01, 2021-12-11:
> It work now but need to use the "strict" DDF mode, "hybrid" still not wroking for me.

Same here...DDF settings only seem to be applied when in Strict mode. We are making progress 👍

closed 2022-01-09 by github-actions[bot]

jcarusone1650, 2022-05-23:
@Smanar a note that we purchased a new batch of these shades with the "same" P440 motor as the first batch. They definitely changed the motor. The code change that addressed the reversed LIFT/BRI had to be removed in order for the new batch to operate correctly. Fortunately we have these in two different homes but they did use the same D10110 designator. Happy to help test if useful. I am good for now but may help others.

jcarusone1650, 2022-05-23:
Yep. I compiled the current code base commenting out the "reversal" and it works correctly with the new Yoolax motors/firmware. Happy to test new code as well.