dresden-elektronik/deconz-rest-plugin #5449: New Firmware breaks Ikea-Switches

HenningGithub, 2021-11-05:
## Describe the bug
vendor 0x117c
image 0x11c5
version 0x23079631

Since these got the above mentioned firmware update, I don't get any deconz_event on key presses in home assistant any more. Multiple affected units. The not yet updated ones continue to work fine.

## Steps to reproduce the behavior
update firmware on an ikea dimmer switch or blinds switch

## Expected behavior
## Screenshots
## Environment
 - Host system: Raspberry Pi 3B+
 - Running method: Home Assistent deCONZ Add-on 
 - Firmware version: 26390500
 - deCONZ version: 2.12.06
 - Device: RaspBee II (or I? dunno)

## deCONZ Logs
I can't find anything relevant

## Additional context

Gloomyeye, 2021-11-06:
Hello,

I can confirm the bug with the new IKEA TRÅDFRI firmware. I updated my TRÅDFRI on/off switch remote as well as my TRÅDFRI open/close remote to firmware V-2.3.079 (binary URL of the firmware = http://fw.ota.homesmart.ikea.net/global/GW1.0/01.16.026/bin/10005778-10.1-TRADFRI-onoff-shortcut-control-2.3.079.ota.ota.signed) with the help of STD OTAU plugin. After the firmware update no Deconz event is pushed to home assistant. So, both remotes are useless so far.

I updated also my TRÅDFRI remote to V-2.3.080. This remote has no issues. Only the on/off switch remote as well as the TRÅDFRI open/close remote have problems.

Here are screenshots of the on/off switch remote:

[image]
[image]
[image]
[image]
[image]
[image]

[image]

Smanar, 2021-11-07:
Hello, can you show some logs when pressing a button ? (deconz log with "info" and "info_l2")

Gloomyeye, 2021-11-07:
Hello Smanar,

I hope I can help you with that:

deconz           | 12:36:54:725 Sqlite sensors: sid = 57
deconz           | 12:36:54:725 Sqlite sensors: name = Ikea Remote Control KUE
deconz           | 12:36:54:725 Sqlite sensors: type = ZHASwitch
deconz           | 12:36:54:725 Sqlite sensors: modelid = TRADFRI on/off switch
deconz           | 12:36:54:725 Sqlite sensors: manufacturername = IKEA of Sweden
deconz           | 12:36:54:725 Sqlite sensors: uniqueid = 58:8e:81:ff:fe:2b:84:c2-01-1000
deconz           | 12:36:54:725 Sqlite sensors: swversion = 2.3.079
deconz           | 12:36:54:725 Sqlite sensors: state = {"buttonevent":null,"lastupdated":null}
deconz           | 12:36:54:725 Sqlite sensors: config = {"alert":"none","battery":87,"group":"19","on":true,"reachable":true}
deconz           | 12:36:54:725 Sqlite sensors: fingerprint = {"d":2080,"ep":1,"in":[0,1,4096],"out":[6,8,258],"p":260}
deconz           | 12:36:54:725 Sqlite sensors: deletedState = normal
deconz           | 12:36:54:725 Sqlite sensors: mode = 1
deconz           | 12:36:54:725 Sqlite sensors: lastseen = 2021-11-07T11:35Z
deconz           | 12:36:54:725 DB found sensor Ikea Remote Control KUE 57

...

deconz           | 12:39:30:436 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('57', 'Ikea Remote Control KUE', 'ZHASwitch', 'TRADFRI on/off switch', 'IKEA of Sweden', '58:8e:81:ff:fe:2b:84:c2-01-1000', '2.3.079', '{"buttonevent":null,"lastupdated":null}', '{"alert":"none","battery":87,"group":"19","on":true,"reachable":true}', '{"d":2080,"ep":1,"in":[0,1,4096],"out":[6,8,258],"p":260}', 'normal', '1', '2021-11-07T11:39Z', '')

...

deconz           | 12:42:20:907 Poll light node Hue KUE Pendelleuchte (ML)
deconz           | 12:42:20:957 read attributes of 0x001788010B7F6C6D cluster: 0x0006: [ 12:42:20:957 0x0000 12:42:20:958 ]
deconz           | 12:42:20:958 add task 1580 type 19 to 0x001788010B7F6C6D cluster 0x0006 req.id 232
deconz           | 12:42:20:958 Poll APS request 232 to 0x001788010B7F6C6D cluster: 0x0006
deconz           | 12:42:21:023 Poll APS confirm 232 status: 0x00
deconz           | 12:42:21:024 Erase task req-id: 232, type: 19 zcl seqno: 39 send time 0, profileId: 0x0104, clusterId: 0x0006
deconz           | 12:42:21:048 Node data 0x001788010b7f6c6d profileId: 0x0104, clusterId: 0x0006
deconz           | 12:42:21:048 0x001788010B7F6C6D: update ZCL value 0x0B/0x0006/0x0000 after 0 s
deconz           | 12:42:21:343 Websocket 192.168.178.4:57050 send message: {"attr":{"id":"57","lastannounced":null,"lastseen":"2021-11-07T11:42Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI on/off switch","name":"Ikea Remote Control KUE","swversion":"2.3.079","type":"ZHASwitch","uniqueid":"58:8e:81:ff:fe:2b:84:c2-01-1000"},"e":"changed","id":"57","r":"sensors","t":"event","uniqueid":"58:8e:81:ff:fe:2b:84:c2-01-1000"} (ret = -1095381000)
deconz           | 12:42:21:981 poll node 00:17:88:01:06:16:7e:be-0b
deconz           | 12:42:21:982 Poll light node Hue WZM Lampe TV
deconz           | 12:42:22:033 read attributes of 0x0017880106167EBE cluster: 0x0006: [ 12:42:22:033 0x0000 12:42:22:034 ]
deconz           | 12:42:22:034 add task 1585 type 19 to 0x0017880106167EBE cluster 0x0006 req.id 242
deconz           | 12:42:22:034 Poll APS request 242 to 0x0017880106167EBE cluster: 0x0006
deconz           | 12:42:22:117 Poll APS confirm 242 status: 0x00
deconz           | 12:42:22:118 Erase task req-id: 242, type: 19 zcl seqno: 40 send time 0, profileId: 0x0104, clusterId: 0x0006
deconz           | 12:42:22:184 Node data 0x0017880106167ebe profileId: 0x0104, clusterId: 0x0006
deconz           | 12:42:22:184 0x0017880106167EBE: update ZCL value 0x0B/0x0006/0x0000 after 0 s

...

This is also vom Phoscon API information:

{
    "config": {
        "alert": "none",
        "battery": 87,
        "group": "19",
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "2068e7cfcbc9c2d092dd689aee3bee66",
    "lastannounced": null,
    "lastseen": "2021-11-07T11:29Z",
    "manufacturername": "IKEA of Sweden",
    "mode": 1,
    "modelid": "TRADFRI on/off switch",
    "name": "Ikea Remote Control KUE",
    "state": {
        "buttonevent": null,
        "lastupdated": "none"
    },
    "swversion": "2.3.079",
    "type": "ZHASwitch",
    "uniqueid": "58:8e:81:ff:fe:2b:84:c2-01-1000"
}

{
    "12:29:27:656": {
        "attr": {
            "id": "57",
            "lastannounced": null,
            "lastseen": "2021-11-07T11:29Z",
            "manufacturername": "IKEA of Sweden",
            "modelid": "TRADFRI on/off switch",
            "name": "Ikea Remote Control KUE",
            "swversion": "2.3.079",
            "type": "ZHASwitch",
            "uniqueid": "58:8e:81:ff:fe:2b:84:c2-01-1000"
        },
        "e": "changed",
        "id": "57",
        "r": "sensors",
        "t": "event",
        "uniqueid": "58:8e:81:ff:fe:2b:84:c2-01-1000"
    },
    "12:30:39:849": {
        "attr": {
            "id": "57",
            "lastannounced": null,
            "lastseen": "2021-11-07T11:30Z",
            "manufacturername": "IKEA of Sweden",
            "modelid": "TRADFRI on/off switch",
            "name": "Ikea Remote Control KUE",
            "swversion": "2.3.079",
            "type": "ZHASwitch",
            "uniqueid": "58:8e:81:ff:fe:2b:84:c2-01-1000"
        },
        "e": "changed",
        "id": "57",
        "r": "sensors",
        "t": "event",
        "uniqueid": "58:8e:81:ff:fe:2b:84:c2-01-1000"
    }
}

Smanar, 2021-11-07:
Ok, so thx, but there is badly nothing visible in log ^^.
Can you make the same test but with logs with the flag "APS" ? Hoping for something visible.

Gloomyeye, 2021-11-07:
This is the log with DEBUG_APS = 2 and DEBUG_INFO=1 parameter: [APS Log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7492709/APS.Log.txt)

Smanar, 2021-11-07:
Sorry, perhaps someone with the device updated will take a look, but there is still nothing in logs about this device (Have search with the MAC adress 58:8e:81:ff:fe:2b:84:c2)

manup, 2021-11-07:
I have one available for testing. Perhaps this is the same binding mess reported for the round remotes so we need to fall back to unicast bindings. I'll schedule some tests.

ebaauw, 2021-11-07:
I'll try and have a look this week.  Currently upgrading my devices; looks like there are upgrades for:

Image | Device(s) | Firmware string
-- | -- | --
110C | ? | GBL 10082261-zingo-lds-starkvin
11C1 | TRADFRI remote control | EBL tradfri_controller
11C5 | TRADFRI on/off switch<br>TRADFRI open/close remote | GBL GBL_tradfri_onoff_shortcut_
11C6 | TRADFRI SHORTCUT button | GBL GBL_tradfri_shortcut_button
11CA | SYMFONISK Sound Controller | GBL GBL_tradfri_dimmer
2102 | ? | GBL 10076691-zingo-lds-bulb-hwp
2205 | ? | GBL zingo-lds-bulb-hwpwmcs-ws

I don't know what the devices with `?` are, I don't have these.

ebaauw, 2021-11-07:
Hm, having a look at the upgraded 5-button remote (newer model E1810).  Don't like what I see.  It reports its binding table correctly after the upgrade (server 0x0001 to coordinator; client 0x0006, 0x0008, 0x0005 to group), but it only sends unicasts to 0x0000.  The ZLL commissioning cluster is still reporting the group.  When I removed the unicast binding for the power configuration cluster, the device seems dead in the water, and no longer sends anything nor polls its parent on any button action.

Tried resetting and re-pairing the remote a couple of times, but no joy.  Looks like the REST API Plugin creates the power configuration binding first, which causes the button events to be created, since it now sends the on/off. level control, and scene commands to the coordinator with only that binding.  I wonder what would happen if one of the client clusters is bound first, to a group.

@manup, would you be able to sniff how the Trådfri hub sets up these devices?  The way IKEA works, I cannot image they would break direct control of a group.  The unicast commands ask for a default response (which the plugins sends alright) - I doubt they want half a dozen or more unicast bindings to control multiple lights at once.

As always, make sure to replace the battery after the firmware upgrade, never mind what level it's reporting.

HenningGithub, 2021-11-08:
FWIW, 11c1 and 11ca continue to work fine after the upgrade.
11c5 has the issue, I don't have the shortcut button or the ? ones from your table.

Mimiix, 2021-11-08:
> I'll try and have a look this week. Currently upgrading my devices; looks like there are upgrades for:
> 
> Image	Device(s)	Firmware string
> 110C	?	GBL 10082261-zingo-lds-starkvin
> 11C1	TRADFRI remote control	EBL tradfri_controller
> 11C5	TRADFRI on/off switch
> TRADFRI open/close remote	GBL GBL_tradfri_onoff_shortcut_
> 11C6	TRADFRI SHORTCUT button	GBL GBL_tradfri_shortcut_button
> 11CA	SYMFONISK Sound Controller	GBL GBL_tradfri_dimmer
> 2102	?	GBL 10076691-zingo-lds-bulb-hwp
> 2205	?	GBL zingo-lds-bulb-hwpwmcs-ws
> I don't know what the devices with `?` are, I don't have these.

I think the first 1 (starkvin) is this one: https://www.ikea.com/nl/nl/p/starkvind-luchtreiniger-zwart-60502013/  which isn't supported by deCONZ (yet): https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5351

The others i ahve no clue of. Google couldn't be of any help

ebaauw, 2021-11-08:
The 11C1 "works", because it has a unicast binding to the coordinator for the battery reporting.  Because of this, it sends its commands to the coordinator as well, so the API plugin can translate these to button events.  The direct control of lights using a bound group no longer seems to work.

We don't create group bindings for the 11CA, as the rate of commands is too high for group messages.  If it has the same issues as the 11C1, we wouldn't see them.

>  think the first 1 (starkvin) is this one: https://www.ikea.com/nl/nl/p/starkvind-luchtreiniger-zwart-60502013/ which isn't supported by deCONZ (yet): #5351

Ah, I've been eyeing that one for a while.  Still haven't won the lottery :-( but received enough donations on Paypal and IKEA are currently offering free shipping.  ETA by next weekend.

manup, 2021-11-08:
[ConBee_v21301_Ikea_on_off_switch_23079.pcapng.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7497418/ConBee_v21301_Ikea_on_off_switch_23079.pcapng.gz)

Here is the on/off switch joined to and configured by ConBee II in my test setup:

Network key: cccccccccccccccccccccccccccccccc

So it looks we lose group functionality by doing group bindings :)
In the Ikea gateway group casts still work, but are not configured. Only Basic and Power Configuration clusters are bound by Ikea gateway.

manup, 2021-11-08:
[PhilipsHueV1_v21301_Ikea_on_off_switch_23079.pcapng.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7498204/PhilipsHueV1_v21301_Ikea_on_off_switch_23079.pcapng.gz)

Network key: 82a90bebdbb7201efb7cb89a7e9e1ad4

Same again with Philips Hue V1 bridge. Also ends up in unicasts towards the bridge. This test was made to rule out that the firmware might check if this is a distributed trust center network. It doesn't.

The remaining assumption is that the Ikea gateway assigns groups via ZLL Touchlink command during joining. However I can't test this or see it in sniffer, looks like the switch keeps it network specific configuration, even after resetting it.

ebaauw, 2021-11-08:
Damn.  We explicitly ignore unicasts from a number of IKEA switches, since they used to send both groupcast and unicast commands, causing double button events: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/84e7dccdd58b9fca86dba0383edb43d721eacb37/de_web_plugin.cpp#L4516-L4519

It concerns these switches: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/84e7dccdd58b9fca86dba0383edb43d721eacb37/de_web_plugin.cpp#L4498-L4503

HenningGithub, 2021-11-08:
> Checking the firmware per device doesn't seem like fun. We would probably need to do that and remove config.group from the resource
Well, if both old and new send unicast, only old sends groupcast and double events are the problem - couldn't you just do that unconditionally and act on the unicast?

manup, 2021-11-08:
Hmm good question, yes either we filter out group casts or have some debounce mechanic, but that might introduce other difficulties.

fribse, 2021-11-13:
So I guess this is why my brand new IKEA Shortcut button (fw 2.3.015) doesn't do anything besides getting enrolled 😢

petergng, 2021-11-18:
same is affecting me with tradfri on/off switch with firmware 2.3.079. any workaround here?

Farmand77, 2021-11-21:
I have the same issue as well. Will try and see if i can downgrade and restore funktionality tomorrow.

mattiasottosson, 2021-11-22:
I'm having this issue too. Just completed my first deconz setup and pairing the buttton was the last remaining step. Just thought i did something wrong in the setup process ;) 

@Farmand77, what firmware are you downgrading to? I tried 10005778-10.1-TRADFRI-onoff-shortcut-control-2.2.010.ota.ota.signed (hoping that is the right fw), but progress is stuck on Queued. Might be missing something obvious here though. 

[image]

Farmand77, 2021-11-22:
> I'm having this issue too. Just completed my first deconz setup and pairing the buttton was the last remaining step. Just thought i did something wrong in the setup process ;)
>
> @Farmand77, what [...]

@mattiasottosson Updating buttons is a bit strange cause of the manual steps needed what I do is.
Locate one button chose the firmware i want to use in files. Then I press update wait 5 sec then query. When is says Queued. I take out its battery for 5 sec. and wait 3 min. then I quary the button again and it changes to idle and then 1 min later starts updating.

mattiasottosson, 2021-11-22:
> Locate one button chose the firmware i want to use in files. Then I press update wait 5 sec then query. When is says Queued. I take out its battery for 5 sec. and wait 3 min. then I quary the button again and it changes to idle and then 1 min later starts updating.

[image]

@Farmand77  That worked. Thank you!

stromrickard, 2021-11-22:
Also looking for a solution.

Regarding downgrade, not sure if the information is correct but from what i've found in other threads, they are blocking downgrades. They will revert to the new firmware after flash and reboot. Only way to downgrade seems to be to open it up, solder in a programmer and flash it that way. I've tried downgrading using Deconz and after success and reboot, they're still on the new firmware.

Farmand77, 2021-11-22:
> Also looking for a solution.
>
> Regarding downgrade, not sure if the information is correct but from what i've found in other threads, they are blocking downgrades. They will revert to the new [...]

@stromrickard Unfortunatly downgrading did not work for me. how do you check what firmware they are on. it seams that they list the old firmware version not matter if they are upgraded or downgraded. based on your error deleting and readding would not help either after a downgrade...

stromrickard, 2021-11-22:
> 

You can check firmware version in Phoscon if you select the switch. I've tried removing, resetting it and adding it back again after flashing 2.2.010 with Deconz but when i add it back, it's still 2.3.079.
To me, this confirms all information i've found. The switch itself blocks downgrades. If you try to flash an older firmware OTA, it will just ignore the firmware and stay on the new one.

Farmand77, 2021-11-22:
@stromrickard I am affraid you are right. And found another Old but open bug where firmware info is not updated from installed version even if device is updated. Hopefully they can resolve This soon. Currently im two switches down

mattiasottosson, 2021-11-23:
@stromrickard , @Farmand77 

Redid the 2.2.010-downgrade procedure and it reverted to 2.3.079, so you guys are definitely right.  

I will need another switch for another room later on, so maybe I should just buy a new one as a quick fix and hope that the stock that Ikea has isn't updated with new the firmware yet. ;)

smartmatic, 2021-11-24:
It also effects me and my question is if there is any chance to fix it with the next stable release?

Ar-eh, 2021-11-30:
@stromrickard , @Farmand77, @mattiasottosson 

The [release notes](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html) for v. 2.3.079 on/off remote include "Increased security regarding upgrade of firmware" - perhaps this includes blocking downgrades?

manup, 2021-11-30:
Unfortunately group casts can't be used with new firmware, but unicast to the gateway is now fixed with above PR to get button events working again.

vbaros, 2021-12-04:
Is there any workaround before the fix gets released?

manup, 2021-12-04:
Perhaps downgrading the switch firmware version works, but I haven't testet this.

smartmatic, 2021-12-04:
> Is there any workaround before the fix gets released?

My current workaround is to use the switch with zigbee2mqqt.

closed 2021-12-05 by manup