dresden-elektronik/deconz-rest-plugin #5450: Tuya ZigBee TM616EGBZTN Pull Bead Curtain Motor

P2W2, 2021-11-05:
## Device

- Product name: Tuya ZigBee TM616EGBZTN 
- Manufacturer: _TZE200_gubdgai2
- Model identifier: TS0601
- Device type : Please remove all unrelated device types. 
  - Other: Cover

## Screenshots
[image]
[image]

### Basic
[image]
### Identify

### Alarms

### Device Temperature

### Groups
[image]

### Scenes
[image]

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

### Tuya Specific clusters:
[image]

### OTAU
[image]

### Time
[image]

P2W2, 2021-11-15:
The solution seems to be simple, since it is the same system as the following issues:  
[#5238](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5238) & https://github.com/dresden-elektronik/deconz-rest-plugin/issues?q=is%3Aissue+is%3Aclosed+M515EGB.
@Smanar maybe you can help since you have solved all issues related to curtain motors above?

I tried to solve it myself, but ran into a problem:
If I understood it correctly it is solved by adding following line:
`{"_TZE200_gubdgai2", "TS0601", "Tuya", "Tuya_COVD TM616EGBZTN"},`
to this [file](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/product_match.cpp).

However I tried to implement and test it but I get a `E: Unable to locate package deconz-dev` if I try to `sudo apt install deconz-dev`
I have tried the solutions that are written [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2811), but unfortunately did not work either

Smanar, 2021-11-15:
Hello, what is your OS ?

You need to switch to beta package https://www.phoscon.de/en/conbee2/install

P2W2, 2021-11-17:
So I installed the beta package with a new raspbian-buster image and i was able to install the deconz-dev package and build the project. Unfortunately, my above-mentioned change did not bring my hoped-for success. In fact nothing changed at all the device is still a smart plug and not shown in the phoscon app or HA.

closed 2021-12-17 by github-actions[bot]

stevewull, 2022-08-21:
I have this cover and am having trouble getting deconz to recognize it properly

The cover is simliar to _TZE200_xuzcvlku. I have three of those. Phoscon displays them as fairy lights. They work for me with no problems.

seems like _TZE200_gubdgai2 is a new rev of their cover. I tried two cycles of adding a new light.

Screenshots:

[image]
[image]
[image]

Smanar, 2022-08-22:
Sorry but for the moment it's not possible adding new tuya covering.

stevewull, 2022-08-23:
Thanks Smanar,  I'll check back later.

mdcobra, 2024-04-04:
I also have this _TZE200_gubdgai2, but cannot bind it, is this solved?

Smanar, 2024-04-04:
Nope, sorry, not possible adding tuya covering using DDF.