dresden-elektronik/deconz-rest-plugin #5451: stale link in comment in device.cpp

fkuep, 2021-11-05:
## Describe the bug
a comment in a cpp uses an old url 
## Steps to reproduce the behavior
Go to : 
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/6088c22394309c44a642a9607e3c8dd9dfd17476/device.cpp#L36 
and try to follow the link. 
## Expected behavior
Instead display the link:
`https://dresden-elektronik.github.io/deconz-dev-doc/modules/class-device/#state-machine`
in the comment.

## Screenshots
not applicable

## Environment
not applicable commit hash version given.

 - Host system: PC
 - Running method: chrome
 - Firmware version: not aplicable
 - deCONZ version: (head)
 - Device: not applicable
 - Do you use an USB extension cable:  no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
not available

## Additional context
no relevant additional context

manup, 2021-11-06:
Good catch, will update the comment and the bindings URL in the same file.

The content has moved to: https://dresden-elektronik.github.io/deconz-dev-doc/modules/class-device/#state-machine

fkuep, 2021-11-06:
Actually, since it's You.. forgive me to posting like this on the wrong project:
https://dresden-elektronik.github.io/deconz-dev-doc/modules/class-device/#device-class
At the top of the page You link to device.cpp and .h ;
Should that links lead to the main branch instead of Your fork ?

closed 2021-12-05 by github-actions[bot]