dresden-elektronik/deconz-rest-plugin #5459: LiXee ZLinky_TIC

chpego, 2021-11-06:
## Device

- Product name: ZLinky_TIC.
- Manufacturer: LiXee
- Model identifier: ZLinky_TIC
- Link : https://github.com/fairecasoimeme/Zlinky_TIC
- Device type :
  - Sensor

## Screenshots
  - Endpoints and clusters of the node
[image]

  - Node Info panel
[image]

### Simple Metering
[image]

### Electrical Measurement
[image]

Smanar, 2021-11-06:
Ok so if I make a resume.

Cluster to bind
- 0x702
- 0xB04

Usefull Attributes.
- cluster 0x0b04 : Attribute 0x050F = apparent power
- cluster 0x0702 : EP = 01 : Attributes 0x0100 0x0102 = usage update (Base, or HP,HC, or BBRHCJB, BBRHPJB)
- cluster 0x0702 : EP = F2 : Attributes 0x0104 0x0106 = usage update (BBRHCJW, BBRHPJW)
- cluster 0x0702 : EP = F3 : Attributes 0x0108 0x010A = usage update (BBRHCJR, BBRHPJR)

Edit:
Ok, I give up, not comprehensible, not possible make a clear tutorial ?
Seriously a classic user don't need 20 attributes to use this kind of device ....

chpego, 2021-11-06:
@fairecasoimeme author of project, may be able to answer these questions ?

fairecasoimeme, 2021-11-06:
@Smanar seriously, did you really read https://github.com/fairecasoimeme/Zlinky_TIC ?

Mimiix, 2021-11-06:
> @Smanar seriously, did you really read https://github.com/fairecasoimeme/Zlinky_TIC ?
> 
> There is 1 endpoint , the optarif values are listed and the optarif is given by Linky.

No need to become hostile.

Smanar, 2021-11-06:
My bad.
For the 3 Endpoints, it s from a project that already support it (have usefull comment on code to explain), but IDK if his code is still in date, as on the capture the device use effectively only 1 endpoint, the second one is only for GP.

And haven't see the value for optarif  was read only.

fairecasoimeme, 2021-11-06:
> > @Smanar seriously, did you really read https://github.com/fairecasoimeme/Zlinky_TIC ?
> > There is 1 endpoint , the optarif values are listed and the optarif is given by Linky.
> 
> No need to become hostile.

?

Smanar, 2021-11-06:
Ok, I undertand better, For exemple If I m right

 If you use the "mode historique" with the opt tarif  = "BASE" you need only 

<html><body>

BASE | 0x0702 | 0x0000 | RP |   | Uint32 | 9 car | Wh | Index Base | 0
-- | -- | -- | -- | -- | -- | -- | -- | -- | --

</body>
</html>

fairecasoimeme, 2021-11-06:
> Ok, I undertand better, For exemple If I m right
> 
> If you use the "mode historique" with the opt tarif = "BASE" you need only
> 
> BASE	0x0702	0x0000	RP	 	Uint32	9 car	Wh	Index Base	0
> @chpego Can I have some information on your electricity contract ?

Yes, you can observe the color to know what attributes are used

Smanar, 2021-11-06:
And usefull too

<html><body>

IINST | 0x0B04 | 0x0508 | RP |   | Uint16 | 3 car | A | Courant efficace | 0xFFFF
-- | -- | -- | -- | -- | -- | -- | -- | -- | --

</body>
</html>

And some alarm on Cluster 0xFF66

<html><body>

ADPS | 0xFF66 | 0x0005 | RP |   | Uint16 | 3 car | A | Avertissement de Dépassement De Puissance Souscrite | 0
-- | -- | -- | -- | -- | -- | -- | -- | -- | --

</body>
</html>

So in reality was not as terrific than the table size.

fairecasoimeme, 2021-11-06:
You can get the protocol here : https://www.enedis.fr/media/2035/download

Smanar, 2021-11-06:
So in reality was not as terrifying than the table size, sorry for my reaction @fairecasoimeme .
I still hope that no one will use the "Mode standard" soon.

fairecasoimeme, 2021-11-06:
> And usefull too
> 
> IINST	0x0B04	0x0508	RP	 	Uint16	3 car	A	Courant efficace	0xFFFF
> And some alarm on Cluster 0xFF66
> 
> ADPS	0xFF66	0x0005	RP	 	Uint16	3 car	A	Avertissement de Dépassement De Puissance Souscrite	0
> And some other informations in Read Only mode. So in reality was not as terrifying than the table size, sorry for my reaction @fairecasoimeme . I still hope that no one will use the "Mode standard" soon.

No PB @Smanar. Standard mode is not used a lot.

Smanar, 2021-11-06:
@chpego Can I have some information on your electricity contract ?

We will first add the new cluster/attribute to the XML file, for you will be able to write and read data according to your contract.
This is easy part, just need to edit an xml file.

chpego, 2021-11-06:
> @chpego Can I have some information on your electricity contract ?
> 
> We will first add the new cluster/attribute to the XML file, for you will be able to write and read data according to your contract. This is easy part, just need to edit an xml file.

yes , this is information of contract :
[image]

Smanar, 2021-11-07:
Ok so I think it s possible adding this device without editing the c++ code.

I presume you are using "Mode historique" ?

Here you have some change to try on the xml file (/usr/share/deCONZ/zcl/general.xml on raspbian) 
https://github.com/Smanar/deconz-rest-plugin/blob/zlinky/general.xml

The code to try is just that
```
    <!-- LiXee -->
    <cluster id="0xff66" name="LiXee specific cluster" mfcode="0x1037">
      <description>LiXee specific cluster.</description>
      <server>
        <attribute id="0x0000" type="cstring" name="Option tarifaire" access="r" required="m"/>
        <attribute id="0x0002" type="u8" name="Horaire HP-HC" access="r" required="m" default="0x00"/>
        <attribute id="0x0003" type="u8" name="Présence des potentiels" access="r" required="m" default="0x00"/>
        <attribute id="0x0004" type="u8" name="Préavis début EJP" access="r" required="m" default="0x00"/>
        <attribute id="0x0005" type="u16" name="Avertissement de Dépassement De Puissance Souscrite" access="r" required="m" default="0x00"/>
        <attribute id="0x0006" type="u16" name="Avertissement de Dépassement D'intensité P1" access="r" required="m" default="0x00"/>
        <attribute id="0x0007" type="u16" name="Avertissement de Dépassement D'intensité P2" access="r" required="m" default="0x00"/>
        <attribute id="0x0008" type="u16" name="Avertissement de Dépassement D'intensité P3" access="r" required="m" default="0x00"/>
      </server>
    </cluster>
```

The only ones availables for you will be 0x0000 an perhaps 0x0005

For the second part, do you know what is DDF file on deconz ?
You have a sample here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5248#issuecomment-945174475

For your device :
- set "basic cluster information" if needed (can use drag and drop from cluster 0x0000)
- drag and drop from the right to left ZHAcomsumption and ZHApower
- set the status to bronze
- save the DDF
- try on the menu "hot reload"

If I m right at least 2 values are native, 
- Cluster 0x0702, attribute 0x0000 > puissance consommée
- Cluster 0x0B04, attribute 0x0508 > courant 

If it not working can you show your DDF file ?

chpego, 2021-11-08:
### Meter Identification
[image]

Smanar, 2021-11-08:
Nice, thx.
"Intensité souscrite" can be 30A, seem normal.
I have added more field, but they are not mandatory (and some of them not used for you).  I think it s for that your have 0

Have you tried to make a DDF file ? So you need more explanation ?

chpego, 2021-11-08:
For DDF file, i think i can't because i'm on 2.12.6 deconz version, no ?

Smanar, 2021-11-08:
Ha yep, you need a recent version.
I haven't asked, what is your OS ? Can you update deconz ?

chpego, 2021-11-08:
I'm usinge HA with the [addon deconz](https://github.com/home-assistant/addons/tree/master/deconz)

Smanar, 2021-11-08:
Ha bad luck.
HA use only stable version of deconz ?
Using DDF don't need to recompile the code (nice for HA users), but you need a recent version.

chpego, 2021-11-09:
> Ha bad luck.
> HA use only stable version of deconz ?
> Using DDF don't need to recompile the code (nice for HA users), but you need a recent version.

Yes, only stable. I'll waiting 🤷🏼‍♂️

chpego, 2021-11-09:
@Smanar , i'm curious , how will the device be detected in the Phoscon Web App afterwards as a sensor?

Smanar, 2021-11-09:
I think it will be not displayed in phoscon ATM.
Phoscon don't support power measurement for exemple, you will have it in HA, same result than a simple plug that display current and consumption

AdrienBigot, 2021-11-18:
I've just receive today a Lixee Zlinky_TIC.  I'm running deconZ  outside of HomeAssitant OS so I have :
 `ii  deconz                         2.13.02-raspbian-buster-beta armhf        A generic ZigBee monitoring and control tool`

I can help with an DDF file if you still need one. Let me know.

Adrien

chpego, 2021-11-18:
> I've just receive today a Lixee Zlinky_TIC.  I'm running deconZ  outside of HomeAssitant OS so I have :
>  `ii  deconz                         2.13.02-raspbian-buster-beta armhf        A generic ZigBee monitoring and control tool`
> 
> Tomorrow, I'll try to modify general.xml file with the config you provided above and I can help with an DDF file if you still need one. Let me know.
> 
> Adrien

Yes , please 💪

Smanar, 2021-11-18:
Yep pls, forr DDF you have a starting point here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5248#issuecomment-945174475

But can explain better, I think you will need only ZHAPower and ZHAConsumption.

AdrienBigot, 2021-11-18:
Is it what you need ?

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "LiXee",
  "modelid": "ZLinky_TIC",
  "product": "ZLinky_TIC",
  "sleeper": false,
  "status": "Bronze",
  "subdevices": [
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "description": "The measured consumption (in Wh).",
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "description": "The measured current (in mA).",
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "description": "The measured power (in W).",
          "refresh.interval": 300
        },
        {
          "name": "state/voltage",
          "description": "The measured voltage (in V).",
          "refresh.interval": 300
        }
      ]
    }
  ]
}
`

Smanar, 2021-11-19:
If the name was changed it's fine, it s mean the device was added in the API.
Are you using third app ? Else you can use the beta phoscon http://phoscon.de/pwabeta in help / API information.

If all is fine you will have 2 new sensor in it.

For information, as DDF are still beta, for the moment it s disabled at every restart, so don't forget to save the file.

AdrienBigot, 2021-11-19:
I'm using deconz with the home Assitant integration. For the moment, I think this device is not supported by integration so not visible in HA. 
But I also have Phoscon installed and I confirm that in the Help / API info  I can see 2 new sensors (Power 6 and Consumption 5)
[image]

` "5": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": "57689bdbea80225c741004f532ed4ed5",
    "lastannounced": null,
    "lastseen": "2021-11-18T19:06Z",
    "manufacturername": "LiXee",
    "modelid": "ZLinky_TIC",
    "name": "Consumption 5",
    "state": {
      "consumption": null,
      "lastupdated": "none"
    },
    "swversion": "4000",
    "type": "ZHAConsumption",
    "uniqueid": "00:15:8d:00:05:be:9b:92-ff-0702"
  },
  "6": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": "af89342494e1234c8957f64f8d9b5b15",
    "lastannounced": null,
    "lastseen": "2021-11-18T19:06Z",
    "manufacturername": "LiXee",
    "modelid": "ZLinky_TIC",
    "name": "Power 6",
    "state": {
      "current": null,
      "lastupdated": "none",
      "power": null,
      "voltage": null
    },
    "swversion": "4000",
    "type": "ZHAPower",
    "uniqueid": "00:15:8d:00:05:be:9b:92-ff-0b04"
  }
`

Could you please give me more informations about what is Devices Description Files ?  Why do I need to generate it ?

Thanks !

Adrien

Smanar, 2021-11-20:
>I think this device is not supported by integration so not visible in HA.

No problem, it s already used sensor a ZHAPower and a ZHAConsumption, thoses one are used for exemple for plugs
And you have it in the API, so there is no reason to don't see them in HA. I will ask to Robban if you still have problem with it.

>Could you please give me more informations about what is Devices Description Files ? Why do I need to generate it ?

But you have already started. You have saved the file ?
DDF file are the future deconz feature, every devices will have one, it will help for device reconization, and make the support easier, as it will be json file, so less work to do on c++ code (and less problem for us, new devices will not broke olders for exemple).
Just using a text editor if you don't have the GUI and the editor) you will be able to make almost all on devices (setting reporting, binds, reverse covering, ect ...)

For the moment it still on beta, so for exemple DDF are disabled at every deconz restart.

For your device, it seem value are not filled, so we will continue the DDF file (and don't forget to save the file)
- go on the second tab "binding"
- drag and drop the cluster 0x0b04 and the cluster 0x0702
- go on the first tab "items"
- select the state consumption and configure it (cluster 0x0702, attribute 0x0000)
- select the state power and configure it (cluster 0x0b04, attribute 0x0508)
- select "hot reload" in menu

It's possible you have some value auto filled, so idk exactly values your need or not, if you are not sure, just show how look the DDF editor for the device.

AdrienBigot, 2021-11-20:
Hi,

Thank you for your explanations.
I continue the DDF file by adding bindings as you ask  and you were right when you said "It's possible you have some values auto filled".
I think all is automatically auto filled without errors. See screenshots :

[image]
[image]

I save the DDF File and hot reload (multiple times) but I still don't see any value when I request the API, even after waiting for more than 300 seconds.

```json
curl -s -X GET -H "Content-Type: application/json" http://xxxxxxxxxxxxxxxxxxx/api/XXXXXXXXX/sensors/5 | jq .
{
  "config": {
    "on": true,
    "reachable": true
  },
  "etag": "9aef951c6ca18eed9bb2438d49f5a1e7",
  "lastannounced": null,
  "lastseen": "2021-11-18T19:06Z",
  "manufacturername": "LiXee",
  "modelid": "ZLinky_TIC",
  "name": "Consumption 5",
  "state": {
    "consumption": null,
    "lastupdated": "none"
  },
  "swversion": "4000",
  "type": "ZHAConsumption",
  "uniqueid": "00:15:8d:00:05:be:9b:92-ff-0702"
}
```

```json
curl -s -X GET -H "Content-Type: application/json" http://xxxxxxxxxxxxxxxxxxx/api/XXXXXXXXX/sensors/6 | jq .
{
  "config": {
    "on": true,
    "reachable": true
  },
  "etag": "d01530209e0f1f4a5bed1d75698c45a6",
  "lastannounced": null,
  "lastseen": "2021-11-18T19:06Z",
  "manufacturername": "LiXee",
  "modelid": "ZLinky_TIC",
  "name": "Power 6",
  "state": {
    "current": null,
    "lastupdated": "none",
    "power": null,
    "voltage": null
  },
  "swversion": "4000",
  "type": "ZHAPower",
  "uniqueid": "00:15:8d:00:05:be:9b:92-ff-0b04"
}
```
Maybe there is a problem with bindings ? What the endpoint address means ?

[image]

Smanar, 2021-11-21:
>Maybe there is a problem with bindings ? What the endpoint address means ?

Ha yes sorry, my bad. On the second tab "bindings"
- Select the cluster "Simple metring cluster"
- Drag and drop (from the attribute list) the attribute 0x0000 in the right panel to be able to configure the reporting, you will have something like that
[image]

And use this setting for exemple (it s the defaut value in c++ code)
```
        rq.minInterval = 1;
        rq.maxInterval = 300;
       rq.reportableChange48bit = 1;
```

And yes have forget the power too.
Cluster 0x0b04 attribute 0x050F

AdrienBigot, 2021-11-21:
Great, it works !

Current, power and consumption are now visible through the API. 
IDK why the lastupdated attribute is never updated.

Now I have to understand why these values are not visible in Home Assitant.
I think it's related to a starnge thing I noticed. When I curl the /sensors endpoint  API All values are OK. But when I curl each sensors directly values are null. It's the same behaviour with curl and in the Phoscon app.

OK :
```json
curl -s -X GET -H "Content-Type: application/json" http://xxxxxxxxxxxxxxxxxxx/api/XXXXXXXXX/sensors/ | jq .
"5": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": null,
    "lastannounced": null,
    "lastseen": "2021-11-21T10:36Z",
    "manufacturername": "LiXee",
    "modelid": "ZLinky_TIC",
    "name": "Consumption 5",
    "state": {
      "consumption": 1.02747E+7,
      "lastupdated": "none"
    },
    "swversion": "4000",
    "type": "ZHAConsumption",
    "uniqueid": "00:15:8d:00:05:be:9b:92-01-0702"
  },
  "6": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": null,
    "lastannounced": null,
    "lastseen": "2021-11-21T10:36Z",
    "manufacturername": "LiXee",
    "modelid": "ZLinky_TIC",
    "name": "Power 6",
    "state": {
      "current": 2,
      "lastupdated": "none",
      "power": 500,
      "voltage": null
    },
    "swversion": "4000",
    "type": "ZHAPower",
    "uniqueid": "00:15:8d:00:05:be:9b:92-01-0b04"
  }
```

NOT OK : 
```json
curl -s -X GET -H "Content-Type: application/json" http://xxxxxxxxxxxxxxxxxxx/api/XXXXXXXXX/sensors/6/ | jq .
{
  "config": {
    "on": true,
    "reachable": true
  },
  "etag": "0b163249539ceb854550127b47b28b65",
  "lastannounced": null,
  "lastseen": "2021-11-20T22:47Z",
  "manufacturername": "LiXee",
  "modelid": "ZLinky_TIC",
  "name": "Power 6",
  "state": {
    "current": null,
    "lastupdated": "none",
    "power": null,
    "voltage": null
  },
  "swversion": "4000",
  "type": "ZHAPower",
  "uniqueid": "00:15:8d:00:05:be:9b:92-ff-0b04"
}
```json

In Phoscon : 

![sensors](https://user-images.githubusercontent.com/12624309/142759875-30224aaf-45ec-4343-9403-04d576f8d607.png)

![direct_6](https://user-images.githubusercontent.com/12624309/142759874-a8e64324-330d-475e-a568-d320e861e062.png)

Smanar, 2021-11-21:
Ha, effectively, this is funny ...

I will ask to other devs, I hope it s not a DDF issue.

Smanar, 2021-11-21:
I have no explanation yet but there is something strange, for exemple on the ZHAPower you have 2 differents uniqueID.
Try to force the endpoint to 0x01 instead of "auto"

AdrienBigot, 2021-11-21:
Good news, it works !

I can now graph for example the Power attribute in Home Assistant because the API return good data.
[image]

As you noticed, the problem was in the subdevice UID. This morning I wanted to restart a DDF from scratch and when you open the editor, it automatically add in the Left Column ZHApower and ZHAConsumption subdevices. BUT , by default, their template UID are $address.ext-0xff-0x0b04 and $address.ext-0xff-0x0702 . IDK why  ?

[image]

Smanar, 2021-11-21:
I think you will have too some issue for the consumption ?

>"consumption": 1.02747E+7

In the field "Expression" , you can add "/10" for exemple. IDK what is the good value you need.

Smanar, 2021-11-21:
I m reading the page https://github.com/fairecasoimeme/Zlinky_TIC

It seem the only usefull thing you miss is 
<html><body>

ADPS | 0xFF66 | 0x0005 | RP |   | Uint16 | 3 car | A | Avertissement de Dépassement De Puissance Souscrite | 0
-- | -- | -- | -- | -- | -- | -- | -- | -- | --

</body>
</html>

But idk if it s possible to simulate it, and the utility. If you exceed the value too long time, it cut the power nope ?

And we will have a problem too, the device is easy to make it working, it s zigbee compliant. But according to the contract or the mode (historical/standard), bind and report will not be the same.
IDK if tis will be possible without touching the c++ code.

Smanar, 2021-11-21:
BTW for the issue with lastupdated, try to check the case "Awake on receive" for some attributes.

AdrienBigot, 2021-11-21:
Yes the "Avertissement de Dépassement De Puissance Souscrite"  in cluster 0xFF66 is difficult to test and as I imagine, when you reach the subscribed power, the linky cut the power. So, It's not very useful since if if the power is cut, home assistant / internet box  etc ...  are down :-D   But maybe the power is cut after a certain amount of time and it can be useful to send an alert during the period the power is still ok :) . Or in another case the limit can flap because the subscriber use just the limit of its contract ... but IMHO it's very rare.
I tried to add it but I don't have LiXee specific cluster in the column available items in DDF editor and edit the DDF file by hand not works anymore.

Of Course I can share my DDF File.

For a contract :  BASE (no HC HP) Monophasé 6KVA   (same as @chpego )  DDF file is : 

```json
{
  "schema": "devcap1.schema.json",
  "manufacturername": "LiXee",
  "modelid": "ZLinky_TIC",
  "product": "ZLinky_TIC",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/zlinky_tic.json",
  "subdevices": [
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 300,
          "read": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "if (Attr.val != 65535) { Item.val = Attr.val; }"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 300,
          "read": {
            "at": "0x050f",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x050f",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "if (Attr.val != -32768) { Item.val = Attr.val; }"
          }
        },
        {
          "name": "state/voltage",
          "description": "The measured voltage (in V).",
          "refresh.interval": 300
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "description": "The measured consumption (in Wh).",
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x050F",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x23",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

Smanar, 2021-11-22:
Ok will be more sportive. So waiting for new deconz version.

Can try to add the missing state/alam, taked from here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5503/files (devices/generic/items/state_alarm_item.json)

And make something like tha
[image]
t

>I don't have LiXee specific cluster in the column available items in DDF editor

To have the lixee cluster, you need to edit the xml file  https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5459#issuecomment-962577707

AdrienBigot, 2021-11-22:
Sorry but today, DDF doesn't want to work ...   IDK why, but sometimes  DDF are not taken into account when I "hot reload" or partially working.
At the begining, I thought it was an API cache problem because after 3 or more try, I was able to read a good value. But today, no value are visible by home assistant because when I call sensors directly (/sensors/6  for example)  no values are present (same problem I had 2 days ago).  After manny tries / restart DDF doesn't work anymore for me. All attributes are set to "null"  (it's not the case when I request the endpoint /sensors )

Result : 

```json
curl -s -X GET -H "Cache-Control: no-cache" -H "Content-Type: application/json" http://xxxxxxxxxxxxxxxxxxxx/api/XXXXXXXX/sensors/6/ | jq .
{
  "config": {
    "on": true,
    "reachable": true
  },
  "etag": "011af670a593a71a3a738d4fa9a6ea5f",
  "lastannounced": null,
  "lastseen": "2021-11-22T20:24Z",
  "manufacturername": "LiXee",
  "modelid": "ZLinky_TIC",
  "name": "Power 6",
  "state": {
    "current": null,
    "lastupdated": "none",
    "power": null,
    "voltage": null
  },
  "swversion": "4000",
  "type": "ZHAPower",
  "uniqueid": "00:15:8d:00:05:be:9b:92-ff-0b04"
}
```

I did a last test to try something with the DDF file and it was a bit confusing.

What I did :  
1. I first remove my DDF file placed in my folder ~/.local/share/dresden-elektronik/deCONZ/devices/zlinky_tic.json. I suspect deCONZ try to read it when it starts but because DDF are disabled by default it turns into bad things (it's just a guess).
2. I curl API (with DDF disabled in the controle pane) and I saw old values in the `current` and `power` attributes  (I think the last value found in the API cache).
Results : 
```json
curl -s -X GET -H "Cache-Control: no-cache" -H "Content-Type: application/json" http://xxxxxxxxxxxxxxxxxxxx/api/XXXXXXXX/sensors/6/ | jq .
{
  "config": {
    "on": true,
    "reachable": true
  },
  "etag": "36cfbe30f682b932a2a14a1549900e4f",
  "lastannounced": null,
  "lastseen": "2021-11-22T20:38Z",
  "manufacturername": "LiXee",
  "modelid": "ZLinky_TIC",
  "name": "Power 6",
  "state": {
    "current": 2,
    "lastupdated": "none",
    "power": 560,
    "voltage": null
  },
  "swversion": "4000",
  "type": "ZHAPower",
  "uniqueid": "00:15:8d:00:05:be:9b:92-ff-0b04"
}

```
3. Then, I enabled DDF in the controle pane , load my saved DDF file and Hot Reload and Curl 
Results : 
```json
curl -s -X GET -H "Cache-Control: no-cache" -H "Content-Type: application/json" http://xxxxxxxxxxxxxxxxxxxx/api/XXXXXXXX/sensors/6/ | jq .
{
  "config": {
    "on": true,
    "reachable": true
  },
  "etag": "930ab037a556d467dbe256216230c244",
  "lastannounced": null,
  "lastseen": "2021-11-22T20:38Z",
  "manufacturername": "LiXee",
  "modelid": "ZLinky_TIC",
  "name": "Power 6",
  "state": {
    "current": 2,
    "lastupdated": "none",
    "power": 560,
    "voltage": 520
  },
  "swversion": "4000",
  "type": "ZHAPower",
  "uniqueid": "4000"
}
```

WTF ?  The updated power value (520) appears in the Voltage attribute !
It verify why this feature is still in beta state :-D 

But it's not the goal of this issue so I won't go in details but I'm available to fill a bug report if you are interested.

ivico94, 2021-11-22:
Hello @AdrienBigot and thank you for this ongoing project! I would like to complete you're answer. The "Avertissement de Dépassement De Puissance Souscrite" does not lead to an instantaneous cut of power. Without knowing the exact duration before the power is cut, I was able to see via my Linky data on the site of Enedis (electricity provider) that I have several times exceeded my power over a year, without having the Linky cut.

On the contrary, having this information would allow to do load shedding (i.e. to cut off energy consuming devices) and thus optimize the power selected with the electricity subscription. 

I would find it very relevant to have this information sent to Home Assistant in order to trigger an automation. It would allow us to optimize the max power selected with the electricity subscription and thus, lower our bill :-)

AdrienBigot, 2021-11-23:
Thanks @ivico94  for these details. Indeed, it's quite interesting and following your comment, I've dowloaded my monthly max power and I observed that me too, I've several time exceeded my 6kVA subscription.

You're right :
> it's very relevant to have this information sent to Home Assistant in order to trigger an automation

Smanar, 2021-11-23:
Same issue than before 

>"uniqueid": "00:15:8d:00:05:be:9b:92-ff-0b04"

There is something wrong with the endpoint, but your ddf is fine, you are using 0x01. Yes there is a conflict somewhere but I don't see why the api have take FF without using the ddf core .... You have deleted and re-included the device ? (because a uniqueid can't be modified and the one in the API don't look like one that can be created by the ddf.)

And from my memory the device don't use the "voltage" at all ? it s not visible in the GUI.
Have you updated the xml file to have the lixee cluster available in deconz ? Even I think the value is 0 for 0xFF66 / 0x0005

And BTW, look at the last uniqueid, it s the swversion.
>"uniqueid": "4000"

I think you have found a new issue.

Smanar, 2021-11-23:
Ha yes I understand now, will check for lixee cluster tommorow.
And for the issue Mimiix have fowarded it to manup, I m waiting for his reaction, because for me too, I think this issue can expose something more critic.

manup, 2021-11-23:
Pretty interesting find. I think what might have happened is that in the database a the invalid sensor resources with the 0xff endpoint still exist. Not sure if they are also deleted when the device is deleted via REST-API or GUI. Need to make a few tests here...

After this likely I'll add a small check that such resources won't be added in first place and a warning to the editor that the unique id needs to be edited if endpoint is still 0xff (which is meant as a invalid placeholder).

AdrienBigot, 2021-11-24:
Hi,

I made some tests this morning.

1. Here what I have in database when I start deCONZ with DDF disabled. Old sensors 5 and 6 are still in the database with bad UID  : 
```
sqlite> SELECT "_rowid_",* FROM "main"."sensors" WHERE "sid" >= 5;
rowid|sid|name|type|modelid|manufacturername|uniqueid|swversion|state|config|fingerprint|deletedState|mode|lastseen|lastannounced
8385|5|Consumption 5|ZHAConsumption|ZLinky_TIC|LiXee|00:15:8d:00:05:be:9b:92-ff-0702|4000|{"consumption":null,"lastupdated":null}|{"on":true,"reachable":false}||normal|1|2021-11-23T18:26Z|
8384|6|Power 6|ZHAPower|ZLinky_TIC|LiXee|00:15:8d:00:05:be:9b:92-ff-0b04|4000|{"current":null,"lastupdated":null,"power":null,"voltage":null}|{"on":true,"reachable":false}||normal|1|2021-11-23T18:26Z|
```

2. If I enable DDF + hot reload + query each sensor their endpoint is good : 
```
sqlite> SELECT "_rowid_",* FROM "main"."sensors" WHERE "sid" >= 5;
rowid|sid|name|type|modelid|manufacturername|uniqueid|swversion|state|config|fingerprint|deletedState|mode|lastseen|lastannounced
8483|6|Power 6|ZHAPower|ZLinky_TIC|LiXee|00:15:8d:00:05:be:9b:92-01-0b04|4000|{"current":2,"lastupdated":null,"power":500,"voltage":null}|{"on":true,"reachable":true}||normal|2|2021-11-24T08:45Z|
8484|5|Consumption 5|ZHAConsumption|ZLinky_TIC|LiXee|00:15:8d:00:05:be:9b:92-01-0702|4000|{"consumption":1.03214e+07,"lastupdated":null}|{"on":true,"reachable":true}||normal|2|2021-11-24T08:45Z|
```

3. If I stop deCONZ , values UID are still OK in the database.

4. And Finally if I start deCONZ with DDF disabled, UID are back to bad value : 
```
sqlite> SELECT "_rowid_",* FROM "main"."sensors" WHERE "sid" >= 5;
rowid|sid|name|type|modelid|manufacturername|uniqueid|swversion|state|config|fingerprint|deletedState|mode|lastseen|lastannounced
8486|6|Power 6|ZHAPower|ZLinky_TIC|LiXee|00:15:8d:00:05:be:9b:92-ff-0b04|4000|{"current":null,"lastupdated":null,"power":null,"voltage":null}|{"on":true,"reachable":false}||normal|1|2021-11-24T08:45Z|
8487|5|Consumption 5|ZHAConsumption|ZLinky_TIC|LiXee|00:15:8d:00:05:be:9b:92-ff-0702|4000|{"consumption":null,"lastupdated":null}|{"on":true,"reachable":false}||normal|1|2021-11-24T08:45Z|
```

I also notice that bad values returned when I curl API don't reflect values present in the database.

Example :

What I have in the database : 
```
8494|6|Power 6|ZHAPower|ZLinky_TIC|LiXee|00:15:8d:00:05:be:9b:92-ff-0b04|4000|{"current":null,"lastupdated":null,"power":null,"voltage":null}|{"on":true,"reachable":false}||normal|1|2021-11-24T09:12Z|
```

And what is return by curl : 
```json
curl -s -X GET -H "Content-Type: application/json" http://xxxxxxxxxxxxxxxxxxxxxxxxx/api/XXXXXXXXX/sensors/6 | jq .
{
  "config": {
    "on": true,
    "reachable": true
  },
  "etag": "c5d9b8c234afd8a752355da9bcba2036",
  "lastannounced": null,
  "lastseen": "2021-11-24T09:12Z",
  "manufacturername": "LiXee",
  "modelid": "ZLinky_TIC",
  "name": "Power 6",
  "state": {
    "current": 2,
    "lastupdated": "none",
    "power": 560,
    "voltage": 610
  },
  "swversion": "4000",
  "type": "ZHAPower",
  "uniqueid": "00:15:8d:00:05:be:9b:92-ff-0b04"
}
```

Smanar, 2021-11-24:
So this  issue happen on device added using DDF, but during the deconz launching/restart with DDF disabled.

And I have learned something, the uniqueid is not static ? During the working mode deconz can change/update a uniqueid ?

Less important, but about the Lixee cluster, I have made a try with a Legrand cluster, I have same result than you, but even the attribute fonction and the cluster name don't appear, it s seem the procedure is working.
```
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xFC40",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x30",
          "min": 0,
          "max": 0
        }
      ]
```

You have added the state/alarm state too ?

AdrienBigot, 2021-11-24:
@Smanar yes the problem appears on devices added throught DDF editor and not available at reboot because DDF are disabled by default.
I've just reproduced the problem with a new device added to test the  `state/alarm`.
I've added a device AirQuality since the LiXee Specific cluster is not available in the "available devices" column (the right column in the DDF editor) and I modified it by adding the new state/alarm and the correct binding / reporting.
At the beginning it worked fine : 
```json
{
	"schema": "resourceitem1.schema.json",
	"id": "state/alarm",
	"datatype": "Bool",
	"access": "R",
	"public": true,
	"description": "True when an alarm is detected.",
	"parse": {"fn": "ias:zonestatus", "mask": "alarm1,alarm2"}
} 
```
At reboot, with DDF disabled my device has a bad endpoint : 
```json
 "7": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": "8f063cd77aa3789448084cacdecc590a",
    "lastannounced": null,
    "lastseen": "2021-11-24T17:23Z",
    "manufacturername": "LiXee",
    "modelid": "ZLinky_TIC",
    "name": "AirQuality 7",
    "state": {
      "airquality": null,
      "airqualityppb": null,
      "lastupdated": "none"
    },
    "swversion": "4000",
    "type": "ZHAAirQuality",
    "uniqueid": "00:15:8d:00:05:be:9b:92-ff-ff66"
  }
```

And the worse is here. When I loaded my DDF file a new AirQuality device has been created with good values but the old AirQuality device remained with values belonging to others clusters. I'm going to be crazy.

```json
{
  "1": {
    "config": {
      "configured": true,
      "on": true,
      "sunriseoffset": 30,
      "sunsetoffset": -30
    },
    "etag": "4582f4991e9a137e77bc66f5199cafaa",
    "manufacturername": "Philips",
    "modelid": "PHDL00",
    "name": "Daylight",
    "state": {
      "dark": true,
      "daylight": false,
      "lastupdated": "2021-11-24T18:01:20.602",
      "status": 230,
      "sunrise": "2021-11-24T06:51:28",
      "sunset": "2021-11-24T16:14:42"
    },
    "swversion": "1.0",
    "type": "Daylight",
    "uniqueid": "00:21:2e:ff:ff:06:9b:04-01"
  },
  "2": {
    "config": {
      "battery": 100,
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "8a1ad480e436edb6e0a022257e2a7359",
    "lastannounced": "2021-11-03T17:53:52Z",
    "lastseen": "2021-11-24T17:58Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.weather",
    "name": "Multi Sensor",
    "state": {
      "lastupdated": "2021-11-24T17:58:26.051",
      "temperature": 1732
    },
    "swversion": "20191205",
    "type": "ZHATemperature",
    "uniqueid": "00:15:8d:00:01:9d:18:b7-01-0402"
  },
  "3": {
    "config": {
      "battery": 100,
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "d25990142062450ead597dab307404ad",
    "lastannounced": "2021-11-03T17:53:52Z",
    "lastseen": "2021-11-24T17:58Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.weather",
    "name": "Humidity 3",
    "state": {
      "humidity": 5808,
      "lastupdated": "2021-11-24T17:58:26.085"
    },
    "swversion": "20191205",
    "type": "ZHAHumidity",
    "uniqueid": "00:15:8d:00:01:9d:18:b7-01-0405"
  },
  "4": {
    "config": {
      "battery": 100,
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "f3af3465854183de1a4344ebdae1acf4",
    "lastannounced": "2021-11-03T17:53:52Z",
    "lastseen": "2021-11-24T17:58Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.weather",
    "name": "Pressure 4",
    "state": {
      "lastupdated": "2021-11-24T17:58:26.104",
      "pressure": 1001
    },
    "swversion": "20191205",
    "type": "ZHAPressure",
    "uniqueid": "00:15:8d:00:01:9d:18:b7-01-0403"
  },
  "5": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": null,
    "lastannounced": null,
    "lastseen": "2021-11-24T18:13Z",
    "manufacturername": "LiXee",
    "modelid": "ZLinky_TIC",
    "name": "Consumption 5",
    "state": {
      "consumption": 1.03294E+7,
      "lastupdated": "none"
    },
    "swversion": "4000",
    "type": "ZHAConsumption",
    "uniqueid": "00:15:8d:00:05:be:9b:92-01-0702"
  },
  "6": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": null,
    "lastannounced": null,
    "lastseen": "2021-11-24T18:13Z",
    "manufacturername": "LiXee",
    "modelid": "ZLinky_TIC",
    "name": "Power 6",
    "state": {
      "current": 3,
      "lastupdated": "none",
      "power": 640,
      "voltage": null
    },
    "swversion": "4000",
    "type": "ZHAPower",
    "uniqueid": "00:15:8d:00:05:be:9b:92-01-0b04"
  },
  "7": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": "75fa180982ab55c7c3b97e3fe8095e98",
    "lastannounced": null,
    "lastseen": "2021-11-24T17:23Z",
    "manufacturername": "LiXee",
    "modelid": "ZLinky_TIC",
    "name": "AirQuality 7",
    "state": {
      "airquality": null,
      "airqualityppb": 640,
      "lastupdated": "none"
    },
    "swversion": "4000",
    "type": "ZHAAirQuality",
    "uniqueid": "4000"
  },
  "8": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": "5447e52291f200f39320392a55fb1767",
    "lastannounced": null,
    "lastseen": "2021-11-24T18:13Z",
    "manufacturername": "LiXee",
    "modelid": "ZLinky_TIC",
    "name": "AirQuality 8",
    "state": {
      "alarm": false,
      "lastupdated": "none"
    },
    "swversion": "4000",
    "type": "ZHAAirQuality",
    "uniqueid": "00:15:8d:00:05:be:9b:92-01-ff66"
  }
}
```

Smanar, 2021-11-25:
@manup but it s normal the DDF core is able to create new entry (outside a inclusion procedure) or edit an uniqueid ?

AdrienBigot, 2021-11-30:
Hi,

@manup  if some informations are missing, I can help and do more testing.
And if needed, I can arrange you a ssh access to my testing deCONZ instance linked to LiXee device.
Thank you.

Adrien

Smanar, 2021-12-10:
Hello, the last beta is out, if you have some time to check if you still have this kind of issues ?

AdrienBigot, 2021-12-10:
Hi,

It's not very clear for me what diferent DDF mode (bronze, silver gold ....)  means exactly but I've tested with  latest v2.13.3-beta. 

WHen I start deconz and request I've first : 

```json
curl -s -X GET -H "Content-Type: application/json" http://xxxxxxxxxxxxxxxxxxxxxx.fr/api/XXXXXXXXXXXXX/sensors/ | jq .
{
  "1": {
    "config": {
      "configured": true,
      "on": true,
      "sunriseoffset": 30,
      "sunsetoffset": -30
    },
    "etag": "f4923cbea8aa922e857ac86dda24f419",
    "manufacturername": "Philips",
    "modelid": "PHDL00",
    "name": "Daylight",
    "state": {
      "dark": true,
      "daylight": false,
      "lastupdated": "2021-12-10T21:21:40.297",
      "status": 230,
      "sunrise": "2021-12-10T07:08:31",
      "sunset": "2021-12-10T16:09:45"
    },
    "swversion": "1.0",
    "type": "Daylight",
    "uniqueid": "00:21:2e:ff:ff:06:9b:04-01"
  },
  "2": {
    "config": {
      "battery": 100,
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "853c074b3ecf65b4144bfed9fb8da51b",
    "lastannounced": "2021-12-06T22:55:19Z",
    "lastseen": "2021-12-10T21:24Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.weather",
    "name": "Multi Sensor",
    "state": {
      "lastupdated": "2021-12-10T21:24:36.939",
      "temperature": 1459
    },
    "swversion": "20191205",
    "type": "ZHATemperature",
    "uniqueid": "00:15:8d:00:01:9d:18:b7-01-0402"
  },
  "3": {
    "config": {
      "battery": 100,
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "56208e7082341f70427084c12786c6a5",
    "lastannounced": "2021-12-06T22:55:19Z",
    "lastseen": "2021-12-10T21:24Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.weather",
    "name": "Humidity 3",
    "state": {
      "humidity": 4596,
[... 274 lines cut ...]

```json
curl -s -X GET -H "Content-Type: application/json" http://xxxxxxxxxxxxxxxxxxxxx.fr/api/XXXXXXXXX/sensors/8 | jq .
{
  "config": {
    "on": true,
    "reachable": true
  },
  "etag": "9353eb328d25543633835736498b8636",
  "lastannounced": null,
  "lastseen": "2021-12-10T21:53Z",
  "manufacturername": "LiXee",
  "modelid": "ZLinky_TIC",
  "name": "LiXee 8",
  "state": {
    "alarm": false,
    "lastupdated": "2021-12-10T21:59:16.203"
  },
  "swversion": "4000",
  "type": "ZHALiXee",
  "uniqueid": "00:15:8d:00:05:be:9b:92-01"
}
```

So I think there is again problem inside the database.

Smanar, 2021-12-11:
>It's not very clear for me what diferent DDF mode (bronze, silver gold ....) means exactly but I've tested with latest v2.13.3-beta.

I haven't confirmation, but I think "gold" will enable the DDF even you are on "basic" mode.

And if you close and restart deconz, you will have again the "bad uniqueid" devices ?

AdrienBigot, 2021-12-11:
To be sure not to have side effect caused by bad values into database, I've purged from the sqlite database all LiXee references (mac address) from tables ressource_item , sub_devices an sensors tables.

At the first start of deCONZ my device is detected like in this :
[image]

A curl of  /sensors endpoint doesn't contains anything about the LiXee device.

Then, I load my DDF File, the device is seen as LiXee. Good point.

[image]

A curl contains all sensors with good uniqueid : 

```json
"5": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": "072761e12e339ceee5eac44e23b49d28",
    "lastannounced": null,
    "lastseen": "2021-12-11T18:24Z",
    "manufacturername": "LiXee",
    "modelid": "ZLinky_TIC",
    "name": "Power 5",
    "state": {
      "current": 4,
      "lastupdated": "2021-12-11T18:28:15.132",
      "power": 840,
      "voltage": null
    },
    "swversion": "4000",
    "type": "ZHAPower",
    "uniqueid": "00:15:8d:00:05:be:9b:92-01-0b04"
  },
  "6": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": "c64019fcf2c89bd800ecdac5eb4671ba",
    "lastannounced": null,
    "lastseen": "2021-12-11T18:24Z",
    "manufacturername": "LiXee",
    "modelid": "ZLinky_TIC",
    "name": "Consumption 6",
    "state": {
      "consumption": 1.06366E+7,
      "lastupdated": "2021-12-11T18:28:30.106"
    },
    "swversion": "4000",
    "type": "ZHAConsumption",
    "uniqueid": "00:15:8d:00:05:be:9b:92-01-0702"
  },
  "7": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": "8db4bcf0863e8b0d9dd3df250d05c1c3",
    "lastannounced": null,
    "lastseen": "2021-12-11T18:24Z",
    "manufacturername": "LiXee",
    "modelid": "ZLinky_TIC",
    "name": "LiXee 7",
    "state": {
      "alarm": false,
      "lastupdated": "2021-12-11T18:28:33.702"
    },
    "swversion": "4000",
    "type": "ZHALiXee",
    "uniqueid": "00:15:8d:00:05:be:9b:92-01-ff66"
  }
```

All is good for now.

If I stop and start deconz without loading DDF file API returns sensors with bad uniqueid. This bad uniqueID is only returns by API, I cannot find this bad uniqueID in the database.

```json
 "5": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": "faa24682514dd962658c8778e719f4ae",
    "lastannounced": null,
    "lastseen": "2021-12-11T18:24Z",
    "manufacturername": "LiXee",
    "modelid": "ZLinky_TIC",
    "name": "Power 5",
    "state": {
      "current": 4,
      "lastupdated": "2021-12-11T18:25:20.460",
      "power": 880,
      "voltage": 0
    },
    "swversion": "4000",
    "type": "ZHAPower",
    "uniqueid": "00:15:8d:00:05:be:9b:92-ff-0b04"
  },
  "6": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": "6a7225cef9401dc5e3e0a47fbf2b829f",
    "lastannounced": null,
    "lastseen": "2021-12-11T18:24Z",
    "manufacturername": "LiXee",
    "modelid": "ZLinky_TIC",
    "name": "Consumption 6",
    "state": {
      "consumption": 1.06366E+7,
      "lastupdated": "2021-12-11T18:25:29.204"
    },
    "swversion": "4000",
    "type": "ZHAConsumption",
    "uniqueid": "00:15:8d:00:05:be:9b:92-ff-0702"
  },
  "7": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": "dcda74a8c65adb65319e8a6efce64242",
    "lastannounced": null,
    "lastseen": "2021-12-11T18:24Z",
    "manufacturername": "LiXee",
    "modelid": "ZLinky_TIC",
    "name": "LiXee 7",
    "state": {
      "lastupdated": "2021-12-11T18:25:28.331"
    },
    "swversion": "4000",
    "type": "ZHALiXee",
    "uniqueid": "00:15:8d:00:05:be:9b:92-ff-ff66"
  }
```

And if I load again DDF, all is back to normal state / uniqueid.

I don't see anymore my zombie sensors as I think it was a due to a bad record in my database caused by many many tests ...

So now, I can share I'll try to power on many many electrical devices and test if the state/alarm change from false to true. I think we are closed to have a functional device. But I can't imagine to ask people to use DDF files like I did.

Smanar, 2021-12-12:
>So now, I'll try to power on many many electrical devices and test if the state/alarm change from false to true

BTW, for information, 1 week ago I was on a client with a linky, He need to use 45A on a 30A lnstallation before something cut the power and it was the circuit breaker not the linky.

>But I can't imagine to ask people to use DDF files like I did. So what would be the best solution ?

I think the issue is perhaps not visible if DDF are enabled all the time, but for me there is still something bad somewhere.

AdrienBigot, 2021-12-12:
Hi,

I've checked the alarm.
In the Enedis documentation it says "L’intensité de référence est calculée de la façon suivante : IR = P Référence en VA/200 V."  So My IR max is 6000/200 = 30A.

With a current of 32A (2amps above my 30A allowed) I successfully trigger a change from false to true on the alarm state.

```json
curl -s -X GET -H "Content-Type: application/json" http://xxxxxxxxxxxxxxxxxxxxxxxxxxxx.fr/api/XXXXXXXXXXXX/sensors/ | jq '.["5","7"].state'
{
  "current": 32,
  "lastupdated": "2021-12-12T17:55:11.209",
  "power": 7180,
  "voltage": null
}
{
  "alarm": true,
  "lastupdated": "2021-12-12T17:55:11.453"
}
```

But now, my alarm is always set to true because the ADPS (Avertissement de dépassement de Puissance Souscrite) is set to 32
[image]

I don't know how many time I have to wait for an update of the 0x0005 id by the Linky ...   @fairecasoimeme maybe are you aware about that ? I didn't see this information neither in the Enedis documentation nor yours.

AdrienBigot, 2021-12-13:
Checked just now (more than 24hours after the first test), the value of the attribute 0x0005 in the cluster 0xff66 is still the same as yesterday : 32A.  Therefore, the subscribed power exceeded alarm is always set to true.

@fairecasoimeme  : did you ever face this situation ? 

Thank you in advance.

Adrien.

[61 more replies]

closed 2022-03-15 by github-actions[bot]