dresden-elektronik/deconz-rest-plugin #5462: Danfoss Ally - Can not set negative offset, always results in 0

tatec22, 2021-11-07:
## Describe the bug
When setting a negative number for the offset of the Danfoss Ally, the value is accepted but isn't really set. 

## Steps to reproduce the behavior
  1. Set a negative value (e.g. -160) via Home Assistant and "deconz.configure" command
  2. Check Rest Api for correct value -> It is the correct value
  3. Open deconz and read "Regulation SetPoint Offset" (0x404B)
  4. The value read will be 0
  5. Check Rest Api again -> Value will also now be 0

## Expected behavior
It is expected that a negative value can be set via "deconz.configure" and can be read directly via the "Regulation SetPoint Offset" (0x404B)
Manually setting a negative value at "Regulation SetPoint Offset" (0x404B) is working without any problem and the value is reported correctly to the rest api.

## Screenshots
[image]
[image]
[image]
[image]

## Environment
 - Host system: NAS
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26720700
 - deCONZ version: 2.12.06
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
When setting a negative value the following message appears in the deconz log:
[image]

Maybe just a wrong type?

SwoopX, 2021-11-07:
Lookslike this one is on me. Can you try omitting the trailing 0?

tatec22, 2021-11-07:
If I got you right: There is no difference between -160 and -16. Both values cause the same error message

closed 2021-11-07 by manup