dresden-elektronik/deconz-rest-plugin #5464: Connection issues with Aqara Smart Switch (WXKG12LM)

tilman, 2021-11-07:
## Describe the bug

My Aqara  Smart Switch (WXKG12LM) looses connection after 3-4h of operation. Then I need to re-pair it in order to get it running again (but only for 3-4h again).

## Steps to reproduce the behavior

1.) Pair an WXKG12LM Switch with an conbee 2 stick
2.) Wait some time
3.) Check deConz app. The green line from conbee stick to the smart switch is missing
4.) The blue LED on the smart switch starts to blink (indicating it is not paired)

## Expected behavior

No connection loss

## Screenshots

Debug log shows some connection issues as well
[image]

## Environment

 - Host system: Raspberry Pi, Hass.io
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.12.06 / 19.8.2021
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Max! Cube flashed with culfw and a HDD.

## deCONZ Logs

## Additional context

no router, only aqara devices

Mimiix, 2021-11-07:
How many routers do you have? Can you post a screen of your network?

Mimiix, 2021-11-07:
Also, please share full logs.

suncore, 2021-11-09:
I have the same problem with Xiaomi Smart Switch WXKG01LM. It was OK before I upgraded to 2.12.6 firmware (via the web gui) but now it drops after a while. Considering to downgrade... I think I hade the previous stable version 2.11.5 before.

ctrix, 2021-11-18:
I am in the same situation, since a few months ago. But i have decided to dig into this only yesterday.

I have two Mi Magic Cube MFKZQ01LM devices.

One of them has been in the box since yesterday. The other one has been working for a few months then it stopped working.
I have not related the malfunctioning to a specific release of the firmware, as i don't remember. But i was on the 26660700 firmware too when i noticed the issue.

What happens is that i pair the cube and it works for a few tens of minutes. Then it stops.
In order to rule out a malfunctioning of the cube, i unpacked the brand new one and the same happens. After a while the green line connecting the "configuration tool"in deconz with the cube disappears and the cube requires another pair.
Then back to the malfunctioning after some time.

I have updated today to 26720700 and the same problem happens.

Running Home assistant deCONZ Add-on latest release on an amd64 debian machine.

What can i do to help debugging the regression?

Mimiix, 2021-11-18:
@SwoopX Can you check in? I am not really want this to go into a "Device dropped #metoo" issue 😅 .

closed 2021-12-17 by github-actions[bot]

suncore, 2021-12-17:
I gave up on the Xiaomi button. Instead I used an Ikea on/off button...