dresden-elektronik/deconz-rest-plugin #5470: Windows covering inverted direction for _TZ3000_1dd0d5yi

samsoul63, 2021-11-09:
## Describe the bug
UP and Down command are inverted for _TZ3000_1dd0d5yi 

When calling "Down" command from deconz or through API from home assistant, "lift" is going from 90 to 0 whereas according to the API implementation it should be from 0 to 90 as "lift" represent the percentage of closing : 100 meaning fully closed and 0 fully open.
For up command this is the same issur lift is going from 0 to 100 where it should be reverse.

This produce an issue in Home Assistant described here : https://github.com/home-assistant/core/issues/58651

## Steps to reproduce the behavior

Here is an extract of log when triggering "down" command
2021-11-01 17:18:31 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"3","r":"lights","state":{"bri":228,"lift":90,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"60:a4:23:ff:fe:ff:1b:82-01"}
2021-11-01 17:18:32 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"3","r":"lights","state":{"bri":203,"lift":80,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"60:a4:23:ff:fe:ff:1b:82-01"}
2021-11-01 17:18:33 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"3","r":"lights","state":{"bri":177,"lift":70,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"60:a4:23:ff:fe:ff:1b:82-01"}
2021-11-01 17:18:34 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"3","r":"lights","state":{"bri":152,"lift":60,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"60:a4:23:ff:fe:ff:1b:82-01"}
2021-11-01 17:18:35 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"3","r":"lights","state":{"bri":127,"lift":50,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"60:a4:23:ff:fe:ff:1b:82-01"}
2021-11-01 17:18:36 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"3","r":"lights","state":{"bri":101,"lift":40,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"60:a4:23:ff:fe:ff:1b:82-01"}
2021-11-01 17:18:37 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"3","r":"lights","state":{"bri":76,"lift":30,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"60:a4:23:ff:fe:ff:1b:82-01"}
2021-11-01 17:18:38 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"3","r":"lights","state":{"bri":50,"lift":20,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"60:a4:23:ff:fe:ff:1b:82-01"}
2021-11-01 17:18:39 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"3","r":"lights","state":{"bri":25,"lift":10,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"60:a4:23:ff:fe:ff:1b:82-01"}
2021-11-01 17:18:40 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"3","r":"lights","state":{"bri":0,"lift":0,"on":false,"open":true,"reachable":true},"t":"event","uniqueid":"60:a4:23:ff:fe:ff:1b:82-01"}

## Expected behavior

## Screenshots

## Environment

 - Host system: (NUC)
 - Running method: (Home Assistent deCONZ Add-on)
 - Firmware version: (26680700)
 - deCONZ version: (2.12.06)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes)
 - Is there any other USB or serial devices connected to the host system? If so: Yes rflink

Mimiix, 2021-11-09:
@Smanar

Mimiix, 2021-11-09:
Wont work for this user either, as he is on 12.06.. So need to wait for next stable.

samsoul63, 2021-11-14:
Hello

> like that users can do what they want for the mouting side they want, but for the moment it is not working,.

I don't understand 'the mounting side' part of your reply.
In deed I can invert the wiring of the motor so when I trigger Up command the cover is actually going Up
But my issue is that after a 'up' command API returns a closed status. So I cannot manage this from home assistant (cf. Issue mentionned above).

I will wait for next stable.

Thank you.

Smanar, 2021-11-15:
>I don't understand 'the mounting side' part of your reply.

Don't worry, not specific to you or your device, but we relay need a setting for user can reverse covering themself. For exemple some curtain have 4 mounting mode, and for the same command, same wiring can have different result.
It seem too on some device the calibration can reverse the covering
And sometime for the same device, same model, can have different result.

So I m a little bored to reverse them. If the device is a switch with "up" and "down", or if lift/bri are reversed compared to on/open, we can be sure there is a problem, but for other ....

And ATM DDF are not working enought to be used to reverse a covering.

Can make a PR but some PR are in waiting list since 2 month, DDF will work before.

samsoul63, 2021-11-16:
Hello,
Thank you! this is clear.
I more question : what is DDF ?
Sorry for my ignorance...
Have a nice day

Smanar, 2021-11-16:
DDF are a future deconz feature, you will be able to add/edit/share a JSON file by device, just with a test editor, or using the GUI.
It s starting to work https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5248#issuecomment-946378140

samsoul63, 2021-11-16:
Great feature!
thanks

closed 2021-12-16 by github-actions[bot]

forast3r, 2025-05-16:
Sorry for reviving this old issue. 
Just bought a zigbee cover switch and I am experimenting the same behaviour as the OP.

It is recognized as window cover module, actions from deconz (up, down, stop, opened and closed as well as current lift position are reported right in deconz as well as in phoscon app)

But as stated before, the information (states) and outputs to-from homeassistant are all inverted.

Is there a DDF file where I can check how it is working?

Vendor: _TZ3000_xdo0hj1k
Product: TS130F
Version: unknown

phoscon:
[image]

HA:
[image]

Smanar, 2025-05-17:
Hello, I don't found this device in DDF so it's probably managed by legacy core.
If you send order, it's fine but ONLY return are reversed ? If yes, can be solved by a DDF.

There is request to "reverse" device, but it only work on some device and the result is random
You can see it on the "covering" cluster if you have the GUI else using the API

`curl -H 'Content-Type: application/json' -X PUT -d '{"reverse": true}' http://IP:PORT/api/KEY/lights/ID`

forast3r, 2025-05-18:
Thanks for the help @Smanar !

Double checked again and I've edited my previous reply. Only the lift and open states were inverted. The outputs are right.

Tests made:
1) The **motor reversal** parameter han no effect on states, only swaps the two outputs (by the way the covering switch I bought can control two motors)Checked with voltimeter.
So when **motor reversal**=off, Open command activates relay L1 and Close command activates relay L2.
When **motor reversal**=on, Open command activates relay L2 and Close command activates relay L1.
Therefore this parameter could be useful to invert the outputs without changing electrical connections.
Same happens with 2nd motor ouputs L3 and L4.

2) Modified the auto generated DDF, adding the parse to the lift and open states.
`

[_TZ3000_xdo0hj1k_2gang_covering.json](https://github.com/user-attachments/files/20272278/_TZ3000_xdo0hj1k_2gang_covering.json)

        {
          "name": "state/lift",
          "description": "The lift state of a roller blind.",
          "refresh.interval": 5,
          "read": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 0,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 0,
            "eval": "Item.val = 100 - Attr.val;",
            "fn": "zcl:attr"
          },
          "default": 0
        },

...

        {
          "name": "state/open",
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 0,
            "eval": "Item.val = Attr.val > 1",
            "fn": "zcl:attr"
          },
          "default": false
        },
`
Result: it is working.

3)Also tried reducing the **refresh.interval** from 30 to 5, but the lift status apparently updates when the outputs change (I can live with that)

I have attached the resulting DDF in case someone else wants to use this device.

Smanar, 2025-05-18:
If it don't have impact better to remove " refresh.interval'
When it's not in the DDF deconz use the defaut value, wich is 300 for state/lift.

```
      "report": [
        {
          "at": "0x0008",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
```

In the DDF there is setting for reporting, minimum one report every 300s and maximum 1 by second.
It's natural reporting, made by the device.
The refresh.interval is a poll make by deconz ONLY if the natural report don't work, so this value need to be superior at the value used for natural report.

forast3r, 2025-05-18:
Thank you, removed the refresh.interval parameter