dresden-elektronik/deconz-rest-plugin #5477: Mercator Smart Zigbee Isaac AC Fan Speed Controller Switch

xdss, 2021-11-11:
Fan controls not exposed to Home Assistant, type is On/Off light
`'manufacturername': '_TZ3210_lzqq3u4r',
'modelid': 'TS0501',
'name': 'On/Off light 18',
'state': {'alert': 'none',
'on': False,
'reachable': True,
'speed': 2},
'swversion': None,
'type': 'On/Off light',`
## Device

- Product name: Mercator Smart Zigbee Isaac AC Fan Speed Controller Switch
- Manufacturer: _TZ3210_lzqq3u4r
- Model identifier: TS0501
- Device type : 
  - Light
  - Fan

## Screenshots

[image]

### Basic
[image]

### Identify
[image]

### On/Off
[image]

### Fan Control

### Other clusters that are not mentioned above
[image]
[image]

Mimiix, 2021-11-11:
Some screens are missing .  Also: Make sure to make a forum post.

Mimiix, 2021-11-11:
Also, does this work correctly when you control it with the rest api? https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/lights/#parameters_2

xdss, 2021-11-11:
> Some screens are missing . Also: Make sure to make a forum post.

Sorry, first time, all other clusters appear to be zeros. 

> Also, does this work correctly when you control it with the rest api? https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/lights/#parameters_2

Yes rest api speed control works, 
Should the device type and class be grayed out in basic?

Mimiix, 2021-11-11:
@xdss Ah okay. Than i think it's not a deCONZ issue but a HA issue.

@kane610 If the rest-api works, than this issue is solely for HA is suppose? Can you check allong?

Kane610, 2021-11-11:
This issue originates from https://community.home-assistant.io/t/deconz-official-thread/100504/2752?u=robban

I would expect an endpoint with zhatype="fan" would be created, so in my opinion the implementation on deconz side is lacking.

Mimiix, 2021-11-11:
@SwoopX @manup @ebaauw can you elaborate on this?

xdss, 2021-12-06:
bump, screens all updated.

xdss, 2022-01-04:
still awaiting response

carryonrewardless, 2022-01-29:
Any movement on this?  I have the same device/issue.
Thnx

xdss, 2022-01-30:
> Any movement on this? I have the same device/issue. Thnx

No updates from devs yet, still keen to get this integrated.

carryonrewardless, 2022-02-01:
On this page https://github.com/Koenkk/zigbee2mqtt/issues/1357  some-one has posted 
"You should be able to control the fan by sending to zigbee2mqtt/0x0022a3000016d039/set payload:
{
"fan_mode": "medium"
}"

So we should at least have a work-around but zigbee2mqtt is a package I have not needed previously.

carryonrewardless, 2022-02-22:
Still hoping for some dev work to resolve this.  Is there anything I can provide to help with this e.g. from logs etc.?

carryonrewardless, 2022-03-20:
Still an issue pls keep open for investigation of fans.

Mimiix, 2022-04-11:
@Smanar Can you take a look here :)?

Smanar, 2022-04-11:
There is state/speed to control the Fan speed, it don't work ?

carryonrewardless, 2022-05-02:
So this is a multi-function device incorporating light on/off control and fan speed control.
Is the required dev work to be done in the HA Addon or in the Integration?

Kane610, 2022-05-02:
> So this is a multi-function device incorporating light on/off control and fan speed control.
> Is the required dev work to be done in the HA Addon or in the Integration?

Theoretically it can be done in any of them. Though as the developer of the integration I would very much prefer a bit more clear cut rules than how it is now that any light could potentially have fan controls, thus this issue is not resolved yet.

Otoh it might be that I will add generic fan speed control to lights due to other refactorizations, but that does not mean I'd still not want a clear type to identify the support by.

carryonrewardless, 2022-05-02:
The device is primarily a fan speed controller but it supports those ceiling fans that have an in-built light so the new type would be something like "fan controller with light".

Kane610, 2022-05-12:
While waiting for any other improvement @carryonrewardless you can use the deconz.configure service in Home Assistant to control the fan speeds, check integration documentation for generic suggestions and just swap stuff out according to the deconz rest api documentation for setting fan speeds

xdss, 2022-06-09:
> There is state/speed to control the Fan speed, it don't work ?
> 
> Edit: Ok , have see it s working using the API
> 
> Have see the Kane610 post too [#5477 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5477#issuecomment-966516617)
> 
> But I don't see what to do ... Make a new type ? There is not an ikea fan wip ATM ?
> 
> Edit2:
> 
> Find it #5351 And I don't think someone have planned a new device type.

Still hoping for a permanent solution, I see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5351 has been resolved.

xdss, 2022-07-06:
@Kane610 thanks for your assistance!

Kane610, 2022-07-08:
> @Kane610 thanks for your assistance!

I haven't yet published the change as there where quite some rework of the integration i want to get feedback that warrants patching the current release prior to me publishing breaking changes between dev and 2022.7. I do plan to publish it next week.

Kane610, 2022-07-15:
> @Kane610 thanks for your assistance!

@xdss support has been merged to home assistant dev. You can try it out now

carryonrewardless, 2022-07-15:
@Kane610 Thanks for your work on this!

xdss, 2022-08-09:
> > @Kane610 thanks for your assistance!
> 
> @xdss support has been merged to home assistant dev. You can try it out now

Perfect, tested and working in 2022.8.2. Shows up as a light and fan with speed control.

closed 2022-08-09 by xdss

Kane610, 2022-08-09:
> > > @Kane610 thanks for your assistance!
> > 
> > 
> > @xdss support has been merged to home assistant dev. You can try it out now
> 
> Perfect, tested and working in 2022.8.2. Shows up as a light and fan with speed control.

Thanks for verifying!

carryonrewardless, 2022-08-09:
@Kane610 You say everything is working? 
I only get 2 entities in HA. One for the light on-off and the other for the fan on-off.  There are three buttons so I have no control over fan speed Low/Med/High. What entities do you see?

xdss, 2022-08-09:
I see the same entities, only need two for full functionality. The fan
entity contains state on/off with speed attributes % slider for speed
off/low/med/high

It corresponds to hardware button 1 toggle light on/off, button 2 toggle
fan on/off, button 3 cycle low/med/high

carryonrewardless, 2022-08-09:
Thanks. I had started to work that out. Got it happening with an automation. A slider sounds like a good idea.

carryonrewardless, 2022-08-09:
Thanks again @Kane610