dresden-elektronik/deconz-rest-plugin #5483: Tuya ZigBee 3 Temperature Humdity sensor

JaXx4ever, 2021-11-13:
- Product name: Temperatura & Humidity Sensor
- Manufacturer: _TZ3000_dowj6gyi
- Model identifier: TS0201
- Device type : 
  - Sensor

## Screenshots

[image]

[image]

[image]

[image]

[image]

[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

supersebbo, 2021-11-16:
I have just purchased 4 of these identical sensors!  Bumping this request please!

supersebbo, 2021-11-16:
[image]

Nothing showing in Phoscon and nothing flows through the API to HomeAssistant

supersebbo, 2021-11-16:
Appears they are similar to these: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4517 but with a different manufacturer ID.

Smanar, 2021-11-16:
Someone know if this kind of sensor can work with DDF file ? It don't use tuya cluster so it s a good thing.

supersebbo, 2021-11-17:
@Smanar I would love to edit the DDF and get this set up, but I cannot find the options you mention to edit the DDF. 

What version of deCONZ are you using?  I'm using 2.12.6 that is in the HomeAssistant plugin.

Smanar, 2021-11-17:
> I'm using 2.12.6 that is in the HomeAssistant plugin.

It's for that, you need to wait for a better version in HA.

julidau, 2021-11-28:
just fyi, PR  #5299 should add support for this device

manugparedes, 2021-11-28:
I installed last stable version, delete node and search again, but nothing.

[image]

[image]

Should I do anything more?

julidau, 2021-11-28:
Since this pr is not merged at this moment you would have to build the branch from source.

Sorry if i caused confusion. I am also not part of the developers which means i cannot do that.

Smanar, 2021-11-28:
The PR is not yet validated.
But you can try with DDF https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5483#issuecomment-970582252 if your deconz version permit that.

manugparedes, 2021-11-28:
> The PR is not yet validated. But you can try with DDF [#5483 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5483#issuecomment-970582252) if your deconz version permit that.

Yes!! Thank you! I thinked that deconz version post was special, and only was beta... sry for my confussion. Installed in my Raspberry Pi 3 from [instructions](https://phoscon.de/en/raspbee2/install#raspbian)

I got edit ddf and device now appears in my Phoscon app, but values are 0.

[image]

I tryed set temperature Parse as zcl with same result (zero in phsocon app).

[image]

Any ideas?

Smanar, 2021-11-29:
Yep, probably missing the binding part on the second tab.
You have take a look here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet ?
For the temperature you need to drag&drop the cluster 0x0402, and the attribute 0x0000
For the setting the standards one are

```
            rq.minInterval = 10;
            rq.maxInterval = 300; 
            rq.reportableChange16bit = 20;
```

manugparedes, 2021-12-01:
Thanks!! But I think that we get small step.

[image]

Now, node name and battery status appears in deconz, but phoscon values are yet 0.

[image]

[image]

Smanar, 2021-12-02:
Even after some time ? You have used "hot reload" ?
Can you show the complete DDF pls ? Tab "Preview"

Perhaps something in logs ? (in deconz / help / debug view, with "info" and "info_l2")

manugparedes, 2021-12-03:
First! Thank you very much your help.

Second: Yes, I have used "Hot Reload" and have triyed remove and added too.

I cant show you all debug log because I have tons of traces (50 devices aprox). I found traces searching by MAc Address

20:15:44:185 ZCL attribute report 0xA4C138091016F512 for cluster: 0x0402, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:15:44:189 discard double entry in binding queue (size: 1) for for 0xA4C138091016F512, cluster 0x0001
20:15:44:191 discard double entry in binding queue (size: 1) for for 0xA4C138091016F512, cluster 0x0001
20:15:44:370 ZCL attribute report 0xA4C138091016F512 for cluster: 0x0405, ep: 0x01, frame control: 0x08, mfcode: 0x0000 

...

20:35:44:088 ZCL attribute report 0xA4C138091016F512 for cluster: 0x0402, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
20:35:44:091 discard double entry in binding queue (size: 1) for for 0xA4C138091016F512, cluster 0x0001
20:35:44:094 discard double entry in binding queue (size: 1) for for 0xA4C138091016F512, cluster 0x0001

This is DDF content:

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_dowj6gyi",
  "modelid": "TS0201",
  "product": "TS0201",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/ts0201.json",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "refresh.interval": 120,
          "read": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 1,
            "eval": "Item.val = Attr.val * 100",
            "fn": "zcl"
          }
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "refresh.interval": 120,
          "read": {
            "at": "0x0000",
            "cl": "0x0405",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0405",
            "ep": 1,
            "eval": "Item.val = Attr.val * 100",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_BATTERY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0001"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/battery",
          "refresh.interval": 3602,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 10,
          "max": 300,
          "change": "0x00000014"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 10,
          "max": 300,
          "change": "0x00000014"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 10,
          "max": 300,
          "change": "0x00000014"
        }
      ]
    }
  ]
}

timmriesen, 2021-12-04:
Hi Smanar et al.,
Thank you for your work and support. I have similar Tuya sensors in use with slightly different ending (_TZ3000_i8jfiezr) and read all related postings here (including the ones linking this thread). I can confirm that after installing deconz-beta, and working with ddf files (adding ZHATemperature and ZHAHumidiy, and doing bindings as decribed by Smanar), I have the exact same result as described by manugparedes two posts above: I can see the sensor now in phoscon, but the values for temperatures and humidity are both 0.

As Smanar asked for logfile output, I add it here below, hoping that _TZ3000_i8jfiezr is representative for _TZ3000_dowj6gyi.

The questions that I currently have are:

1. Is it possible to fix the DDF to get the sensors to work with my current installation of deconz-beta?
2. Is the pull request already included in the tarball, would the steps described in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5292 already solve my issue?

Again, many thanks, this is cool stuff, and it is fun being part of such a community.

DDF-Preview
`{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_i8jfiezr",
  "modelid": "TS0201",
  "product": "TS0201",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/ts0201tr.json",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
[... 152 lines cut ...]
17:47:50:485 Force binding of attribute reporting for node TRÅDFRI an/aus Taster
17:47:51:147 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
17:47:51:483 Idle timer triggered
17:47:51:484 Force read attributes for ZHATemperature SensorNode Temperature 6
17:47:51:484 Force binding of attribute reporting for node Temperature 6
17:47:52:483 Idle timer triggered
17:47:52:483 Force binding of attribute reporting for node Configuration tool 1
17:47:52:484 Force binding of attribute reporting for node TRÅDFRI an/aus Taster
17:47:52:983 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
17:47:53:482 Idle timer triggered
17:47:53:483 Force read attributes for ZHATemperature SensorNode Temperature 6
17:47:53:484 Force binding of attribute reporting for node Temperature 6
17:47:54:482 sql exec SELECT conf FROM zbconf ORDER BY rowid desc limit 1
17:47:54:484 Idle timer triggered
17:47:54:484 Force binding of attribute reporting for node TRÅDFRI an/aus Taster
17:47:54:508 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
17:47:55:483 Idle timer triggered
17:47:55:484 Force read attributes for ZHATemperature SensorNode Temperature 6
17:47:55:484 Force binding of attribute reporting for node Temperature 6
17:47:56:483 Idle timer triggered
17:47:56:483 Force binding of attribute reporting for node TRÅDFRI an/aus Taster`

[image]
[image]
[image]

Smanar, 2021-12-04:
All look fine for me, there is a flag "DDF" for logs, no ?

Smanar, 2021-12-04:
@timmriesen your dddf is not finished, you miss the more important part.

- Select the state/temperature and state/humidity
- Configure them according to capture here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5483#issuecomment-983936409
- You need to have on your preview something like 

```
"name": "state/humidity",
"refresh.interval": 120,
"read": {
"at": "0x0000",
"cl": "0x0405",
"ep": 1,
"fn": "zcl"
},
"parse": {
"at": "0x0000",
"cl": "0x0405",
"ep": 1,
"eval": "Item.val = Attr.val * 100",
"fn": "zcl"
}
```

But idk yet why it's not working, perhaps with DDF flag in logs

timmriesen, 2021-12-04:
>your ddf is not finished, you miss the more important part.

@Smanar many thanks for pointing out the missing _read_ and _parse_ settings. I followed your instructions and the DDF includes now the following 2 sections for state/humidity and state/temperature:

SELECTED SECTIONS FROM THE DDF FILE:

          {
          "name": "state/humidity",
          "refresh.interval": 120,
          "read": {
            "at": "0x0000",
            "cl": "0x0405",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0405",
            "ep": 1,
            "eval": "Item.val = Attr.val * 100",
            "fn": "zcl"
          }
( . . . )

          {
          "name": "state/temperature",
          "refresh.interval": 120,
          "read": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 1,
            "eval": "Item.val = Attr.val * 100",
            "fn": "zcl"
          }

**I did a hot reload, but for the moment, the result in phoscon is unchanged, the values show 0 for both temperature and humidity.**

>This PR solve a problem with the device mac adress not in whitelist. The DDF core not take care of the Mac adress.
If your device MAC adress don't start by 0xa4c138 this PR will change nothing for you.

My MAC Address is 0xa4c138f3c10d8cd4 so it does indeed start with 0xa4c138.

- Does this explain, why PHOSCON is not finding the device while pairing?
- Does this explain the zero values?

I add some screenshots showing the new confic and the MAC address below. Thank you very much!

[image]
[image]
[image]

Smanar, 2021-12-05:
```
My MAC Address is 0xa4c138f3c10d8cd4 so it does indeed start with 0xa4c138.

    Does this explain, why PHOSCON is not finding the device while pairing?
    Does this explain the zero values?
```

Yes, but not the zero value.

From manup it need to work, the only thin I m seeing different than a working mode is

```
"max": 300,
"change": "0x00000014"
}
```

and 

```
"max": 300,
"change": "0x14"
}
```

But I don't think it make a difference

Nothing in logs (with ddf flag too) about a missed binding ?

manugparedes, 2021-12-05:
Ouh yeah! Murphy's law! Last try was correct!

Thank you @Smanar!

Finally I did this steps:
* Applied "sleeper"
* The best: "awake on receive"
* Removed arithmetic operations for temp and humidity

I let here ddf for helping:

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_dowj6gyi",
  "modelid": "TS0201",
  "vendor": "Tuya",
  "product": "TS0201",
  "sleeper": true,
  "status": "Bronze",
  "path": "/devices/ts0201.json",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true,
          "refresh.interval": 120,
          "read": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true,
          "refresh.interval": 120,
          "read": {
            "at": "0x0000",
            "cl": "0x0405",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0405",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_BATTERY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0001"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/battery",
          "awake": true,
          "refresh.interval": 3602,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 1,
          "max": 6000,
          "change": "0x00000014"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 10,
          "max": 300,
          "change": "0x00000014"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 10,
          "max": 9999,
          "change": "0x00000014"
        }
      ]
    }
  ]
}
```

Smanar, 2021-12-06:
>The best: "awake on receive"

So this setting is important too ?

timmriesen, 2021-12-09:
I can confirm that the DDF version posted by manugparedes works for my sensor too (after editing the model string). But "awake on receive" is not the only difference from our files, I also lacked the ZHABattery part. If I remove this, the sensors show 0 only, so it is not yet clear, whether if work without the awake on receive as well. In manus bind sections of the DDF files I see three different paris of values for "min" and "max". Is this just for testing, or does has to be exactly like this?

I will try more test regarding ZHABattery and "awake on receive" with my DDF and report back.

Smanar, 2021-12-10:
>I realized also, that this can be added in different ways

Yep, one by state for exemple.

Min is the minimum time beetween 2 reports, and max the maximimum.

timmriesen, 2021-12-11:
>Min is the minimum time beetween 2 reports, and max the maximimum.
Thank you.
There are a few basic things, that I might misunderstand, so maybe one of you has some insight:
- Are these sensors typically sending temperature, humidity, and battery level at the same time? 
- Are these sensors actively triggered to send (let's call it polling for now), or are they sending their reports at certain events (like a temperature difference largen than X) or max time without sending?

And dependent on the answers above, how are the min and max settings as decribed changing the behavior?

Again thank you very much for your support.

Smanar, 2021-12-11:
>Are these sensors typically sending temperature, humidity, and battery level at the same time?

I think no, some brand make a periodic report with all values in same time, but not all values change in same time, and more  a value change, more you wil have report.

>Are these sensors actively triggered to send (let's call it polling for now), or are they sending their reports at certain events (like a temperature difference is becoming larger than X) or a certain max time without sending?

They are using report, polling on sensor is not realy usefull (they can "hibernate")

>And dependent on the answers above, how are the min and max settings as decribed changing the behavior?

Lower value make more report but decrease battery life. Values I have give are the one used by HUE bridge (found on deconz code)

timmriesen, 2021-12-15:
>Lower value make more report but decrease battery life. Values I have give are the one used by HUE bridge (found on deconz code)

I guess what I don't understand is why there is max setting and min settings. Is the sensor generating reports in a fequency that depends on the DDE file, if yes, does it take min or max, or both?

Smanar, 2021-12-15:
Both (or third for some attributes)
The device try to make at least 1 report every "max" value (but can make on before), but wait at least for the "min" value without report.
And some attributes have too a value for "change", it s the difference value that  can trigger a report.

timmriesen, 2021-12-19:
I have a new problem :(. I restarted the computer, on which deCONZ and Conbee Stick are. After running deCONZ again, it comes up with DDF support deactivated, which I can swich back on using the Control tab. However, all sensors report now 0 values again, even though I did not change anything with the configuration. Hot reload of DDF does not help. restarting deCONZ and Phoscon does not help. What did I overlook? I can not afford to delete the nodes and pair them again, each time my deCONZ restarts. Is there other solution?

Smanar, 2021-12-19:
Yep can you show the uniqueid of the device pls ?

Have you tried to set the status to "gold" ?

Remember you can need 1 hour before the first update from those sensors.

timmriesen, 2021-12-19:
>Yep can you show the uniqueid of the device pls ?
 Are these the MAC Addresses? I attach an image with all the sensors listed.

> Have you tried to set the status to "gold" ?
It was on Bronze so far, I now try with gold

>Remember you can need 1 hour before the first update from those sensors.
Yes, I tried over night (total more than 3 days now)

Again, thanks for your help!

[image]

Smanar, 2021-12-20:
>Are these the MAC Addresses? I attach an image with all the sensors listed.

Nope, you can find it on a log, or can use phoscon with help / API information, it s realy called "uniqueid". I have an issue ATM where the uniqueid use 0x01 for the working device and 0xff for the not working.

To check if it's not a bind/report issue, go on the cluster 0x0402, and read the attribute 0x0000 (this one with the temperature) and check if it have updated the value in the API.

timmriesen, 2021-12-22:
>Nope, you can find it on a log, or can use phoscon with help / API information, it s realy called "uniqueid". I have an issue ATM where the uniqueid use 0x01 for the working device and 0xff for the not working. It s something composed by the Mac adress + Endpoint + cluster

Ok, using the Filter for "Sensors" I find the original 15 entries, (3 per harware sensor, that each comes with temperature (0402), humitidy (0405), and battery (0001):
- Sensor 11: "uniqueid": "a4:c1:38:f3:c1:0d:8c:d4-ff-0402"
- Sensor 12: "uniqueid": "a4:c1:38:f3:c1:0d:8c:d4-ff-0405"
- Sensor 13: "uniqueid": "a4:c1:38:f3:c1:0d:8c:d4-ff-0001"
- Sensor 14: "uniqueid": "a4:c1:38:97:0b:a2:2e:ef-ff-0402"
- Sensor 15: "uniqueid": "a4:c1:38:97:0b:a2:2e:ef-ff-0405"
- Sensor 16: "uniqueid": "a4:c1:38:97:0b:a2:2e:ef-ff-0001"
- Sensor 17: "uniqueid": "a4:c1:38:3d:d6:dc:2b:e9-ff-0402"
- Sensor 18: "uniqueid": "a4:c1:38:3d:d6:dc:2b:e9-ff-0405"
- Sensor 19: "uniqueid": "a4:c1:38:3d:d6:dc:2b:e9-ff-0001"
- Sensor 20: "uniqueid": "a4:c1:38:10:5e:e6:33:bb-ff-0402"
- Sensor 21: "uniqueid": "a4:c1:38:10:5e:e6:33:bb-ff-0405"
- Sensor 22: "uniqueid": "a4:c1:38:10:5e:e6:33:bb-ff-0001"
- Sensor 23: "uniqueid": "a4:c1:38:55:3c:db:8d:b1-ff-0402"
- Sensor 24: "uniqueid": "a4:c1:38:55:3c:db:8d:b1-ff-0405"
- Sensor 25: "uniqueid": "a4:c1:38:55:3c:db:8d:b1-ff-0001"

Above, is what was before the reboot of deCONZ and what I expected.
After the reboot, I have now another set of 15 entries. It seems that the sensor have been replicated with new numbers: 

- Sensor 27: "uniqueid": "a4:c1:38:f3:c1:0d:8c:d4-01-0402"
- Sensor 28: "uniqueid": "a4:c1:38:f3:c1:0d:8c:d4-01-0405"
- Sensor 29: "uniqueid": "a4:c1:38:f3:c1:0d:8c:d4-01-0001"
- Sensor 30: "uniqueid": "a4:c1:38:97:0b:a2:2e:ef-01-0402"
- Sensor 31: "uniqueid": "a4:c1:38:97:0b:a2:2e:ef-01-0405"
- Sensor 32: "uniqueid": "a4:c1:38:97:0b:a2:2e:ef-01-0001"
- Sensor 33: "uniqueid": "a4:c1:38:10:5e:e6:33:bb-01-0402"
- Sensor 34: "uniqueid": "a4:c1:38:10:5e:e6:33:bb-01-0405"
- Sensor 35: "uniqueid": "a4:c1:38:10:5e:e6:33:bb-01-0001"
- Sensor 36: "uniqueid": "a4:c1:38:55:3c:db:8d:b1-01-0402"
- Sensor 37: "uniqueid": "a4:c1:38:55:3c:db:8d:b1-01-0405"
- Sensor 38: "uniqueid": "a4:c1:38:55:3c:db:8d:b1-01-0001"
- Sensor 39: "uniqueid": "a4:c1:38:3d:d6:dc:2b:e9-01-0402"
- Sensor 40: "uniqueid": "a4:c1:38:3d:d6:dc:2b:e9-01-0405"
- Sensor 41: "uniqueid": "a4:c1:38:3d:d6:dc:2b:e9-01-0001"

Phoscon only lists the sensors from the first set (Temperature 11, Temperature 14, Temperature 17, Temperature 20 and Temperature 23)

>To check if it's not a bind/report issue, go on the cluster 0x0402, and read the attribute 0x0000 (this one with the temperature) and check if it have updated the value in the API.

In deCONZ (right panel with graphical representation of the sensors), I select sensor with title "Temperature 11" and click on right most circle. On the context menu, I click on 0402 (temperaure measurement). In the left panel of deCONZ, I select "Cluster Info" and mark attribute 0x0000. It already shows the correct temperature (e.g. 2272). I click read. The temperature does not change (this type of sensor gets reports, and cannot be actively read, right?). However, putting the sensor to a warm place generates next report with higher temperature which shows correclty in Cluster Info.

Now I check Phoscon: Still shows 0 values. 
In the API, I can filter for sensors and find the correct values when I look with the new numbers, here entry "Temperature 27" instead of original "Temperature 11":
                    {
    "config": {
        "battery": 87,
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "etag": "b49ac7c831a8d013c297f24bfed2050c",
    "lastannounced": "2021-12-22T03:35:45Z",
    "lastseen": "2021-12-22T14:37Z",
    "manufacturername": "_TZ3000_i8jfiezr",
    "modelid": "TS0201",
    "name": "Temperature 27",
    "state": {
        "lastupdated": "none",
        "temperature": 2363
    },
    "type": "ZHATemperature",
    "uniqueid": "a4:c1:38:f3:c1:0d:8c:d4-01-0402"
}

Using Node-Red, and reading the objects that the deCONZ adapter generates, I can use the new sensor numbers (27-41) to get correct values for Temperature, Humitidy and Battery, the old numbers (11-25) do no longer work and show 0 values.

My preliminary conclusions:
- Somehow the reboot generated new internal objects for the sensors, now with numbers 27-41)
- Phoscon still polls the old values (11-25) and as these are 0 and no longer updated, displays 0 values
- The API has 2 sets of sensors, the old "stale" set with 0 values, which are not updated, and the new set that works
- deCONZ has a mixed state, it shows only 5 sensors with the old numbering / old names (e.g. Temperature 11) BUT it connects to the right values internally, as the cluster info for "Temperture 11" reports in fact the temperature for sensor 27.

I hope these observations help in finding out, what is going on.

Smanar, 2021-12-22:
Yep, it s exactly the same issue I have here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5459#issuecomment-991744990

Generally if you have "ff" in the unique id, there is an issue 

>Sensor 20: "uniqueid": "a4:c1:38:10:5e:e6:33:bb-ff-0402"

You can set the DDF status to gold as workaround for the moment.

timmriesen, 2021-12-30:
>You can set the DDF status to gold as workaround for the moment.

I had it on Bronze before, what changes with Gold? Is this the to load DDF at boottime of deCONZ?

>manup added this to the v2.14.0-beta milestone 8 days ago

Will I get this automatically once it is released, with apt-get update (I am currently on v2.13.2-beta, using the beta repository)

Smanar, 2021-12-30:
>I had it on Bronze before, what changes with Gold? Is this to load DDF at boottime of deCONZ or is there another way to achieve that?

Depend of setting but generaly setting the status to gold, force the use of the DDF even at start. This issue appear on start when DDF are disabled.

>Will I get this automatically once it is released, with apt-get update (I am currently on v2.13.2-beta, using the beta repository)

Yes, on beta branch, you will have it.

closed 2022-02-10 by github-actions[bot]

Smanar, 2022-07-07:
The code was merged 21h ago ^^
So not yet, but will be in next version for sure

Value are in seconds

"refresh.interval" is the polling, "report" is the reporting.
The polling is used when the report don't work, so always use bigger value for it.