dresden-elektronik/deconz-rest-plugin #5484: Config/offset not working for some Tuya TRV

Smanar, 2021-11-14:
## Describe the bug
Some Tuya TRV like
- _TZE200_aoclfnxz
- _TZE200_ckud7u2l 
- _TZE200_cwnjrr72 

Display a bad temperature.
They have an offset setting but not/bad working

## Steps to reproduce the behavior
Just use the device

## Expected behavior

## Screenshots

## Environment
- deconz version : v2.13.1-beta
- 
## deCONZ Logs

## Additional context

Smanar, 2021-11-14:
To test code, you have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch tuya_offset https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2"

Value need to be *100, so for an offset of 2 °C use 200.
If it don't work can you try to set the value manualy (if the device support it) and give deconz logs (with "info" and "info_L2") during the manipulation.

Or at least logs when doing the request, to see how the device react to it.

cwm77, 2021-11-14:
Ok, I now loaded the compiled plugin and tried to set the offset. In the display I could see that the current temperature increased by 1 degree Celsius, although I set -400. Other values did not lead to any further adjustment in the display. Even a value of 0 did not. Attached is the log file.

[50:32:5f:ff:fe-1.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7533696/50.32.5f.ff.fe-1.txt)

Smanar, 2021-11-14:
For the "auto" mode there is a PR, idk if it concern your device https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5471

For the device
`{"_TZE200_aoclfnxz", "TS0601", "Moes", "Tuya_THD BTH-002 Thermostat"},`

From the actual code, the "preset" support "auto" and "program" and for "mode" only "off" and "heat".
This device haven't the mode set in the code, from my memory the "mode" is only used to enable/disable the display.

>13:13:31:470 Send Tuya request 0x50325FFFFE289F7F : Dp_type: 0x02, Dp_identifier 0x1B, data: 00000000

Something is bad in the code, data are always 00000000 even you are using -450 and - 430.

Edit:
Ok have found the issue, all offsets was broken.
IDK if it will work, but this time, you will have other value than 00000

To update the code

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Smanar, 2021-11-15:
>Interesting is that instead of value 100 = 1 degree Celsius now value 10 = 1 degree Celsius has to be taken

It mean you are using config/offset = 10 to add 1 °C ? If yes I m wrong in my convertion, need to keep the same convertion for all value (need to use 100 for 1°C)

>Increments of 0.5 degrees Celsius are not taken over

Yep, but it's from the device.

> for an offset of -4 degrees C -0.4 degrees C is displayed, although the desired offset of 4 degrees remains on the thermostat.
You mean 4 °C on the TRV but 0.4 on the application ? (probably because my convertion error ?)

For the missing "auto" I think it s normal for this device, on the z2m they are doing same than us, they use only "off" and "heat"
https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html
`RConfigModeValuesTuya2 = { { {QLatin1String("off"), {0x00}}, {QLatin1String("heat"), {0x01}} } };`

>You can only adjust the heatsetpoint for thermostat via ioBroker

So it mean it work or not ? ioBroker use the API too ?
I have checked the code again and for exemple the locking need to work too, you have some logs when making the request ?

Smanar, 2021-11-16:
First log, but you have used 400 not 40 ? (in the API I mean, as it was the value updated in the websocket return)
```
13:21:27:133 Send Tuya request 0x50325FFFFE289FB3 : Dp_type: 0x02, Dp_identifier 0x1B, data: 00000004
13:21:27:155 Websocket 192.168.4.118:58996 send message: {"config":{"heatsetpoint":2100,"locked":false,"mode":"heat","offset":400
```

But later
```
13:21:28:395 Tuya debug 4 : Address 0x50325FFFFE289FB3 Payload 009e1b02000400000004
13:21:28:396 Tuya debug 5 : Status: 0 Transid: 158 Dp: 539 (0x02,0x1B) Fn: 0 Data 4
13:21:28:426 Websocket 192.168.4.118:58996 send message: {"config":{"heatsetpoint":2100,"locked":false,"mode":"heat","offset":40
```

So the return need to be * 10, But on the code I m doing `temp = temp * 100;` How 4 *100 can be 40 ....
You are sure you are using the last version ?

```
                    case 0x021B : // temperature calibration (offset in degree) for Moes
                    {
                        qint16 temp = static_cast<qint16>(data & 0xFFFF);
                        
                        if (temp > 2048)
                        {
                            temp = temp - 4096;
                        }
                        
                        temp = temp * 100;
                        
                        ResourceItem *item = sensorNode->item(RConfigOffset);

                        if (item && item->toNumber() != temp)
                        {
                            item->setValue(temp);
                            Event e(RSensors, RConfigOffset, sensorNode->id(), item);
                            enqueueEvent(e);
                        }
                    }
```

On the Second log, you never make the zigbee request, so the API have probably send an error.

cwm77, 2021-11-16:
https://user-images.githubusercontent.com/9320205/142074578-6c005234-1309-44b8-bebb-8d6da9768541.mp4

[50:32:5f:ff:fe:28:9f:b3-01-0201.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7550188/50.32.5f.ff.fe.28.9f.b3-01-0201.txt)

Smanar, 2021-11-17:
Lol, first time someone make a video to debug ^^.

I m not guilty for all your issues, I think some of them are from the application, like the locked command, if you can't intercept the http request, you can see it in deconz if you enable the flag "http" but take care this flag is realy talkative.
Sometime the value is written in red, it s don't mean there was an error ?

For the offset issue it s still a mystery, can you make atry with the last version (just to have log), I have added some more debug line.

The variable temp is a quint, so can' be a decimal value, and *100, I realy don't see how i can have a value < 100.

cwm77, 2021-11-17:
I have now tried it with the latest version. The log file contains the information about what I did with the thermostat with ID 50:32:5f:ff:fe:28:9f:b3-01-0201.

[50325ffffe289fb3-01-0201.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7556761/50325ffffe289fb3-01-0201.txt)

Smanar, 2021-11-17:
Ok I give up, have made a *1000 without understand, hopping it work now.

crabmanX, 2021-11-17:
I just tested with df391951aa6c8e3f9e2f754170718507a512ebed and a _TZE200_ckud7u2l and can confirm the offset works as expected. `curl` -X PUT -i http://xxx:8082/api/yyy/sensors/15 --data '{  "config": { "offset": 100 } `}'`  increases the temperature by 1°C (Displayed in the phoscon API Information and home assistant).

cwm77, 2021-11-17:
I have applied the changes and tested them. First with an offset of 0 then with -300. From my point of view nothing has changed. It is still possible to set an offset and it is taken over by the thermostats. Enclosed the logfile for 50:32:5f:ff:fe:28:9f:b3-01-0201.

[50:32:5f:ff:fe:28:9f:b3-01-0201.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7558525/50.32.5f.ff.fe.28.9f.b3-01-0201.txt)
01

Smanar, 2021-11-18:
Find it ^^
Was because the missing "break" so th code execute the 0x021B part AND the 0x022c one.

This error message can be produced if ioBroker  use "-300" instead of 300

cwm77, 2021-11-18:
Hey, great. Can the problem be solved, or does the error originate from the hardware, i.e. the firmware would have to be adapted?

Smanar, 2021-11-19:
For the error in ioBroker  ? Try to ask to their devs (If it's not stupid as answer ? I realy don't know what is it).
But they make a bad request, malformed one.

But if it work on deconz side, I can make the PR ? Even it s not tested for all devices.

cwm77, 2021-11-19:
Ok, I'll ask the ioBroker devs about this when I find time. Thanks for everything and yes, from my point of view a PR is a good next step.

Smanar, 2021-11-20:
The PR is done, if someone still have not working device that support offset ...

embeddeddev, 2021-11-23:
I can confirm this bug of not remotely applied offset in the Hama TRV aka "_TZE200_yw7cahqs", too.
Is this even included?

The device was integrated w/ this device request (detailed infos if required):
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4841

Smanar, 2021-11-24:
I have just modified something for your device.
Can you make a try with the new code ?

If it s still not working, can you share logs when making the command ?

embeddeddev, 2021-11-24:
Thanks a lot so far. Got it to work in my docker and tested it w/ following results.

First sending an offset request, the device connection crashed. Unavailable in Home Assistant, in deCONZ GUI via VNC not reachable any more. Device connection was reestablished when I manually changed the target temperature at the thermostat itself.

After that I digged a bit deeper:
- The offset was changed afterwards in REST frontend
- Sending another offset change request to ``0`` correctly changed offset back.

**But:**
The thermostat delivers no current temperatures to frontends any more. In Home Assistant it shows up as ``null`` value.
So I cannot validate, if the device really accepted/received the value and notified back the adjusted current temperature.

Finding the corresponding log lines for reading (?) the node was not as easy.
I hope, you can deal w/ it. (Log timestamps in CET)

```
22:21:48:882 Tuya debug Request : Address 0xCC86ECFFFEA0FDD2, Endpoint 0x01, Command 0x01, Payload 000367020004000000be
22:21:48:883 Tuya debug 4 : Address 0xCC86ECFFFEA0FDD2 Payload 000367020004000000be
22:21:48:884 Tuya debug 5 : Status: 0 Transid: 3 Dp: 615 (0x02,0x67) Fn: 0 Data 190
.........
22:21:50:162 Tuya debug Request : Address 0xCC86ECFFFEA0FDD2, Endpoint 0x01, Command 0x01, Payload 000367020004000000a0
22:21:50:163 Tuya debug 4 : Address 0xCC86ECFFFEA0FDD2 Payload 000367020004000000a0
22:21:50:164 Tuya debug 5 : Status: 0 Transid: 3 Dp: 615 (0x02,0x67) Fn: 0 Data 160
22:21:50:167 Websocket 172.16.0.8:47078 send message: {"config":{"heatsetpoint":1600,"locked":true,"mode":"heat","offset":-210,"on":true,"preset":null,"reachable":true,"schedule":{"W124":[{"heatsetpoint":0,"localtime":"T17:18"},{"heatsetpoint":0,"localtime":"T00:00"}],"W3":[{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":2,"localtime":"T00:01"},{"heatsetpoint":0,"localtime":"T17:18"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"}]},"schedule_on":false,"setvalve":false,"windowopen_set":false},"e":"changed","id":"47","r":"sensors","t":"event","uniqueid":"cc:86:ec:ff:fe:a0:fd:d2-01-0201"} (ret = 779)
```

Some basic cluster retrieval logs (?):
```
22:25:28:247 Node data 0xcc86ecfffea0fdd2 profileId: 0x0104, clusterId: 0x0000
22:25:28:248 Update Sensor 0xCC86ECFFFEA0FDD2 Basic Cluster
22:25:28:259 Websocket 172.16.0.8:47078 send message: {"attr":{"id":"47","lastannounced":"2021-11-24T21:13:31Z","lastseen":"2021-11-24T22:25Z","manufacturername":"_TZE200_yw7cahqs","modelid":"TS0601","name":"therm_schlafz","productid":"Tuya_THD Smart radiator TRV","swversion":null,"type":"ZHAThermostat","uniqueid":"cc:86:ec:ff:fe:a0:fd:d2-01-0201"},"e":"changed","id":"47","r":"sensors","t":"event","uniqueid":"cc:86:ec:ff:fe:a0:fd:d2-01-0201"} (ret = 392)
```

And even for restoring/writing (?) the zero value:
```
22:26:04:611 Send Tuya request 0xCC86ECFFFEA0FDD2 : Dp_type: 0x02, Dp_identifier 0x2C, data: 00000000
22:26:04:611 add task 4283 type 37 to 0xCC86ECFFFEA0FDD2 cluster 0xEF00 req.id 84
22:26:04:613 Websocket 172.16.0.8:47078 send message: {"config":{"heatsetpoint":2000,"locked":true,"mode":"heat","offset":0,"on":true,"preset":null,"reachable":true,"schedule":{"W124":[{"heatsetpoint":0,"localtime":"T17:18"},{"heatsetpoint":0,"localtime":"T00:00"}],"W3":[{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":2,"localtime":"T00:01"},{"heatsetpoint":0,"localtime":"T17:18"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"}]},"schedule_on":false,"setvalve":false,"windowopen_set":false},"e":"changed","id":"47","r":"sensors","t":"event","uniqueid":"cc:86:ec:ff:fe:a0:fd:d2-01-0201"} (ret = 776)
```

embeddeddev, 2021-11-24:
Thanks so far for the quick response and adaption!
In my previous comment I'd mistakenly used the original plugin in my docker image, so I deleted it directly.

Need to spend more time on building a fresh docker image tomorrow, which doesn't seem to be so easy.

embeddeddev, 2021-11-25:
Finally got it.
Started w/ a rejoined thermostat as it was reporting only ``null``/``NaN`` values before for ``current_temperature``.
In my setup I'm using Home Assistant as frontend for viewing device state and issuing REST commands.

1. Set offset to 100
  --> value kept persistent in device attributues
  --> but _no_ reaction/change of ``current_temperature``
2. Set offset to 50
  --> value kept for a few moments but was resetted to 0 after a few moments and vanished from device atttributes (since 0)
  --> _no_ reaction on ``current_temperature``

Here the log sequence grepped by device MAC + Tuya keyword.
I hope it's OK so, because my network is very noisy.
[tuya_offset_hama.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7599346/tuya_offset_hama.log)

embeddeddev, 2021-11-25:
You're right.
I tried about an hour to detect, that the device communication was really broken and TRV didn't report any changes until I repaired it a few times and resetted it in deCONZ GUI finally.
Thanks!

Smanar, 2021-11-26:
And offset is working ?

embeddeddev, 2021-11-26:
Offset was working w/ your branch after resetting the Hama TRV node.

Smanar, 2021-11-27:
@cwm77, if you have some time, can you try the actual code with positive and negative values pls ?

cwm77, 2021-11-28:
Hi, I'm trying to compile the new plugin but I get the following error. 

alarm_system.cpp:13:10: fatal error: deconz/timeref.h: file or directory not found

Smanar, 2021-11-28:
Perhaps 
`sudo apt install deconz-dev`

cwm77, 2021-11-28:
Thanks, that's it. I don't do it that often. Now, I have once set the offset with 400 and then with -400, and it works perfectly. In iobroker I see now also no more the 10s, but the 100 values (Instead of 40 now 400 ...).

Here is the logfile.

[50325ffffe289fb3-01-0201.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7613491/50325ffffe289fb3-01-0201.txt)

Smanar, 2021-11-28:
```
13:00:53:017 Send Tuya request 0x50325FFFFE289FB3 : Dp_type: 0x02, Dp_identifier 0x1B, data: 00000004
3:00:53:044 Websocket 192.168.4.118:36178 send message: {"config":{"heatsetpoint":1800,"locked":false,"mode":"heat","offset":400
13:00:53:601 Tuya debug 4 : Address 0x50325FFFFE289FB3 Payload 00c21b02000400000004
13:00:53:605 Tuya debug 5 : Status: 0 Transid: 194 Dp: 539 (0x02,0x1B) Fn: 0 Data 4
```

```
13:01:01:944 Send Tuya request 0x50325FFFFE289FB3 : Dp_type: 0x02, Dp_identifier 0x1B, data: 00000ffc
13:01:01:976 Websocket 192.168.4.118:36178 send message: {"config":{"heatsetpoint":1800,"locked":false,"mode":"heat","offset":-400,
13:01:02:931 Tuya debug 4 : Address 0x50325FFFFE289FB3 Payload 00c91b02000400000ffc
13:01:02:937 Tuya debug 5 : Status: 0 Transid: 201 Dp: 539 (0x02,0x1B) Fn: 0 Data 4092
```

Ok so from code it's working, on the device too ? I m asking because something was not working previously.
For positive value you send exactly the value, but for negative one the value is changed.

cwm77, 2021-11-28:
Yes, it also works directly on the thermostat. The indication of the current temperature is adjusted accordingly on the display.

Because of the problem that the offset cannot be set in ioBroker I have not yet had time to report this to the devs. I hope to get to it in the next few days.

closed 2021-11-28 by manup

fi-sch, 2021-12-14:
Guys, do you plan to release this fix anytime soon? This is a pretty big issue for many depending on Moes TRV in our homes. It seems to be broken for a long time, also the fix is in a "limbo" for nearly a month now...

Mimiix, 2021-12-14:
It's in the latest beta.

fi-sch, 2021-12-14:
> It's in the latest beta.

Awesome. I am asking about non-beta release, however. Something that I can safely run on my production server as a DE client, not beta tester.

MrMarkus42, 2022-05-01:
I hope you don't mind if I ask a question:
I've got the same problem with my new Hama TRV aka "_TZE200_yw7cahqs" devices. I wan't to set an offset via REST-API to 40 (0.4 degrees). I only can set values to multiples of 100 (100/200/300...). Why can't I set values to 40/50/60?

If I set for example 40 the offset is changed back to 0.

Smanar, 2022-05-01:
Are you sure the device is able to support a decimal value ?
I m reading the code, and deconz:
- take the value from the request
- divide by 10
- divide by 10 a send time (???)
- compare it, it need to be >-6 and < 6
- send it to the device

so using the 400 value mean deconz send the value 4 (and it's the minimum value, there is no decimal in request)

MrMarkus42, 2022-05-01:
Mh.
That's what I'm doing:

Looking in Phoscon API Help:

"12:16:22:381": {
        "e": "changed",
        "id": "107",
        "r": "sensors",
        "state": {
            "lastupdated": "2022-05-01T10:16:21.724",
            "lowbattery": false,
            "on": false,
            **"temperature": 2020,**
            "valve": 0
        },
        "t": "event",
        "uniqueid": "cc:86:ec:ff:fe:c2:00:7c-01-0201"
    },

- Reported Temperature was "2020" -> 20,2 Degrees (this is displayed within Home Assistant)
- This Temperature is 0.7 degrees to low. Therefore I tried to use an offset of "70" (not working)
- Then I changed the offset to "100"

    "12:16:30:457": {
        "config": {
            "heatsetpoint": 1300,
            "locked": true,
            "mode": "off",
            **"offset": 100,**
            "on": true,
            .........snip........

- now the reported Temperature is 1 degree higher. Therefore I've thought that the device supports decimal values...

    "12:16:59:785": {
        "e": "changed",
        "id": "107",
        "r": "sensors",
        "state": {
            "lastupdated": "2022-05-01T10:16:21.724",
            "lowbattery": false,
            "on": false,
            **"temperature": 2130,**
            "valve": 0

Smanar, 2022-05-01:
The reaction is strange, it's like when you set the offset it use "tuya rule", but for the return it use the "classic rules"

Are you using the GUI ? Can you share some logs with "info" and "info_l2" for this device, heat the device to trigger a temperature notification.

steljwagh, 2023-11-15:
If you want an offset for humidity and or temperature sensors in deconz, it's easy to do so through the API.
To get a API key login on Phoscon, go to /help/API information and get a existing key from the whitelist (search for whitelist and you'll find the keys)

Filter on sensors to find the right ID for your sensor.

Then open a command/terminal box and construct the following command to your needs.

`curl -X PUT -d "{\"config\": {\"offset\": 100 } }" http://ipaddress:port/api/key/sensors/10`

offset examples:

-80 = minus 0,8 degrees
100 = plus 1 degree