dresden-elektronik/deconz-rest-plugin #5486: Innr plugs SP120 reporting "on": true while they are off when DDF usage is enabled (2.13.2)

wvuyk, 2021-11-14:
## Describe the bug
If I set the DDF handling to "Normal", then my INNR plugs are changing to `"on":true` in the state object, despite the setting on the  hardware device itself. Sending a command '"on":false' does not change the REST API response back.
Once the DDF handling is disabled again, the status is as expected again.

## Steps to reproduce the behavior

Enable the DDF handling and wait for a minute or so. Then check the REST API result for the state object, while making sure the plug is physical off. It will report the device is on `"on":true`
## Expected behavior

## Screenshots

"etag": "785fd79fbb5650b7314cb74f29d8cf3d",
"hascolor": false,
"lastannounced": "2021-09-12T15:30:31Z",
"lastseen": "2021-11-14T13:20Z",
"manufacturername": "innr",
"modelid": "SP 120",
"name": "Plug mobiel keuken White",
"state": {
"alert": "none",
"on": true,
"reachable": true
},
"swversion": "2.0",
"type": "On/Off plug-in unit",
"uniqueid": "00:15:8d:00:03:2a:e8:be-01"

## Environment

 - Host system: any
 - Running method: Raspbian/ Windows e)
 - Firmware version: 26390500 (for raspbee)
 - deCONZ version: 2.13.02
 - Device:  ConBee II / RaspBee I
 - Do you use an USB extension cable: not relevant
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

Mimiix, 2021-11-14:
Can you add your ddf file?

wvuyk, 2021-11-14:
I am sure it is the same as present in the install of 2.13.02? 

I am very lefthanded when it comes to the raspberry pi, where the INNR plug is in use, so I copied the file from my Windos install here assuming it is the same. 

I had to rename it to txt to be able to copy it here

[sp_120.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7533924/sp_120.txt)

SwoopX, 2021-11-14:
Looks like the DDF uses the wrong cluster. 0x0008 instead of 0x0006. You can amend the file and then it's recommendable to restart deconz.

wvuyk, 2021-11-14:
Swoopx, thanks. But how can we put this correction in the package? As it is wrong there? I did not fiddle with these files yet?

SwoopX, 2021-11-14:
Well, you should fiddle now ;)

This is the place where you need to change the value as mentioned above

[image]

wvuyk, 2021-11-14:
I have used the editor in deCONZ for this and can confirm this solves the issue. But I think you are missing my worry here. How to get this in a next reslease of deCONZ? We do not want everyone to make their own adjustment I suppose?

Mimiix, 2021-11-14:
Need to make a pr on that file

wvuyk, 2021-11-15:
Thanks @SwoopX

closed 2021-11-26 by manup