dresden-elektronik/deconz-rest-plugin #5493: Lidl SilverCrest Smart USB Extension Lead HG06338

fana605, 2021-11-15:
## Device

- Product name: Lidl SilverCrest Smart USB Extension Lead
- Manufacturer: Unknown
- Model identifier: unknown
- Device type : 
  - Other: Smart Plug / Power Strip

## Screenshots

[image]
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]

### Scenes
[image]

### On/Off
[image]

This is the portuguese version of this Lidl Smart USB Extension Lead, model HG06338. There's a previous request to the model HG06338-DK, which I guess has different sockets.

The extension has 3 power sockets. It works, but the sockets all turn on/off at the same time. It would be great to control each one individually.

Thank you.

Smanar, 2021-11-15:
But I m looking the code fast, and I dont see something specific to the device HG06338
On the capture you have 3 endpoint, so deconz have created 3 entries, one by plug, no ?

fana605, 2021-11-15:
Ok, sorry for the confusion. HG06338 is the model stated on the instructions manual. And yes, it seems to be 3 endpoints, one for each plug, but they seem to behave like only one. If I turn one on, all the other turn on.

[image]

Here I turned on the first one, and the other two became on too.

Smanar, 2021-11-15:
Strange, not a group issue ?
Can you show a device json ? (You can have them in http://phoscon.de/pwabeta in help/API Information.

fana605, 2021-11-15:
I'm using Home Assistant, not sure where to get the device json. But regarding the group issue, what could I do to check that out?

Smanar, 2021-11-15:
You can use the link on the previous post, it will not work for all user but can make a try.

fana605, 2021-11-15:
@Smanar Got it.

Plug 1:
`{
    "etag": "3b9611b6b247a41860c0ae37a0450886",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-11-15T21:40Z",
    "manufacturername": "LIDL Silvercrest",
    "modelid": "HG06338",
    "name": "Extensão Lidl Tomada 1",
    "state": {
        "alert": "none",
        "on": true,
        "reachable": true
    },
    "swversion": "70",
    "type": "On/Off plug-in unit",
    "uniqueid": "50:32:5f:ff:fe:4d:fb:fb-01"
}`

Plug 2:
`                    {
    "etag": "a755895db9e81560d8f65d0b54b5e6ec",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-11-15T21:40Z",
    "manufacturername": "LIDL Silvercrest",
    "modelid": "HG06338",
    "name": "Extensão Lidl Tomada 2",
    "state": {
        "alert": "none",
        "on": true,
        "reachable": true
    },
    "swversion": "70",
    "type": "On/Off plug-in unit",
    "uniqueid": "50:32:5f:ff:fe:4d:fb:fb-02"
}`

Plug 3:
`                    {
    "etag": "ebbfdd79d98458ae6bc3b6984f9127ec",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-11-15T21:43Z",
    "manufacturername": "LIDL Silvercrest",
    "modelid": "HG06338",
    "name": "Extensão Lidl Tomada 3",
    "state": {
        "alert": "none",
        "on": true,
        "reachable": true
    },
    "swversion": "70",
    "type": "On/Off plug-in unit",
    "uniqueid": "50:32:5f:ff:fe:4d:fb:fb-03"
}`

fana605, 2021-11-18:
Hi @Smanar, Copy/paste is not working on the debug logs in order to share with you the info and info_l2 flags...  
But from what I can see, when making a "on" request, 3 events appear on the logs, one for each plug, at the same time.

On API information, the groups part shows (I removed groups not related to this device):

`{
    "5": {
        "action": {
            "alert": "none",
            "bri": 127,
            "colormode": "hs",
            "ct": 0,
            "effect": "none",
            "hue": 0,
            "on": true,
            "sat": 127,
            "scene": null,
            "xy": [
                0,
                0
            ]
        },
        "devicemembership": [],
        "etag": "90111d420aa417646f63b88c2b15adb7",
        "hidden": false,
        "id": "5",
        "lights": [
            "11",
            "10",
            "13",
            "14",
            "15"
        ],
        "lightsequence": [],
        "multideviceids": [],
        "name": "Room",
        "scenes": [],
        "state": {
            "all_on": false,
            "any_on": false
        },
        "type": "LightGroup"
    }
}`

Smanar, 2021-11-18:
This group is normal, it s the "all device" group.
Seriously I don't understand, and if you turn "on" a plug using the GUI, for exemple on endpoint 0x01, cluster 0x0006, it turn on all plugs or only one ?

fana605, 2021-11-18:
Yes, it turns all plugs at the same time. I checked with a friend that bought the same device and he has the same issue using zigbee2mqtt. But the device does support control of each individual plug, using LIDL app, as you can see on this video (UK version): [https://youtu.be/PASZ-p9EuJs?t=1481](url)

Smanar, 2021-11-18:
Ok, so there is something magic to do.
But what ? and this device don't have tuya cluster for magic ....

Element2, 2021-12-08:
Have the same issue, any progress?

hcastanho, 2021-12-15:
Same here. It turns on/off all 3 power sockets at the same time.

Smanar, 2021-12-15:
I haven't found more info.
Need someone with the gateway sniff the traffic to find the "magic trick"

JvD12, 2022-01-19:
Any updates on this?

Mimiix, 2022-01-19:
Its funny, i have the same one and it works flawlessly.

Element2, 2022-01-19:
I have an 'old' one. Bought it a year ago. This one works.
A 'new' one which I bought about a month ago, has the problems described above.
So probably a firmware update.

Element2, 2022-01-21:
@Smanar 
I have access to root at the original Lidl gateway.
Can you do anything with the TUYA debug logs when switching this device?
I can supply them

Smanar, 2022-01-21:
You have something to log request done by an original tuya gateway ?
For that you need to sniff the zigbee traffic when the tuya gateway send a working request, with wireshark (for exemple)

Element2, 2022-01-21:
Wireshark is not an option since my wifi-card doesn't do promiscuous mode.
Are you looking for some kind of string, because the debug log on the gateway has a lot of info.
So maybe I can see if the log has any info you need?

Smanar, 2022-01-22:
IDK, how look the debug log on the gateway ?

Element2, 2022-01-22:
This is part of the debug log when switching outlet 3 of 3 to ON:

[tuya_debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7918208/tuya_debug.txt)

Smanar, 2022-01-22:
Honnestly, I m not sure to understand the log

>[01-22 12:18:00:450 TUYA Debug][z3_slabs.c:339] status[0x00] dev[50325ffffe4dfe63] cluster[0x0006] node[0x9c32] gId[0x0000] dEp[0x03] cmd[0x00] type[0x00] znum[98] anum[106].

It seem it's the on/off request but znum ? anum ?

And it s log from the gateway, so it can send 5 zigbee request but print only 1 in logs. The better way is realy sniffing the zigbee request to mimic zigbee request.

Element2, 2022-01-22:
01-22 12:18:00:690 TUYA Debug][app_agent.c:1710] JSON string:{"data":{"cid":"50325ffffe4dfe63","ctype":0,"dps":{"3":true}},"protocol":5,"t":1642850280}

This looks like a form of a command in json. The cid is the mac address of the device.
But I understand the log is not the same as the real data that is send.
I do have Wireshark, but I can't capture anything which is not send to and from it's own address. 
It can't do promiscuous mode. It's a known problem with the ethernet/wifi card of that model.
So bummer I guess.

Smanar, 2022-01-23:
Just by curiosity you don't have a zigbee sniffer ?
Else just need to wait for someone else sniff the device and give result.

Element2, 2022-01-23:
With sniffer I presume you mean capturing the Zigbee radio protocol?
I don't have that.
I got another laptop and now can capture TCP packets from the Lidl Android app to the Lidl Gateway.
Can we do anything with that?

Smanar, 2022-01-23:
Nope, only the zigbee traffic is usefull.

Element2, 2022-01-29:
So, I have a sniffer running and can decoded Zigbee traffic.
I can see the On/Off frames in Wireshark if I switch one of the the extension lead outlets.
What do you need to see?

Smanar, 2022-01-30:
If you can send me the decoded traffic (to not share your network key) or at least show a picture of the echange I will give you the interesting packed.
I can't be more precise, because they can use a "magic request" just before.

I think the on/off frame is usefull for sure, can start with this one, just check if you don't have a "strange" frame just before.

Element2, 2022-01-30:
This is the 'switch outlet 3 to ON' frame.
I will look for something like a 'magic request'

[ON_frame.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7966142/ON_frame.txt)

Element2, 2022-01-30:
This is part of the exchange.
OFF = frame 496
ON = frame 514

[capture.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7966202/capture.txt)

Element2, 2022-01-30:
with a Whireshark Src/Dst filter with Src = Lidl Coordinator and Dst = Lidl HG06338, during 1 hour
I see, beside the OnOff frames, a lot of 'Read Attributes' frames.
One 'Default Response' frame and just a few 'Unknown Command' and 'Data' frames
But the're not send just before the OnOff frames.

Smanar, 2022-01-31:
But it s strange we need the "Unknow command", if we need a special setting, it was done at start, just after the pairing, only 1 time, not repetively.

Can you show the frame 514 ? to see if it use a special parametre ?
The 516 is the normal report after the action.
And just before sending the "on" request, the gateway ask for attribute on frame 508, can we see this one too ?

Element2, 2022-01-31:
I removed the old capture files. But now I have saved everything. So let's start with the pairing.
This is the exchange of the pair process. Let me know if you want to see any individual frames.
Frame 91 gives status: succes 

[exchange.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7973903/exchange.txt)

Element2, 2022-01-31:
Exchange of an OFF and ON switching of 1 outlet.

[exchange.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7973945/exchange.txt)
.

Element2, 2022-01-31:
514 was an ON-frame. I copied an ON-frame in a previous post. But this is another:
[ONframe.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7974039/ONframe.txt)

Element2, 2022-01-31:
This is frame 30, the attribute-frame just before the ON-frame.

[attr.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7974108/attr.txt)

Element2, 2022-01-31:
This is frame 21. Read attribute from gateway before OFF.

[read_attr.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7974124/read_attr.txt)

Smanar, 2022-02-01:
I don't see something special on "classic request"
- The frame 30 is the report state from the previous on/off request (frame28), so was too late
- The frame 21 is more curious, but you have have so much "Read Attributes" before, the device was probably added recently and the gateway is asking for information ?

Else if the gateway do the same request every time it mean, the gateway ask for Application Version (0x0001) on cluster 0x0000 just before making a on/off request ?

You don't have something different when you toogle the plug 1 compared to the plug 2 ? (except the endpoint 0x01 or 0x02)

Element2, 2022-02-02:
Correct, this device was added recently.
I have removed all other non related devices from the Zigbee network. 
So now I only have the Lidl coordinator and one HG06338.
This is the exchange of switching all plugs first to Off and then quickly to On again.

Smanar, 2022-02-02:
There is still some "Read Attributes" but you have a cycle just with on/off

```
47	26.580456	0x0000	0x33e8	ZigBee HA	48	ZCL OnOff: Off, Seq: 11
51	27.809571	0x0000	0x33e8	ZigBee HA	48	ZCL OnOff: Off, Seq: 12
55	30.969862	0x0000	0x33e8	ZigBee HA	48	ZCL OnOff: On, Seq: 13
59	32.289428	0x0000	0x33e8	ZigBee HA	48	ZCL OnOff: On, Seq: 14
63	33.630473	0x0000	0x33e8	ZigBee HA	48	ZCL OnOff: On, Seq: 15
```

But without "special" request between them.

You have tried to compare a "on/off" request for the plug 1 and the same for the plug 2, to check if you have not something special other than the endpoint ?

Can try too making the same thing but with the device on a deconz network, it s possible on this situation the device use some "group" ou " broacast" request.

Element2, 2022-02-03:
It doesn't pair correctly in Phoscon. Instead of 3 outlets, Phoscon shows only one after pairing.  
That one switches all outlets. So no individual outlets appear in Phoscon.
Phoscon is my 'production' network, with 40+ devices. This gives the sniffer a few thousand packets every minute.
Therefor I can't filter out the pairing frames. I can only filter something useful when the device has been given an address. I can show the On/Off frames of switching the three outlets on one?

Mimiix, 2022-02-03:
What does the restAPI provide?

Element2, 2022-02-03:
Under lights:

41": {
        "etag": "f78166b05671fdadcf8be6fd62bd9d5f",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2022-02-03T12:30Z",
        "manufacturername": "LIDL Silvercrest",
        "modelid": "HG06338",
        "name": "On/Off plug-in unit 41",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off plug-in unit",
        "uniqueid": "50:32:5f:ff:fe:4a:91:13-01"
    }

Smanar, 2022-02-03:
>Therefor I can't filter out the pairing frames. I can only filter something useful when the device has been given an address. I can show the On/Off frames of switching the three outlets as one?

Better to use direclty deconz for tests, not phoscon.
But perhaps you haven't the GUI too ? and yes can have so much traffic, but it's easy to use filter on wireshark, you can filter only data from and to this device network id ?
But I dn't think you will see something special during the on/off request.

I don't think it's from deconz because there is the ame issue on z2m
It's perhaps some that happen with a "group" or "broadcast" request, but this one can be send from and to another network id than the device.
Or a special "magic" during the inclusion, and here you realy need to check all requests.

Element2, 2022-02-06:
Maybe let's start with the pairing. It does not pair correctly. In Phoscon it shows up as only 1 outlet, instead of 3.

I do not exactly know what to look for. How does special 'magic' look like in terms of frames or content of frames?
 
Only the frames with a Zigbee network address are relevant right?
 
Are there any frames specially interested? 

The Extention Lead reports to the gateway it has 4 Active Endpoints:
Endpoint: 1
Endpoint: 2
Endpoint: 3
Endpoint: 242

ebaauw, 2022-02-06:
My Schuko (DE/NL) extension lead almost looks the same as the one in the first post, except that reports _Application Version_ of 69.  This is also the _Version_ reported in the OTAU panel.  It is exposed as three `/lights` resources (for endpoints 0x01, 0x02, and 0x03); each resource/endpoint controls the corresponding outlet, independently from the other two outlets.  The physical button on the device controls all three outlets simultaneously.

Except for the mutilated _Model Identifier_ and _Manufacturer Name_, there's nothing special about the device, and it should be recognised by the C++ code in the REST API plugin.  I did pair it quite some versions ago, not sure if any recent updates might have broken it.

Could you double-check in the GUI, that the _On/Off_ cluster on each endpoint controls only one of the three endpoints?  If not, the issue is with the device hardware and/or firmware, and there's nothing we can do.  If it does, the issue is with the REST API plugin and should be fixable.

Element2, 2022-02-06:
It seems there are now 2 versions of this Lidl HG06338 extention lead. The first is sold about a year ago and was then added to the api. That one (still) pairs ok and still works ok. At the end of 2021 Lidl sold a 'new' version, also with the name HG06338 and looking the same, but with new firmware?. This one does not work/pair correct with the old api implementation. The problems reported are with this new version. I have one old (still working ok) and two new. Both the new versions have the same problem. The new versions work correct with the Lidl app en Lidl gateway. So I don't think it's a hardware problem.
Like others reported, the GUI only report 1 endpoint which switches all three outlets as one.

ebaauw, 2022-02-06:
> Like others reported, the GUI only report 1 endpoint which switches all three outlets as one.

I must have missed that, sorry.  Where are the corresponding screenshots?

Element2, 2022-02-06:
For the new one the're at the top of this  thread.
For the old one I found: #3612  #4503 (search for HG06338)

ebaauw, 2022-02-06:
> For the new one the're at the top of this thread.

That one shows four endpoints?!  Just like the old one I have.

Element2, 2022-02-06:
That seems correct. I guess that's 1 for every outlet plus one for all three outlets together. Which agrees with the frame that reports them: 1,2,3 and 242
I don't know why only one sows up in pairing. I have paired and removed both of the new devices I have now about 20 times.  I have even tried pairing them within 1 meter radius of my Conbee-II stick (which works flawlessly with all other Zigbee devices).
Only once appeared a second outlet in Phoscon GUI, which did nothing. I checked the Help->API information if there was something left in the database, but that seems clean. Every time I pair/remove it in the Lidl App, it is discovered correctly and all three outlets (plus 1 button for switching them all) works.

ebaauw, 2022-02-06:
The 0xF2 (242) endpoint is for ZGP, nothing to do with the outlet function.

Can you control the individual sockets from the _On/Off_ cluster of the corresponding endpoint, using the GUI?

Element2, 2022-02-06:
I'm sorry, can you please explain that a bit more in detail?

ebaauw, 2022-02-06:
See the user manual (under _Help_).  When you select _On_, _Off_, or _Toggle_ in the _Cluster Info_ panel, while one of the _On/Off_ clusters is selected, does that act on a single or on all outlets?

ebaauw, 2022-02-07:
That would suggest it’s really an issue with the device (hardware and/or firmware).  Did you use this very device with the LIDL hub and app?

Element2, 2022-02-07:
Yes, this device pairs and works ok with the Lidl hub and app.
The second one I have is the same.

ebaauw, 2022-02-07:
Then we really need to sniff the LIDL hub to figure out what magic it’s doing to get the device in the right mode.

I’ll see if I can try my v69 extension cord this weekend.  I’m not sure I ever linked it to my LIDL hub - I acquired the hub to reverse engineer the Xmas light strip.  Also not sure if the hub does the same magic to the older extension cord firmware.

Element2, 2022-02-07:
It shows 4 On/Off clusters in the GUI. HA 01 - HA 03 and F2 (Green power) All 3 HA On/Off clusters switch all outlets On/Off at the same time.

I was in the process of sniffing it with help from Smanar, but since I'm new to that process it's difficult to find the 'magic'.

ebaauw, 2022-02-07:
> It shows 4 On/Off clusters in the GUI. 

Huh?!?  Please add a screenshot, as it doesn’t in the screenshot in the first comment in this issue.

Element2, 2022-02-07:
[image]

[55 more replies]

closed 2023-03-26 by github-actions[bot]