dresden-elektronik/deconz-rest-plugin #5495: Hihome Smart Thermostat not working with deconz

besiktas97, 2021-11-16:
## Device

- Product name: Hihome Zigbee
- Manufacturer: _TZE200_zuhszj9s
- Model identifier: TS0601
- Device type : Please remove all unrelated device types. 
  - Thermostat

## Screenshots

[image]

### Basic

### Groups

### Scenes

Mimiix, 2021-11-17:
@Smanar Another one!

Smanar, 2021-11-17:
We have already it, but the other model

>    {"_TYST11_zuhszj9s", "uhszj9s", "HiHome", "Tuya_THD WZB-TRVL TRV"},

So we just need to edit a line.
Do you have a linux machine with real OS to test code modification ?

besiktas97, 2021-11-17:
No I dont have it. Only Windows and a Pi with Home Assistant.thank you for the effort that you are gonna take.

Smanar, 2021-11-17:
Will add it in blind mode, hoping for same working mode

besiktas97, 2021-11-17:
What do you mean with blind mode? I guess that I will hear from you, when its ready to test it?

Smanar, 2021-11-17:
It mean "without test" ^^
You can alredy test it, but you need a real linux OS else need to wait for PR validation, and ATM with the DDF core I realy can't give a adate.

besiktas97, 2021-11-17:
Thank you for your support. Can you tell me how I can test it with linux? I will install tomorrow a linux OS as a virtual machine to check it out. I hope that the PR validation will come soon, so I can use the devices 😁

Smanar, 2021-11-17:
You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch add_clone_1 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Can need 2 inclusions like lot of tuya stuff.

cdlh, 2021-11-20:
Hi there,

have apparently the same device here, manufacturer according to the package is Unitec, item number is 30946. The difference I see is the Manufactor Name in deconz: _TZE200_ps5v5jor.

I use a Raspberry Pi 3 with the RaspBee SD-card image.

The commands listed by Smanar work until the qmake-line.
When I enter the command "qmake && make -j2" I get the error message: 

Info: creating stash file /home/pi/deconz-rest-plugin/.qmake.stash
Project ERROR: Unknown module(s) in QT: qml

Any idea?

Smanar, 2021-11-20:
Yep from z2m, all thses devices are clone

```
        zigbeeModel: ['ivfvd7h', 'eaxp72v\u0000', 'kfvq6avy\u0000', 'fvq6avy\u0000', 'fvq6avy'],
        fingerprint: [
            {modelID: 'TS0601', manufacturerName: '_TZE200_zivfvd7h'},
            {modelId: 'TS0601', manufacturerName: '_TZE200_kfvq6avy'},
            {modelId: 'TS0601', manufacturerName: '_TZE200_hhrtiq0x'},
            {modelId: 'TS0601', manufacturerName: '_TZE200_ps5v5jor'},
            {modelId: 'TS0601', manufacturerName: '_TZE200_owwdxjbx'}],
        model: 'GS361A-H04',
        vendor: 'Siterwell',
```

So you can just add them in the productmach.cpp file

You don't have error when you have used 

>sudo apt install deconz-dev
>deconz-dev est déjà la version la plus récente (2.13.01-raspbian-buster-beta).

You can try too
>make clean

cdlh, 2021-11-20:
Thank you for your help, i cloned the lines in product_match.cpp and now i can see the thermostate in HA. 
The Temperature is shown and when changed manually, will be updated in HA. BUT it's not possible to change temperature in HA and send to device. Neither it's possible to change the preset ("auto", "holiday", ... , "manual"). When trying, error is shown "Could not set attribute".
And the device type in deconz is not correct. Still named "Home Automation Smart Plug".
[image]

besiktas97, 2021-11-20:
Cdlh, did you test this on your production machine? I couldn't test it, because I didnt have the time for it. Inguess when I install a fresh linux machine, I must use my conbee 2 stick from my production machine.

cdlh, 2021-11-21:
Besiktas97, yes, it's the production machine, i've got the RaspBee SD-card image. Simple to restore if something fails.

Smanar, 2021-11-21:
>And the device type in deconz is not correct. Still named "Home Automation Smart Plug".

Yes this is normal, the device realy announce it as plug.

Can you show the change you have done ? You need to use exaclty

`    {"_TZE200_ps5v5jor", "TS0601", "Siterwell", "Tuya_THD GS361A-H04 TRV"},`

Because the code make test on "Tuya_THD GS361A-H04 TRV"

Else you can check to device json here http://phoscon.de/pwabeta in help / api information

cdlh, 2021-11-21:
I changed the line in product_match.cpp accordingly the one Smanar posted. (I had a different Manufacturer).
Deleted the Nodes out of Deconz and reassigned them. But still the same behaviour. You can change the temperature manually at the device. Value is synced to HomeAssistant. When I change temperature in HomeAssistant it won't be synced. When I change preset in HA same error occurs ("Could not set attribute").
Here is the Information from API:
"29": {
            "config": {
                "battery": 97,
                "heatsetpoint": 1650,
                "locked": null,
                "offset": 0,
                "on": true,
                "preset": "manual",
                "reachable": true,
                "schedule": {},
                "schedule_on": null,
                "setvalve": null,
                "windowopen_set": true
            },
            "ep": 1,
            "etag": "60088059e03443e861040b23b4fbb5dc",
            "lastannounced": null,
            "lastseen": "2021-11-21T10:11Z",
            "manufacturername": "_TZE200_ps5v5jor",
            "modelid": "TS0601",
            "name": "Thermostat 29",
            "state": {
                "lastupdated": "2021-11-21T10:05:57.264",
                "on": null,
                "temperature": 2100
            },
            "type": "ZHAThermostat",
            "uniqueid": "xx:xx:xx:ff:fe:16:17:f5-01-0201"
        }
    }
}

besiktas97, 2021-11-21:
> Besiktas97, yes, it's the production machine, i've got the RaspBee SD-card image. Simple to restore if something fails.

I am wondering if it will also work on my production machine when I am gonna use the terminal option of home assistant. I am using the default home assistant image that you can download from there website.

Smanar, 2021-11-21:
You can change the manufacture, the important string is "Tuya_THD GS361A-H04 TRV".

>When I change temperature in HomeAssistant it won't be synced

Can you show logs during this procedure with "info" and "info_l2" ?

And if you can catch at a moment the productid in the json "Tuya_THD GS361A-H04 TRV", because it s hidden in the json on the api (so can be visible only in logs). And all tests are based on this string.

cdlh, 2021-11-21:
Wow - Log is filled VERY quickly with a lot of stuff. These are the lines which i think belong to the action, changed temperature to 25 degrees, and tried to set preset to "auto".
Strange thing is: Two different IP-Adresses used, first one is from my HA-Installation, the second ... idk ...

14:24:09:371 add task 49640 type 37 to 0x847127FFFE1617F5 cluster 0x0201 req.id 234
14:24:09:382 Websocket <<HA-IP>>:57862 send message: {"config":{"battery":97,"heatsetpoint":2500,"locked":null,"offset":0,"on":true,"preset":"manual","reachable":true,"schedule":{},"schedule_on":null,"setvalve":null,"windowopen_set":true},"e":"changed","id":"29","r":"sensors","t":"event","uniqueid":"84:71:27:ff:fe:16:17:f5-01-0201"} (ret = 2128569288)
14:24:09:385 Websocket <<Unknown IP>>:51374 send message: {"config":{"battery":97,"heatsetpoint":2500,"locked":null,"offset":0,"on":true,"preset":"manual","reachable":true,"schedule":{},"schedule_on":null,"setvalve":null,"windowopen_set":true},"e":"changed","id":"29","r":"sensors","t":"event","uniqueid":"84:71:27:ff:fe:16:17:f5-01-0201"} (ret = 2128569288)

14:24:10:711 don't create binding for attribute reporting of sensor Thermostat 29

14:24:11:252 API error 608, /sensors/29/config/preset, Could not set attribute

I

Smanar, 2021-11-21:
Yeah depend of how much you have.

>Strange thing is: Two different IP-Adresses used, first one is from my HA-Installation, the second ... idk ...

Can be phoscon ?

preset=auto seem supported by the device, you are sure you have not mixed "mode" and "preset" ?

Log I need are logs starting by 
```
Send Tuya request xxxxxxxxxxx
Tuya debug 4 : xxxxxxxxxxxxxxxx
Tuya debug 5 : xxxxxxxxxxxxxxxxx
```

Inside them we can check the command make by deconz and all return from the device.
The temperature 25°C is updated in the API, but not on the device ?

cdlh, 2021-11-21:
> The temperature 25°C is updated in the API, but not on the device ?!
Yes, i can change the temperature in HomeAssistant, there is no error showing up. But the device is not reacting. 
When i change temperature at the device, the Info in HomeAssistant is updated.

> preset=auto seem supported by the device, you are sure you have not mixed "mode" and "preset" ?
In the Device configuration in HomeAssistant it's named preset:
[image]

I found out: The 2nd IP is my Mac, from where i'm controling Home Assistant.

In the Log i cannot find any lines starting with Tuya, i even cannot find the word "Tuya" in the whole log.

Please remember that i just changed the product_match.cpp - the Linux commands you posted i was not able to execute - is that the problem?

Smanar, 2021-11-21:
>Please remember that i just changed the product_match.cpp - the Linux commands you posted i was not able to execute - is that the problem?

Ha, you have just edited the line ? you need too compile the code.
All lines need to be used, you still have an error message during compilation ?

cdlh, 2021-11-21:
No, still problems with that. I use the image provided by dresden electronics. I use vnc and log me in on the raspberry the conbee ii is connected to.
first error occurs:
pi@phoscon:~ $ git clone --branch add_clone_1 https://github.com/Smanar/deconz-rest-plugin.git
fatal: Zielpfad 'deconz-rest-plugin' existiert bereits und ist kein leeres Verzeichnis.
Meaning: Targetpath 'deconz-rest-plugin' already exists and is not an emtpy directory.

Next error:
~/deconz-rest-plugin $ qmake && make -j2
Project ERROR: Unknown module(s) in QT: qml

Smanar, 2021-11-22:
The first error is normal, you have already cloned the github

For the second one
```
sudo apt-get install qtdeclarative5-dev
sudo apt-get install libqt5qml5
```

besiktas97, 2021-11-22:
Do you guys know if I can test this also on a production machine in HA?

Smanar, 2021-11-22:
>Do you guys know if I can test this also on a production machine in HA on the terminal add-on?

It s possible, but not easy at all, and I can't help you for that, you have the procedure on the link to compile on docker.

>are not displaying the battery-status anymore.

Can just wait, you need the device report a battey first to have it, and on some devices can take 1 hour.
The inclusion for tuya stuff is realy random, so if you can avoid to re do it, it s better.

besiktas97, 2021-11-25:
> It s possible, but not easy at all, and I can't help you for that, you have the procedure on the link to compile on docker.
>
>
>
>
>
>
> Can just wait, you need the device report a battey first to have it, [...]

That means that I need to unplug my conbee 2 stick from home assistant and test it out on the new build OS for the new device. I hope that I can check it out next week! Thanks for now Smanar.

Smanar, 2021-11-25:
For information :
- the network setting is stored on the conbee
- the devices setting are stored on the host
- can use a phoscon feature to migrate all the database if needed.

closed 2021-11-26 by manup

besiktas97, 2021-12-02:
Hey Smanar, sorry for all the delay.
I did test it and I have noticed that the device is recognized as a smart plug.
See screenshots below after I let deconz read the cluster info:
[image]

At the begin I saw also the information like manufacturer and model, but this time it doesn't show it:

[image]

Smanar, 2021-12-02:
Yep, the device present itself as a smart plug, so you will have that in deconz (but as thermostat in the API)
But it haven't worked (else the name will be other thing than 0xXXXX). You are using phoscon for inclusion ?

You probably need 2 inclusions, without deleting the device.

And BTW the PR was validated, so you can use the official code too.

besiktas97, 2021-12-02:
I see, thata great news. Do you mean with inclusions, to pair it twice and not deleting the device to pair it?
If so I will try it now to see if its gonna work.
Thanks!

besiktas97, 2021-12-02:
I see, thata great news. Do you mean with inclusions, to pair it twice and not deleting the device to pair it?
If so I will try it now to see if its gonna work.
And yes I am using phoscon as inclusion.

Smanar, 2021-12-02:
>I see, thata great news. Do you mean with inclusions, to pair it twice and not deleting the device to pair it?

Yep

besiktas97, 2021-12-02:
I did like you tell to do, but its like the first time. Everything is showing up, but cant control the device like on or off or set a temperature. All the functions are greyed out and when I am forcing it to open the valve it immediatly close it on the developer tools of HA.
[image]

[image]
[image]
[image]

besiktas97, 2021-12-02:
I just checked the deconz log files and keep getting those error lines for the specific thermostat:

19:00:45:330 Tuya : Schedule command
19:00:45:331 Tuya debug 4 : Address 0x5C0272FFFEE55C4A Payload 000973000018000000000000000000000000000000000000150000000000
19:00:45:332 Tuya debug 5 : Status: 0 Transid: 9 Dp: 115 (0x00,0x73) Fn: 0 Data 545460846592
19:00:45:332 Tuya : Schedule data error
19:00:45:552 Tuya : Schedule command
19:00:45:553 Tuya debug 4 : Address 0x5C0272FFFEE55C4A Payload 000d6e000018000000000000000000000000000000000000120000000000
19:00:45:553 Tuya debug 5 : Status: 0 Transid: 13 Dp: 110 (0x00,0x6E) Fn: 0 Data 0
19:00:45:554 Tuya : Schedule data error
19:00:45:784 Tuya : Schedule command

Smanar, 2021-12-02:
Don't worry, realy few device support the shedule command.
You can't use the GUI to control the device, it use Tuya cluster with exotic requests, it s only possible using direclty the api or the third app (HA for you).

ATM you have 19 °C set, you can't modify this value ?

besiktas97, 2021-12-02:
> Don't worry, realy few device support the shedule command.
> 
> You can't use the GUI to control the device, it use Tuya cluster with exotic requests, it s only possible using direclty the api or the third app (HA for you).
> 
> 
> 
> ATM you have 19 °C set, you can't modify this value ?

I am indeed using the third app HA to test it out. When I change the temperature it doesn't change it on the device. Also on off state doesn't work.

Smanar, 2021-12-03:
>I am indeed using the third app HA to test it out. When I change the temperature it doesn't change it on the device

Ok, so bad start.
Can you show the same logs you have sho just before (with "Tuya debug X") when you make the request using HA ?

besiktas97, 2021-12-04:
Do I need to use the command tuya debug x on the linux machine or on HA production machine?
If you mean the same logs, I tried to change the degrees and get those logs:

02:05:33:674 delay sending request 254 dt 1 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:33:683 delay sending request 254 dt 1 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:33:784 delay sending request 254 dt 1 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:33:883 delay sending request 254 dt 1 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:33:983 delay sending request 254 dt 1 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:34:083 delay sending request 254 dt 1 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:34:184 delay sending request 254 dt 1 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:34:283 delay sending request 254 dt 2 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:34:384 delay sending request 254 dt 2 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:34:484 delay sending request 254 dt 2 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:34:584 delay sending request 254 dt 2 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:34:683 delay sending request 254 dt 2 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:34:784 delay sending request 254 dt 2 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:34:883 delay sending request 254 dt 2 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:34:898 delay sending request 254 dt 2 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:34:983 delay sending request 254 dt 2 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:35:084 delay sending request 254 dt 2 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:35:183 delay sending request 254 dt 2 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:35:284 delay sending request 254 dt 3 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:35:321 Replace task 520383 type 37 in queue cluster 0x0201 with newer task 520392 of same type. 1 runnig tasks
02:05:35:322 delay sending request 10 dt 3 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:35:383 delay sending request 10 dt 3 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:35:483 delay sending request 10 dt 3 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:35:583 delay sending request 10 dt 3 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:35:683 delay sending request 10 dt 3 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:35:783 delay sending request 10 dt 3 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:35:884 delay sending request 10 dt 3 ms to 0x5C0272FFFEE55C4A, ep: 0x01 cluster: 0x0201 onAir: 1
02:05:35:983 delay sending request 10 dt 3 ms to 0x5C0272FFFEE55C4A, ep: 0x01

Smanar, 2021-12-04:
Nothing about "tuya debug X" ?

It seem the code try to use the thermostat cluster 0x0201 instead of the tuya one. The device is reconised because of the thermostat cluster but not sure all is finished.
Are you sure using the good branch ? You can use the official code actually on the github.

Nothing about a "product id" for this device ?

besiktas97, 2021-12-04:
I am using the official code from github, atleast if I can understanf what you mean. I am resetting the device and set it in pair mode, after that deconz have recognised it, like the previous screenshots I put it here. I didn't saw anything like "tuya debug x" and all this I am trying on the production machine. Do I need to do something else to get the tuya debug x information ? And the branch is hihome like it says on the box where the device came out. Thanks again Smanar.

Smanar, 2021-12-04:
I realy don't understand, all look fine in the code.
Are you able to try to change the heatsetpoint using direclty the API ?

Can use the browser https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/
Or command line

`curl -H 'Content-Type: application/json' -X PUT -d '{"heatsetpoint": 20000}' http://IP:PORT/api/APIKEY/sensors/ID/config`

IP and PORT are the same used for phoscon
ID is the device ID
APIKEY is an APIKEY, can use the same used by the third app.

besiktas97, 2021-12-04:
This is what i see when I try to change the trv degrees:

Logger: homeassistant.components.websocket_api.http.connection
Source: components/deconz/climate.py:252
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: December 2, 2021, 6:26:41 PM (6 occurrences)
Last logged: 8:06:08 PM

[547855491024] Invalid content type: text/html
[547831543840] Invalid content type: text/html
[547694446000] Invalid content type: text/html
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 185, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1495, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1530, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
    await self.hass.helpers.service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 667, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 863, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 704, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/climate/__init__.py", line 586, in async_service_temperature_set
    await entity.async_set_temperature(**kwargs)
  File "/usr/src/homeassistant/homeassistant/components/deconz/climate.py", line 252, in async_set_temperature
    await self._device.set_config(**data)
  File "/usr/local/lib/python3.9/site-packages/pydeconz/sensor.py", line 1045, in set_config
    return await self.request(field=f"{self.deconz_id}/config", data=data)
  File "/usr/local/lib/python3.9/site-packages/pydeconz/api.py", line 137, in request
    return await self._request("put", path=field, json=data)
  File "/usr/local/lib/python3.9/site-packages/pydeconz/gateway.py", line 141, in request
    return await self._request(
  File "/usr/local/lib/python3.9/site-packages/pydeconz/gateway.py", line 160, in _request
    raise ResponseError(
pydeconz.errors.ResponseError: Invalid content type: text/html

besiktas97, 2021-12-04:
This is on the production machine what I see, so you should say everything should work fine...

[image]

besiktas97, 2021-12-04:
> Nothing about "tuya debug X" ?
>
> It seem the code try to use the thermostat cluster 0x0201 instead of the tuya one. The device is reconised because of the thermostat cluster but not sure all is [...]

Sorry I look wrong, now I see tuya debug 4 and 5 in the log files:

21:01:13:643 Tuya debug 4 : Address 0x5C0272FFFEE55C4A Payload 001177000018000000000000000000000000000000000000000000000000
21:01:13:643 Tuya debug 5 : Status: 0 Transid: 17 Dp: 119 (0x00,0x77) Fn: 0 Data 545460846592
21:01:13:643 Tuya : Schedule data error

Smanar, 2021-12-04:
Those one are just schedule request. The one interest me is the Set heatpoint request.

Can you try using direclty the API, or can you at least show me the device json (visible here http://phoscon.de/pwabeta in help / api information)

besiktas97, 2021-12-04:
How can I use the direct api? I am really getting confused now. I will try the link that you have send to me.

besiktas97, 2021-12-04:
> I realy don't understand, all look fine in the code, the device need to be reconised as "Tuya_THD WZB-TRVL TRV" for the product id (not visible in the json, but perhaps in logs) Are you able to try to change the heatsetpoint using direclty the API ?
> 
> Can use the browser https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/ Or command line
> 
> `curl -H 'Content-Type: application/json' -X PUT -d '{"heatsetpoint": 20000}' http://IP:PORT/api/APIKEY/sensors/ID/config`
> 
> IP and PORT are the same used for phoscon ID is the device ID APIKEY is an APIKEY, can use the same used by the third app.

When I am using the command line: curl -H 'Content-Type: application/json' -X PUT -d '{"heatsetpoint": 20000}' http://IP:PORT/api/APIKEY/sensors/ID/config and going after that to the browser with the link, I am getting the next info that I see: 

[{"error":{"address":"/sensors/5c:02:72:ff:fe:e5:5c:4a-01-0201/config","description":"resource, /sensors/5c:02:72:ff:fe:e5:5c:4a-01-0201/config, not available","type":3}}]

But when I am going to the following link, I get the next info:

http://192.168.2.30:40850/api/CF78A23A5C/sensors/5c:02:72:ff:fe:e5:5c:4a-01-0201/

{"config":{"heatsetpoint":3150,"offset":0,"on":true,"reachable":true,"schedule":{},"schedule_on":false},"ep":1,"etag":"e0daa10bf0396bcff6ea0e6673db29b2","lastannounced":"2021-12-02T17:42:51Z","lastseen":"2021-12-04T22:51Z","manufacturername":"_TZE200_zuhszj9s","modelid":"TS0601","name":"Thermostat 42","state":{"lastupdated":"none","on":null,"temperature":null},"type":"ZHAThermostat","uniqueid":"5c:02:72:ff:fe:e5:5c:4a-01-0201"}

Smanar, 2021-12-05:
???

And if you try with the ID number, instead of the ressource id (5c:02:72:ff:fe:e5:5c:4a-01-0201) ? it s a small number

You can see them in http://192.168.2.30:40850/api/CF78A23A5C/sensors it s the number on top of json

You are using deconz on a separate machine ? or trying to update the HA plugin ?

besiktas97, 2021-12-05:
I will try that tonight and check it out. I was trying this yesterday on the production machine. I will try this time on a linux machine with deconz on it (raspbeen).

Smanar, 2021-12-05:
Yep because using compiled library file on the HA OS is not something easy to do.