dresden-elektronik/deconz-rest-plugin #5496: DIY CC2530+CC2591 with ptvo firmware - RESTAPI not updated

docloulou, 2021-11-17:
## Describe the bug
Hello, i flashed a CC2530+CC2591 chip with ptvo firmware, the purpose is to return a on/off  switch status.
When i read the on/off cluster and trigger the input pin, false/true attribute change but the restapi isn't updated and remain false.
i used DDF. (code below)

## Device

- Product name: DIY CC2530+CC2591 with ptvo firmware
- Manufacturer: N/A
- Model identifier: N/A
- Device type : 
  - On/Off output

## Screenshots
[image]
[image]
[image]
[image]

## DDF i use

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "ptvo.info",
  "modelid": "ptvo.switch",
  "product": "ptvo.switch",
  "sleeper": false,
  "status": "Draft",
  "path": "/devices/ptvo.switch.json",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5,
          "read": {
            "at": "0x0000",
            "cl": "0x0006",
            "ep": 2,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0006",
            "ep": 2,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}`

Smanar, 2021-11-17:
I think you have something to do with the endpoint.
You have the on/off cluster on the endpoint 0x02 but the device in the API use the 0x01

docloulou, 2021-11-17:
thanks for your reply, i have tried to move on/off cluster on the endpoint 0x01 without any result

Smanar, 2021-11-18:
>to move on/off cluster on the endpoint 0x01

No this is not possible, but on the DDF editor try change `$address.ext-0x01` and use 0x02

docloulou, 2021-11-19:
> It's possible that ? (or you mean moving on the firmware itself ?)
Yes i mean on the firmware :p 

> And can you show the second panel the "binding" one, because your ddf seem fine
The table is empty, i already tried to add binding in ddf without any result :(

Thanks for helping

i will try on zigbee2mqtt and see if its working

Smanar, 2021-11-19:
>The table is empty, i already tried to add binding in ddf without any result :(

So I think it s the problem.
Drag and drop the cluster 0x0006 inside to configure the bind and the report.

docloulou, 2021-12-09:
its finally working with the last version 2.13.3.
Thanks to dev team :)

closed 2021-12-09 by docloulou