dresden-elektronik/deconz-rest-plugin #5499: _TZ3000_ogceypug TS0502B

filipkotian, 2021-11-19:
## Device

- Product name: Immax neo (tuya)
- Manufacturer: Tuya _TZ3000_ogceypug
- Model identifier: TS0502B
- Device type :
  - Light
 

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2021-11-19:
Missing all screenshots.

Closing

closed 2021-11-19 by Mimiix

filipkotian, 2021-11-20:
@Mimiix I am sorry, could I still add theme to this topis? It was my first request.

Mimiix, 2021-11-20:
Feel free to do so.

Smanar, 2021-11-20:
But it 's a light device ? Not supported natively ?
The mesh support is native for all router.

filipkotian, 2021-11-20:
It is a light device. I can add it through phoscon, but always just few lights very close to conbee II stick. (I have around 20 of them in office)

Smanar, 2021-11-20:
So you mean the device is supported, but you have problem to pair them ?
Or if the device is too far away from the conbee, the device disconnect ?

So yes, some capture will be fine. The device can have mesh issue.

filipkotian, 2021-11-20:
Yes I can add it and also control it. But I am not able to connect devices which are too far away. I had it installed through hue gateway and it is working only when I use hue smart plugs as mesh repeaters.

Smanar, 2021-11-21:
Can you add some captures pls to check device caracteristics ?

filipkotian, 2021-11-21:
[image]
I only could get this screenshot in deconz, would it be enough? everything else was empty.

Mimiix, 2021-11-21:
The instructions are really clear how to read the clusters.

filipkotian, 2021-11-21:
Sorry, please find them attached:

Basic
[image]
Identify
[image]
Groups
[image]
[image]
ON/OFF
[image]
[image]
Level control
[image]
[image]
ZLL
[image]

Smanar, 2021-11-22:
I m sorry but all seem fine.
I don't see why the meshing is not working for this device, and I don't see similar issue elsewhere.
Perhaps a bug in the device firmware ?