dresden-elektronik/deconz-rest-plugin #5502: Device name

nicksterz, 2021-11-20:
## Device

- Product name: Stelpro Maestro(TM)
- Manufacturer: Stelpro
- Model identifier: SMT402AD
- Device type : 
  - Thermostat

## Screenshots

[image]

### Basic
[image]

### Identify
[image]

### Thermostat
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Thermostat User Interface Configuration
[image]

### Groups
[image]
[image]

### Relative humidity measurement
[image]

Mimiix, 2021-11-24:
Can you please also forum post?

nicksterz, 2021-11-24:
Done, sorry.

nicksterz, 2021-12-16:
*keep active*

nicksterz, 2022-01-30:
:(

closed 2022-03-01 by github-actions[bot]

BabaIsYou, 2022-03-26:
Wanted to buy this thermostat soon. Is it supported ? Just add DDF is sufficient ?

nicksterz, 2022-03-26:
I renewed this a bunch of times and let it go dead. What is DDF?

SwoopX, 2022-03-26:
It's not supported, but creating a DDF for it is sufficient to make it work with the common/generic thermostat capabilities. If there's anything very specific to the device (which I don't see here) that might be a bit more challenging.

BabaIsYou, 2022-03-26:
Hi. 
Seems difficult to have full support of Sinope and Stelpro thermostats even if they are two of the leaders in 240V baseboard smart thermostats domain  in North America. At around 100 bucks each, that's not a cheap try. Those more or less fully supported are already discontinued by manufacturer (that's North American market ! ;-)

Hope DDF will permit soon native writing on specific attributs/cluster like recently for Tuya.

BabaIsYou, 2022-03-26:
Device Description File. The still in progress future for deConz support of new devices using json in place of code.

nicksterz, 2022-03-26:
Apologies @BabaIs I have a novice level of expertise hence my device request. I have since noticed the DDF editor thanks to you and have been trying to make sense of it.

nicksterz, 2022-03-27:
@BabaIsYou I can report that I am able to at the very least set temperature and view temperature/humidity.

[image]

Can't get the humidity sensor to pass through to HA:

[image]

And I need to see an example DDF for a thermostat if I'm going to stand any chance of figuring out MODES and ON/OFF. As @SwoopX stated it is probably sufficient for people who know what they are doing lol

nicksterz, 2022-03-27:
Despite figuring that out as well, it appears as though mode selection doesn't turn off thermostat. As much as I now can turn it off in HA, if the requested temperature is higher than actual it will still stay on. This leads me to believe that mode selection doesn't actually do anything. As my thermostat is in the garage I need to physically turn off the breaker in winter as the temperature drops below device minimum of 5C. Regret my purchase

BabaIsYou, 2022-03-27:
It could be easier with an template of DDF. The other smart thermostats I use Z-Wave instead of Zigbee and the other one is partially supported to but the off function is working (it’s from Sinope).

I’m quite sure this Stelpro one is a good thermostat too and other plateform seems to be able to integrate it, but I don’t know if it’s partially or not.

If you can read the actual temp in HA may be you can send an off command to the thermostat rather than relying on mode ?

I don’t know HA at all, using Domoticz.

May be someone will have time to integrate this one … I hope.

nicksterz, 2022-03-27:
Home Assistant aside direct setting of mode to off within deconz does not turn off the thermostat. 

[image]

In most basic scenarios this is fine as one would never allow their home to drop below 5C however in a garage that is simply not the case. Also there is no changing the minimum heat setpoint limit as it appears hardcoded to the device.

[image]

Smanar, 2022-03-27:
IDK if thermostat can work only using DDF.
He is making one here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5793 and I think there is some code edition needed for that.

BabaIsYou, 2022-03-27:
I think that it’s normal not to be able to overwrite minimum setpoint. May be the thermostat accuracy has it’s proper limit as this point to !

On my Sinope thermostat I can put it off using mode modification and it turns off.

nicksterz, 2022-03-28:
I believe what he is trying to solve here is schedules which is more advanced 

> IDK if thermostat can work only using DDF. He is making one here #5793 and I think there is some code edition needed for that.
> 
> Here you have the @BabaIsYou one [#5848 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5848#issuecomment-1074403206) extract from his own thermostat, but his thermostat is realy more advanced than yours, so much thing to remove, I think it will be faster starting from scratch.
> 
> Just start the DDF and show the result you have, I will edit the text file.

nicksterz, 2022-03-28:
The reason for modifying the minHEATsetpoint is to try to set the temperature lower than anticipated temperature as a way to turn off the thermostat. My suspicion is that there isnt a way to turn the thermostat off as a mode setting. Therefore with 5 C as a minimum, the thermostat would be hitting that low of a temperature in a garage setting in winter. As the thermostat controls a fan heater on the ceiling below bedrooms, and that it is rather loud, I dont want it to cycle through the night. 

> Probably nothing to see with the error message you have , but for information, the attribute 0x0017 "min cool set point limit"
> 
> ```
> 
> The MinCoolSetpointLimit attribute specifies the minimum level that the cooling setpoint MAY be set to.
> The value is calculated as described in the LocalTemperature attribute. It must be greater than or equal to
> AbsMinCoolSetpointLimit. If this attribute is not present, it SHALL be taken as equal to
> AbsMinCoolSetpointLimit.
> ```
> 
> AbsMinCoolSetpointLimit is the attribute 0x0005

Smanar, 2022-03-28:
Ha ? you mean the attribute 0x001C don't work for you ?

nicksterz, 2022-03-28:
> t do that.

Correct, I can set it to off, read to confirm change, heater still runs

Smanar, 2022-03-28:
Strange, I m looking on Z2M and they are doing same, have you wait a little after sending the request, some TRV are realy lazy.

BabaIsYou, 2022-03-30:
I seems that Zigbee2MQTT are effectiveley most advanced on managing this sort of thermostat [Stelpro SMT402AD](https://www.zigbee2mqtt.io/devices/SMT402AD.html).
I'm looking at giving this a try as it could allow me to have more native control, even on Sinopé thermostat for setting outdoor temp or baclklight option.

BabaIsYou, 2022-04-07:
> > t do that.
> 
> Correct, I can set it to off, read to confirm change, heater still runs

Strange, as I can see [here](https://api.knack.com/v1/applications/54e658034b4f44e42fb18201/download/asset/5cd466420657e100067f6dd0/picssmt402ad.zip) SystemMode is implemented but I couldn't find a document giving the possible values.

BabaIsYou, 2022-04-07:
> I believe what he is trying to solve here is schedules which is more advanced

Absolutely not ... I'm not at this point ;-)
The DDF file is straight from Phoscon DDF editor interface without modifications.

BabaIsYou, 2022-04-09:
Well, I got one off this thermostat and can have it connected to my Zigbee network and visible in deConz exactly as @nicksterz described it in the beginning of this thread. But it's clearly not recognized in Phoscon App.

The DDF file is quite empty 
`{
  "schema": "devcap1.schema.json",
  "manufacturername": "Stelpro",
  "modelid": "SMT402AD",
  "product": "SMT402AD",
  "sleeper": false,
  "status": "Draft",
  "subdevices": []
}`

@Smanar, regarding the manufacturer documentation I found above, and a template from other Stelpro DDF that are perhaps supported, would it be easy to build one for this thermostat ?

BabaIsYou, 2022-04-09:
> Ha ? you mean the attribute 0x001C don't work for you ? It's rare, I know only the "Super TR" that do that.

I can confirm that it doesn't work at all.

nicksterz, 2022-04-09:
> > Ha ? you mean the attribute 0x001C don't work for you ? It's rare, I know only the "Super TR" that do that.
> 
> I can confirm that it doesn't work at all.

Thanks for checking, feel better knowing I'm not missing something obvious but I'm still forced to use this darn thermostat.

BabaIsYou, 2022-04-09:
You're welcome. However, it seems an issue that's note sahred by other software : https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/stelpro.js.

I guees we could find an issue for accessing the features of this thermostat.

BabaIsYou, 2022-04-11:
> @BabaIsYou I can report that I am able to at the very least set temperature and view temperature/humidity.
> 
> [image]
> 

Did you get this in Phoscon App (web)  ? Cause I couldn't get it that way. I could just address the thermostat using deConz GUI.

BabaIsYou, 2022-04-22:
> It's not supported, but creating a DDF for it is sufficient to make it work with the common/generic thermostat capabilities. If there's anything very specific to the device (which I don't see here) that might be a bit more challenging.

May be adressing Outdoor temp attribute from API as a /config ?
<attribute id="0x4001" name="Outdoor temp" type="s16" access="rw" required="m" mfcode="0x1185"></attribute>

nicksterz, 2022-04-23:
> > @BabaIsYou I can report that I am able to at the very least set temperature and view temperature/humidity.
> > [image]
> 
> Did you get this in Phoscon App (web) ? Cause I couldn't get it that way. I could just address the thermostat using deConz GUI.

Once I switched the revision it just popped up in Phoscon App.

BabaIsYou, 2022-05-04:
@nicksterz, waiting for some PR about Sinope to be reviewed I begin to have some read about Stelpro thermostats.
It seems that Stelpro uses a specific "system mode" , "setpoint mode" or "CustomThermostatMode" (depending on litteracy) attribute (0x401c in Thermostat cluster) to manage its modes between "heat", "off" or "eco" ([](https://github.com/stelpro/Ki-ZigBee-Thermostat/blob/master/devicetypes/stelpro/stelpro-ki-zigbee-thermostat.src/stelpro-ki-zigbee-thermostat.groovy#L481). May it's not the case for SMT402AD but I'll give it a try in a few days. I have to plug mine again (may be in a few days) to test that with just a valid DDF.

BabaIsYou, 2022-06-02:
Well, afer nearly a month I retry to connect my SMT402AD but I can't have it appearing in Phoscon like yours. I'll continue this on the specific thread for this device.

Smanar, 2022-06-02:
But it still working ?

BabaIsYou, 2022-06-02:
I'm still testing, some functionnalities seems working using Deconz GUI, even setting output temp on attribute 0x4001 cluster Thermsotat. But : 
- Item is still displayed with adress and not name, even after setting a DDF file

[image]

- Can't write on rw attribute 0x401c in cluster Thermostat (0x0201), seems like the trouble I had on Sinope with other definitiosn in general.xml superseed the Stelpro specific definiton (I have to make further test on it in general.xml). I think that accessing this specific attribute will perhaps help to put the thermostat in "off" state. Something that is not possible using standard System Mode attribute as @nicksterz and me have previously tested.
[image]
- Ultimately I wanted it to be accessible in Domoticz, when integration in Deconz will be completed, then I could integerate it in my schedules.

I began a DDF file but I still have trouble with bindings. May be someone can help a bit ;-)
`{
  "schema": "devcap1.schema.json",
  "manufacturername": "Stelpro",
  "modelid": "SMT402AD",
  "product": "SMT402AD",
  "sleeper": false,
  "status": "Silver",
  "path": "/devices/smt402ad.json",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/externalsensortemp",
          "description": "External sensor temperature used as Outdoor temperature on display.",
          "refresh.interval": 300,
          "read": {
            "at": "0x4001",
            "cl": "0x0201",
            "ep": 0,
            "fn": "zcl",
            "mf": "0x1185"
          },
          "parse": {
            "at": "0x4001",
            "cl": "0x0201",
            "eval": "Item.val = Attr.val;",
[... 150 lines cut ...]
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "description": "The current relative humidity in percent.",
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
`

BabaIsYou, 2022-06-02:
After making a little change in general.xml I can write in Stelpro specific system mode. I'll see this evening (back home) if it has some incidence on the thermostat ;-)

And the device appears in deConz wirth a name and Domoticz want to create new devices according to ZHATemperature, ZHAThermostat and ZHAHumidity (yes this thermostat has this sort of sensor).
[image]

[image]

That sounds good !

BabaIsYou, 2022-06-03:
You're right, there was no binding in my DDF, that's why I needed some help ;-)
I added some bindings as you suggested but that's doesn't chnage anything for the values to be reported to REST API.
The only thing I can do is to change the heatsetpoint but no humidity or temperature are reported.
When I try to update externalsensortemp I got a success but the value is not updated. If I change it in deCONZ GUI the value is displayed on thermostat display.
Here is my DDF :

> {
>   "schema": "devcap1.schema.json",
>   "manufacturername": "Stelpro",
>   "modelid": "SMT402AD",
>   "product": "SMT402AD",
>   "sleeper": false,
>   "status": "Silver",
>   "path": "/devices/smt402ad.json",
>   "subdevices": [
>     {
>       "type": "$TYPE_THERMOSTAT",
>       "restapi": "/sensors",
>       "uuid": [
>         "$address.ext",
>         "0x01",
>         "0x0201"
>       ],
>       "items": [
>         {
>           "name": "attr/id"
>         },
>         {
>           "name": "attr/lastannounced"
>         },
>         {
>           "name": "attr/lastseen"
>         },
>         {
>           "name": "attr/manufacturername"
>         },
>         {
>           "name": "attr/modelid"
>         },
>         {
>           "name": "attr/name"
>         },
>         {
>           "name": "attr/swversion"
>         },
>         {
>           "name": "attr/type"
>         },
>         {
>           "name": "attr/uniqueid"
>         },
>         {
>           "name": "config/externalsensortemp",
>           "description": "External sensor temperature used as Outdoor temperature on display.",
>           "refresh.interval": 300,
>           "read": {
>             "at": "0x4001",
>             "cl": "0x0201",
>             "ep": 1,
>             "fn": "zcl",
>             "mf": "0x1185"
>           },
>           "write": {
>             "at": "0x4001",
>             "cl": "0x0201",
>             "dt": "0x29",
>             "ep": 1,
>             "eval": "Item.val",
>             "fn": "zcl",
>             "mf": "0x1185"
[... 142 lines cut ...]
>           "dt": "0x29",
>           "min": 300,
>           "max": 3600,
>           "change": "0x00000001"
>         },
>         {
>           "at": "0x0012",
>           "dt": "0x29",
>           "min": 1,
>           "max": 300,
>           "change": "0x00000001"
>         },
>         {
>           "at": "0x4001",
>           "dt": "0x29",
>           "min": 300,
>           "max": 3600,
>           "change": "0x00000001"
>         }
>       ]
>     }
>   ]
> }

Regarding System mode, I tried using 0x0401C in addition to 0x001C because it seems that it's the way it is implemented in another Stelpro Thermostat ([Ki](https://github.com/stelpro/Ki-ZigBee-Thermostat/blob/master/devicetypes/stelpro/stelpro-ki-zigbee-thermostat.src/stelpro-ki-zigbee-thermostat.groovy#L481) model), and it was defined in general.xml before my changes, but unfortunately none of these attributes seems to be able to cut the thermostat Off.

BabaIsYou, 2022-06-03:
Complement ... Here is sensors view from Phoscon, it seems that externalsensortemp and heatsetpoint are updated, but even if the second go to the device, the first clearly not ... And humidity and tempratures (from ZHAThermostat or ZHATemperature) are still 0 or null (because never announced)
`    "20": {
        "config": {
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "etag": "3cd244b2373983196eacc63512ebd94a",
        "lastannounced": "2022-06-02T22:27:16Z",
        "lastseen": "2022-06-03T20:44Z",
        "manufacturername": "Stelpro",
        "modelid": "SMT402AD",
        "name": "Temperature  CdM",
        "state": {
            "lastupdated": "none",
            "temperature": null
        },
        "type": "ZHATemperature",
        "uniqueid": "f8:f0:05:ff:ff:70:7e:95-ff-0402"
    },
    "21": {
        "config": {
            "externalsensortemp": 2330,
            "heatsetpoint": 2200,
            "mode": "heat",
            "offset": 0,
            "on": true,
            "reachable": true,
            "unoccupiedheatsetpoint": null
        },
        "etag": "f29048b1e61c4e7443c09f4e4f065a67",
        "lastannounced": null,
        "lastseen": "2022-06-03T20:44Z",
        "manufacturername": "Stelpro",
        "modelid": "SMT402AD",
        "name": "Thermostat CdM",
        "state": {
            "lastupdated": "none",
            "on": null,
            "temperature": 0,
            "valve": 0
        },
        "type": "ZHAThermostat",
        "uniqueid": "f8:f0:05:ff:ff:70:7e:95-01-0201"
    },
    "22": {
        "config": {
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "etag": "ed692eab47baa29c4a3156c06accfbe7",
        "lastannounced": null,
        "lastseen": "2022-06-03T20:44Z",
        "manufacturername": "Stelpro",
        "modelid": "SMT402AD",
        "name": "Humidité CdM",
        "state": {
            "humidity": 0,
            "lastupdated": "none"
        },
        "type": "ZHAHumidity",
        "uniqueid": "f8:f0:05:ff:ff:70:7e:95-01-0405"
    }
}

                    
   `

BabaIsYou, 2022-06-03:
Looking to solve my issue I found some weird things for me 
- The fingerprint part is missing in my DDF compared to the Sinope one I have, then I added this 

>             
"fingerprint": {
        "profile": "0x0104",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0003",
          "0x0201",
          "0x0204",
          "0x0405"
        ]
that changed nothing ...
- Looking to other repo (https://github.com/Philippe-Charette/Hubitat-Stelpro-Maestro-Thermostat/blob/master/stelpro-maestro-thermostat.groovy) it seems that Endpoint is not 0x01 but 0x19 (see line 19)
- When I use deCONZ interface to simulate bindings rather than using text editor, the EP is also 0x19 ?!
[image]
Does it make sense ?
What do I have to change in the DDF if its 0x19 and not 0x01 because there are a lot of places with ep (source or dest)

Smanar, 2022-06-04:
Ha yes, this device use the cluster 0x19, not the 0x01

[image]

I don't think fingerprint is usefull for thermostat (but I can be wrong)

But using the good endpoint will change lot of thing, so much corrective to do on the DDF.

BabaIsYou, 2022-06-06:
Hi,

I'm trying to modify the DDF but just to be sure : 
- Do I have to add "ep": "0x19" for each attribute or juste for those that are not reported (humidity, temperature and externalsensortemp that return success but that is not updated ?
- Or may be just updating bindings is sufficient ? For bindings what ep do I have to update, dst.ep, src.ep or both to value              "0x19" ?

Another question, why is heatsetpoint updated event if endpoint is wrong ?

Smanar, 2022-06-06:
```
{
"name": "config/heatsetpoint",
"description": "Target temperature of a thermostat.",
"default": 0
},
```

There is no endpoint specified here, so it s the legacy code that handle it.

```
"name": "config/externalsensortemp",
"description": "External sensor temperature used as Outdoor temperature on display.",
"refresh.interval": 300,
"read": {
"at": "0x4001",
"cl": "0x0201",
"ep": 1,
"fn": "zcl",
"mf": "0x1185"
},
```

Here you need to use 0x19

```
"uuid": [
"$address.ext",
"0x01",
"0x0405"
],
```

Here it's just cosmetic, but will be logic to use the good one too.

And ofc need to chnage all "src.ep" in binds.

BabaIsYou, 2022-06-06:
I'm progressing. Modfiying DDF file I got : 
- Humidity seems to report now,

>                     {
>     "config": {
>         "offset": 0,
>         "on": true,
>         "reachable": true
>     },
>     "etag": "d09312e7ae5977cf4c4d2625430cecc9",
>     "lastannounced": null,
>     "lastseen": "2022-06-06T14:36Z",
>     "manufacturername": "Stelpro",
>     "modelid": "SMT402AD",
>     "name": "Humidité CdM",
>     "state": {
>         "humidity": 4500,
>         "lastupdated": "2022-06-06T14:36:38.235"
>     },
>     "type": "ZHAHumidity",
>     "uniqueid": "f8:f0:05:ff:ff:70:7e:95-01-0405"
> }
>                     
>                 

- Temperature is still not reporting, 
>                     {
>     "config": {
>         "externalsensortemp": 1950,
>         "heatsetpoint": 1800,
>         "mode": "heat",
>         "offset": 0,
>         "on": true,
>         "reachable": true,
>         "unoccupiedheatsetpoint": 1700
>     },
>     "etag": "8bc7632f7c1181c76bdd75b9ee7b2ae5",
>     "lastannounced": null,
>     "lastseen": "2022-06-06T14:41Z",
>     "manufacturername": "Stelpro",
>     "modelid": "SMT402AD",
>     "name": "Thermostat CdM",
>     "state": {
>         "lastupdated": "2022-06-06T14:29:54.083",
>         "on": null,
>         "temperature": 0,
>         "valve": 0
>     },
>     "type": "ZHAThermostat",
>     "uniqueid": "f8:f0:05:ff:ff:70:7e:95-01-0201"
> }
>                     
>                 
- externalsensortemp is updating on thermostat, even if debug wirtes errors : 

> 10:36:50:080 SC tick --> StateRead
> 10:36:50:151 f8:f0:05:ff:ff:70:7e:95-01-0201/config/externalsensortemp expression: Item.val = Attr.val; --> 1950
> 10:36:50:152 No thermostat sensor found for 0xF8F005FFFF707E95, endpoint: 0x19
> 10:36:50:173 SC config/externalsensortemp: synced
> 10:36:50:174 SC --> StateFinished
> 10:36:50:175 SC state change finished: f8:f0:05:ff:ff:70:7e:95-01-0201

My DDF become : 

> {
  "schema": "devcap1.schema.json",
  "manufacturername": "Stelpro",
  "modelid": "SMT402AD",
  "product": "SMT402AD",
  "sleeper": false,
[... 219 lines cut ...]
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 25,
      "cl": "0x0405"
    },
    {
      "bind": "unicast",
      "src.ep": 25,
      "cl": "0x0204"
    },
    {
      "bind": "unicast",
      "src.ep": 25,
      "cl": "0x0201"
    }
  ]
}

BabaIsYou, 2022-06-06:
I got it : Temperature is reporting now, changing

> {
> "name": "state/temperature",
> "description": "The current temperature in °C (*100).",
> "parse": {
> "at": "0x0000",
> "cl": "0x0402",
> "ep": 25,
> "eval": "Item.val = Attr.val + R.item('config/offset').val",
> "fn": "zcl"
> },
> "default": 0
> },

to

```
{
"name": "state/temperature",
"description": "The current temperature in °C (*100).",
"parse": {
"at": "0x0000",
"cl": "0x0201",
"ep": 25,
"eval": "Item.val = Attr.val + R.item('config/offset').val",
"fn": "zcl"
},
"default": 0
},

Continue testing ... Many thanks @Smanar

BabaIsYou, 2022-06-06:
Thanks. Corrected and PR openened.

0201 is the thermostat but the endpoint is still 0x01, the code can probably handle it (as it's just cosmetic), but you will prevent issue using 0x019 too for the unique id creation.

PR #6877 (2023-04-16): Add fingerprint in SMT402AD thermostat DDF
Add fingerprint in smt402ad DDF to avoid error in log like "no thermostat found in cluster 0x19" but with all seemed to report anyway.

Fix #5502