dresden-elektronik/deconz-rest-plugin #5505: Device name

orionevo14, 2021-11-21:
## Device

- Product name: Tuya Moes ZTS-EUB
- Manufacturer: _TZE200_amp6tsvy
- Model identifier: TS0601
- Device type : 
  - Switch

Seems to be similar to this one : 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2879

## Screenshots

[image]

[image]

### Basic

[image]

### Groups

[image]

### Scenes

[image]

### Other clusters that are not mentioned above
Tuya specific cluster
[image]

Smanar, 2021-11-21:
Hello, it s a wired device ?

orionevo14, 2021-11-21:
Hello, yes it is.
I'm running deCONZ on a RPi4 with Home Assistant integration so I don't have the latest version, only the 2.12.6 so no DDF I think

chatainsim, 2021-11-22:
Hello, same here with a 2 gangs.
[image]
Basic
[image]
Groups
[image]
Scenes
[image]
Others Clusters
[image]

Same version, Deconz add-on on HA 2.16.6

Thanks

Smanar, 2021-11-22:
I can explain a way to make it working in 5mn for almost all tuya switch (that use already supported tuya command) and without compiling the code, but you need at least a version that support DDF.

And this method will be beta (disabled at every restart for the moment, waiting for next deconz version)

chatainsim, 2021-11-22:
> I can explain a way to make it working in 5mn for almost all tuya switch (that use already supported tuya command) and without compiling the code, but you need at least a version that support DDF. [...]

How can we check if there is DDF support ?

orionevo14, 2021-11-22:
> I can explain a way to make it working in 5mn for almost all tuya switch (that use already supported tuya command) and without compiling the code, but you need at least a version that support DDF. [...]

Hello @Smanar 
I can wait. When the method you're talking about will be considered as stable ? In the next 2.13 stable version ?

Smanar, 2021-11-22:
No sure, will take some time to have the DDF core full working. But if the autodisable at sartup is removed (it s for me the more boring) you can already start to use it.
But yes on the next stable DDF will be usable, even still beta. It will work for exemple for wired switch (already tested), but not battery one.

orionevo14, 2021-11-22:
Ok thank you for the detailed information, and fast answers too.
Do you have a link or something to some documentation about the way to fix it ? Will surely test that when I'll be able to use 2.13.x

chatainsim, 2021-11-23:
2.13 stable should be release soon I hope:

```
Current Beta: v2.13.2-beta
Current Stable: v2.12.6

Next Beta: v2.13.3-beta
Next Stable: v2.13.x expected in November.
```
We also have to wait the HA addon update then.

Mimiix, 2021-11-23:
@chatainsim I shouldn't get my hopes up for november. I think december or perhaps january is a more appropriate timeframe.

Smanar, 2021-11-23:
For the documentation about DDF (sorry the documentation is WIP too)
The hard part > https://dresden-elektronik.github.io/deconz-dev-doc/modules/ddf/
The easy one > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

You can find some device already done with capture on the forum or the github

Mimiix, 2021-11-24:
@orionevo14 Missing a forum post aswell. Can you please make one?

orionevo14, 2021-11-24:
> @orionevo14 Missing a forum post aswell. Can you please make one?

Of course !
[https://forum.phoscon.de/t/tuya-moes-zts-eub-1-gang-wired-switch/946](url)

Der-WU, 2021-12-16:
Hat anyone the ddf Files ready dir the 1 and 2 was Switch?
Are they still in Progress?

Great

Smanar, 2021-12-16:
If it s a wired device, it can be possible if you have a deconz version that support DDF.

Take a look here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

As device type take on/off  light, and you can let all the rest empty, because the device don't use report, don't use bind, don't use classic cluster.

Der-WU, 2021-12-16:
I do it. And add 2 on/off light.
But in phoscon it is shown as lightchain and only one light.

Smanar, 2021-12-16:
Can you show your DDF ?
You need to use fake Endpoint if you have 2 gangs, because I think at the base you have only one.

Der-WU, 2021-12-16:
My DDF looks :

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_g1ib5ldv",
  "modelid": "TS0601",
  "product": "TS0601",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/ts0601.json",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}

The Cluster  are like the 4th Post.

Smanar, 2021-12-17:
Ha nice DDF, I would have done the same 

>But in phoscon it is shown as lightchain and only one light.

I m not sure to understand but phoscon can "custom" display.
Can you checks the 2 devices json ? You can find them in phoscon / help / api information, the result will be more "raw" without phoscon interpretation.

Der-WU, 2021-12-18:
I found some info about the state of ON/OFF.

https://www.zigbee2mqtt.io/devices/TS0601_switch_3_gang.html#tuya-ts0601_switch_3_gang

This is for the 3 way switch. 

The 2 way switch used Endpint 1 and 3.

Smanar, 2021-12-18:
>The 2 way switch used Endpint 1 and 3.

But it's  virtual endpoint, your device have only one in reality.
It s easy to check, enable deconz log with "info", you will see line with "tuya debug", and press the 2 buttons, to mimic the request.

chatainsim, 2022-01-04:
@Der-WU did you manage to make it work ?
I've tried to edit the DDF but it's not working.
To light appear in Phoscon but nothing in HA.
They appear in Phoscon, Help, API information.

Smanar, 2022-01-04:
>To light appear in Phoscon but nothing in HA.

Not normal, HA use the same API than deconz

But BTW you have used the same DDF than him ?
How they are working in phoscon ?
Can you show the 2 device json as result of this DDF ?

chatainsim, 2022-01-10:
> > To light appear in Phoscon but nothing in HA.
> 
> Not normal, HA use the same API than deconz
> 
> But BTW you have used the same DDF than him ? How they are working in phoscon ? Can you show the 2 device json as result of this DDF ?

Yes, i'ved used the same DDF has @Der-WU 
In Phoson, state is not changing.

json from Phoscon API?

If yes:
```
{
    "etag": "bda2f966ede11c8e1f2e6978eee55da8",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2022-01-04T12:08Z",
    "manufacturername": "_TZE200_g1ib5ldv",
    "modelid": "TS0601",
    "name": "On/Off light 36",
    "state": {
        "on": false,
        "reachable": true
    },
    "swversion": null,
    "type": "On/Off light",
    "uniqueid": "5c:02:72:ff:fe:22:90:fd-02"
}
```
```

                    
{
    "etag": "bda2f966ede11c8e1f2e6978eee55da8",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2022-01-04T12:08Z",
    "manufacturername": "_TZE200_g1ib5ldv",
    "modelid": "TS0601",
    "name": "On/Off light 37",
    "state": {
        "on": false,
        "reachable": true
    },
    "swversion": null,
    "type": "On/Off light",
    "uniqueid": "5c:02:72:ff:fe:22:90:fd-01"
}
```

Smanar, 2022-01-10:
You have the 2 gangs version ?
Do you have the deconz application (with the GUI) to check logs when making the request (with flag "info") ?

chatainsim, 2022-01-10:
Yes I have the 2 gangs version.
I have the deconz app with the GUI in home assistant. I don't know how can
I export logs with this version.

Smanar, 2022-01-11:
No need to export, a screen capture can be enought (if logs are not too much talkative), take only the flag "info_l2".

chatainsim, 2022-01-17:
Here are the logs.
[image]
I hope the usefull infos are captured. There is a lot's of logs.
Thanks

chatainsim, 2022-01-17:
Here is my DDF
[image]
[image]
Maybe I made some mistake.

Smanar, 2022-01-17:
You only have logs with "info" and not "infol2" so all is not complete but we can see the device sending report and the good one.
But nothing appear after ...
I think the DDF is good, but why the code don't proceed the data.

You are sure not having a sensor with same address ? If I add more debug line on code, are you able to recompile the code ? (not possible on HA)

chatainsim, 2022-01-24:
> You are sure not having a sensor with same address ? If I add more debug line on code, are you able to recompile the code ? (not possible on HA)

Maybe this is my issue. I haven't checked that. How can I see address of the sensor ? Thanks

Smanar, 2022-01-24:
Just go in phoscon / help / api information / sensors, and look if you have not one with the same uniqueid

`"uniqueid": "5c:02:72:ff:fe:22:90:fd-01"`

Need to check only for 5c:02:72:ff:fe:22:90:fd

chatainsim, 2022-01-27:
I've found it two times:
```
        "36": {
            "etag": "9ec59053bdb0773ffe6ab39848108a56",
            "hascolor": false,
            "lastannounced": null,
            "lastseen": "2022-01-18T06:24Z",
            "manufacturername": "_TZE200_g1ib5ldv",
            "modelid": "TS0601",
            "name": "On/Off light 36",
            "state": {
                "on": false,
                "reachable": true
            },
            "swversion": null,
            "type": "On/Off light",
            "uniqueid": "5c:02:72:ff:fe:22:90:fd-02"
        },
        "37": {
            "etag": "194121a36ea6fc29f66417a56f849d5c",
            "hascolor": false,
            "lastannounced": null,
            "lastseen": "2022-01-18T06:24Z",
            "manufacturername": "_TZE200_g1ib5ldv",
            "modelid": "TS0601",
            "name": "On/Off light 37",
            "state": {
                "on": false,
                "reachable": true
            },
            "swversion": null,
            "type": "On/Off light",
            "uniqueid": "5c:02:72:ff:fe:22:90:fd-01"
        },
```
But it's what i've defined mylself.

chatainsim, 2022-02-18:
Hello,
Sorry about the delay.
Here are some screenshots.
Logs:
[image]

DDF:
[image]
[image]

Thank you.

chatainsim, 2022-02-18:
ok, some news, it appear in HA now.

One of the switch is inverted.
It appear on in HA but it's off and viceversa.
And the other one is not working at all.

It's a good start :)

Smanar, 2022-02-19:
>One of the switch is inverted.

Here it s bad luck, the DDF core don't support yet the tya cluster, so no way to rever it

>And the other one is not working at all.

Here we need the fake entry, on your DDF I m seing you are creating 2 entrie, one at the endpoint 0x01 (the real one) and one at the endpoint 0x02.
Both are created on the API ?

chatainsim, 2022-02-19:
Yes both are created on the API.

Smanar, 2022-02-19:
Can you show the device JSON of both entries ? (Visible for exemple in phoscon / help / API Information) please ?

chatainsim, 2022-02-19:
Here is the device JSON:
```
"36": {
        "etag": "c8fee577534c3c751746b142cb5c15cc",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2022-02-18T23:01Z",
        "manufacturername": "_TZE200_g1ib5ldv",
        "modelid": "TS0601",
        "name": "On/Off light 36",
        "state": {
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "5c:02:72:ff:fe:22:90:fd-02"
    },
    "37": {
        "etag": "c8fee577534c3c751746b142cb5c15cc",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2022-02-18T23:01Z",
        "manufacturername": "_TZE200_g1ib5ldv",
        "modelid": "TS0601",
        "name": "On/Off light 37",
        "state": {
            "on": true,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "5c:02:72:ff:fe:22:90:fd-01"
    },
```

Smanar, 2022-02-19:
Ok so on the debug line with command (0x01 0x02) with data = 1 you need to have a websocket notification too ...

`"uuid": [ "$address.ext", "0x02"],`

Perhaps this part is just cosmetic ... I will ask to manup.

Smanar, 2022-03-13:
So I have answer, this part is realy just cosmetic.
So not possible to create fake entry with fake endpoint to support this kind of device yet.

chatainsim, 2022-03-13:
Too bad. Thank you for your answer! I will store it in my closet and some day maybe I will able to use it. 😉

TheAlphaLaw, 2022-09-05:
Is there any additional info on the above switch. Still not able to utilize in GUI or turn light off or on.

Smanar, 2022-09-05:
Hello, just to try, can you try this DDF
```

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_g1ib5ldv",
  "modelid": "TS0601",
  "product": "TS0601",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {
            "fn": "tuya",
            "dpid": 1,
            "eval": "Item.val = Attr.val;"
          },
          "write": {
            "fn": "tuya",
            "dpid": 1,
            "dt": "0x10",
            "eval": "Item.val == 1 ? 1 : 0;"
          },
          "read": {
            "fn": "tuya"
          },
          "refresh.interval": 300
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {
            "fn": "tuya",
            "dpid": 2,
            "eval": "Item.val = Attr.val;"
          },
          "write": {
            "fn": "tuya",
            "dpid": 2,
            "dt": "0x10",
            "eval": "Item.val == 1 ? 1 : 0;"
          },
          "read": {
            "fn": "none"
          },
          "refresh.interval": 300
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

Mimiix, 2022-09-05:
Reopening.

TheAlphaLaw, 2022-09-07:
> Hello, just to try, can you try this DDF ? (long time we haven't check with recent deconz version)
> 
> ```
> 
> {
>   "schema": "devcap1.schema.json",
>   "manufacturername": "_TZE200_g1ib5ldv",
>   "modelid": "TS0601",
>   "product": "TS0601",
>   "sleeper": false,
>   "status": "Gold",
>   "subdevices": [
>     {
>       "type": "$TYPE_ON_OFF_LIGHT",
>       "restapi": "/lights",
>       "uuid": [
>         "$address.ext",
>         "0x01"
>       ],
>       "items": [
>         {
>           "name": "attr/id"
>         },
>         {
>           "name": "attr/lastannounced"
>         },
>         {
>           "name": "attr/lastseen"
>         },
>         {
>           "name": "attr/manufacturername"
>         },
>         {
>           "name": "attr/modelid"
>         },
>         {
>           "name": "attr/name"
>         },
>         {
>           "name": "attr/swversion"
>         },
>         {
>           "name": "attr/type"
>         },
>         {
>           "name": "attr/uniqueid"
>         },
>         {
>           "name": "state/on",
>           "parse": {
>             "fn": "tuya",
>             "dpid": 1,
>             "eval": "Item.val = Attr.val;"
>           },
>           "write": {
>             "fn": "tuya",
>             "dpid": 1,
>             "dt": "0x10",
>             "eval": "Item.val == 1 ? 1 : 0;"
>           },
>           "read": {
>             "fn": "tuya"
>           },
>           "refresh.interval": 300
>         },
>         {
>           "name": "state/reachable"
>         }
>       ]
>     },
>     {
>       "type": "$TYPE_ON_OFF_LIGHT",
>       "restapi": "/lights",
>       "uuid": [
>         "$address.ext",
>         "0x02"
>       ],
>       "items": [
>         {
>           "name": "attr/id"
>         },
>         {
>           "name": "attr/lastannounced"
>         },
>         {
>           "name": "attr/lastseen"
>         },
>         {
>           "name": "attr/manufacturername"
>         },
>         {
>           "name": "attr/modelid"
>         },
>         {
>           "name": "attr/name"
>         },
>         {
>           "name": "attr/swversion"
>         },
>         {
>           "name": "attr/type"
>         },
>         {
>           "name": "attr/uniqueid"
>         },
>         {
>           "name": "state/on",
>           "parse": {
>             "fn": "tuya",
>             "dpid": 2,
>             "eval": "Item.val = Attr.val;"
>           },
>           "write": {
>             "fn": "tuya",
>             "dpid": 2,
>             "dt": "0x10",
>             "eval": "Item.val == 1 ? 1 : 0;"
>           },
>           "read": {
>             "fn": "none"
>           },
>           "refresh.interval": 300
>         },
>         {
>           "name": "state/reachable"
>         }
>       ]
>     }
>   ]
> }
> ```

Is there a simple way to copy this info in? I see edit DDF. Thanks!

TheAlphaLaw, 2022-09-08:
> Yep you can
> 
> * make "edit DDF"
> * Save the file, with looking for the name and the path
> * Edit the file with a text editor and paste this contain
> 
> Or can just create a json file in the "devices" folder, what is your OS ?

Win 11. I used visual basic to make a JSON files and edited DDF. No change.

Smanar, 2022-09-08:
Deconz check all DDF only at start, so perhaps with restarting deconz.
If you make "edit DDF" now, you will have the edited DDF no ? If you use the editor to make so changes, you can use the "hot reload" fonction in the editor to not have to restart deconz.

TheAlphaLaw, 2022-09-09:
> Deconz check all DDF only at start, so perhaps with restarting deconz. If you make "edit DDF" now, you will have the edited DDF no ? If you use the editor to make so changes, you can use the "hot reload" fonction in the editor to not have to restart deconz.
> 
> If the modified DDF is not used, create a json file in the "devices" folder, it s the defaut folder used by deconz.

Already done. Does not appear in Home Assistant or Phosconz.

Smanar, 2022-09-10:
And if you make "Edit DDF" on the GUI on the node ?
It open the good file ? (the file and the name is visible on the editor title)

TheAlphaLaw, 2022-09-14:
> And if you make "Edit DDF" on the GUI on the node with the right clic menu ? It open the good file ? (the file and the name is visible on the editor title)

Correct, but still not picked up by Phoscon or Home Assistant.

Smanar, 2022-09-14:
Can you enable log in deconz with "DDF" and "info" ("info_l2 if possible) and on the DDF editor make "hot reload"

TheAlphaLaw, 2022-09-16:
> Can you enable log in deconz with "DDF" and "info" ("info_l2 if possible) and on the DDF editor make "hot reload"

Nothing shows in either log for that device after hot reload...

Smanar, 2022-09-16:
With "info" you need to see
```
12:35:47:792 update ddf TS0601 index 1
12:35:47:794 Hot reload device: 84:fd:27:ff:fe:89:ae:21
12:35:47:902 DEV found DDF for 0x84FD27FFFE892A21, path: /home/pi/.local/share/dresden-elektronik/deCONZ/devices/dimmer_module.json
```

Else it mean the DDF is not used, why why it's the one selected on the editor when you make "edit DDF", you are sure it's the same and path in the window title bar ?

Thoses value are good, visible in "node information" or cluster 0x0000 ?

```
  "manufacturername": "_TZE200_g1ib5ldv",
  "modelid": "TS0601",
```

nixagit, 2022-09-22:
When I try this the json file doesn't load on restart. I can open the json from Edit DDF menu but node remains 0xXXXX even after hot reload. It's correctly named in the /devices folder. Hot reload isn't noted in the info debug. It just says:
`update ddf TS0601 index 0`
Nothing more than that.
I have a 1-gang switch so only included the first endpoint.

Smanar, 2022-09-27:
For me it mean the DDF is not used at all.
- The DDF is not at the good place
- The device have different model id / manufacture name, perhaps a typo somewhere ...

nixagit, 2022-09-28:
Deconz version is 2.17.1

I can't see any errors/typos.

Smanar, 2022-09-28:
Ha ? Oups, yeah after reading there is 2 devices in this post, sorry.
`"modelid": "TS0601"` is good ?

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_amp6tsvy",
  "modelid": "TS0601",
  "product": "TS0601",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {
            "fn": "tuya",
            "dpid": 1,
            "eval": "Item.val = Attr.val;"
          },
          "write": {
            "fn": "tuya",
            "dpid": 1,
            "dt": "0x10",
            "eval": "Item.val == 1 ? 1 : 0;"
          },
          "read": {
            "fn": "tuya"
          },
          "refresh.interval": 300
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {
            "fn": "tuya",
            "dpid": 2,
            "eval": "Item.val = Attr.val;"
          },
          "write": {
            "fn": "tuya",
            "dpid": 2,
            "dt": "0x10",
            "eval": "Item.val == 1 ? 1 : 0;"
          },
          "read": {
            "fn": "none"
          },
          "refresh.interval": 300
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}

nixagit, 2022-09-29:
Sorry for ths confusion. I assumed the manufacturer name would be the same for all devices. Apparently not. Anyway, it now works. The switch shows up in the web gui as "fairy lights" but works as expected and I can control it through Home Assistant.

My switch is 1-gang so this is the json I used:

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_amp6tsvy",
  "modelid": "TS0601",
  "product": "TS0601",
  "sleeper": false,
  "status": "Gold",
  "path": "/home/pi/.local/share/dresden-elektronik/deCONZ/devices/ts0601.json",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 300,
          "read": {
            "fn": "tuya"
          },
          "write": {
            "dpid": 1,
            "dt": "0x10",
            "eval": "Item.val == 1 ? 1 : 0;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 1,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}`

Smanar, 2022-09-29:
Nice, thx, Pr submitted https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6366

[4 more replies]

closed 2022-10-16 by manup