dresden-elektronik/deconz-rest-plugin #5512: Philips Hue Outdoor Motion Sensor

mgrn0, 2021-11-24:
## Philips Hue Outdoor Motion Sensor

- Product name: Motion Sensor EU (SLR) | Philips Hue
- Manufacturer: Philips
- Model identifier: SML002
- Device type : 
  - Sensor
  
## Screenshots

### Basic

[image]

[image]

### Identify
[image]

### Device Temperature

[image]

### Illuminance

[image]

### Occupancy

[image]

### OTAU

[image]

### Power Configuration

[image]

### Node Info

[image]

mgrn0, 2021-11-24:
Ah, I forgot: this is what the API shows for that specific sensor with latest beta (2.13.02):

[image]

mgrn0, 2021-11-24:
And this is how the sensor looks like (just not to mismatch it with the philips indoor motion sensor):

[image]

Mimiix, 2021-11-24:
This device is already supported? 

Can you provide the macAdress please?

mgrn0, 2021-11-24:
00:17:88:01:06:49:fa:48-02-0406 that should be.

I know that the 'compatibile devices list' states compatibility, I thought that mabye I received a new version of the sensor since it does not work..

Mimiix, 2021-11-24:
00:17:88:01:06:49:fa:48-02-0406

In your edit, is this really what the rest api showed?

mgrn0, 2021-11-24:
uh. Maybe I should have just waited. Sensor works now, maybe because I have manually read all values of all clusters one by one.. ? Sorry for the hassle. Will close this request now.

Mimiix, 2021-11-24:
There people reporting issues with them. That's why I ask. Can you please share a rest api message again?

bdickhaus, 2021-11-30:
I don't know if this helps, or not, but I have one working with this id 00:17:88:01:06:44:b6:45-02-0406

closed 2021-12-30 by github-actions[bot]