dresden-elektronik/deconz-rest-plugin #5515: Centralite Temp and Humidity Sensor

colin-haber, 2021-11-25:
## Device

- Product name: Temp and Humidity Sensor
- Manufacturer: CentraLite
- Model identifier: 3310-G
- Device type:
  - Sensor

## Screenshots

[image]

### Basic
[image]

### Identify
[image]

### Diagnostics
[image]

### Other clusters that are not mentioned above
[image]
[image]
[image]

633kd4d, 2022-02-05:
Looks like this was closed without any action...
Can I request support for this device as well?

Mimiix, 2022-02-05:
Reopened

Smanar, 2022-02-05:
I think it can be done using DDF.
Take a look here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5738

633kd4d, 2022-02-05:
I can get part of it working that way. There is an unknown cluster 0xfc45 that is probably the humidity sensor. No idea how to make that work through DDF.

Stim04, 2022-02-06:
> I can get part of it working that way. There is an unknown cluster 0xfc45 that is probably the humidity sensor. No idea how to make that work through DDF.

@633kd4d If you can read the humidity value on a the cluster info and that you can see it fluctuating there is a big chance that you will find a way to get it work with DDF, otherwise you'd need a zigbee sniffer to look for cluster info and endpoints for this value. 

@Smanar I speak under your control as I'm just a noob in all that stuff.

633kd4d, 2022-02-06:
Yeah. The cluster doesn't show anything for the humidity. I also noticed that the DDF stuff I set up only returned one value and never updated after that. I've never messed with zigbee protocols, so I'm sure I just missed something in the setup.

Stim04, 2022-02-06:
Maybe not, might have to reset and integrate it again while noot deleting the node. Worked for my sensor that i created my DDF. I had some issues to make it report temp at first pairings, but now i know the trick to reset without deleting the node on deconz. Hope it helps

633kd4d, 2022-02-06:
Still not responding. No idea why.
This is the only device I have with a Poll Control cluster. Maybe that has something to do with it?
The little blue/green indicator at the top left of the node seems to indicate that the device is constantly communicating.

Stim04, 2022-02-06:
then you maybe messed up while creating the DDF, the part where you drag and drop can be a little tricky, and also don't forget to put some min and max req. intervals values.

Do you see the temperature fluctuating when you read the temperature cluster and press the refresh button on your end device?

633kd4d, 2022-02-06:
Yes. If I click the read button, the value shows up in the temperature measurement cluster, but that is the only time it changes.

Are you referring to the min max values in the polling cluster?
[image]

Stim04, 2022-02-06:
no i'm talking about the temperature cluster in the DDF, after you drag and drop. Look the last screenshot I posted you can see it for the battery voltage

633kd4d, 2022-02-06:
I think I'll have to move the hardware to a different system to do any of this. It is currently running under Home Assistant. Unfortunately that means it is running in VNC with a window manager that only allows ONE window to be visible at a time. So, no way to drag between windows. Annoying
[image]
.

Stim04, 2022-02-06:
No Draging = no Easy coding of the JSON

You can try your luck modifying my JSON with your manufacturer and stuff. I'm using same 402 cluster for temperature.

https://pastebin.com/EQF35gLk

Stim04, 2022-02-06:
try this it might work

https://pastebin.com/qfegcwac

Smanar, 2022-02-06:
> I can get part of it working that way. There is an unknown cluster 0xfc45 that is probably the humidity sensor. No idea how to make that work through DDF.

Ha yes, right sorry, this device Use the attribute 0x0000 on the cluster 0xFC45. IDK if it can work.
Drag and drop a ZHAhumidity, clic on it and edit the cluster to use this one instead.

closed 2022-03-07 by github-actions[bot]