dresden-elektronik/deconz-rest-plugin #5516: Ikea Styrbar E2002 - 1002 ans 2002 block after long press left or right

VentiloFred, 2021-11-25:
## Describe the bug
  Hello, i have 2 Styrbar E2002 connected with my Home Assistant v2021.11.5 and deCONZ v6.10.0
1001, UP shoort 
1002, UP long press
1003, UP long press release
2001, Down shoort 
2002, Down long press
2003, Down long press release
3002, Shoort Left
4002 Shoort Right
all work but i don't have long press left and right
If i press long left or right, the 1002 and 2002 will no longer work
I listen the message with Node Red pelette deCONZ

## Steps to reproduce the behavior
Puch long on left or right button

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi 4
 - Running method: Home Assistent deCONZ Add-on v6.10.0
 - Firmware version:  26660700
 - deCONZ version: 2.12.06
 - Device: ConBee II 
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? no
 - 
## deCONZ Logs

## Additional context

tsadan, 2021-11-28:
Hi, 
I had the same problem. Flashing firmware 26720700 ( Raspbee II ) and upgrade deCONZ Gateway to 2.13.02 / 10.11.2021, 
largely resolved the problem. 
Sometimes, when left or right button have been pressed long, but not long enough, up and down signals 1002 and 2002 are still missing. They come back when left or right button are held as long as 3001 and 3003 ore 4001 and 4003 are seen.

Hope that helps, regards
tsadan

LetmeCheckthatforme, 2021-12-23:
Same Issue here - Conbee II with firmware 26660700. The up down buttons of the styrbar suddenly stopped working.
Left right are still functioning fine. The only way i've got them working again is removing and re-adding the switch in Phoscon.

exi, 2021-12-29:
Can confirm that long-pressing the left/right buttons unblocks the on/off buttons for me.

KurvaJmenoVole, 2021-12-29:
I confirm that the long-pressing the left/right buttons blocks and unblocks the on/off buttons for me as well.

I am also prety sure that it is not just about the left/right buttons because I was a few days out of my house. When left my home, all worked perfectly. When I came back the On/Off buttons were blocked.

exi, 2021-12-29:
I can semi-reliably trigger the hanging by pressing the left or right buttons multiple times in fast succession.

mcormick, 2021-12-30:
Same here. Sometimes cannot turn off the associated light group after the scenes or brightness has been long-pressed or in succession.

fsedarkalex, 2021-12-30:
I can also confirm the staleness, but seems to not affect the API.
Rebooting my server solves the issue. I am not using phoscon anymore, as I am an OpenHAB user so I cant tell for sure is rebooting the gateway alone is also enough

JoaoCruz89, 2021-12-31:
After reading the suggestion here, i tried holding the left and right buttons and it been working for a day now.
Thanks for the hint.

fsedarkalex, 2022-01-01:
> I can also confirm the staleness, but seems to not affect the API.

Okay it seems as if was only lucky. Ran into this very same issue with OpenHAB today.

benyaming, 2022-01-02:
Same here, long-pressed left/right blocks and unblocks up/down AND left/right long/release events

Jackie78De, 2022-01-06:
I am using FHEM and deCOnz on a Raspi 3 and I am so glad I found this thread, because I am having the same problem and searching for the error for hours.

Dresden Elektronik. can you please investigate this issue? This is my first Styrbar remote, and I immediately slipped into this issue :-(

Mimiix, 2022-01-06:
Asked manup to check.

manup, 2022-01-06:
If I remember correctly the switch was a bit strange in what it sends for left/right buttons, not sure if it supports long press/hold at all for these buttons. Need to make some tests with the sniffer on.

MGr238, 2022-01-07:
Pi4
I've got the problem that all telegrams are not reliable.

slootjes, 2022-01-08:
Just got a Styrbar too and experiencing the same issue, overall it's a very unreliable remote at the moment. Is there any way I can help with this?

horbix28, 2022-01-09:
I got this Styrbar too, now but I'm expecting the simmilar issues mentioned up. I also can't fix this with rebooting my system pressing one of the button very often. The only way to solve, is delete it from deConz and reconnect it. But when the side buttons get pressed again, none of the other buttons work either. Thats very annoying and I hope it get fixed soon.

doets001, 2022-01-09:
I have the same issues. I logged an issue because i could not find any other logged issue.
[https://github.com/dresden-elektronik/phoscon-app-beta/issues/439](url)

LindyBalboa, 2022-01-12:
I am also having some issues with Styrbar. It seems to work for the most part, but "long press for colder" seems to be broken. Using "toggle warm/cold" appears to be a workaround.

vbaros, 2022-01-15:
I have the same issue. Here is the debug from line I get when the press on the on button doesn't work:
[INFO] No button handler for: Remote Control N2, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (Ox0006), command: ON(0x01), payload: None, zclSec: 24

Jackie78De, 2022-01-15:
Is there any chance that somebody from Dresden-Elektronik will fix this annoying issue please? :-)

Mimiix, 2022-01-15:
Already forwarded and it's on the list. Nothing more I can do atm.

manup, 2022-01-16:
The switch sits already on the table for testing - before the next release :) 
Currently busy with fixing the ubisys J1 after that the Styrbar is next.

manup, 2022-01-18:
Hmm the device is pretty strange for the side buttons.

The 300x and 400x commands do actually work, but compared to the top/bottom buttons you need to press them longer until long press starts, like 2–3 seconds. After This the top/down buttons work again.

If the side buttons are released before long press starts I can confirm the top/down buttons no longer work until long press again for > 2–3 seconds, at least this part seems to be a hickup in the plugin code, not sure yet what it is.

The side button long press sends surprisingly a lot of commands during until long release:

```
No.	Time	MAC Source	NWK Source	MAC Destination	NWK Destination	Destination PAN	Delivery Mode	Cluster	Source Endpoint	Info
24	01:14:40,507	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x09, Seq: 103
25	01:14:40,507	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x09, Seq: 103
31	01:14:40,586	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x09, Seq: 103
33	01:14:40,586	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x09, Seq: 103
39	01:14:40,971	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	On/Off	1	ZCL OnOff: On, Seq: 104
41	01:14:41,002	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	On/Off	1	ZCL OnOff: On, Seq: 104
43	01:14:41,034	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	On/Off	1	ZCL OnOff: On, Seq: 104
45	01:14:41,034	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	On/Off	1	ZCL OnOff: On, Seq: 104
47	01:14:41,480	0x3a67	0x3a67	0x0000	0x0000	0xde20				Data Request
49	01:14:41,480	0x0000	0x0000	0x3a67	0x3a67	0xde20	Unicast	Scenes	1	ZCL: Default Response, Seq: 103
51	01:14:41,497	0x3a67	0x3a67	0x0000	0x0000	0xde20				Data Request
53	01:14:41,528	0x0000	0x0000	0x3a67	0x3a67	0xde20	Unicast	Scenes	1	ZCL: Default Response, Seq: 103
55	01:14:41,528	0x3a67	0x3a67	0x0000	0x0000	0xde20				Data Request
57	01:14:41,559	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x07, Seq: 105
59	01:14:41,559	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x07, Seq: 105
61	01:14:41,590	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x07, Seq: 105
63	01:14:41,606	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x07, Seq: 105
67	01:14:41,654	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	APS: Ack, Dst Endpt: 1, Src Endpt: 1
70	01:14:42,454	0x3a67	0x3a67	0x0000	0x0000	0xde20				Data Request
72	01:14:42,487	0x0000	0x0000	0x3a67	0x3a67	0xde20	Unicast	Scenes	1	ZCL: Default Response, Seq: 103
74	01:14:42,518	0x3a67	0x3a67	0x0000	Broadcast	0xde20	Group	Scenes	1	Unknown Command: 0x08, Seq: 106
76	01:14:42,518	0x3a67	0x3a67	0x0000	0x0000	0xde20				Data Request
78	01:14:42,549	0x0000	0x0000	0x3a67	0x3a67	0xde20	Unicast	Scenes	1	ZCL: Default Response, Seq: 103
80	01:14:42,549	0x3a67	0x3a67	0x0000	0x0000	0xde20				Data Request
82	01:14:42,564	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x08, Seq: 106
84	01:14:42,594	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x08, Seq: 106
86	01:14:42,626	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x08, Seq: 106
87	01:14:42,626	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x08, Seq: 106
89	01:14:42,658	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x08, Seq: 106
90	01:14:42,658	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x08, Seq: 106
92	01:14:42,704	0x3a67	0x3a67	0x0000	Broadcast	0xde20	Group	Scenes	1	Unknown Command: 0x08, Seq: 106
93	01:14:42,704	0x3a67	0x3a67	0x0000	Broadcast	0xde20	Group	Scenes	1	Unknown Command: 0x08, Seq: 106
95	01:14:42,735	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	APS: Ack, Dst Endpt: 1, Src Endpt: 1
98	01:14:42,750	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	APS: Ack, Dst Endpt: 1, Src Endpt: 1
135	01:14:43,474	0x3a67	0x3a67	0x0000	0x0000	0xde20				Data Request
137	01:14:43,505	0x0000	0x0000	0x3a67	0x3a67	0xde20	Unicast	Scenes	0	ZCL: Default Response, Seq: 106
139	01:14:43,505	0x3a67	0x3a67	0x0000	0x0000	0xde20				Data Request
141	01:14:43,537	0x0000	0x0000	0x3a67	0x3a67	0xde20	Unicast	Scenes	1	ZCL: Default Response, Seq: 106
145	01:14:43,552	0x3a67	0x3a67	0x0000	0x0000	0xde20				Data Request
147	01:14:43,583	0x0000	0x0000	0x3a67	0x3a67	0xde20	Unicast	Scenes	1	ZCL: Default Response, Seq: 106
149	01:14:43,583	0x3a67	0x3a67	0x0000	0x0000	0xde20				Data Request
151	01:14:43,614	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	APS: Ack, Dst Endpt: 0, Src Endpt: 1
179	01:14:44,455	0x3a67	0x3a67	0x0000	0x0000	0xde20				Data Request
181	01:14:44,486	0x0000	0x0000	0x3a67	0x3a67	0xde20	Unicast	Scenes	1	ZCL: Default Response, Seq: 106
183	01:14:44,486	0x3a67	0x3a67	0x0000	0x0000	0xde20				Data Request
187	01:14:44,518	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	APS: Ack, Dst Endpt: 1, Src Endpt: 1
190	01:14:45,461	0x3a67	0x3a67	0x0000	0x0000	0xde20				Data Request
193	01:14:46,453	0x3a67	0x3a67	0x0000	0x0000	0xde20				Data Request
277	01:14:50,535	0x3a67	0x3a67	0x0000	Broadcast	0xde20	Group	Scenes	1	Unknown Command: 0x09, Seq: 107
279	01:14:50,535	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x09, Seq: 107
281	01:14:50,566	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x09, Seq: 107
282	01:14:50,582	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	Unknown Command: 0x09, Seq: 107
[... cut ...]
366	01:14:52,015	0x3a67	0x3a67	0x0000	0x0000	0xde20	Unicast	Scenes	1	APS: Ack, Dst Endpt: 1, Src Endpt: 1
386	01:14:52,507	0x3a67	0x3a67	0x0000	0x0000	0xde20				Data Request
388	01:14:53,500	0x3a67	0x3a67	0x0000	0x0000	0xde20				Data Request
```

Next I'd like to check if the Ikea gateway applies some different configuration to make this useable.
Note also that the switch requests a ZCL Default Response for group casts (I think it shouldn't do this, more badly it throws the unicasts after this until the ZCL Default Response is received).

manup, 2022-01-18:
I made a few tests with the Ikea gateway and think we misunderstood how the "long press" and "long release" actually works.
First the Ikea gateway doesn't any other configuration as we do, and the remote also spams all these messages on the native gateway.

But more importantly the long press of the side buttons is completely symmetric, meaning left + right can't be distinguished.
On the Ikea gateway they use them to start and stop a color temperature loop, which works not really well as the ct value jumps. It doesn't matter if this is done with the left or right button as both spill the exact same commands for long press.

The long press command sequence is always:  0x09 &rarr; 0x07 &rarr; 0x08 &rarr; 0x09
The payload for first and last 0x09 differs, the start can be detected with payload 0.

Only the short press commands are different for left and right.

Basically this means we need to change the implementation, for example we could add another virtual button which only has 5001 and 5003 and remove 3001, 3003 and 4001, 4003.

Currently we also detect the long press too late at the stage of the 0x08 command, this needs to be changed to handle the 0x09 start properly.

manup, 2022-01-18:
Ah now I see why we use 0x08, this actually does contain the info for left / right in the first payload byte. But this command comes 5 seconds after the initial 0x09 command, way too late to be usable.

Mimiix, 2022-02-04:
AS there's still issues with it, Re-opening.

Mimiix, 2022-02-04:
Copy from the merge:

@de-k 
Bug not fixed for me.
I installed 2.14.00-raspbian-buster-beta on my raspberry pi.

Behaviour:
Click left|right button longer than "shortpress" but shorter than "longpress"
click "on" Button -->Signal 1002 is not received.

Healing:
Click left|right long enough to receive x001 and x002 signals -> After that 1002 is received again.

Mimiix, 2022-02-04:
Notified @manup

justindekeizer, 2022-02-04:
I can confirm this is still an issue on the HA version of Deconz 2.13.4 with a just unboxed STYRBAR.

seackone, 2022-02-04:
Same here. Using Deconz 2.13.4 and bought the styrbar 2 hours ago..

hoodlumj3, 2022-02-05:
Adding my comments, same for me -
STYRBAR
Deconz 2.13.4 / Firmware: 0x26660700
missing messages in nodered / HA - no info on when this started sorry
holding left and right (seperately) fixes issue

robertklep, 2022-02-05:
The issue is (supposed to be) fixed in v2.14.0-beta, so reporting it doesn't work in older versions like v2.13.4 is rather superfluous.

justindekeizer, 2022-02-05:
> The issue is (supposed to be) fixed in v2.14.0-beta, so reporting it doesn't work in older versions like v2.13.4 is rather superfluous.

No clue as to how I(/we) should know that to be honest? The post in #5713 indicates its not fixed in the 2.14, that made me post in this original one. It was not intended to be rude or harsh comment, just wanted to point out that the issue was still there on current release and new hardware.

@all that are working on the issue, thanks for all the efforts.

robertklep, 2022-02-05:
> > The issue is (supposed to be) fixed in v2.14.0-beta, so reporting it doesn't work in older versions like v2.13.4 is rather superfluous.
> 
> No clue as to how I(/we) should know that to be honest?

I found out by reading [the comments for the PR](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5713#issuecomment-1023218951), and also [the release notes of the v2.14.0-beta](https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.14.0-beta) 😊

closed 2022-02-06 by manup